Wouter1/EMU-driver #21: stuttering when changing volume

Wouter1, 2015-01-03:
after changing volume the playback stutters  for about 10 seconds.

twixst, 2015-01-09:
Its almost ok with playback with the latest update, but there are strange repeat lags (for example i open browser while my DAW is playing, and it starts repeating small fragment of audio, until i change sample rate or off-on the card) and small click (rarely)

Wouter1, 2015-01-09:
Thanks for the conformation of this issue.
Does opening of your browser result in playing a short sound or modification of the volume (which also seems to have this effect)? Then it's this known issue.

twixst, 2015-01-09:
It sound like repeating one sec. sound, aslo, for example, when i preview from desktop any audio file

Wouter1, 2015-01-09:
In some cases, it seems the driver gets screwed up completely once it gets in stutter.

Even turning off the EMU or re-loading the driver won't work then. You need to reboot.

Wouter1, 2015-01-09:
For now, please avoid changing the volume with the software volume sliders in OSX, it seems a primary cause of this stutter problem. Just use the volume knob on the EMU.

Wouter1, 2015-01-09:
Actually the software volume setting is not working #27. But it does not necessarily hang up the driver.

Wouter1, 2015-01-09:
twixst, I think I could reproduce the problem you have with the browser

But instead of a browser I need to use Blender, see #29. I have no idea yet. 

For a workaround, you have the latest firefox? I'm on Firefox 31.0 and Safari 7.0.6 and both are running at the same time now while playing audio through EMU.

Wouter1, 2015-01-09:
The stutter seems caused by the following

If there is only a brief sound, I guess the OS leaves the streams running a bit longer instead of closing the stream immediately.

Our output stream does not use an erase head. 

So the previous sample frames are resent a few times as we re-use the buffer without having good new content.

Wouter1, 2015-01-09:
This problem also occurs while starting up, here the buffer is also used ( to start up the streams) without clearing.

Wouter1, 2015-01-09:
twixst your reported problem with running a 2nd application (your browser) seems to be issue #28. Maybe you can check if that is indeed your problem.

Wouter1, 2015-01-26:
This should now be fixed as we have the erase head on. It's now the job of the OS to clean the output buffer where necessary.

closed 2015-01-26 by Wouter1