Wouter1/EMU-driver #23: midi issues

Wouter1, 2015-01-04:
trambouter reported in #18

Wouter1, when i change sample rate in "Audio MIDI Setup app" nothing happens, and sample rate in select box not each time changed.

Wouter1, 2015-01-04:
I can not test or fix this as I do not have any midi devices. I will add this to the readme. I suppose the control panel is sufficient to change the sample rate? Or is there a difference with Midi?

trambouter, 2015-01-05:
emu0204 don't have midi. On creative control panel we can only change the sample rate. In "Audio MIDI Setup app" we can see also number of inputs and outputs channels.

Wouter1, 2015-01-05:
Ah thanks. I thought you were trying to get MIDI working. This is what this current ticket is about.

I never used a "Audio MIDI Setup app" panel so I have no idea what would happen. The driver seems to be screwing up with 0204 because the number of input and output channels are not matching. Maybe you can try with a balanced number of input and output.

Wouter1, 2015-01-20:
I have no means here to test anything midi. 
I will close this ticket. Maybe someone that has midi reopens it and then we can do something about this.

closed 2015-01-20 by Wouter1