Wouter1/EMU-driver #39: checking reaper reported latency

Wouter1, 2015-02-17:
A user has given me feedback on the reaper reported latencies.

This presumedly is with the default driver 

 reaper says:

```[44.1kHz 24bit WAV : 2/2ch 64spls ~ 1.4/3.4ms]```

44.1Khz 64 samples 1.4/3.4ms I think the 2 figures are input and output latency, and they include the buffer size reported.

I can get figures for different sample rates and buffer sizes if it is helpful. They change as you would expect with the buffer sizes (I think so anyway - I haven’t actually calculated how many samples is 1ms at each rate and worked it out)

I’m not sure why the input latency is less. Different interfaces vary quite a bit in these figures, with the same buffer size settings.

This doesn’t change when I change the plist.

I rebooted after changing it - I presume that reloads it ok? or is there a cache that needs updating? 

I also tried unloading and reloading from the terminal using the same commands that are in your install script. It still reports above figures however the plist is set.  Since the above figures include the 64 samples reaper buffer I think it is actually using the 2.2ms latency, even when it is set to 4.2ms.

These are very good figures so I have no complaints, but it is strange that it doesn’t change when the plist is changed.

Wouter1, 2015-02-17:
Yes the 1.4ms is the latency due to a buffer size of 64 samples:

64/44100 = 1.45124716553 ms

Wouter1, 2015-02-17:
The output latency and input latency should be equal I suppose? So the 2nd number must be something different.

Wouter1, 2015-02-17:
Think I found it. 
I am REPORTING in the code only an output sample latency , no input sample latency.

The output sample latency is estimated at 		

setOutputSampleLatency(2*samplesPerFrame);

Each USB frame is 1ms, so this makes 2ms. 

So what we probably see is the latency sum of (1) buffer size (2) the latency that my driver reports.

The first value 1.4ms is then the estimated input latency, the second value the estimated output latency.

Wouter1, 2015-02-17:
Both values are probably not OK. The values should be estimated from the actual offset value that is being used. I thought that this would be done automagically but apparently not. 

The best estimate of course should be the latency as set in the plist as this is the actual target distance to the read/write head.

TomDrinkwater, 2015-02-17:
so -

* Can your code be modified to report a input latency too?

* It sounds like you are saying that the driver reports an input latency of 0 and an output latency of 2ms regardless of the real values. Is that right?

* side note for a later time, can the USB frame rate be changed?

* are the buffers the driver uses one for input and one for output? ie it isn't round trip latency? So, minimum round trip latency is 4.4ms? Is there a hard lower limit or just that you start to get dropouts/buffer underruns?  Are the input and output latencies always the same? It might well be useful to be able to set them separately.

* Is the buffer in the DAW conceptually separate from the buffer in the driver? Am I right in thinking that we can't reduce the driver buffer to 0 and increase the buffer in the DAW to compensate?

It is a shame that Reaper isn't somehow magically measuring the real latency but just reporting what the driver says. However if the driver can be made to report accurately that will be good enough.

If you need input from the Reaper devs or the reaper community about what reaper is actually reporting, their forum is pretty good.

http://forum.cockos.com/forumdisplay.php?f=20

Wouter1, 2015-02-17:
Checked apple's doc. 

```Call setOutputSampleLatency or setInputSampleLatency (or both methods, if appropriate) to indicate how much latency exists on the input and output streams. The Audio family makes this information available to the Audio HAL so it can pass it along to its clients for synchronization purposes.```

I read this as that this info is ONLY used to infor the clients for sync purposes. Therefore I will not entirely re-test the driver.

Wouter1, 2015-02-17:
Tom,

This info should be passed onward by the kernel to your client. So it should now report the latency correctly.

I'd love it to hear back from you if this fix works and what you see now in reaper.

Wouter1, 2015-02-17:
To answer your questions I created a FAQ section. Thanks for the questions!
The FAQ section is here

https://github.com/Wouter1/EMU-driver/blob/master/FAQ.md

TomDrinkwater, 2015-02-17:
with the new version reaper reports correctly 44.1Khz 64 samples 5.6/5.6ms

that's with the default 4.2ms buffer.

TomDrinkwater, 2015-02-17:
and with the low latency it reports 3.6/3.6ms so also correct.

for reference of a professional interface, my Fireface 800 gives 2.5/2.9ms with the same 44.1Khz 64samples settings. 

However that is connecting with firewire so it's a whole different thing! 

nevertheless the 1ms in 1.5ms out figures implied by that are a good benchmark to aspire to in a driver, even though it may be impossible in this case.

TomDrinkwater, 2015-02-17:
thanks,

that is pretty much what I thought.

a few comments and clarification questions. I can repeat any of these that might be useful on github.

when I asked about the USB framerate I think maybe I used the wrong term, I mean can the rate at which the emu is polled over USB be increased, that is the 1ms USB polling rate. if this is increased there should be less jitter in the input and a smaller buffer should be possible.

what is the relationship between the baseline USB polling rate and the “framerate” are they the same thing or just related things?

does a faster cpu allow lower driver latencies? or is some of this stuff happening at a speed that is fixed independently of the cpu such as the USB polling rate? if so is that fixed in software or hardware? is it part of the driver or lower level again?

I think this rate must be in the software at some level  - either the driver or coreaudio because there exist other devices (some gaming mice, or the eigenharp pico for example) which connect over USB at faster rates i.e. 0.5ms accuracy and so on. However they don’t use coreaudio. Would giving control over this basic rate in this driver be very difficult?

I will experiment with even lower latency settings and I’ll let you know when it gives problems.

a lot of this driver calculation seems like it is only required because it is using coreaudio - it would presumably be easy in principle for a DAW to deliver 24bit integers in a single stream with sole control over the device, but the extra processes and latency are the price we pay for the convenience of integrated system audio - is that correct?

Wouter1, 2015-02-17:
Thanks Tom for this confirmation

>with the new version reaper reports correctly 44.1Khz 64 samples 5.6/5.6ms. that's with the default 4.2ms buffer.

That's exactly what I was expecting. 1.45 ms from from the 64 sample buffer and 4.2ms in the driver default value.

Wouter1, 2015-02-17:
>for reference of a professional interface, my Fireface 800 gives 2.5/2.9ms with the same 44.1Khz 64samples settings.

>However that is connecting with firewire so it's a whole different thing!

Yes . 

>nevertheless the 1ms in 1.5ms out figures implied by that are a good benchmark to aspire to in a driver, even though it may be impossible in this case.

1.5ms is almost impossible as the 64 sample buffer in reaper already gives you 1.45ms That leaves nothing for the driver itself.

TomDrinkwater, 2015-02-17:
the 1.5ms is my guess at the latency in the FF800 driver excluding the reaper buffer. the total output latency on the fireface is 2.9ms with 64 sample buffer on reaper.

Wouter1, 2015-02-17:
Ah yes. At some sample rates in some conditions I think the EMU can also deliver 1.5ms driver latency (plus 1.45ms reaper latency). I have seen cases with low jitter on the USB timing. But these are isolated situations, probably highly dependent on the exact crystal frequency of my device, the selected sample rate etc. It would be much work to make timing filters made specifically for these cases, and then it might still break if it would get slightly warmer or colder, if the CPU speed would be slightly different, etc.

 I chose a more general path of reliability and generally robust code. I only have so much time and low latency was not top priority.

Wouter1, 2015-02-17:
I have added your questions to the FAQ

Wouter1, 2015-02-17:
Looking forward to Tom's feedback whether recording works ok on the Tracker Pree.

Wouter1, 2015-02-18:
I close this issue as the reported latency now matched the computed latency of the software. What is NOT yet fixed is that the computed latency does not match the measured latency. But this is ticket #40 .

closed 2015-02-18 by Wouter1