Wouter1/EMU-driver #41: phase inversion

Wouter1, 2015-02-18:
TomDrinkwater reported that the phase is being reversed in a roundtrip test connecting output directly to input.

Wouter1, 2015-02-18:
Tom, would it be possible if you play back using a different device, eg your alesis, and record using the EMU. Assuming the alesis does no inversion, this would then tell us if the input or output is inversed

TomDrinkwater, 2015-02-18:
I wll try later today.

TomDrinkwater, 2015-02-18:
OK, I set up an aggregate device.

I then sent a sound out of the alesis and recorded it on the emu via a TRS cable.

44.1Khz 64 samples.

it was recorded from the emu out of phase and 47ms behind.

this seems to indicate that both the phase reversal and the ~10x higher than expected latency are in the emu inputs not outputs.

That is consistent with my earlier tests.

TomDrinkwater, 2015-02-18:
PS the aggregate device reported latency is the same as the emu reported latency. I guess this makes sense since the aggregate device might be expected to use the highest reported latency of its component devices.

Wouter1, 2015-02-18:
Great, thanks for testing this. I hope I have time later this week to dig deeper into this

Wouter1, 2015-02-19:
I checked the input code convertFromEMUUSBAudioInputStreamNoWrap and it seems ok,  no inversion there.

Wouter1, 2015-02-19:
I still need to check the Volume() code.

TomDrinkwater, 2015-02-20:
It is of course possible there is an inversion in hardware, i have seen such things surprisingly often in audio gear. But phase inversion is mostly only a problem if you don't know it is happening, as it is easy to flip back.

Wouter1, 2015-02-21:
I checked the input side.
I connected my DAT recorder with both S/PDIF and with normal line to the EMU inputs and recorded some arbitrary music with different left and right channels.

connected only the red connector of the DAT lineout to input 1 of the EMU.

The data looks correct, no phase inversion.

Wouter1, 2015-02-21:
I looked up the color codings of these connectors. http://en.wikipedia.org/wiki/RCA_connector

It seems that Red should be RIGHT channel. So it's a bit weird that it matches with the FIRST SP/DIF channel, assuming that first = left in SP/DIF and that my cable is correctly color coded. Need some more checks here.

Wouter1, 2015-02-21:
Found it, I was using the wrong mini jack plug, I need a special 4-conductor that came with the DAT recorder if I plug in the DAT line-out. Now it correctly maps the WHITE connector into the left channel. Again, no phase inversion.

TomDrinkwater, 2015-02-21:
It would not surprise me much if we found that the tracker pre inputs were out of phase in hardware.

however , the best test would be to use a single cable to loop from the emu output to input.

a TRS cable is best but a guitar cable would do, the phase should be the same either way.

the phase is unimportant as it is easy to correct at any stage. it would be nice to know where it is happening though....

TomDrinkwater, 2015-02-21:
we are also assuming your dat player has correct phase.

Wouter1, 2015-02-22:
Tested EMU 0404 with mic on input1, mains/headph output back into input2.

It appears the mains line output is inverted.

However the headphone output is not inverted.

This must be some hardware bug.

Can you check this is also the case with the tracker pre?

TomDrinkwater, 2015-02-22:
OK. I will check. the headphone output is of course unbalanced stereo trs and the inputs are balanced trs, but i think i have a cable that converts ok.

Wouter1, 2015-02-22:
Ah that could be. I'm using multiple conversion cables, just old cables I found in a drawer here. Nothing to convert from balanced to unbalanced

Wouter1, 2015-02-22:
I found something else puzzling. I'm using very short clicks to test the latency. The headphone output is looped back into input 2.Now of course this gives echos, because the input 2 sound will again reach the direct monitor output and then re-appear. As in the picture below.

I'm using the headphone as loopback  as it seems there the phase is correct. 

However, the phase is correct only the first click! The second click is inverted. The third click again is normal. The fourth click again inverted. Etc. 

[image]

TomDrinkwater, 2015-02-22:
you should be able to plug a unbalanced TS cable into the TRS input and get correct results, as it simply grounds the out of phase wire. however if you plug a trs cable between a headphone jack and a balanced input you will get very strange results. a TS cable from headphones to input should work, it shorts one channel at the headphone jackthen sends the other unbalanced into the in phase input of the balanced input.

TomDrinkwater, 2015-02-22:
>However, the phase is correct only the first click! The second click is inverted. The third click again is normal. The fourth click again inverted. Etc. 

if the input jack in inverted this is what you would expect I think, the first click you hear direct then each echo passes through the input jack one more time? does that make sense?

Wouter1, 2015-02-22:
Yes you're right, stupid. The TRS is no problem I suppose but I should not have connected BOTH outputs, only the one going to the tip ...

TomDrinkwater, 2015-02-22:
do you have a regular guitar type TS cable? using that from headphone to input should give correct phase, by shorting one channel of the headphones and shorting the out of phase input of the balanced input.

Wouter1, 2015-02-22:
I have one such cable but it is hard wired to a broken PZM mic... 

The other two I have and use now are TRS to dual cinch converters.

Wouter1, 2015-02-22:
Ok that solves also the phase inversion with the mains output :+1:

Wouter1, 2015-02-22:
Here someone also reports an issue with inversion on tracker pre. 

http://www.diyaudio.com/forums/equipment-tools/250860-e-mu-tracker-pre-phase-issues.html

Wouter1, 2015-02-28:
>Tom wrote in #40 

See the above.

I'm still getting the phase inversion too, did we decide that was just the tracker pre input?

I can't reproduce it here. I thought I had something but it turned out a wrong connector. 

Did you double check your cables? 
Do you have this issue also with reaper?

TomDrinkwater, 2015-02-28:
yes it is in reaper and I have tried a few different cables.....

Wouter1, 2015-02-28:
Ah then that confirms this is an issue with the Tracker Pre.

Shall I put it in the FAQ or on the main page with the exclamation mark? How important is this for a normal user? Is this the mic input, line input or both?

Wouter1, 2015-02-28:
>I don't really understand how the line "if (pollInterval == 4)" changes the variable pollinterval, seems like it just checks its value. what is going on there?

That code determines which configurations are considered for use

TomDrinkwater, 2015-02-28:
Shall I put it in the FAQ or on the main page with the exclamation mark?

maybe, if I confirm it with some more tests.

 How important is this for a normal user? 

potentially quite important to know it is there, it depends on what the user is doing. but so easy to correct in the daw, that it is not worth much effort to fix.

Is this the mic input, line input or both?

the line, but I would imagine they are the same. they are both on one of those combo XLR/TRS jacks.

Wouter1, 2015-02-28:
Yes but they might have messed up jus the TRS , not the XLR? They are NOT straight connections, there is extra resistors on the TRS and also there must be some switch to turn on phantom power. Do you have a mic orsomething with XLR to test that?

TomDrinkwater, 2015-02-28:
I have mics, but with a mic there is no reference for the input phase.... so hard to tell if it is flipped.

best would be a trs to xlr loopback cable, I will try it...

Wouter1, 2015-02-28:
Just playback something rhrough speakers? Or can they be reversed too?

Also I think if you clap hands or make a pop with a balloon etc the first thing to arrive should be a HIGH pressure front. There prpbably is some trick to check this

Wouter1, 2015-02-28:
I just see the tip to say "pop" to the mic. The pressure wave from that should be positive.