Wouter1/EMU-driver #44: midi support

Wouter1, 2015-02-26:
It is not known if the driver supports midi

Wouter1, 2015-02-26:
TomDrinkwater remarked in #40 about

http://support.apple.com/en-gb/HT201530

this from the apple doc is sort of wrong/incomplete:

    The roundtrip latency when playing software instruments live through Logic will typically be less than when monitoring audio. It is the sum of:

    Audio driver latency
    Output portion of the I/O Buffer
    DA converter latency

They are ignoring the midi input buffer for the software instrument. That is not directly relevant to what we are dealing with here, except it is indirectly relevant because some DAWs set the incoming midi buffer to the same delay as the incoming audio buffer. (makes sense to have acoustic instruments with a mic and software instruments with a midi controller to have the same input latency)

Quite a few audio apps don't deal with live incoming midi correctly, you actually need a buffer and time stamping on the midi to avoid jitter.

this article deals with the issue quite well:

http://www.eigenzone.org/2012/12/04/midi-jitter

note that as of 2012 logic (and mainstage) did this wrong. I can't stand logic myself and I don't use it so I don't know if it has improved since then.

Wouter1, 2015-05-22:
In #62 Whyiamdead  confirmed that MIDI does not work

Whyiamdead, 2015-05-23:
Ready to help with testing a midi.

Wouter1, 2015-05-24:
Thanks for your offer!
I am going to look into the old midi code and post back here when I have something testable.

Wouter1, 2015-05-24:
The very first step here is to create a new target in the compiler 

The old HulaDrvr (check out the first version of this project) creates a EMUMIDIDriver.plugin

So the *Wrapper Extension* must be .plugin

Unfortunately it's nowhwere documented what type of targets are associated with which extensions

I suppose this is a "system plugin" and not a framework or application plugin so that leaves us with 8 options: generic c++ plugin, generic kernel extension ,  iokit driver, image unit plugin, preference pane, quick look plug-in, screen saver and spotlight importer. Only the first three make any sense for midi

generic plugin gives me a ".bundle" extension

kernel gives me a ".kext" extension

iokit gives me a ".kext" extension

Wouter1, 2015-05-24:
Checking some OSX documentations show that midi runs on top of IOKit. Su best chances seem to use an IOKit type

Wouter1, 2015-05-24:
The original HulaDrvr is linked with 

* CoreMIDIServer.framework
* CoreFoundation.framework
* IOKit.framework

Wouter1, 2015-05-25:
For some reason, I do not get the CoreMIDIServer framework as an option in the project setup.

Wouter1, 2015-05-25:
Something seems wrong with that framework. FAIK I need to include 
```
#include <CoreMIDIServer/CoreMIDIServer.h>
```

But the files available in /System/Library/Frameworks/CoreMIDIServer.framework (all other resources apparently are linking to this) are

```
CoreMIDI.h
MIDISetup.h
MIDIThruConnection.h
MIDIDriver.h
MIDIServices.h
```

Wouter1, 2015-05-25:
Looking at /Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk/System/Library/Frameworks/CoreMIDIServer.framework  the Headers folder is just a symbolic link to 
/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk/System/Library/Frameworks/CoreMIDI.framework/Versions/A/Headers

That may explain why it does not appear?

Wouter1, 2015-05-25:
Tried to include stuff using the CoreMIDI instead of CoreMIDIServer framework. getting not found errors too.

Wouter1, 2015-05-25:
There does not exist a CoreMIDIServer.h anymore.

Wouter1, 2015-05-25:
Fixed the build issues.. 

Checked the EMUMIDIDriver.

It seems it has to be installe in /Library/Audio/MIDI Drivers/EMUMIDIDriver.plugin (or maybe System/Library/...)

I already have that file there. That may be the original. 

Another possibility that I did not yet think of is that this file maybe just works?

Wouter1, 2015-05-25:
added the orig EMUMIDIDriver to the project, as ```EMUMIDIDriver orig.plugin```

Wouter1, 2015-05-25:
@Whyiamdead I added ```EMUMIDIDriver orig.plugin``` to the project (master project, last revision)

Can you do the following in the console
* download the EMU-driver latest version
* ```cd EMU-driver```
* ```cp -R EMUMIDIDriver\ orig.plugin /Library/Audio/MIDI\ Drivers/EMUMIDIDriver.plugin```
* reboot

Can you then check if MIDI is working?

Whyiamdead, 2015-05-25:
I had to create a folder midi drivers manually, but everything else was fine. Midi works, thank you so much!
Tested on Yosemite 10.10.3, Logic Pro 10.0.7

Wouter1, 2015-05-26:
@Whyiamdead that is great, thanks for the confirmation that this works!

Wouter1, 2015-05-26:
@Whyiamdead do you know if the reboot is necessary? Or is it sufficient to put the .plugin in /Library/Audio/MIDI Drivers?

Wouter1, 2015-05-26:
The install script has been updated.

Wouter1, 2015-05-26:
NOTE: the code in the midi branch is currently not used. The code there compiles but has not even been tested.

Whyiamdead, 2015-05-27:
Yes, reboot is necessary.

Wouter1, 2015-10-17:
It seems midi is working, so I close this ticket.

closed 2015-10-17 by Wouter1