Wouter1/EMU-driver #50: EMU 0404 does not start

vitalibondar, 2015-03-08:
*I am sorry for terrible English - I have to use Google Translate.*
Today, after turning on the computer, I found that Mac OS X Yosemite does not "see" EMU.
That's part of the system log messages:
`3/8/15 10:34:50.000	kernel[0]	*** kernel exceeded 500 log message per second limit  -  remaining messages this second discarded ***
3/8/15 10:34:51.000	kernel[0]	+EMUUSBAudioDevice[0xffffff802a012800]::queryXU
3/8/15 10:34:51.000	kernel[0]	DigitalIOStatus changed
3/8/15 10:34:51.000	kernel[0]	Digital IO SampleRate
3/8/15 10:34:51.000	kernel[0]	EMUUSBAudioDevice::deviceRequest result=-536854449
3/8/15 10:34:51.000	kernel[0]	unitID 14
3/8/15 10:34:51.000	kernel[0]	clearing pipe stall result 0
3/8/15 10:34:51.000	kernel[0]	++EMUUSBAudioDevice[0xffffff802a012800]::deviceRequest(0xffffff811de83d28, 0) = e000404f
3/8/15 10:34:51.000	kernel[0]	DigitalIO Sync Lock
3/8/15 10:34:51.000	kernel[0]	EMUUSBAudioDevice::deviceRequest result=-536854449
3/8/15 10:34:51.000	kernel[0]	clearing pipe stall result 0
3/8/15 10:34:51.000	kernel[0]	++EMUUSBAudioDevice[0xffffff802a012800]::deviceRequest(0xffffff811de83d28, 0) = e000404f
3/8/15 10:34:51.000	kernel[0]	-EMUUSBAudioDevice[0xffffff802a012800]::QueryXU`
This part of the constantly repeated until the EMU is turned off.
I removed the driver strictly according to your instructions, reinstall it. The same error all the time.
Instal "stock" driver - without success. Without effect.
Because with "stock" driver EMU also does not work, so it is assumed that the problem is not in the drivers, and in my system. Maybe you have any ideas how to solve this problem?

Wouter1, 2015-03-08:
Thanks for reporting.

This is a USB issue.

What computer do you have? 

The error I see there is result=-536854449 = 0xE000404F. I have never seen that error before. 

Google gives me this

http://lists.apple.com/archives/usb/2010/Apr/msg00028.html

and the reply

http://lists.apple.com/archives/usb/2010/Apr/msg00029.html

>On Apr 26, 2010, at 3:10 PM, Mattiello Joseph wrote:

> The main architectural difference is the MBP is 32bit only, but compiling and running fails on the new iMac as both a 32bit and 64bit binary.

>The main architectureal difference between the 27" i7 iMac (as well as any of the new MacBook Pro's 15" and 17") is that there are no OHCI or UHCI USB controllers;  there are 2 EHCI controllers, along with 2 high speed hubs inside the machine.  That means that any device that you attach will be attached to the high speed hub.    All devices are still supported, but now they are ultimately attached to the high speed controller and so the low level timing might be different.  They are also attached through a hub and not directly to the root hub (even high speed devices).

>The errors that you describe indicate that the device is either not responding to the requests or stalling the requests.   We cannot tell which it is, so we alternate returning the 404f and the 2ed errors.

>You will have to look with an analyzer to see what exactly is going on.  You might have the same issue if you attached the device through a high speed hub to your old Mac.

>Fernando Urbina

>USB Technology Team

>Apple Inc.

vitalibondar, 2015-03-08:
I guess that the problem is not in the driver. I have a MacBook Pro Retina late 2013. Looks like I will have to reinstall the operating system. Do not see another solution, given my technical skills. @Wouter1 thank you for response.

Wouter1, 2015-03-08:
I have the same model so this should work.

Are you connecting theEMU directly to the mac? Or are you using a hub?

Wouter1, 2015-03-08:
Apparently the code is not in IOReturn.h but somewhere else. The error itself does not tell much

>    #define kIOUSBPipeStalled iokit_usb_err(79) // 0xe000404f Pipe has stalled, error needs to be cleared

vitalibondar, 2015-03-08:
EMU connected directly to the mac. I tried connect to second USB port. Same result.
I will try to connect E-MU to the Windows PC. What if the problem is in the sound card?

Wouter1, 2015-03-08:
BTW your computer DOES see the EMU, otherwise the driver would not even start. And you can see from the messages that the driver is starting. But the communication does not work ok.

If it works on the windows PC it's not the EMU. So yes good plan. 

Also you could try if it's maybe a bad cable.

vitalibondar, 2015-03-08:
My E-MU working again (thanks to the old Windows laptop).
I remembered that before listening to music in high quality (192/24). Sound card was configured as appropriate (set to 192 kHz sample rate).
Also notice that the sound card does not respond to most pressing buttons (Direct Monitor, S / PDIF).
What I did:
1. I tried to reset the sound card to the factory firmware - *did not help*.
2. E-MU connected to the Windows-laptop, installed the driver from the CD-ROM that was in the box - during installation installer updated sound card firmware from version 4.x to version 8.x. - *But the card still does not work*.
3. I launched the E-MU Control Panel for Windows. I saw that in the settings 192 kHz. I changed the settings to 44.1 kHz - *sound card immediately "alive" and started working*.
4. Then I connected to the E-MU Mac — **card worked, all right!** But the system log still contains the following errors:
`3/8/15 14:22:05.000	kernel[0]	+EMUUSBAudioDevice[0xffffff80200ca000]::performPowerStateChange(1, 2, 0xffffff811100b8ac)
3/8/15 14:22:05.000	kernel[0]	+EMUUSBAudioEngine[0xffffff8020163800]::performAudioEngineStart ()
3/8/15 14:22:05.000	kernel[0]	about to start USB stream(s)
3/8/15 14:22:05.000	kernel[0]	averageFrameSamples=48
3/8/15 14:22:05.000	kernel[0]	averageFrameSamples=48
3/8/15 14:22:05.000	kernel[0]	Isoc Frames / usbCompletions
3/8/15 14:22:05.000	kernel[0]	EMUUSBAudioEngine::SetSampleRate 48000
3/8/15 14:22:05.000	kernel[0]	using SampleRateXU
3/8/15 14:22:05.000	kernel[0]	createInputPipe
3/8/15 14:22:05.000	kernel[0]	+UsbInputRing::init bytesize=98304 byterate=576000
3/8/15 14:22:05.000	kernel[0]	-UsbInputRing::init 170666666
3/8/15 14:22:05.000	kernel[0]	create output pipe 
3/8/15 14:22:05.000	kernel[0]	check for associated endpoint
3/8/15 14:22:05.000	kernel[0]	frameSizeQueue empty, guessing some queue size. May need fix..
<truncated>
3/8/15 14:22:05.000	kernel[0]	frameSizeQueue empty, guessing some queue size. May need fix..
3/8/15 14:22:05.000	kernel[0]	++EMUUSBAudioEngine[0xffffff8020163800]::performEngineStart result is 0x0 direction 1
3/8/15 14:22:05.000	kernel[0]	warning way-out ring wrap position
3/8/15 14:22:05.000	kernel[0]	warning way-out ring wrap position
3/8/15 14:22:05.000	kernel[0]	frameSizeQueue empty, guessing some queue size. May need fix..
<truncated>
3/8/15 14:22:05.000	kernel[0]	RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
3/8/15 14:22:05.000	kernel[0]	UsbInputRing::notifyWrap 0
3/8/15 14:22:05.000	kernel[0]	warning way-out ring wrap position
<truncated>
3/8/15 14:22:06.000	kernel[0]	warning way-out ring wrap position
3/8/15 14:22:06.000	kernel[0]	RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
3/8/15 14:22:06.000	kernel[0]	UsbInputRing::notifyWrap 4
3/8/15 14:22:06.000	kernel[0]	LowPassFilter::filter init 170666666
3/8/15 14:22:06.000	kernel[0]	USB timer started
3/8/15 14:22:06.000	kernel[0]	RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
3/8/15 14:22:06.000	kernel[0]	**** Output Hiccup!! firstSampleFrame=1248, nextExpectedOutputFrame=0 bufsize=98304
3/8/15 14:22:06.000	kernel[0]	RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
<truncated>`

Anyway, thank you for your patience.
And again sorry for my terrible English.

Wouter1, 2015-03-08:
It sounds likethe EMU was stuck in some bad mode. Maybe some battery going low or some flash memory that flipped. Yes switching the rate usualy brings it back. 

It is normal to get a bunch of messages during startup of the device. But you should not get any errors while it's running (unless of course you start switching rates etc). It seems it's a bit more than usual, looks like I forgot to turn off the debug messages when uploading the latest driver. I will fix that

Wouter1, 2015-03-08:
>Anyway, thank you for your patience.
>And again sorry for my terrible English.

Your English is quite OK!
Glad you could figure it out!

closed 2015-03-08 by Wouter1