Wouter1/EMU-driver #53: report latency to Core Audio

Wouter1, 2015-03-08:
The latency as reported to Core Audio is currently incorrect. We only report driver latency but apparently we should add the EMU latency to it to get the total latencies.

Wouter1, 2015-03-08:
I have no means to determine the exact input and output latency. I will use the roundtrip latency and estimate input and output latencies from that.

Wouter1, 2015-03-08:
We discussed and measured a lot regarding this in #40. I will duplicate the relevant measurements here.

The formula is
```latency = latency of driver + measured internal latency of EMU```

latency of the driver is just the safetyOffset value as in the plist.

To estimate the internal latency of the EMU we found this approximation

```internal emu latency of 0404 (ms) = 140000/rate + 1.54 + ( 0.32 if rate=48k or 96k)```

This is close but not perfect.

For the Tracker Pre this is not ok.

Wouter1, 2015-03-08:
I re-measured all latencies, 1000us driver latency, 64samples DAW buffer.

| rate | 0404 latency (roundtrip #samples)  EMU internal | Tracker Pre |
| --- | --- | --- |
| 44k | (385)  | 3.83 | - |
| 48k | (391) 3.48ms | - |
| 88k | - | - |
| 96k | - | - |
|176k | - | - |
| 192k | - | - |

Wouter1, 2015-03-08:
This is consistently 2ms below the values we found for the normal (pollInterval=8/  normal speed) USB driver.

Wouter1, 2015-03-08:
The plan is to put all the values just in the plist. Just as a number of sampleframes, ready to feed into setInputSampleLatency etc.
Like
LatencyIn44 = 84   // half of 3.83ms
LatencyOut44 = 84 // idem.

etc

Wouter1, 2015-03-10:
The new EMU0404 internal latency formula is extremely simple linear
```
latencysamples = 217+ 0.00370 rate
```

Wouter1, 2015-03-10:
Because that linear formula is extremely simple I went for that approach.

Wouter1, 2015-03-13:
The Tracker Pre 48k measurement seems incorrect.

Wouter1, 2015-03-13:
Tom could you also do a measurement9 at 176 and 192?

TomDrinkwater, 2015-03-13:
i know it repeats the above data, but I'm repostng my latency results here for comparison:

these are with comp 2

these are all with 1ms in the plist

assuming reaper is correct the driver is now over-reporting its latency.

| rate | daw buffer | latency in samples | in ms | roundtrip -daw | difference when using compensation in reaper |
| ---- | ---- | ---- | ---- | ---- | ---- |
| 44 | 128 | 496 | 11.2472 | 5.442 | -12 |
| 48 | 128 | 409 | 8.5208 | 3.188 | -113 |
| 88 | 128 | 666 | 7.5510 | 4.649| -4 |
| 96 | 128 | 682 | 7.1042 | 4.438 | -18 |

TomDrinkwater, 2015-03-13:
the above 48k measurement is anomalous but I did repeat the test a few times and got the same result. I will try again later on that comp to see if I still get the weird 48k result.

TomDrinkwater, 2015-03-13:
here are my results on comp 1 - much more sensible!

these are still all with 1ms in the plist
| rate | daw buffer | latency in samples | in ms | roundtrip -daw | difference when using compensation in reaper |
| ---- | ---- | ---- | ---- | ---- | ---- |
| 44 | 128 | 489 | 11.0884 | 5.283 | -18 |
| 48 | 128 | 503 | 10.4792 | 5.146 | -19 |
| 88 | 128 | 650 | 7.3696 | 4.467| -20 |
| 96 | 128 | 682 | 7.1042 | 4.438 | -18 |

I tested latency in this test and the above test on comp2 using your 3 clicks file. that means that reaper was performng SRC on the 96k file when testing at other rates. This may account for some of the small differences seen, but not for the 48k result on comp2.

Wouter1, 2015-03-13:
Thanks Tom! I need to look at it in more detail. That 1.18ms latency is really anomalous but thanks for confirming it's not a fluke. 

Your other results are interesting. So for 48k this different for the computers?!? The other rates look close enough, within the range of noise?

The hf noise, that is #43. I also have a noise issue #46 but it seems related to whether or not I attach a headphone.

TomDrinkwater, 2015-03-13:
yes so far it looks like 48k is different on the computers. I redid the 48k test a few times, but I didn't change much in between, so I will check it again on comp 2 later tonight maybe... and see if it is still the same. It could be some sort of error, though I can't think what!

TomDrinkwater, 2015-03-13:
>The other rates look close enough, within the range of noise?

my guess is that nose and analog inaccuracy is small just around 0-5 samples, and that there is also a consistent slight over-reporting of the latency now. But maybe the usb latency is different on different usb hardware too.

Wouter1, 2015-03-13:
I can use different formulas for tracker pre and 0404. So the overestimated latency is because I did not yet have the measurements for tracker pre.

But to compensate for the vastly differences at 48k we need to know what exactly is behind this. For now I can go with the measurements from compu 1 that look smooth and ignore those of computer 2.

Wouter1, 2015-03-13:
I updated the table above.

TomDrinkwater, 2015-03-14:
retest of 48k on comp2, it is now in line with the other results.

I don't know what was happening before it must have been in some inconsistent state maybe different places thought there were different rates or something?

| rate | daw buffer | latency in samples | in ms | roundtrip -daw | difference when using compensation in reaper |
| ---- | ---- | ---- | ---- | ---- | ---- |
| 48 | 128 | 505 | 10.5208 | 5.188 | -17 |

TomDrinkwater, 2015-03-14:
here's full results of latency retest on comp 2 using latest non-test driver, making sure to restart reaper after every sample rate change.

| rate | daw buffer | latency in samples | in ms | roundtrip -daw | difference when using compensation in reaper |
| ---- | ---- | ---- | ---- | ---- | ---- |
| 44.1 | 128 | 499 | 11.3152 | 5.510 | -9 |
| 48 | 128 | 505 | 10.5208 | 5.188 | -17 |
| 88 | 128 | 673 | 7.6304 | 4.728 | +3 |
| 96 | 128 | 682 | 10.5208 | 5.188 | -18 |

TomDrinkwater, 2015-03-14:
i suspect that the anomalous result before might have been related to the sample rate changing bugs. maybe it wasn't fully changed to 48k.

Wouter1, 2015-03-14:
> maybe different places thought there were different rates or something?

Yes I have been thinking along that line too. #45 gives even some evidence. So far I could not put my finger on it though.

Wouter1, 2015-03-14:
Great to see that comp1 and comp2 have almost identical results. I put them in the first table as well but see no need to change anything else there.

Wouter1, 2015-03-14:
For the tracker pre, we need the formula ```72.23 + 0.00169 rate``` for the internal latency

Wouter1, 2015-03-14:
So the 0.0017 is a constant factor , the only change wrt the 0404 formula is the 90 (0404) versus 72 (Tracker Pre).

Wouter1, 2015-03-14:
number of samples delay from group delay of adc and dac: 

ADC AK5385 group delay 43.2 * rate (38.2@192)
DAC AK4396 group delay 28 * rate
DAC CS4392 groupdelay 4.7 * rate

| device | ADC | DAC | other |
| --- | --- | --- |
| 0404 | 43.2 | 28 | 19 |
| Tracker Pre | 43.2 | 4.7 | 22 |

Wouter1, 2015-03-14:
A simple guess seems to take  "other" = 20 in both cases. 10 for input and 10 for output?

Wouter1, 2015-03-15:
I will do it that way. Then, we get the following values for latencyDAC and latencyADC

| device | latencyADC | latencyDAC |
| --- | --- | --- |
| 0404 | 53 | 38 |
| Tracker Pre | 53 | 15 |

Wouter1, 2015-03-15:
We still do not have a rationale and a proper breakdown of the part ```0.00169 rate``` in the formula. Therefore we still use a simplistic breakdown: half of it for the input, half of it for the output. This part gives a significant portion of the total latency (about 2ms).

Wouter1, 2015-03-15:
Fixed the code.
Checking the reaper reported latencies, latency samples (using HALLab) and actual. Using reaper 64ms buffer

|rate | HALLab in / out (excl DAW buf) | reaper in/out | actual roundtrip samples/ms |
| -- | --- | --- | --- |
| 44 | 275/260  | 7.6/7.3 (14.9) | 695 = 15.19 |
| 48 | 294 / 279 | 7.4/7.1 (14.5) | 15.19 |
| 88 | 497/482 | 6.3/6.1 (12.4) | - |
| 96 | 537/522 | 6.2/6.1 (12.3) | 1186 = 12.36 |
|176 | 942 / 927 | 5.7/5.6 | - |
|192 | 1021/1006 | 5.6/5.5 (11.1) | 2152=11.2 |

Wouter1, 2015-03-15:
Reaper seems to be truncating the values instead of rounding. This causes a bit optimistic reporting of 0.1 or even 0.2ms below the actual

The 48k estimation is off a bit too much

Other values are ok.

TomDrinkwater, 2015-03-15:
I think reaper actually uses the value reported by the driver and just truncates for the display?

Wouter1, 2015-03-15:
Hi Tom, you enter right at the right moment! I was just closing off this ticket.
I don't know what it actually uses, yes it could be that only the display is truncated.

Wouter1, 2015-03-15:
I just uploaded a new version 3.5.3 that contains this fix. I assume the correct latencies are now also being reported with the tracker pre. 

Using my estimations and assumptions above I also determined the hopefully correct latencies for the other EMU devices.

Tom, can you please briefly check if reaper reports the correct values with the tracker pre? If it's correct we can close this ticket.

TomDrinkwater, 2015-03-16:
will check , but it will have to be tomorrow, or maybe tuesday...

I seem to remember from my previous results that the unaccounted latency was different when using reaper compensation and when manually deducting the displayed latency from the uncompensated results, and I attributed this difference to reaper truncating the display but not the actual value it uses.

Wouter1, 2015-03-16:
>I seem to remember from my previous results that the unaccounted latency was different when using reaper compensation and when manually deducting the displayed latency from the uncompensated results, and I attributed this difference to reaper truncating the display but not the actual value it uses.

I will put that in the FAQ

Wouter1, 2016-07-11:
I believe we fixed this long ago

closed 2016-07-11 by Wouter1