Wouter1/EMU-driver #55: clicking (v3.5.2)

Wouter1, 2015-03-11:
TomDrinkwater reported clicking issues with the current version of the driver

Wouter1, 2015-03-11:
TomDrinkwater wrote in #40  about the clicking. 

>however when i boot another app while the test is running and cpu goes up for a few seconds as a result it does not cause a glitch.

>here's a pic of a typical new type glitch (total length about 520 samples at 88k)
[image]

>and here's an activity monitor screenshot
[image]

>today's test are without hub, and unplugged all unnecessary usb devices, seems to make no difference.

>note the new type glitch has a piece of misplaced audio, and also some flatline - that is typical - the flatline is new. the exact length now varies, and it seems unrelated to daw buffer size, wheras the old glitch was always one daw buffer exactly.

>glitches at 88k 64 samples daw buffer seem to always be the same length 526 samples and at 96k 552 samples. give or take a few samples for my measuring accuracy. so glitch length s related to daw buffer size.

>however the proportion of misplaced audio and flatline in the glitch is not constant.

TomDrinkwater, 2015-03-12:
i'm about to do some more testing on a different computer.

to confirm, i should be testing 3.5.2 with the plist latency set to 1000uS is that right?

Wouter1, 2015-03-12:
Great!

Maybe you can first test with the default settings 4200us. Just a few rates. If that's also clicking it makes no sense to go down to 1000.

TomDrinkwater, 2015-03-12:
ok, testing on computer 2

I get clicks at 88k and 96k at all plist settings up to and including 4200uS, but only at 64samples daw buffer. 96k is much worse than 88k.

I haven't tested at 1x sample rates as they were ok on the other computer even at 1000uS set in the plist.

no clicks at 1200uS and 128 samples daw buffer
no clicks at 1200uS and 96 samples daw buffer

 i guess I will check at 1000uS

I haven't checked the latency figures yet, i'm still just testing for clicks with a sine generator.

Wouter1, 2015-03-12:
Mmm. I gues you are testing with reaper. 
Can you try quitting and restarting reaper after every change in sample rate? SO you switch in reaper to a new rate. Then quit. Then wait about 10 seconds and then restart. Does that help the clicking?

TomDrinkwater, 2015-03-12:
i have been restarting reaper for each sample rate change, because it breaks the sound now to change sample rate, but I have not been restarting for each daw buffer size change. I have also been restarting reaper each time i set the plist setting as then i have to reinstall the dtiver.

restarting reaper doesn't seem to help. but the problem only exists on computer 2 at 2x sample rates and 64 samples buffer. so it is acceptable on this machine - you can just increase to 96 samples daw buffer.

what concerns me more is the clicking at 128 samples buffer on computer 1.

Wouter1, 2015-03-12:
So if I get it right, it's 64 samples DAW that causes the clicks?  Is the computer maybe too slow to handle it?

Are you seeing something particular in the logs?

Wouter1, 2015-03-12:
>i have been restarting reaper for each sample rate change
good. There's an issue there but I will make separate ticket for that

>what concerns me more is the clicking at 128 samples buffer on computer 1. 

ok let's start focusing on that first then

TomDrinkwater, 2015-03-12:
on comp 2 it is 64 samples buffer that causes the clicks and I see many output hiccup messages, but i hear no output problem, then I get "READ HICKUP" every time there is a click on the input.

when i increase the buffer size i get no errors while the test is actually running.

however on comp 1 I had clicks and "READ HICKUP" even at 128 samples. just a few. on comp 2 i have had no clicks so far at 128 or 96 samples daw buffer.

TomDrinkwater, 2015-03-12:
I have now tested on comp2 96k, 96 samples daw buffer, 1000uS plist - no clicks.

TomDrinkwater, 2015-03-12:
so overall I would say that the performance on comp2 is acceptable. I haven't checked the latency and reporting yet, but assuming that is ok then i think it is fine.

maybe comp1 just has a slight problem... it is 5 years old. i will test it again.

but first i will check the latency figures on comp 2.

Wouter1, 2015-03-12:
READ HICKUP appears when there are no USB bytes available when CoreAudio asks them. 
This is another sign that the clock is not syncing properly. I propose that we test how it behaves.

I just uploaded a test driver that dumps clock info to /var/log/system.log
Can you install the ```EMUUSBAudio test.kext``` and do a 2 minute sampling at 44k, 4.2ms (the default), and a big enough daw buffer.

Wouter1, 2015-03-12:
I will then need the /var/log/system.log output while you are sampling. You can copy the relevant minutes from the system.log into  atext file and then zip that file. We need to figure out however how to transfer this file,, as you can only attach images here FAIK

Wouter1, 2015-03-12:
I think we need gist to share these files https://gist.github.com/

Wouter1, 2015-03-12:
off to bed now, I'll check tomorrow evening what happened

TomDrinkwater, 2015-03-12:
I'm not clear if you want this log for when the clicks are occurring, or when they are not occurring, i think not since you are just logging how far out the clock gets when there is no error?

TomDrinkwater, 2015-03-12:
there doesn't seem to be anything in the log for the relevant time. is the driver correct?

the logging doesn't use apsd or a tcp connection i hope?

Wouter1, 2015-03-13:
You're right, somethign went wrong with this test driver.. I forgot to turn on the logging :-(  I built it again now with logging enabled and I also checked that it's working ok now.... Sorry for the inconvenience

Wouter1, 2015-03-13:
>I'm not clear if you want this log for when the clicks are occurring, or when they are not occurring, i think not since you are just logging how far out the clock gets when there is no error?

Yes I just want to see what is happening with the PLL (the sync mechanism). I'm not sure if there are sync issues and if they appear in certain circumstances. If you have time to measure BOTH conditions that would be great. Just let avoid other issues like CPU overload cases. So please keep the DAW buffer big (eg 128) and keeo latency setting of the driver wide enough (default 4.2 should be plenty).

TomDrinkwater, 2015-03-13:
OK here is the log for 2 mins at standard settings, let me know if it is OK... it was a lot of lines so i hope the copy and paste worked correctly!

https://gist.github.com/TomDrinkwater/ecd68e4e1f6dc8b41fbf

TomDrinkwater, 2015-03-13:
this test the cpu use was much higher than usual for such a test, presumably because of the logging itself? I had the console window open so maybe it would be better if I closed it for the test.

Anyway there were no clicks during this test. I have not had clicks with the current driver at 44.1khz in any circumstances...

TomDrinkwater, 2015-03-14:
ok here's another log, there were at least 2 clicks during this test, 96k 64 sample buffer, same driver as before.

https://gist.github.com/TomDrinkwater/3497c4fbea8afec76a0f

TomDrinkwater, 2015-03-14:
however, i get the message: 

>*** kernel exceeded 500 log message per second limit  -  remaining messages this second discarded ***

so that might mess up the results.

do you know how to increase the limit?

Wouter1, 2015-03-14:
Thanks Tom for posting the measurement results. I'm analysing it now. Can you add the used sample rate to the posts?

Wouter1, 2015-03-14:
analysed https://gist.github.com/TomDrinkwater/ecd68e4e1f6dc8b41fbf
I assume you had a rate with ringbuffersize 8192 in the driver (PENDING RATE CONFIRMATION).

Under that assumption, it all looks fine.
[image]

Tom, did you get any clicks here? I assume it would work all fine.

Wouter1, 2015-03-14:
>https://gist.github.com/TomDrinkwater/3497c4fbea8afec76a0f

[image]

There are a few "outliers" around the -60.

Wouter1, 2015-03-14:
These outliers can cause clicks if you are going for the lowest latency. At 96k the best range of the act-est is within [-48, 48]. 

So we do have a little noise in the USB timestamps here.

Wouter1, 2015-03-14:
Tom, with the 96k sample, you had the driver latency set to defautl (4.2ms)? Then this should not have caused clicks

Wouter1, 2015-03-14:
>this test the cpu use was much higher than usual for such a test, presumably because of the logging itself? I had the console window open so maybe it would be better if I closed it for the test.

Yes that's because of the logging. You can try closing the window but I think it's no problem

Wouter1, 2015-03-14:
>kernel exceeded 500 log message per second limit - remaining messages this second discarded

It's fine, no need to fix that. We miss a bit of data (so we miss some clicks) but I assume that if the clicks occur frequently we still will catch a bunch of them

Wouter1, 2015-03-14:
Tom, the 96k case is interesting

> there were at least 2 clicks during this test, 96k 64 sample buffer, same driver as before.

now I would not expect this when seeing the graph. 

Are these clicks of the "flat line" type or of the "duplicate data" type?

TomDrinkwater, 2015-03-14:
>Tom, with the 96k sample, you had the driver latency set to defautl (4.2ms)? Then this should not have caused clicks

yes I think so, unless I made some mistake...

>Are these clicks of the "flat line" type or of the "duplicate data" type?

they have a piece of duplicate data followed by a section of flat line. I think i posted a pic of the type of click earlier in the other thread.

there is one click from that test ~38 seconds in and another ~61 seconds in. but with only 2 clicks could we have missed the data for the clicks due to the 500 message limit?

TomDrinkwater, 2015-03-14:
>The 44k sample (PENDING RATE CONFIRMATION)
>analysed https://gist.github.com/TomDrinkwater/ecd68e4e1f6dc8b41fbf

yes it is 44k, i think the conditions are all in the gist title

Wouter1, 2015-03-14:
AH yes I never noticed. I  copied the titles.

Wouter1, 2015-03-15:
Tom, I'm puzzled with the 96k where you reported clicking but I do not see any error message in the trace. It might be that the messages are indeed missed due to the limit. 

Can you try to capture a new trace where the clicking occurs (using the same test driver), but that now contains error messages at that same time when the clicking occurs? 
Also please copy the startup and stop messages so that I can see if there is something strange at startup or stop.

TomDrinkwater, 2015-03-16:
OK, i will do a longer one, and attempt to capture more clicks in the hope that some show up in the logs.
can we do anything to increase the 500 messages per second limit?

how many messages per second does the driver actually generate?

Wouter1, 2015-03-16:
I don't know how to increase that limit.

But you can increase the DAW buffer size to reduce the number of messages.

The driver generates 1 message for each DAW call. So it depends on the rate (higher=more messages) and the DAW buffer size (smaller=more messages).

TomDrinkwater, 2015-03-16:
increasing the daw buffer may reduce the number of clicks, so we still would have trouble capturing the clicks.

can the logging in the driver be changed to combine several messages into one before sending to reduce the number, or only send messages within a couple of buffers of an READ HICKUP error?

Wouter1, 2015-03-16:
>increasing the daw buffer may reduce the number of clicks, so we still would have trouble capturing the clicks.

We only need to capture full log info for  a few clicks. It does not matter really how large the DAW is; but you asked about how to reduce the number of messages per second.

>can the logging in the driver be changed to combine several messages into one before sending to reduce the number, or only send messages within a couple of buffers of an READ HICKUP error?

No, this is a lot of work just for a single measurement. I don't want to spend much time on a single measurement that may not even give us the clue we are looking for. 

I suggest to just take a longer log until it captures the problem.

TomDrinkwater, 2015-03-17:
here's a log showing a click

there were other clicks but I found no corresponding READ HICKUP error in the files.

https://gist.github.com/TomDrinkwater/200201d507efc70135b6

i have loads of logs but finding an error that I can tell corresponds to a known click is difficult. if you just want lots of data to analyse I can give you plenty more though.

it logs the whole time reaper is running, but to see/hear a click I have to record a file from the input, then work out what time it occurred, find the right logfile and then see if there is an error at about that time.

TomDrinkwater, 2015-03-17:
i had to log using 64 samples, when I tried 256 i got no clicks for several minutes.

Wouter1, 2015-03-17:
Thanks tom! I hope to have time for it tonight

What I usually do is to run this
```tail -f /var/log/system.log```

while sampling.

Then I can see the error messages right away.
If there are errors there usually is a click.

if as you say we have a case here with clicks but without error messages at all (there are others than READ HICKUP eg, ring empty), then it's a new situation.

Wouter1, 2015-03-17:
96k 64 samples 4200plist log showing a click at 1:11:44, other clicks didn't show up

Here is the graph for https://gist.github.com/TomDrinkwater/200201d507efc70135b6

[image]

The READ HICKUP is at sample 15370. There is nothing special in the graph there. 

Tom, is there a correlation between this pattern and the clicks?

Wouter1, 2015-03-19:
Tom I have a question about the clicks when you use the normal (non-test) driver.

Do you always get a message in the system.log if you have a click? Or do you also there get clicks without messages?

What error messages are associated with clicks? Always READ HICKUP?

When READ HICKUP messages occur, this indicates that the CoreAudio system is jumping around unexpectedly. This is not something directly related to our driver - it can be another process doing some audio, a system process being too slow or late, etc. 

I also notice a very small "oscillation" (overshoot) around the click figure you posted above. 

Maybe it is the OUTPUT that has a click, and not the input?

Wouter1, 2015-03-20:
I think I can reproduce this now at 96k and it seems like an issue with the playback, not with the input.

Wouter1, 2015-03-20:
Checked if input and output streams are still in sync when clicking occurs. See #33 how to do that
Here is a result at 44k

Like we had before at 44k, but exactly half of it

[image]

It looks pretty constant like this

```
Mar 20 22:00:46 vlieland kernel[0]: write wrap in usbframe 17910263 list 44 byte 240
Mar 20 22:00:46 vlieland kernel[0]: input wrap in list 0 at frame 44 byte 120
```

So the write head is 120 bytes behind the input head.

Wouter1, 2015-03-20:
And here at 48k

[image]

Again as before so all OK.

Wouter1, 2015-03-20:
I can reproduce the clicking also at 48k. And the sync is then still OK like in the picture above

TomDrinkwater, 2015-03-20:
Hi, sorry I have been busy, I still am really, I don't have definite answers to your questions unless I test with those questions in mind, and actually listen and watch the console as I test rather than just look at the loopback recording for clicks as I usually do. I will try to test again soon, if what you want me to test for changes just say here...

Wouter1, 2015-03-21:
The main cause of these playback clicks seems to be the buffer size of audacity. If I put it to 10ms, I have hardly any clciks, but if i put it to 100ms then there are many clicks about 1 in 5 playback attempts.

Wouter1, 2015-03-21:
Tom, thanks for your reply.

I'm not sure if I can fix this issue quickly. 

Is the current drive stable enough for practical use - do you have a good enough workaround to make it all ok for you? If not, I think I have to roll back the release version to the point where we both had no issues. That seems to include the whole low-latency attempt.

TomDrinkwater, 2015-03-21:
I think the current driver is fine for what I actually need it for for myself. If I were using the emu as my main interface for serious recording I might have more problem with it.

I'm not sure if there was ever a version with no issues at all.... when there is a problem that doesn't happen very often it can be hard to tell.

so, it depends on the goal, I am very happy that this driver exists and I will use it in its current form. I would rather use the emu with this driver than other interfaces that don't sound as good with a perfect driver.

We could see if other issues or other users with different needs emerge over the next weeks or months.

Wouter1, 2015-03-21:
Ok, good to hear that it is workable and that it does not need a rollback.

Do you agree that this current clicking issue seems to be in the PLAYBACK side? 

To ensure this, my plan is to  generate a sine wave with another source (eg borrow another computer) to see if there is clicking then on the input.

Tom unfortunately I have no 2nd computer right here; is it easy for you to generate a 10 minute 440Hz sine wave on one computer where there's no clicking issue, and then record that sine wave on the computer that has some clicking issue?

Wouter1, 2015-03-29:
I tried to reproduce the clicks with reaper. Standard driver.
64samples DAW, 96k rate, 440Hz sine tone. 10 minute recording, playback of sine tone also from reaper.

Not a single click in 10 minutes.

Wouter1, 2015-03-29:
CHecked again now with 128samples DAW buffer, other settings the same. 
Not a single click in 10 minutes.

Wouter1, 2015-04-09:
Tom reported in #56 for driver 3.5.4:

96k 64 samples 1ms - about 7 clicks in 3 mins

48k 64 samples 1ms - 1 click in 10 mins

Wouter1, 2015-04-09:
@TomDrinkwater 
I found that I also sometimes have issues with clicking

1. Mainly in the first 1 or 2 minutes I have some clicking. This is because the PLL in some cases is locking in a bit slow (eg maybe your EMU and CPU clock are different a bit more than on my machine).

2. In some cases, eg, causing some other app to start playing sound while sampling, it seems coreaudio gives some hickups which can affect the sampling process.

3. At high data rates and small DAW buffer, it seems sometimes the system is reporting "too many interrupts". I'm not sure but this might cause hickups in the sound.

Since I have a really fast machine, it could be that one of these issues appears earlier on your machine. 

Do you have an idea if one of these is actually the case when you have these clicks?

TomDrinkwater, 2015-04-09:
as far as I can tell so far, these are not the case. I will keep them in mind when testing though.

1. I have not seen any overall change in the number of clicks with time. However when there are are only a few clicks then it is hard to tell.

2. There might be some relation to other activity on the computer, but f so then it is slight, I have some clciks when there is no other activity, and some that seem to come after I do something, but it could be coincidence. Often I do something else cpu intensive and there is no click as a result. I haven't specifically tried to make it click by getting another app to play sound.

3. I have been getting clicks with no errors other than the expected emu errors in console, but again I would have to do more to be sure.

Wouter1, 2015-04-09:
OK thanks. Do you have a method that reasonably well can reproduce this? Here it just works fine 99% of the time and that makes it hard to find the source of teh problem

TomDrinkwater, 2015-04-09:
it seems totally reproducible using normal testing. the only thing is that testing at 48k or 44 k is slow since there are very few clisks.

is there a specific thing we are trying to reproduce? it is easy to do a test and get clicks (or not), but it is hard to get enough data to make meaningful correlations with anything...

[44 more replies]

closed 2016-07-22 by Wouter1