Wouter1/EMU-driver #56: reaper rate switching behaves erratically

Wouter1, 2015-03-12:
Since v3.5.1 switching the driver sample rate behaves erratically. On my computer you have no sound till you either go out of reeaper and then switch back, or sometimes you have to cmpletely quit reaper and restart it. Tom is experiencing 'breaks in the sound'.

Wouter1, 2015-03-12:
This may be related to the crashing control panel #52

Wouter1, 2015-03-12:
You can notice that something broke in reaper: when it did, these messages do not appear in the system.log anymore when you switch out of reaper:

```
Mar 12 22:11:39 vlieland kernel[0]: +EMUUSBAudioEngine[0xffffff802f1e9000]::stopUSBStream ()
Mar 12 22:11:39 vlieland kernel[0]: All playback streams stopped
Mar 12 22:11:40 vlieland kernel[0]: -EMUUSBAudioEngine[0xffffff802f1e9000]::stopUSBStream ()
```

This break seems to happen right when you start your first sampling. 

Reaper seems to break in a lighter way when you play back something and then stop and change the rate. It won't play back then but if you switch out of reaper and back into it, you have it working again at the new rate.

Wouter1, 2015-03-12:
I see messages like this in the console when switching rates in reaper

```
Mar 12 22:18:15 vlieland kernel[0]: -EMUUSBAudioEngine[0xffffff802f1e9000]::stopUSBStream ()
Mar 12 22:18:15 vlieland kernel[0]: +EMUUSBAudioEngine::performFormatChange existing  sampleRate is 44100 direction 0
Mar 12 22:18:15 vlieland kernel[0]: fDriverTag = 0x10002
Mar 12 22:18:15 vlieland kernel[0]: ++about to set output : ourInterfaceNumber = 1 & newAlternateSettingID = 2
Mar 12 22:18:15 vlieland kernel[0]: called setNumSampleFramesPerBuffer with 8192
Mar 12 22:18:15 vlieland kernel[0]: newFormat->fNumChannels = 2, newFormat->fBitWidth 24
Mar 12 22:18:15 vlieland kernel[0]: -EMUUSBAudioEngine::performFormatChange, result = 0
Mar 12 22:18:15 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged value 0
Mar 12 22:18:15 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged result 0
Mar 12 22:18:15 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged value 1
Mar 12 22:18:15 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged result 0
Mar 12 22:18:15 vlieland kernel[0]: +EMUUSBAudioEngine[0xffffff802f1e9000]::performAudioEngineStart ()
Mar 12 22:18:15 vlieland kernel[0]: create output pipe 
Mar 12 22:18:15 vlieland kernel[0]: check for associated endpoint
Mar 12 22:18:15 vlieland kernel[0]: frameSizeQueue empty, guessing some queue size. May need fix..
Mar 12 22:18:15 --- last message repeated 127 times ---
```
this is weird, it is changing only the input rate I think , not the output rate.

TomDrinkwater, 2015-03-13:
>Tom is experiencing 'breaks in the sound'. 

sorry, that is a misunderstanding, I often have no sound after switching sample rate, so I said it "breaks the sound" meaning it is broken.

so I have the same situation as you, or very close.

Wouter1, 2015-03-14:
Thanks for the clarification. I thought his was a different issue.

Wouter1, 2015-03-16:
Here is another sample, I see this when switching to 48k to 44k

```
Mar 16 21:17:15 vlieland kernel[0]: **** Output Hiccup!! firstSampleFrame=760, nextExpectedOutputFrame=0 bufsize=49152
Mar 16 21:17:39 vlieland kernel[0]: +EMUUSBAudioEngine[0xffffff803b0c6000]::stopUSBStream ()
Mar 16 21:17:39 vlieland kernel[0]: All playback streams stopped
Mar 16 21:17:40 vlieland kernel[0]: -EMUUSBAudioEngine[0xffffff803b0c6000]::stopUSBStream ()
Mar 16 21:17:40 vlieland kernel[0]: +EMUUSBAudioEngine::performFormatChange existing  sampleRate is 48000 direction 0
Mar 16 21:17:40 vlieland kernel[0]: fDriverTag = 0x10002
Mar 16 21:17:40 vlieland kernel[0]: Changing sampling rate to 44100
Mar 16 21:17:40 vlieland kernel[0]: ++about to set input : ourInterfaceNumber = 2 & newAlternateSettingID = 2
Mar 16 21:17:40 vlieland kernel[0]: averageSampleRate = 44100
Mar 16 21:17:40 vlieland kernel[0]: ++about to set output : ourInterfaceNumber = 1 & newAlternateSettingID = 2
Mar 16 21:17:40 vlieland kernel[0]: called setNumSampleFramesPerBuffer with 8192
Mar 16 21:17:40 vlieland kernel[0]: newFormat->fNumChannels = 2, newFormat->fBitWidth 24
Mar 16 21:17:40 vlieland kernel[0]: -EMUUSBAudioEngine::performFormatChange, result = 0
Mar 16 21:17:40 vlieland kernel[0]: +EMUUSBAudioEngine[0xffffff803b0c6000]::performAudioEngineStart ()
Mar 16 21:17:40 vlieland kernel[0]: create output pipe 
Mar 16 21:17:40 vlieland kernel[0]: check for associated endpoint
Mar 16 21:17:40 vlieland kernel[0]: frameSizeQueue empty, guessing some queue size. May need fix..
Mar 16 21:17:40 --- last message repeated 127 times ---
Mar 16 21:17:40 vlieland kernel[0]: +EMUUSBAudioEngine[0xffffff803b0c6000]::stopUSBStream ()
Mar 16 21:17:40 vlieland kernel[0]: USBF:    102427.604    AppleUSBXHCI[0xffffff81ae26a000]::PollEventRing2 - Isoc Transfer event with completion code (3) on pEP (0xffffff803c68b800)
Mar 16 21:17:40 --- last message repeated 1 time ---
Mar 16 21:17:40 vlieland kernel[0]: All playback streams stopped
Mar 16 21:17:41 vlieland kernel[0]: -EMUUSBAudioEngine[0xffffff803b0c6000]::stopUSBStream ()
Mar 16 21:17:41 vlieland kernel[0]: +EMUUSBAudioEngine::performFormatChange existing  sampleRate is 44100 direction 0
Mar 16 21:17:41 vlieland kernel[0]: fDriverTag = 0x10002
Mar 16 21:17:41 vlieland kernel[0]: ++about to set output : ourInterfaceNumber = 1 & newAlternateSettingID = 2
Mar 16 21:17:41 vlieland kernel[0]: called setNumSampleFramesPerBuffer with 8192
Mar 16 21:17:41 vlieland kernel[0]: newFormat->fNumChannels = 2, newFormat->fBitWidth 24
Mar 16 21:17:41 vlieland kernel[0]: -EMUUSBAudioEngine::performFormatChange, result = 0
Mar 16 21:17:41 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged value 0
Mar 16 21:17:41 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged result 0
Mar 16 21:17:41 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged value 1
Mar 16 21:17:41 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged result 0
Mar 16 21:17:41 vlieland kernel[0]: +EMUUSBAudioEngine[0xffffff803b0c6000]::performAudioEngineStart ()
Mar 16 21:17:41 vlieland kernel[0]: create output pipe 
Mar 16 21:17:41 vlieland kernel[0]: check for associated endpoint
Mar 16 21:17:41 vlieland kernel[0]: frameSizeQueue empty, guessing some queue size. May need fix..
Mar 16 21:17:41 --- last message repeated 127 times ---
Mar 16 21:17:41 vlieland kernel[0]: warning way-out ring wrap position
```

I see swice a call to performFormatChange! And there is a call to performAudioStart, immediately followed by stopUSBStream and then the 2nd performFormatChange?  Both are for direction 0.

Wouter1, 2015-03-16:
1st call to performFormatChange changes the output rate. The 2nd call changes the input rate.

What may be happening is that reaper sets both input and output rates separately, shortly after another.

Wouter1, 2015-03-16:
It seems that the driver changes only 1 stream rate (eg, output stream) at a time, then restarts the engine (if we're in Reaper), then gets the request to change the other (eg, input) stream, stops the engine, etc.

Wouter1, 2015-03-17:
Complared with switching the rate of the EMU using the control panel while playing with Quicktime Player 7.
That works OK. 

The difference between the two is that Reaper does a second round of 'new format'.  (supposedly through setFormat ). Here is the full dump:
```
--- COMMON REAPER AND CONTROLPANEL+QUICKTIME---

Mar 17 21:24:12 vlieland kernel[0]: +EMUUSBAudioEngine[0xffffff803153e800]::performAudioEngineStop ()
Mar 17 21:24:12 vlieland kernel[0]: +EMUUSBAudioEngine[0xffffff803153e800]::stopUSBStream ()
Mar 17 21:24:12 vlieland kernel[0]: +EMUUSBInputStream::stop
Mar 17 21:24:12 vlieland kernel[0]: EMUUSBOutputStream::stop
Mar 17 21:24:12 vlieland kernel[0]: writeHandler: now stopped 1 of 2
Mar 17 21:24:12 vlieland kernel[0]: writeHandler: now stopped 2 of 2
Mar 17 21:24:12 vlieland kernel[0]: All playback streams stopped
Mar 17 21:24:12 vlieland kernel[0]: +EMUUSBAudioEngine::OurUSBOutputStream::notifyClosed.
Mar 17 21:24:12 vlieland kernel[0]: EMUUSBInputStream::readCompleted all input streams stopped
Mar 17 21:24:12 vlieland kernel[0]: +EMUUSBAudioEngine::OurUSBInputStream::notifyClosed.
Mar 17 21:24:13 vlieland kernel[0]: -EMUUSBAudioEngine[0xffffff803153e800]::stopUSBStream ()
Mar 17 21:24:13 vlieland kernel[0]: -EMUUSBAudioEngine[0xffffff803153e800]::performAudioEngineStop()
Mar 17 21:24:13 vlieland kernel[0]: new format from OUTPUT
Mar 17 21:24:13 vlieland kernel[0]: +EMUUSBAudioEngine::performFormatChange existing  sampleRate is 48000 direction 0
Mar 17 21:24:13 vlieland kernel[0]: newFormat->fBitDepth 24, newFormat->fNumChannels 4
Mar 17 21:24:13 vlieland kernel[0]: +performFormatChangeInternal
Mar 17 21:24:13 vlieland kernel[0]: fDriverTag = 0x1000c
Mar 17 21:24:13 vlieland kernel[0]: Changing sampling rate to 44100
Mar 17 21:24:13 vlieland kernel[0]: ++about to set input : ourInterfaceNumber = 2 & newAlternateSettingID = 12
Mar 17 21:24:13 vlieland kernel[0]: averageSampleRate = 44100
Mar 17 21:24:13 vlieland kernel[0]: alternateFrameSize is 280
Mar 17 21:24:13 vlieland kernel[0]: maxFrameSize 304 alternateFrameSize 280
Mar 17 21:24:13 vlieland kernel[0]: ++about to set output : ourInterfaceNumber = 1 & newAlternateSettingID = 12
Mar 17 21:24:13 vlieland kernel[0]: alternateFrameSize is 280
Mar 17 21:24:13 vlieland kernel[0]: maxFrameSize 304 alternateFrameSize 280
Mar 17 21:24:13 vlieland kernel[0]: EMUUSBAudioConfigObject::GetEndpointPollInterval 3
Mar 17 21:24:13 vlieland kernel[0]: initBuffers mPollInterval=4
Mar 17 21:24:13 vlieland kernel[0]: inputSize= 280 multFactor= 12
Mar 17 21:24:13 vlieland kernel[0]: outputSize= 280 multFactor= 12
Mar 17 21:24:13 vlieland kernel[0]: releasing the input sampleBufferMemory descriptor
Mar 17 21:24:13 vlieland kernel[0]: new bufferSize = 98304 numSamplesInBuffer = 8192
Mar 17 21:24:13 vlieland kernel[0]: disposing the mUSBBufferDescriptor input
Mar 17 21:24:13 vlieland kernel[0]: initBuffers numUSBFrameLists 4
Mar 17 21:24:13 vlieland kernel[0]: using plist value SafetyOffsetMicroSec = 4200
Mar 17 21:24:13 vlieland kernel[0]: sample offset 185 samples
Mar 17 21:24:13 vlieland kernel[0]: disposing the output mUSBBufferDescriptor
Mar 17 21:24:13 vlieland kernel[0]: In the out path, making new buffer with size of 98304 numSamplesInBuffer 8192
Mar 17 21:24:13 vlieland kernel[0]: using plist value latencyDAC = 38
Mar 17 21:24:13 vlieland kernel[0]: using plist value latencyADC = 53
Mar 17 21:24:13 vlieland kernel[0]: completed initBuffers
Mar 17 21:24:13 vlieland kernel[0]: EMUUSBAudioEngine::SetSampleRate 44100
Mar 17 21:24:13 vlieland kernel[0]: using SampleRateXU
Mar 17 21:24:13 vlieland kernel[0]: EMUUSBAudioDevice::deviceRequest bmRequestType=21 bRequest=1 wValue=300 wIndex=c00 wLength=1
Mar 17 21:24:13 vlieland kernel[0]: EMUUSBAudioDevice::deviceRequest result=0
Mar 17 21:24:13 vlieland kernel[0]: ++EMUUSBAudioDevice[0xffffff803463d400]::deviceRequest(0xffffff81faa3b7c8, 0) = 0
Mar 17 21:24:13 vlieland kernel[0]: set result  = 0
Mar 17 21:24:13 vlieland kernel[0]: +EMUUSBAudioDevice::setOtherEngineSampleRate 44100; there are 1 engines
Mar 17 21:24:13 vlieland kernel[0]: -EMUUSBAudioDevice::setOtherEngineSampleRate
Mar 17 21:24:13 vlieland kernel[0]: called setNumSampleFramesPerBuffer with 8192
Mar 17 21:24:13 vlieland kernel[0]: newFormat->fNumChannels = 4, newFormat->fBitWidth 24
Mar 17 21:24:13 vlieland kernel[0]: -performFormatChangeInternal
Mar 17 21:24:13 vlieland kernel[0]: -EMUUSBAudioEngine::performFormatChange, result = 0
Mar 17 21:24:13 vlieland kernel[0]: +EMUUSBAudioEngine[0xffffff803153e800]::performAudioEngineStart ()
Mar 17 21:24:13 vlieland kernel[0]: about to start USB stream(s)
Mar 17 21:24:13 vlieland kernel[0]: averageFrameSamples=22
Mar 17 21:24:13 --- last message repeated 1 time ---
Mar 17 21:24:13 vlieland kernel[0]: Isoc Frames / usbCompletions
Mar 17 21:24:13 vlieland kernel[0]: EMUUSBAudioEngine::SetSampleRate 44100
Mar 17 21:24:13 vlieland kernel[0]: using SampleRateXU
Mar 17 21:24:13 vlieland kernel[0]: createInputPipe
Mar 17 21:24:13 vlieland kernel[0]: +UsbInputRing::init bytesize=98304 byterate=529200
Mar 17 21:24:13 vlieland kernel[0]: -UsbInputRing::init 185759637
Mar 17 21:24:13 vlieland kernel[0]: create output pipe 
[... 84 lines cut ...]
Mar 17 21:24:14 vlieland kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Mar 17 21:24:14 vlieland kernel[0]: UsbInputRing::notifyWrap 1
Mar 17 21:24:14 vlieland kernel[0]: warning way-out ring wrap position
[... cut ...]
Mar 17 21:24:16 vlieland kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Mar 17 21:24:16 vlieland kernel[0]: **** Output Hiccup!! firstSampleFrame=564, nextExpectedOutputFrame=400 bufsize=98304
Mar 17 21:24:16 vlieland kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Mar 17 21:24:16 --- last message repeated 2 times ---
Mar 17 21:24:16 vlieland kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Mar 17 21:24:16 vlieland kernel[0]: **** Output Hiccup!! firstSampleFrame=1620, nextExpectedOutputFrame=1076 bufsize=98304
```

Wouter1, 2015-03-29:
I was not checking&reporting errors if the output stream does not start up properly. Added check.

Wouter1, 2015-03-29:
That did not reveal any issue. So the startup is not failing

Wouter1, 2015-03-29:
When the audio breaks, the calls to writeCompleted still arrive properly. So we are writing data to the EMU and it arrives there.

Wouter1, 2015-03-29:
Version 3.4.0 did not have this problem.
In the output to the console.log, the only difference I see between current and 3.4.0  is the number of channels - we seem to  get 4 channels  with the new driver.

Wouter1, 2015-03-29:
Tested that pollInterval=4 modes do not cause this issue - pollInterval=8 modes also don't work anymore after switching the rate.

Wouter1, 2015-03-29:
I tested recording with reaper after switching the rate, using a different source for the sound (not the EMU). Recording works fine also after changing the rate. 

So it is definitely the OUTPUT ONLY that is not working.

Wouter1, 2015-03-29:
clipOutputSamples is being called also when the sound is not working anymore.

Wouter1, 2015-03-29:
The clipOutputSamples call contains audio data (it's not containing only zeros).

Wouter1, 2015-04-03:
Checked that the frameSizeQueue is cleaned at start. It is, in init() which is called from startUSBStream()

Wouter1, 2015-04-03:
If I add a check to return immediately from performatChange if the rate has not changed, it still does not work  from reaper. 

It seems the audio breaks just from the quick cycling of stop, start, stop, start.

The rattling noises I sometimes get suggest that the problem is related to the erase head.

Wouter1, 2015-04-03:
Confirmed, turning off the erase head solves this issue (but of course that would cause other problems).

This all suggests we cause a problem in  CoreAudio when quick start/stop cycling 

Maybe CoreAudio expects that we keep a sync lock while switching rate??

Wouter1, 2015-04-06:
We are losing sync after the rate is switched from reaper

This is a proper startup
```
Apr  6 19:49:44 vlieland kernel[0]: ++EMUUSBAudioEngine[0xffffff804d382800]::performEngineStart result is 0x0 direction 1
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 --- last message repeated 2 times ---
Apr  6 19:49:44 vlieland kernel[0]: unitID 14
Apr  6 19:49:44 vlieland kernel[0]: mDigitalIOXU
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 vlieland kernel[0]: +EMUUSBAudioDevice[0xffffff804ea2a400]::queryXU
Apr  6 19:49:44 vlieland kernel[0]: DigitalIOStatus changed
Apr  6 19:49:44 vlieland kernel[0]: Digital IO SampleRate
Apr  6 19:49:44 vlieland kernel[0]: EMUUSBAudioDevice::deviceRequest result=0
Apr  6 19:49:44 vlieland kernel[0]: ++EMUUSBAudioDevice[0xffffff804ea2a400]::deviceRequest(0xffffff8209bcbd48, 0) = 0
Apr  6 19:49:44 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged value 0
Apr  6 19:49:44 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged result 0
Apr  6 19:49:44 vlieland kernel[0]: DigitalIO Sync Lock
Apr  6 19:49:44 vlieland kernel[0]: EMUUSBAudioDevice::deviceRequest result=0
Apr  6 19:49:44 vlieland kernel[0]: ++EMUUSBAudioDevice[0xffffff804ea2a400]::deviceRequest(0xffffff8209bcbd48, 0) = 0
Apr  6 19:49:44 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged value 1
Apr  6 19:49:44 vlieland kernel[0]: EMUXUCustomControl::hardwareValueChanged result 0
Apr  6 19:49:44 vlieland kernel[0]: -EMUUSBAudioDevice[0xffffff804ea2a400]::QueryXU
Apr  6 19:49:44 vlieland kernel[0]: USBF:    45111. 83    AppleUSBXHCI[0xffffff81be73b000]::PollEventRing2 - Isoc Transfer event with completion code (3) on pEP (0xffffff804d324000)
Apr  6 19:49:44 --- last message repeated 1 time ---
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 vlieland kernel[0]: frameSizeQueue empty, guessing some queue size. May need fix..
Apr  6 19:49:44 --- last message repeated 63 times ---
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 --- last message repeated 1 time ---
Apr  6 19:49:44 vlieland kernel[0]: write wrap in usbframe 1996184 list 21 byte 36
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 vlieland kernel[0]: input wrap in list 1 at frame 21 byte 36
Apr  6 19:49:44 vlieland kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Apr  6 19:49:44 vlieland kernel[0]: UsbInputRing::notifyWrap 0
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 --- last message repeated 2 times ---
Apr  6 19:49:44 vlieland kernel[0]: write wrap in usbframe 1996344 list 42 byte 84
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 vlieland kernel[0]: input wrap in list 2 at frame 42 byte 84
Apr  6 19:49:44 vlieland kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Apr  6 19:49:44 vlieland kernel[0]: UsbInputRing::notifyWrap 1
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 --- last message repeated 1 time ---
Apr  6 19:49:44 vlieland kernel[0]: write wrap in usbframe 1996504 list 63 byte 132
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 --- last message repeated 1 time ---
Apr  6 19:49:44 vlieland kernel[0]: input wrap in list 3 at frame 63 byte 132
Apr  6 19:49:44 vlieland kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Apr  6 19:49:44 vlieland kernel[0]: UsbInputRing::notifyWrap 2
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 --- last message repeated 2 times ---
Apr  6 19:49:44 vlieland kernel[0]: write wrap in usbframe 1996696 list 21 byte 36
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 vlieland kernel[0]: input wrap in list 1 at frame 21 byte 36
Apr  6 19:49:44 vlieland kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Apr  6 19:49:44 vlieland kernel[0]: UsbInputRing::notifyWrap 3
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:44 --- last message repeated 2 times ---
Apr  6 19:49:44 vlieland kernel[0]: write wrap in usbframe 1996856 list 42 byte 84
Apr  6 19:49:44 vlieland kernel[0]: warning way-out ring wrap position
Apr  6 19:49:45 vlieland kernel[0]: input wrap in list 2 at frame 42 byte 84
Apr  6 19:49:45 vlieland kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Apr  6 19:49:45 vlieland kernel[0]: UsbInputRing::notifyWrap 4
Apr  6 19:49:45 vlieland kernel[0]: LowPassFilter::filter init 170666666
Apr  6 19:49:45 vlieland kernel[0]: USB timer started
Apr  6 19:49:45 vlieland kernel[0]: write wrap in usbframe 1997016 list 63 byte 132
Apr  6 19:49:45 vlieland kernel[0]: input wrap in list 3 at frame 63 byte 132
Apr  6 19:49:45 vlieland kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Apr  6 19:49:45 vlieland kernel[0]: write wrap in usbframe 1997208 list 21 byte 36
[... 110 lines cut ...]
Apr  6 19:50:55 vlieland kernel[0]: input wrap in list 2 at frame 43 byte 30
Apr  6 19:50:55 vlieland kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Apr  6 19:50:55 vlieland kernel[0]: write wrap in usbframe 2067540 list 55 byte 36
[... cut ...]
Apr  6 19:50:56 vlieland kernel[0]: input wrap in list 3 at frame 44 byte 120
Apr  6 19:50:56 vlieland kernel[0]: write wrap in usbframe 2068468 list 56 byte 126
Apr  6 19:50:56 vlieland kernel[0]: input wrap in list 1 at frame 32 byte 66
Apr  6 19:50:56 vlieland kernel[0]: write wrap in usbframe 2068660 list 44 byte 72
Apr  6 19:50:56 vlieland kernel[0]: input wrap in list 3 at frame 20 byte 6
Apr  6 19:50:56 vlieland kernel[0]: write wrap in usbframe 2068852 list 32 byte 12
Apr  6 19:50:57 vlieland kernel[0]: input wrap in list 1 at frame 7 byte 72
```

Wouter1, 2015-04-06:
Checked that the inputstream rinngbuffer is at position 0 when the sampling starts. Seems OK

Wouter1, 2015-04-06:
frameIndex is not set to 0 in start! The reading of the input stream was starting at the wrong point and therefore the first wrap came at a wrong position causing timing mechanisms to resync and probably this was confusing the coreaudio.

Wouter1, 2015-04-06:
First tests look sound OK. This missed init call seems so essential that I committed immediately to version 3.5.4. Thorough testing will take more time

Wouter1, 2015-04-06:
Checked that the reported latency at 96k is correct with default driver.

Wouter1, 2015-04-07:
testing default driver 4.2ms.

full playback-record cycle from reaper. Reaper block size 64. Playing 10minute 440Hz sine (generated from audacity) from reaper. Recording to new track. Right click on recorded track and select 'render item as new take'. The track is then rendered as wav in the Documents/Reaper media directory.

Using audacity on that file to do 440Hz notch filter. Checking for clicks

|rate|result|
|---|---|
|44k|ok|
| 48k (oops reaper buf=128) | ok |
|96k| ok |
|192k| a few clicks around 9 minutes |
 
At 192k I also see very low level noise spikes (around -50dB). They are 1 sample wide and occur sometimes many times per second, it sounds and looks like cosmic radiation pulses.

Wouter1, 2015-04-07:
@TomDrinkwater
TomDrinkwater you may like this new version as it fixes this issue with switching rates in reaper.

Wouter1, 2015-04-07:
Still need to test with 1ms driver  setting

TomDrinkwater, 2015-04-07:
>TomDrinkwater you may like this new version as it fixes this issue with switching rates in reaper.

that's great, thanks for all your work!

sorry I have been out of action, I will try this new version tomorrow.

Wouter1, 2015-04-08:
CHecked that the reported latency at 44.1 is reported ok with 1ms driver. Reaper reports 4.4/4.1 and actual roundtrip latency 8.48ms.

Wouter1, 2015-04-08:
testing driver with 1ms setting.

|rate|result|
|---|---|
|44k|15 clicks first 2 minutes |

Wouter1, 2015-04-08:
@TomDrinkwater  good to hear you are still around. I'm particularly interested if this fix also fixes #55.

TomDrinkwater, 2015-04-08:
what is the version number for this driver? is it 3.5.4? I tested it but I still have clicks at 96k 1ms plist setting. also sometimes have no sound after sample rate change, but sometimes it works...

TomDrinkwater, 2015-04-08:
96k 64 samples 1ms - about 7 clicks in 3 mins

48k 64 samples 1ms - 1 click in 10 mins

sample rate changing is now behaving correctly. but at first I had the same bug with no sound....

it seems to behave better if the control panel is not running, but I haven't tested that enough to be sure.

Wouter1, 2015-04-09:
Thanks Tom for testing! 
Yes 3.5.4 is the current version.

So I understand that the rate switching is now working but you still have clicking?

Then I can close this ticket but need to look more into #55

closed 2015-04-09 by Wouter1

TomDrinkwater, 2015-04-09:
>So I understand that the rate switching is now working but you still have clicking?

at first with the new driver I had the same problems with sample rate switching. but after a while it got better. I'm not sure why! I will have to see if the problem returns to be sure.