Wouter1/EMU-driver #64: The driver can not be loaded in OS X El Capitan: "System extention cannot be used"

vitalibondar, 2015-10-02:
My English is bad, so I have reproduced this problem in the video:
https://cloudup.com/cugKp1TU9d7
Before installing I opened a terminal and run the following command:
 ```sudo nvram boot-args="kext-dev-mode=1"```
Then I rebooted my machine.
[image]
Maybe, your driver not ready yet for El Capitan? I must roll back to Yosemite?

Wouter1, 2015-10-02:
El captain is still in beta mode. This may be a bug in El Captain.

In your video the message says 

>/System/Library/Extensions/EMUUSBAudio.kext failed to load- (libkern/kext) not loadable (reason unspecified); check the system/kernel logs for errors or try kextutil(8).

Can you please post the messages in the logs ? Maybe that gives more details.

Wouter1, 2015-10-02:
Hjalte reported the same issue.

The website of el captain is this

https://www.apple.com/osx/elcapitan-preview/

which is why I assumed El Captain is still in beta mode. 

But it also says "Coming September 30" while it's already october 2. So apple's info on this is confusing.

blusprite, 2015-10-02:
This is the actual El Capitan page: https://www.apple.com/osx/

Wouter1, 2015-10-02:
Maybe someone using ElCaptain can try if this works

```sudo nvram boot-args="rootless=0";sudo reboot```

Wouter1, 2015-10-02:
>This is the actual El Capitan page: https://www.apple.com/osx/

Ah thanks, apparently apple has currently TWO pages on El Captain, one indicating it's in beta and one that it is current.

vitalibondar, 2015-10-02:
I installed the release of El Capitan (non-beta), updated Yosemite.
> ```sudo nvram boot-args="rootless=0";sudo reboot```

So, after this.
Output

    Installing EMU driver
    plz to turn off your EMU USB device for loading of kext, you have 3 seconds to do so
    (kernel) Kext com.emu.driver.EMUUSBAudio not found for unload request.
    Failed to unload com.emu.driver.EMUUSBAudio - (libkern/kext) not found.
    deleting old kext
    installing new kernel extension
    deleting old midi plugin
    installing new midi plugin
    doing permissions stuff
    loading new kext
    Requesting load of /System/Library/Extensions/EMUUSBAudio.kext./System/Library/Extensions/EMUUSBAudio.kext failed to load - (libkern/kext) not loadable (reason unspecified); check the system/kernel logs for errors or try kextutil(8). 
    so long

Kernel log message:

    Oct 2, 2015, 23:29:14 kernel[0]: Kext com.emu.driver.EMUUSBAudio not found for unload request.
    Oct 2, 2015, 23:29:20 kernel[0]: Resetting IOCatalogue.

Wouter1, 2015-10-02:
Seems it has been loaded. 
The actual loading message (in Yosemite) usually comes a minute or so later. Did you see that message in the kernel log?

Can you confirm that it  works?

Wouter1, 2015-10-02:
So still not working. Did you see more messages in the kernel log apart from the "not found for unload"? 

Not loadable - I expect some message complaining about the extension not being signed.

vitalibondar, 2015-10-02:
http://pastebin.com/6DTEJbBC

    Oct 2, 2015, 23:28:04 com.apple.kextd[45]: Untrusted kexts are not allowed
    Oct 2, 2015, 23:28:04 com.apple.kextd[45]: ERROR: invalid signature for com.emu.driver.EMUUSBAudio, will not load

Wouter1, 2015-10-02:
Thanks. That confirms it's indeed Apple's increased security policy in El Capitan making this problem.

Wouter1, 2015-10-02:
@vitalibondar maybe apple has already disabled the rootless=0 option too.

I read on https://pikeralpha.wordpress.com/2015/07/28/apples-kext-signing-bypassed/ that there are more possibilities. Maybe you can also have a look there.

One is to run the Security Configurator utility and disable the System Integrity Protection.

Can you try that?

Wouter1, 2015-10-02:
@vitalbonder

I digged further and found this
https://forums.developer.apple.com/thread/12499

>The kext-dev-mode and rootless boot-args are being removed from OS X El Capitan and will no longer work.

>The supported way to disable System Integrity Protection in those cases where it's truly necessary is to boot into the Recovery partition and turn System Integrity Protection off from there with the csrutil tool.

>you need to disable SIP by booting into recovery mode (cmd+R):

>Utilities > Security Configuration > uncheck "Enforce System Integrity Protection"

Does that work?

Wouter1, 2015-10-02:
In other posts I read that after the cmd-R boot you need to enter this command

```csrutil disable```

vitalibondar, 2015-10-02:
It's okay, I just went to sleep :) but returned.
Well, now the driver works, but after turning on EMU I see an error message from the post 1.
[image]
But the recorded sound is distorted (Reaper, Logic, Quicktime).
https://cloudup.com/ihcV-dV6lWL

vitalibondar, 2015-10-02:
Playback (iTunes, any DAW) ok, despite the error.

djohayn, 2015-10-02:
Used the 'csrutil disable' command and it worked perfectly!

vitalibondar, 2015-10-02:
recovery mode > terminal > csrutil disable

[image]

@djohayn Recording?

djohayn, 2015-10-02:
sorry saying it works perfectly might be an overstatement. I only use my mac setup for playback. So that works smoothly without any error messages or anything. Don't know how the recording might work. But it looks fine when I checkout the input in system pref.

I went to recovery mode holding cmd+r during boot and then put the csrutil disable command in the terminal there.

Wouter1, 2015-10-03:
Thanks for all the testing. So if i wrap up, in El Capitano, the command ```sudo nvram boot-args="kext-dev-mode=1"``` has to be replaced with boot in safe mode (cmd-R) and then enter ```csrutil disable```

However, recording does not work properly. I will make a new ticket for that.

Wouter1, 2015-10-03:
Added instructions in the readme for enabling developer kext on El Capitano. 
Made new ticket for the recording issues.

closed 2015-10-03 by Wouter1

kornerious, 2015-10-20:
Cracks in recording, even with large buffer size(

Wouter1, 2015-10-20:
kornerious, recording issue is known #65.

kornerious, 2015-10-20:
oh. can you tell when will be some news?)

kornerious, 2015-10-20:
2. I am using your driver in Yosemite. Wanted to load low latency but up to
512 samples i got cracks( is it normal?
3. Your normal driver shows 10 msec latency, but with 32 buffer i get some
cracks also (from 64 is all right) - is it also normal?

i5 2500+8gb DDR

2015-10-21 0:43 GMT+03:00 ALexandr Kiliyevich <kornerious@gmail.com>:

Wouter1, 2015-10-21:
@kornerious, this ticket has been closed already 2 weeks ago and it's neither about Yosemite nor about cracks. Please post in the correct ticket or make a new one. I made one for you #73.