Wouter1/EMU-driver #65: El Capitano recording issues

Wouter1, 2015-10-03:
In #64 @vitalibondar reported

Recorded sound is distorted (Reaper, Logic, Quicktime).
https://cloudup.com/ihcV-dV6lWL

Wouter1, 2015-10-03:
@vitalibondar do you see any problem reports in the log files while recording?

vitalibondar, 2015-10-03:
@Wouter1 Yes. http://pastebin.com/gyAkjkNp

And another video with clean driver installation (update after installation native original driver). I see error after while: https://cloudup.com/iV7M_CWCNpU

[image]

Just for information: original EMU driver installed without errors, but don't work (no playback, no recording, no messages in kernel log).

vitalibondar, 2015-10-03:
Many users are reporting problems with different usb-audio interfaces in El Capitan

Wouter1, 2015-10-03:
ok thanks vitalbondar for the log contents.

Basically the log contains this
```
Oct 3, 2015, 12:13:42 kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples err reading ring: e00002e7
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples err reading ring: e00002e7
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples err reading ring: e00002e7
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples err reading ring: e00002e7
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples err reading ring: e00002e7
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples err reading ring: e00002e7
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Oct 3, 2015, 12:13:42 kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
```

Wouter1, 2015-10-03:
It seems that the driver works fine for some but not for others. Normally, recording has to work first, after that playback (which is sync'ed to the recording channel).

@djohayn you reported playback works fine on your machine for playback. What machine  do you have (hardware)?

@vitalibondar what is your machine?

Wouter1, 2015-10-03:
0x2e7 = data underrun. We are not receiving the expected amount of data from the EMU.

@vitalibondar do you have the EMU directly connected to your computer or is there a hub in between?

Wouter1, 2015-10-03:
>Just for information: original EMU driver installed without errors, but don't work (no playback, no recording, no messages in kernel log).

Thanks.

Not sure what to make of that. But it suggests that something important in the USB or audio system has changed. 

The original driver has the original code signature from creative so that explains why it loads without the extra hacks.

Wouter1, 2015-10-03:
My first impression is that the USB system calls back my code with a signal that the data is ready, but that the data is not there.

Wouter1, 2015-10-03:
@vitalibondar I opened your linked audio sample  https://cloudup.com/ihcV-dV6lWL with audacity and I see you have four channels sampled.

Can you try what it does if you sample only 2 channels, eg with audacity?

Wouter1, 2015-10-03:
Checked the code that generates the reported error messages.

This happens when CoreAudio (the mac kernel) calls our code to fetch audio samples but too early, when the samples are not available. 

This can mean that the coreaudio is out of sync, or that there is not enough data coming in from the USB. Or maybe something does not initialize properly.

vitalibondar, 2015-10-03:
@Wouter1 3 files: mono (reaper), stereo (audacity), 4 channels (quicktime) https://cloudup.com/cAPCLc1UF-Y
Something else: after turning on/turning off EMU, driver cannot loaded. I must reinstall it.

vitalibondar, 2015-10-03:
@Wouter1 I have MacBook Pro (Retina, 13-inch, Late 2013) 8 GB RAM.

Wouter1, 2015-10-03:
Thanks for the update. This should all work fine on a late 2013 rMBP, I even can use the rMBP with a hub between the compu and the EMU.

 I hear 'one two' very soft, with loads of echo in the reaper mono. 
The stereo has just loads of clicking. 
The 4 channels has same 'one two', again very soft and only on channel 1. 

There are repeated blocks of silense in the recording, and duplicate blocks of data.

Wouter1, 2015-10-03:
As you mentioned, there seem lots of troubles with audio in El Capitan.

Maybe we should wait a bit and see how this develops after the first update of El Capitan?

vitalibondar, 2015-10-04:
@Wouter1 I agree, that it makes sense.

Wouter1, 2015-10-05:
EmptyFruit also has a rMBP, just a bit newer. 

Some things have changed on the audio side : dual microphones, and support for digital out through the headphone port (I suppose). But on first thought, that would only complicate matters on the NEW rMBP, not on the old one where the drivers were written for. 

I will wait till El Capitan 2 is out before digging deeper into this.

Wouter1, 2015-10-05:
@vitalibondar just a small shot in the dark, but can you confirm that your /System/Library/Extensions/EMUUSBAudio.kext (and all files inside) have mode 755 (drwxr-xr-x), and are owned by "root", group "wheel"?

Wouter1, 2015-10-05:
And, what does this give you
```
kextutil -tn /System/Library/Extensions/EMUUSBAudio.kext/
```

vitalibondar, 2015-10-05:
> @vitalibondar just a small shot in the dark, but can you confirm that your /System/Library/Extensions/EMUUSBAudio.kext (and all files inside) have mode 755 (drwxr-xr-x), and are owned by "root", group "wheel"?

@Wouter1 Sorry, but i can't check this. I have installed Yosemite.

Wouter1, 2015-10-05:
ok thanks, I hoped you had El Capitan installed on a separate partition.

vitalibondar, 2015-10-05:
@Wouter1 I'm sorry, but I just updated my computer when El Capitan has appeared in App Store.

Philipondio, 2015-10-10:
Hi Wouter & Vitali,
so I'm wondering: If I don't use the interface to record, but solely to listen to music, will that work on El Capitan?
I have the Emu Tracker Pre USB btw., and no problems with my other interfaces (RME Babyface, Geekout 1000).
Thanks! Phil

Wouter1, 2015-10-10:
Thanks for your interest in my driver.

I received mixed reports about how it works in El Capitan. For most users it seems fine (there were 135 visitors last week  and I assume most were here to download it). Vitalibondar however never got it working and had to go back to Yosemite. 

I did not yet switch to El Capitan (I wait at least till version 2 I think) so I have no first hand feedback.

If you try, can you please report back here if it works for you.

Wouter1, 2015-10-11:
I got a thank-you message from Joachim from Germany for a nicely working EMU tracker pre under mac os el capitan.

Just mentioning as this is another indication that for most users the driver still works ok.

digdelu, 2015-10-13:
Record audio (EMU-0404 USB) not work on OS X El Capitan. 
Direct Monitor works fine!
Midi IN/OUT without complaints!
S/PDIF not tested.

 MacBook Pro Retina, 15-inch, Mid 2014

Wouter1, 2015-10-13:
@digdelu  thanks for reporting! You're also the first to report that midi is working, great to hear that.

Wouter1, 2015-10-15:
I created an external drive with a comletely fresh install of El Capitan to test this issue.

Unfortunately El Capitan was completely crashing every time I plugged in the network cable. This ate up all my time. Filed a bug report with Apple 23132686

Emptyfruit, 2015-10-22:
Got time to test recording with El Capitan. More precisely:
El Capitan 10.10.11.
MBPr mid 2015, Logic x.
kext signing - disabled.

Midi: works fine.
playback: woks fine in system (music, web, etc.), seems to work fine in DAW, but get some instability. had to restart the device couple of times, especially after settings change.
recording: sound is distorted. It is not clicking, it is distortion of all input audio signal.

More interesting things: i wanted to completely remove the driver (as is was installed before updating to 11.1) and install it again, but permissions to execute commands, provided in Uninstall guide are denied. Even with SIP fully disabled.

Wouter1, 2015-10-22:
@Emptyfruit 
Yes I'm aware of the recording issues. Im doing more detailed work on it in #70 but so far I could not find the source of the issue. 

Playback might become unstable if there are excessive amount of logs while playing. I noticed this particularly with Quicktime latest version. It seems a bug in Quicktime. Can you check your system.log during playback to see if that is the cause of the instability.

Regarding the uninstalling: thanks for reporting. I think  that you need to put SUDO in front of them. Normal users have no permissions there. I will update the uninstall docu page.

Can you retry with the fixed commands?

Emptyfruit, 2015-10-22:
@Wouter1 Yeap, SUDO helped. I reinstalled driver. recording issue is the same, sadly. Midi worked only after rebooting.

Wouter1, 2015-10-22:
Yes recording in El Capitan is broken, rebooting wont help. I will put that on the main page for now..

Emptyfruit, 2015-10-22:
Do you think there's hope? :)

Wouter1, 2015-10-22:
If you need the driver now, stick with Yosemite or Mavericks. 

I only have so an hour or so in some of the evenings, and only 1 machine so debugging is a pain. The driver was never known for its clean code either (it's a fixed relic from Creative)... Also I need some time to answer requests from users and update pages.

Main work on the issue is reported in #70.

Emptyfruit, 2015-10-22:
I was new to OS X. Updated to 10.11 without backing up.. there is no going back for me :( I wish you luck. And more free time :)

Wouter1, 2015-10-22:
I think they have to fix at least 2 fatal issues with Capitan before I can continue working on this.

First, it hangs up if you plug in the network cable radar issue 23223881

Second, it hangs randomly way too often. radar issue 23132686

Got no feedback from Apple on it. I have already lost 3 evenings just filing bug reports and waiting for systems to come alive again and for the moment I'm done with El Capitan.

Wouter1, 2015-10-30:
Just an update for those hoping that I can work on the issue.

I saw that Apple released El Capitan update 10.11.1

I tried to install it on an external USB drive, but the updater crashed/hang up. After that, El Capitan did not boot at all anymore. 

I tried to re install El Capitan on my USB drive. But the installer crashes at 16%

I did a long (half an hour) hardware test of my compu in Mavericks. It ran for half an hour without any reported problem. 

I'm glad that I did this all on the external USB drive, but that's about all the news I have so far. I'm waiting for Apple to respond on the issues I reported (radar  23132686, 23223881, 23339590)

adrian-polglase, 2015-11-09:
I would be so happy if this could be fixed, because I am experiencing the same problem in El Capitan - it simply is impossible to record without the audio being broken and heavily distorted - basically unrecognisable.

Wouter1, 2015-11-10:
Sorry, El Capitan does not even install on my computer since their latest update. Apple has to do its job before I can do anything. I have several bug reports on this open with Apple.

Wouter1, 2015-11-23:
Update: Finally I managed to get Capitan installed on my machine. I think I can make some time to have a look at the issues.

adrian-polglase, 2015-11-23:
This would be amazing. Thank you Wouter1!

Emptyfruit, 2015-11-23:
I wish you luck, man :)

Wouter1, 2015-11-29:
@adrian-polglase 
@Emptyfruit 

The source of the issue has been localized as broken low-latency support in Apple's USB driver.

That allowed me to make a high latency workaround:

if you set the driver latency to 35ms  and the DAW to 5ms then it seems to work.

The driver latency can be set to 35 ms as follows

1. Download a fresh installer from GIT
2. Edit the lowLatency script in your favourite text editor
3. Change the 1000 into 35000
4. save the script
5. run the lowLatency script
6. run the kextInstall script
7. wait a few minutes (OSX needs time to rebuild the kernel)
8. reboot (should not be necessary but I found sometimes the updated kext does not load in Capitan).

Wouter1, 2015-11-29:
@adrian-polglase 
@Emptyfruit 

can you please confirm if this solution works for you? THen I can announce that we at least have a workaround until Apple responds on the USB issue.

adrian-polglase, 2015-11-29:
Thanks Wouter1. I'll give it a try now and get back to you.

Emptyfruit, 2015-11-29:
Wow, that good news. Unfortunately, i don't have a time machine backup right now, so can't risk going to El Capitan again to test. I'll join the "test team" as soon as i can :))

Wouter1, 2015-11-29:
@Emptyfruit  No problem. Again, this is a high-latency workaround, not a real fix. If you need that EMU I would wait for a good fix.

Wouter1, 2015-11-29:
@vitalibondar 
@digdelu 
@Philipondio 

Above you can find a high-latency workaround. Please inform me about results if you try it

adrian-polglase, 2015-11-29:
I can confirm that on my El Capitan machine at least, this workaround seems to work for now and allow me to once again record

Thanks Wouter1!

Wouter1, 2015-11-29:
@adrian-polglase  Thanks for the confirmation!

boredomisacrime, 2015-11-30:
Hi! first thanks for trying to get this to work, i started reading this because i have the same issue as vitalibondar...I have a 15 inch macbook pro with 16 gb ram running el capitan and an 0404 usb...dunno if that info is worth crap, but anyway, after around 10 seconds I get the info that the .kext file cannot be used, even after your last suggestion it still gives me the same error ... any suggestions?

Wouter1, 2015-12-02:
@boredomisacrime  sorry for late response, I was busy working on the recording issue and did not look again here.

Please post relevant log info that says anything about the EMU kext.

Is your machine still in kext-develop mode?

boredomisacrime, 2015-12-02:
@Wouter1  no worries, still awesome that you are trying to get this to work.
Regarding the issues, this is what i got from Terminal
[image]

Wouter1, 2015-12-02:
Yes that seems fine. I was just wondering if it somehow got re-enabled after rebooting.

Furthermore, do you see anything concerning the EMU kext in system.log when you reboot and when the kext stops working?

boredomisacrime, 2015-12-02:
so this is what the system.log had to say about kext :
Dec  2 12:13:25 Eduardos-MBP kernel[0]: Kext loading now disabled.
Dec  2 12:13:25 Eduardos-MBP kernel[0]: Kext unloading now disabled.
Dec  2 12:13:25 Eduardos-MBP kernel[0]: Kext autounloading now disabled.
Dec  2 12:16:29 localhost kernel[0]: kext submap [0x<ptr> - 0x<ptr>], kernel text [0x<ptr> - 0x<ptr>]
Dec  2 12:16:29 localhost kernel[0]: AirPort_Brcm4360::init AirPortFamily_kexts-1101.24 "AirPortFamily_kexts-1101.24" Sep 24 2015 19:56:07
Dec  2 12:16:39 Eduardos-MBP com.apple.xpc.launchd[1] (com.apple.dockextra.server): The HideUntilCheckIn property is an architectural performance issue. Please transition away from it.
Dec  2 12:27:28 Eduardos-MBP sudo[733]:       Ed : TTY=ttys000 ; PWD=/Users/Ed/Downloads/EMU-driver-master ; USER=root ; COMMAND=/sbin/kextunload /System/Library/Extensions/EMUUSBAudio.kext
Dec  2 12:27:28 Eduardos-MBP kernel[0]: Kext com.emu.driver.EMUUSBAudio not found for unload request.
Dec  2 12:27:29 Eduardos-MBP sudo[738]:       Ed : TTY=ttys000 ; PWD=/Users/Ed/Downloads/EMU-driver-master ; USER=root ; COMMAND=/bin/rm -r /System/Library/Extensions/EMUUSBAudio.kext
Dec  2 12:27:29 Eduardos-MBP sudo[744]:       Ed : TTY=ttys000 ; PWD=/Users/Ed/Downloads/EMU-driver-master ; USER=root ; COMMAND=/bin/cp -r EMUUSBAudio.kext /System/Library/Extensions
Dec  2 12:27:29 Eduardos-MBP sudo[754]:       Ed : TTY=ttys000 ; PWD=/System/Library/Extensions ; USER=root ; COMMAND=/usr/bin/chgrp -R wheel EMUUSBAudio.kext
Dec  2 12:27:29 Eduardos-MBP sudo[756]:       Ed : TTY=ttys000 ; PWD=/System/Library/Extensions ; USER=root ; COMMAND=/sbin/kextload -v EMUUSBAudio.kext
Dec  2 12:27:29 Eduardos-MBP com.apple.kextd[44]: Cache file /System/Library/Caches/com.apple.kext.caches/Directories/System/Library/Extensions/KextIdentifiers.plist.gz is out of date; not using.
Dec  2 12:27:29 Eduardos-MBP com.apple.kextd[44]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"
Dec  2 12:27:34 Eduardos-MBP com.apple.kextd[44]: Cache file /System/Library/Caches/com.apple.kext.caches/Startup/KextPropertyValues_OSBundleHelper_x86_64.plist.gz is out of date; not using.
Dec  2 12:27:34 Eduardos-MBP com.apple.kextd[44]: Rescanning kernel extensions.
Dec  2 12:27:34 Eduardos-MBP com.apple.kextd[44]: Cache file /System/Library/Caches/com.apple.kext.caches/Startup/IOKitPersonalities_x86_64.ioplist.gz is out of date; not using.
Dec  2 12:27:35 Eduardos-MBP com.apple.kextcache[773]: rebuilding //System/Library/PrelinkedKernels/prelinkedkernel
Dec  2 12:27:40 Eduardos-MBP com.apple.kextcache[776]: / locked; waiting for lock.
Dec  2 12:27:46 Eduardos-MBP com.apple.kextcache[774]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext EMUUSBAudio.kext
Dec  2 12:28:04 Eduardos-MBP com.apple.kextcache[774]: Created old kernelcache copy "/System/Library/Caches/com.apple.kext.caches/Startup/kernelcache"
Dec  2 12:28:04 Eduardos-MBP com.apple.kextcache[774]: Created prelinked kernel "/System/Library/PrelinkedKernels/prelinkedkernel"
Dec  2 12:28:04 Eduardos-MBP com.apple.kextcache[774]: Created prelinked kernel using "/System/Library/Kernels/kernel"
Dec  2 12:28:04 Eduardos-MBP com.apple.kextcache[773]: /System/Library/PrelinkedKernels/prelinkedkernel not cached.
Dec  2 12:28:05 Eduardos-MBP com.apple.kextcache[773]: Successfully updated disk0s3.
Dec  2 12:28:05 Eduardos-MBP com.apple.kextcache[776]: Lock acquired; proceeding.
Dec  2 12:28:05 Eduardos-MBP com.apple.kextcache[776]: /: helper partitions appear up to date.

Wouter1, 2015-12-02:
The log says that the EMU kext was loaded succesfully. 
It should work ok after that point. 

Do you still have the issue?

boredomisacrime, 2015-12-02:
Yup the device doesn't show up on any software...but it can be human error mind you, i'm gifted that way :), right now i don't have the 0404 next to me, but i'll try it later today and report back.

boredomisacrime, 2015-12-02:
Ok, so EMU now works, in the sense that sound comes out. But input is laggy and all chopped up...but please help others first, as I got a line 6 pod x3 live that will tide me over until the record solution is fixed. Sorry to have wasted your time on a non issue

Wouter1, 2015-12-03:
@boredomisacrime 
Ok that's good to hear and what was expected at this point. 
Can you try the high-latency workaround I described above?

boredomisacrime, 2015-12-03:
I did, but no luck. Still chunky as hell. I had it connected through a hub, then changed to direct connection and heard no difference...

Wouter1, 2015-12-03:
Did you use a fresh downloaded driver when following the high-latency instructions? The lowlatency script can be used only once on a fresh copy.

[36 more replies]

closed 2016-07-22 by Wouter1