Wouter1/EMU-driver #66: El Capitan driver disappearance

Emptyfruit, 2015-10-04:
The driver installs and seems to work fine (tried music playback in VOX and midi in Logic), but after the second restart e-mu disappears from audio devices. The driver is still present in Extensions with status "not loaded". What i did (2 times) was:

1) Reboot to turn off SIP
2) Booted with SIP off and installed driver - worked fine
3) Rebooted to turn SIP back on
4) Booted with SIP on - driver worked fine
5) Rebooted usually after Gutar Rig installation - EMU was gone from audio devices.

Wouter1, 2015-10-05:
I suppose that is to be expected. If you turn SIP on, OSX will not load the driver.

This is because El Capitan does not accept unsigned kernel drivers.

The only weird thing is that the driver actually works the first time with SIP on. I guess that OSX removes the driver only after booting (and at that time the driver was already loaded the first time)

Does it work properly if you keep SIP off?

Emptyfruit, 2015-10-05:
Haven't tried.. But i will later today. I use the device a lot, so the idea of turning SIP off for good bothers me a little. Do you think there'll be a workaround?

vitalibondar, 2015-10-05:
> Does it work properly if you keep SIP off?

@Wouter1 in my case - no, I must to reinstall the driver over and over again.
So I went back to Yosemite for a while.

Wouter1, 2015-10-05:
Yes it bothers me also. Apple has never responded to my request to support freeware kernel drivers.

So Apple is giving us no other option. They seem to head towards allowing only commercial drivers. At some point they may give no options at all to load freeware kernel drivers.

vitalibondar, 2015-10-05:
@Wouter1 There is a way to sign a driver as commercial? What do you need?

Emptyfruit, 2015-10-05:
have you tried contacting Creative? It is so strange that they ignore all the issues with their ancient driver, because the device is still present in stores.

Wouter1, 2015-10-05:
Last year, you needed to buy a Mac Developer Program ID and I think you needed to be some business/firm/organization in the US. 

https://developer.apple.com/developer-id/

https://developer.apple.com/programs/enroll/

I'm not an organization, have no DUNS number, and I'm not sure if I'm a "legal entity" as I'm not in the US.

Wouter1, 2015-10-05:
I think I tried to contact Creative. But that was long ago when I started developing this driver.
Maybe I can try again to see if they can help.

vitalibondar, 2015-10-05:
> I'm not an organization, have no DUNS number, and I'm not sure if I'm a "legal entity" as I'm not in the US.

[image]

[image]

Maybe, all we need it's to rise some money for your enrolling in developer program? I.e. crowdfunding? It will be cheaper than buying a new audio interface.

Wouter1, 2015-10-05:
Looks like that might be a possibility. I vaguely remember the fees were something like $100 per year but that was some year ago, I never read all the terms and conditions.

If it's on a per-year basis, that would involve a new crowdfunding project every year?

Emptyfruit, 2015-10-05:
I'm gonna test if the driver falls off even if i leave SIP off, but if it falls off as vitalibondar says, doesn't it mean that it can be fixed? As i understand without SIP it should be like Yosemite?

vitalibondar, 2015-10-05:
@Wouter1 If at least 10 people will send $10, we shall find the necessary amount of money. $10-20 a year - a little money, especially if a person makes money with his device. New one worth little more :)

Wouter1, 2015-10-05:
>I'm gonna test if the driver falls off even if i leave SIP off, but if it falls off as vitalibondar says, doesn't it mean that it can be fixed? As i understand without SIP it should be like Yosemite?

That's what I also thought, and I expect it would reload if you leave SIP off. I'd like to hear your test results first.

Emptyfruit, 2015-10-05:
2 reboots with SIP off and the driver works fine. I'm gonna leave SIP off for now and see how it goes. Also gonna test recording in Logic. As for now, i get some clicking with Guitar Rig, but it may be not the driver's fault.

Wouter1, 2015-10-05:
@Emptyfruit Good to hear that it all works fine for you. What system do you have? I ask because @vitalibondar  has problems getting it working at all on a rMBP.

Wouter1, 2015-10-05:
I need to spend time on other things now, I will be back here tonight.

Emptyfruit, 2015-10-05:
MacBook Pro retina 15' mid 2015 16gb ram, Inter Iris Pro 1536, El Capitan 10.11

Wouter1, 2015-10-05:
Thanks. Similar specs as @vitalibondar 's computer. Weird that his computer does not work at all with this driver.

Wouter1, 2015-10-05:
Made a new ticket #68 for the code signing issue.

Wouter1, 2015-10-05:
@vitalibondar  does your SIP stay off after disabling or does it gets re-enabled somehow? If it somehow turns back on that might be the simple explanation. Alternatively, it might be that it gets locked out after somethign crashing in the driver.

vitalibondar, 2015-10-05:
> @vitalibondar does your SIP stay off after disabling or does it gets re-enabled somehow? If it somehow turns back on that might be the simple explanation.

@Wouter1 maybe this is answer?
[image]

Wouter1, 2015-10-05:
Yes it's weird. Is that the result of 'csrutil disable'?

vitalibondar, 2015-10-05:
@Wouter1 Yes. I did everything according to instructions:

    reboot in recovery mode (hold cmd-R while booting)
    go to utilities-terminal
    enter this command: csrutil disable
    reboot your machine

Wouter1, 2015-10-05:
The confusing message is a known issue

https://openradar.appspot.com/22361698

Apple ignores the issue by saying it's a duplicate of another issue that you are not allowed to see, and then closing the issue without any further info.

Emptyfruit, 2015-10-05:
I get the same message though on csrutil status. I felt better before seeing it :))))

Wouter1, 2015-10-06:
I suppose it's still ok, as long as "kext signing" is disabled.

Emptyfruit, 2015-10-06:
It made we wondering if it is possible to actually configure SIP leaving everything enabled, except Kext Signing..

Emptyfruit, 2015-10-07:
Some good news. There is a SIP Utility http://cvad-mac.narod.ru/files/SIPUtility.app.v1.0.1.zip that can configure SIP parameters on El Capitan. Link to the thread: https://www.applelife.ru/threads/system-integrity-protection-sip-v-os-x-10-11-el-capitan.536553/page-6#post-537884

I enabled every component of SIP except kext signing, rebooted and the driver is still working. 
csrutil status gives me:

`System Integrity Protection status: enabled (Custom Configuration).
Configuration:
	Apple Internal: disabled
	Kext Signing: disabled
	Filesystem Protections: enabled
	Debugging Restrictions: enabled
	DTrace Restrictions: enabled
	NVRAM Protections: enabled
This is an unsupported configuration, likely to break in the future and leave your machine in an unknown state.`

I guess it's kind of a compromise, you still have most of the SIP working and driver as well. Of course the utility should be used at your own risk.

Wouter1, 2015-10-07:
That's great news, thanks for reporting. Yes this is much more secure than disabling the whole thing.

I wonder how to do this without the separate tool, just using the built-in commands. Maybe with a more specific csrutil command?

Wouter1, 2015-10-08:
I think I found something to do that  https://derflounder.wordpress.com/2015/10/01/system-integrity-protection-adding-another-layer-to-apples-security-model/

Can you try this
```
csrutil enable --without kext
```

Emptyfruit, 2015-10-08:
Strange thing. In order to try, i enabled the whole SIP and rebooted, but the driver survived. I rebooted again and the driver is still working. So now i have SIP fully enabled and the driver working. I don't know the reason :) I'm gonna leave it that way for now, but if it falls off, i'm gonna try "--without kext" command.

Wouter1, 2015-10-08:
@Emptyfruit  Thaniks, that is even nicer :) Please report back if it keeps working over the coming week or so.

Emptyfruit, 2015-10-08:
And it is gone.. Strangely without reboot. I just plugged out the device though. So it is either takes some time for system to kill the driver or the unplugging of the device. But i switched off the device before my reboots, so it's kind of the same as unplug. Unclear...

So i tried --without kext. It works. I get the same configuration as i got with the utility. The driver works again. Interesting thing that the driver appeared by itself. No need to reinstall. I guess SIP just blocks it, not wipes out completely. But i have a feeling that it is not the end of it, so we'll see..

Wouter1, 2015-10-08:
@Emptyfruit  well it was to be expected.

Great to hear that the ```--without kext``` works. 

I'll update the instructions for El Capitan.

>No need to reinstall. I guess SIP just blocks it, not wipes out completely

I suppose that SIP can not remove the files from the directory. So yes, it only can prevent the kext from loading.

Wouter1, 2015-10-08:
@Emptyfruit  I think your original issue is solved and we can close this ticket?

Emptyfruit, 2015-10-08:
In that case, if you just ocassionally  use the device, the SIP Utility saves you some time, cause you can turn off kext, reboot and work. Then turn kext on, reboot and "be protected" if you don't need the device for some time. It saves you booting in recovery (which is longer that usual boot), typing and rebooting again.

As for this ticket, i guess it is, thank you. I got some clicking yesterday using midi, but i'll do more tests and then report. Anyway i doubt it is related to SIP.

Wouter1, 2015-10-08:
Yes. Thanks for the reporting and help in checking the solutions :-)

closed 2015-10-08 by Wouter1