Wouter1/EMU-driver #68: signing the extension

Wouter1, 2015-10-05:
If I could sign the extension, that would make easier installation and better security (without turning off SIP or allowing unsigned extensions).

Wouter1, 2015-10-05:
In #66, @vitalibondar  proposed
>Maybe, all we need it's to rise some money for your enrolling in developer program? I.e. crowdfunding? It will be cheaper than buying a new audio interface. If at least 10 people will send $10, we shall find the necessary amount of money. $10-20 a year - a little money, especially if a person makes money with his device. New one worth little more :) He also shows a page where apparently you can apply for this as an individual.

@Emptyfruit suggested
>have you tried contacting Creative? It is so strange that they ignore all the issues with their ancient driver, because the device is still present in stores.

Wouter1, 2015-10-05:
checked apple's video at 
https://developer.apple.com/videos/play/wwdc2013-707/

For signing, I need a Developer ID certificate. 
1. go to https://developer.apple.com/resources/developer-id
2. request Developer ID certificate for signing kexts
3. fill in the form 
4. once form is approved, request Developer ID certificate in the member center.

Wouter1, 2015-10-05:
If I try the enroll form pointed to by @vitalibondar, I get 

>In order for your company name to be listed as the seller on the App Store, your company must be recognized as a legal entity in your country. Sole proprietors and single person companies located in countries where they are not recognized as legal entities will be enrolled and be listed for download under the enrollee’s personal legal name, if selected for distribution by Apple.

Then proceeding, I get a 73-page agreement form!

They also say
>You acknowledge and agree that it is Your responsibility to consult with Your legal advisors with respect to Your legal obligations hereunder.

I'm not a lawyer ... even let a lawyer check this agreement form probably will take a fortune...

Wouter1, 2015-10-05:
Older versions of OSX can NOT load signed kernel extensions. 
So if we go for the signing route, we would exclude users running older OSs. (I think before 10.8.5)

Wouter1, 2015-10-06:
I read this on https://pikeralpha.wordpress.com/2015/07/28/apples-kext-signing-bypassed/

>Kext Whitelisting
>If all you need is kext whitelisting, with SIP in place, then you can change/add one or more entries under OSKextSigExceptionHashList (see AppleKextExcludeList.kext/Content/Info.plist) in the prelinkedkernel, like we have been doing in Mavericks and Yosemite.

Would that be an option?

Emptyfruit, 2015-10-06:
Can you, please, explain the procedure in details?

Wouter1, 2015-10-06:
This ticket is about how we can sign the extension, as to avoid disabling SIP.
We do not have a working procedure for that yet (if we would, this ticket would have been closed).

Emptyfruit, 2015-10-06:
I get it, but what you wrote is to be done on each machine, so that it allows one particular unsigned kext, or i misunderstood?

Wouter1, 2015-10-06:
It depends. Maybe we can somehow sign the kext; or maybe we need to change the security settings to accept this specific kext. 

Signing it is the neatest way but it takes money and knowledge of legislation. Also it will break backwards compatibility

Emptyfruit, 2015-10-06:
Yeah, i'm on track, i meant this:

>Kext Whitelisting
>If all you need is kext whitelisting, with SIP in place, then you can change/add one or more entries under OSKextSigExceptionHashList (see AppleKextExcludeList.kext/Content/Info.plist) in the prelinkedkernel, like we have been doing in Mavericks and Yosemite.

Is it a way to make some kext available on my machine?

Wouter1, 2015-10-06:
I don't know yet. I posted it as it looks promising but we need to figure out what exactly is ment and if it still works in El Capitan.

Wouter1, 2015-10-20:
I got the impression that Kext whitelisting is only possible if you have a custom bootloader like RevoBoot that he mentions. I didn't try but in theory changing the kext will render the checksum invalid and then the kext (in this case AppleKextExcludeList) won't load anymore.

strogonoff, 2018-05-19:
I wish I could install the kext without disabling SIP after I build and self-sign in Xcode… doesn’t seem to work though.

Focusrite Scarlett Solo (pretty much EMU 0202 which is what I have now) costs just a bit more than Developer ID, and I’m not even sure I’ll be able to properly sign the kext if I do shell out for that ID, heh.

Wouter1, 2018-05-19:
@strogonoff  what does Focusrite have to do with this? 

Are you suggesting that you could make a signed version of this EMU driver?

And notice that besides the costs of the developer ID you also have to sign a 73 page contract, for which you might need a lawyer.

strogonoff, 2018-05-26:
@Wouter1 I was just weighting the costs between

* keeping using my old EMU 0202 & applying for Developer ID just to have working drivers for it without disabling SIP (~$100)
* buying a new and supported audio interface in similar price range (abovementioned model by Focusrite is ~$120)

Since I can’t be sure I’ll be able to sign the kext **even if** I buy the ID and sign contract, I’m considering buying a new audio interface… sadly.

(Yes, I think disabling SIP is more reckless than signing a contract with Apple without a lawyer.)

strogonoff, 2018-05-26:
Originally I was going to suggest to sign your kext for public use with my ID. Without your permission I’d just use it on my own machine. I thought I can use it on my machine if I self-sign it, but apparently that doesn’t work in kext land.

strogonoff, 2018-05-27:
@Wouter1 Yeah I gathered it’d be a long shot, so it’d be pointless to invest in a Developer ID just for this.

Kext whitelisting could’ve been a better option (if I could build a kext, self-sign it with the “virtual” team in Xcode tied to my regular Apple ID, and then explicitly set kext policy to allow kexts with that team ID), I looked it up and seems like corporations might be making it work somehow when they mass-provision staff machines with custom security software and stuff. Or maybe they all really do go to Apple and apply… Either way I couldn’t make kext whitelisting work on my machine.

closed 2021-01-22 by Wouter1