Wouter1/EMU-driver #71: 4 outs midi not working

Wouter1, 2015-10-16:
I received this issue report

Driver work on Emu 0204 only if change in Mac OS "Audio-Midi settings" from 4 outs to 2 outs. In this case lost independent headphone source. Cant make pro cue in daw.

[image]

Wouter1, 2015-10-16:
I think this report is not about midi but about the normal output through headphones and spdif. Has nothing to do with midi except that he uses the midi settings panel.

Wouter1, 2015-10-16:
asked some clarification

1. "in this case lost independent headphone source": can you explain? 4 outs would involve spdif and headphone out. If you select 2, you get only headphone, right?

2. what you mean by "cant make pro cue in DAW".

3. On what OS are you?

Wouter1, 2015-10-16:
I just realize that on EMU 0204  the outputs are probably different. 
Checked the manual, it seems that channel 1 and 2 normally are both headphone and line

If 3 and 4 are used, then ch1 and 2 are for the line output and 3 and 4 for the headphones. I think he is referring to this.

That explains why he is talking about independent headphone loss.

vspvsp, 2015-10-17:
"If 3 and 4 are used, then ch1 and 2 are for the line output and 3 and 4 for the headphones. I think he is referring to this." 
you're right, in this video (low quality montage, but simple language) show how it work
http://www.youtube.com/watch?v=4k9F_DVrDDU

------------
44,1 — not work, no sound
48 — not work, no sound
88,2 — work
96 — work
176,4 — not work, have sound, but with clicks from 1 sec
192 — work, but after 5 min appear clicks
44,1 and 48 — it's a problem, because it's optimal sample rate for work, they have less load on system 
176,4 and 192 — don't need for normal users

Wouter1, 2015-10-17:
Ok thanks, so this is a new issue for Capitan. It may be related to other known issues.

I have no 0204 but maybe this can be replicated to ch3 and ch4 on the 0404.

Please note #72 which may prevent me from fixing issues on El Capitan...

Wouter1, 2016-07-26:
@vspvsp The EMU driver now is working on 0404 with El Capitan. Can you try the new driver with El Capitan on your 0204? I did not do anything about the output channels yet but I have good hope that at least all sample rate work ok.

Wouter1, 2016-12-18:
@vspvsp can we close this?

shtirlic, 2017-04-04:
Another owner of 0204 reporting, 0204 working only in 2 channels mode selected in audio devices. also every 30 or more seconds I hear fast "click-click", tested in every sample rate. Latest macOS and latest build. Installed with low latency 'no' setting.

Wouter1, 2017-04-04:
@shtirlic usually the clicking is caused by the clocks getting in sync slower than usual. Does the clicking get less and finally stop if you play a long track, like 5 minutes?

shtirlic, 2017-04-04:
@Wouter1 nice point, I will try to listen longer in iTunes on hour track, but I heard it everytime in Ableton Live DAW while playing syths.

Wouter1, 2017-04-04:
@shtirlic do you maybe have a very small buffer in Ableton? Or is the CPU load very high while playing? Do you alos hear clicks if you play a simple mp3 with Quicktime?

shtirlic, 2017-04-04:
@Wouter1 it was set to 128 samples, CPU around 20-30%, will check quicktime when I came back home.

shtirlic, 2017-04-06:
@Wouter1 tested the same clicks, even worse  please see the log, audio stopped after 10 minutes 
```
default	11:16:52.112212 +0300	kernel	frameSizeQueue empty, guessing some queue size. May need fix..
default	11:16:52.157488 +0300	kernel	warning way-out ring wrap position
default	11:16:53.039990 +0300	kernel	LowPassFilter::filter init 185759637
default	11:16:53.039994 +0300	kernel	USB timer started
default	11:16:53.214415 +0300	kernel	**** Output Hiccup!! firstSampleFrame=1568, nextExpectedOutputFrame=5632 bufsize=49152
default	11:16:55.508339 +0300	kernel	+ IOAudioLevelControl::validateValue[0xffffff80498cb3c0] (0xffffff805ec06b70)
default	11:16:55.508349 +0300	kernel	  IOAudioLevelControl::validateValue[0xffffff80498cb3c0] - newIntValue = 6735, min = 0, max = 7200
default	11:16:55.508352 +0300	kernel	- IOAudioLevelControl::validateValue[0xffffff80498cb3c0] (0xffffff805ec06b70) returns 0x0
default	11:26:37.061742 +0300	kernel	PIPE write error :0. Stopping OutputStream.
default	11:26:37.093616 +0300	kernel	All playback streams stopped
```

shtirlic, 2017-04-06:
@Wouter1 did you ever considered to rewrite driver as user space driver like here https://developer.apple.com/library/content/samplecode/AudioDriverExamples/Introduction/Intro.html ?

Wouter1, 2017-04-06:
@shtirlic  no this driver is very large and I dont have time for a complete re-impelemtation like that

Wouter1, 2017-04-06:
@shtirlic That log, and the stopped audio, is that with quicktime?

shtirlic, 2017-04-06:
@Wouter1 no, this log from iTunes playing, in Quicktime only click-click

shtirlic, 2017-04-06:
@Wouter1 I see, now I am cleaning up project to left only code for 10.10+ system, after that I will try to build some Apple examples, there are two of them: one is simple in one file and  another one with very rich functionality.

shtirlic, 2017-04-06:
@Wouter1 Update: Tested on my Macbook pro mid 2014, kext working without clicks and crystal clear, maybe something with hardware inside my iMac ?

Wouter1, 2017-04-07:
I would not expect that, but how old is the imac? Do you have good throughput on the USB ports with other devices or is the USB very slow?

shtirlic, 2017-04-11:
@Wouter1 after long testing(listening) I found that connected devices doesn't affect clicks, trying to find another cure. iMac mid 2011 with latest macOS

Wouter1, 2017-04-11:
@shtirlic thanks for keeping us updated. I have no new suggestions at this point. Mid 2011 should be new enough, it does have USB 2.0 ports and 2.5GHz quad core, seems plenty to me. I would focus on simple quicktime playing a wav or MP3 to ensure the problem is not in the player. 

I can make a debug version of the driver but this is a bit of work and I'm really tight on time the coming weeks

shtirlic, 2017-04-11:
@Wouter1 thx for your help, I can make debug version by myself, may you suggest me places where to place debug logs regarding this kind of issue, also what you think about ring buffer implementation, half a year ago I implemented soundio support for shairport-sync(opensource airplay player) and noticed similar issues while tuning ring buffer sizes and sample counts.

Also I know that liniux Alsa project has full support of e-mu cards, did you see their code?

Wouter1, 2017-04-11:
From top of my head, check the debug settings in EMUUSBLogging.h and recompile, install, run and then check the console output while running.

>what you think about ring buffer implementation
What do you mean? I have a ring buffer implementation and it works fine

>half a year ago I implemented soundio support for shairport-sync(opensource airplay player) and noticed similar issues while tuning ring buffer sizes and sample counts

Ok, what does it fix/do different compared to my implementation? Also notice that my ring implementation has a few things that are needed for this specific case

>Also I know that liniux Alsa project has full support of e-mu cards, did you see their code?

I remember I briefly checked but it did not really matter because the EMU driver is very different.

Wouter1, 2018-07-03:
@vspvsp  @shtirlic  In #112 I found that there is something in the driver thinking we have 4 channels available at 44kHz. Apparently that's not the case. This must be a bug somewhere deep down in the driver....