Wouter1/EMU-driver #72: compile with XCode 7

Wouter1, 2015-10-16:
For debugging under EL Capitan, it would be nice if I could use XCode 7 for compilation

Wouter1, 2015-10-17:
The project imports fine in XCode 7 

But the USB libraries have been changed and now basic things like #include <IOKit/usb/USBPipe.h> does not work anymore

I don't want to spend time right now on learning Apple's new USB design so I will stick with XCode 5 for now. The old compiled code should still work on El Capitan, it's just that their new compiler does not allow compiling it...

viktorvos, 2015-10-17:
Hello, first thanks for this driver! I had this EMU 0204 lying in the cupboard and it's working on 10.11.1 now.

I tried to compile with xcode 7 on 10.11.1 but usb has changed a lot in the SDK.

Wouter1, 2015-10-17:
Thanks, good to hear it's working for you.

There are a few issues in El Capitan that I'd like to fix. Being able to compile on Capitan would help that a lot. 

If I would make changes to use the new El Capitan USB system, then the driver would not work anymore on Mavericks and Yosemite which is not desirable as well.

Wouter1, 2015-10-17:
Maybe I can get xcode older versions get to work on El Capitan.

Wouter1, 2015-10-17:
Finally some good news, I CAN still run XCode 5 in Capitan. You have to start it from the command line. And it spits a lot of error messages to the console. But it seems to work nevertheless.

```/Volumes/apps/Xcode.app/Contents/MacOS/Xcode ```

Wouter1, 2015-10-17:
Ok I consider this fixed, under the cheanged issue title.

closed 2015-10-17 by Wouter1