Wouter1/EMU-driver #73: recording cracks in Yosemite

Wouter1, 2015-10-21:
@kornerious posted (#64)

Cracks during recording, even with large buffer size(

oh. can you tell when will be some news?)

2. I am using your driver in Yosemite. Wanted to load low latency but up to 512 samples i got cracks( is it normal? 3. Your normal driver shows 10 msec latency, but with 32 buffer i get some cracks also (from 64 is all right) - is it also normal? i5 2500+8gb DDR 2015-10-21 0:43 GMT+03:00 ALexandr Kiliyevich <kornerious@gmail.com>:

Wouter1, 2015-10-21:
Maybe your computer is not fast enough to pull off the 1 millisecond latency. Maybe you can try 1.5ms 

How fast is your CPU?

To do this download the driver again from git. Change the 1000 to 2000 in the lowlatency script and then follow the complete procedure to (re)install the driver. Note,the lowlatency script runs correctly only 1 time. Therefore you need to download the fresh driver before doing this

128 should be good enough for buffer. Notice that for latency it probably is better to combine normal latency driver with 64 than to have very low latency driver but needing a huge buffer. 

Please report back if this works for you so that I can add some notes to the low latency installation.

kornerious, 2015-10-21:
I have chosen "normal latency driver with 64 buffer"
But there is one interesting thing i noticed. Official driver shows 6 ms latency in 48Khz project and your driver shows 10 ms for 32 and 64 buffer size. Why?

Wouter1, 2015-10-21:
Great. Does it work properly for you now? Any results with trying 2ms?

I don't know what they say on the official driver. Also often drivers report unrealistic low latencies.

Here you can find the details about latency of my driver. 

https://github.com/Wouter1/EMU-driver/blob/master/Latency.md

We measured these latencies thorougly and they should closely match the actual latencies.

kornerious, 2015-10-22:
"Any results with trying 2ms?"
Didn't even try.

For my ears latency is the same with official and your's driver, only repots in different ways and i mentioned that your driver is like most stable.

Waiting for some changes in El Capitan EMU recording. 
Thanks for support!

Wouter1, 2015-11-23:
Update, I got El Capitan working on my machine and may have some time for that. That is issue #65. I will close this ticket as we are on El Capitan now.

closed 2015-11-23 by Wouter1