Wouter1/EMU-driver #76: Installation script error on El Capitan 10.11.4

mcropper, 2016-04-29:
When running the script i get the error..

`Requesting load of /System/Library/Extensions/EMUUSBAudio.kext.
/System/Library/Extensions/EMUUSBAudio.kext failed to load - (libkern/kext) not loadable (reason unspecified); check the system/kernel logs for errors or try kextutil(8).`

Is this an issue with signing. Is there a work around without turning off SIP?

Wouter1, 2016-04-30:
@mcropper 

I suppose you did not set up SIP properly. Please check the installation manual how to prepare your computer for installation

https://github.com/Wouter1/EMU-driver/blob/master/Install.md

Regarding your question, you need to set up SIP properly. This extension is not signed so you can only install with the proper settings for SIP.  

Can you confirm that it installs properly if you follow the installation guide?

mcropper, 2016-04-30:
I followed the installation instructions yes. Still had the same problem. 
However:

I managed to get it working by doing a fresh install of the manufacturers installation package after running your script with some amendments, I duplicated the scripts run on the /System/Library/Extensions file path and ran them on the /Library/Extensions path including the `doing permissions stuff`.  The file /Library/Extensions/EMUUSBAudio.kext is untouched by the fresh install.

I will have to try it again after doing a compete removal to confirm if this action helped me get it installed.

I have done fresh installs of the original software before since upgrading to El Capitan but to no avail. and i was having no luck following the steps as i was getting errors mentioned above. Some combination has worked.

**Note:**
I thought that since El Capitan, Apple has made the command obsolete:

`sudo nvram boot-args="kext-dev-mode=1"`

[As explained here](https://developer.apple.com/library/mac/documentation/Security/Conceptual/System_Integrity_Protection_Guide/KernelExtensions/KernelExtensions.html)

mcropper, 2016-04-30:
Ignore the note. I should get more sleep.

Wouter1, 2016-04-30:
I don't quite follow you

There should not be anything installed in /Library/Extensions. The extension is installed in /System/Library/Extensions. 

Did you install my driver in /Library/Extensions?

Did you do the correct ```csrutil enable --without kext```?

Wouter1, 2016-05-07:
Never heard back from it. I assume it all works as intended.

closed 2016-05-07 by Wouter1