Wouter1/EMU-driver #77: wake up issues El Capitan

Wouter1, 2016-05-11:
@Laximus wrote in #65

Hello, I would like to write a report, that the wake up issues on El Capitan/iMac somehow is related to this driver or to the hardware itself. Or the Apple didnt write properly something in the USB drivers. I used your driver, and was lot of wake up issues with EMU USB 0202 - kernel panics, and have to restart the iMac almost from every second sleep state. Around a week ago I dont use the EMU USB 0202 at all, it's not even pluged to the iMac by usb, and no wake up issues at all. I really apreciate your work of course, I just want to report this. Thank you.

Wouter1, 2016-07-12:
Laximus reported earlier in the same ticket

>I noticed with the high latency workaround, if the iMac goes to sleep for longer period, and when I want to use it again, it is just freezes with a half dark screen, and i have to reset the computer. It start with a screen like this, but after everithing is ok until the next sleep...

```
OSX El Capitan
IMac (27", 2012)
Version 10.11.4
Processor: 3.2 GHz Intel Core i5
Memory: 8 GB 1600MHz DDR3
Graphics: NVIDIA GeForce GTX 675MX 1024MB
```

Wouter1, 2016-07-12:
@Laximus can you retest this with the new driver? Maybe the issue is now resolved since you dont need the high latency workaround anymore

Laximus, 2016-07-12:
@Wouter1 
I installed your driver yesterday evening - here in Hungary - and since I dont have this kind of kernel panic or freezing. =D
Hope it's solved too. If it will come back, I will report it. =)

Wouter1, 2016-07-12:
@Laximus  thanks for the feedback.Great!

Then I close this ticket. Please reopen it if this problem re-appears

closed 2016-07-12 by Wouter1

Laximus, 2016-07-12:
OK! =D