Wouter1/EMU-driver #78: Compile with 10.11 sdk

Wouter1, 2016-06-06:
Maybe the code can work again properly with low latency if we compile against the 10.11 SDK

Wouter1, 2016-06-06:
I worked a bit on this already in #70

Wouter1, 2016-06-06:
Plan is to keep code compilable on XCode 5 as it was, with SDK 10.9 and deploy target 10.9

But to also make it compilable with XCode 7 using SDK 10.11 and deploy target 10.11

The idea is to make adapters for several classes in 10.11, so that I do not have to modify the main bulk of the code. 10.9 does not need the adapters but in 10.11 the driver calls would be rerouted through the adapters that map the old 10.9 calls into the new 10.11 calls

So far for the rough plan. The details will have some problems, I can not really see how far and rough that will get.

Wouter1, 2016-06-06:
I put in adapters for IOUSBPipe, IOUSBDevice, IOUSBInterface, IOUSBController,

and a bunch of structures in USB.h: IOUSBIsocFrame, IOUSBLowLatencyIsocFrame, IOUSBLowLatencyIsocCompletionAction, IOUSBLowLatencyIsocCompletion, IOUSBEndpointDescriptor, IOUSBConfigurationDescriptor, IOUSBCompletionAction, IOUSBIsocCompletionAction, IOUSBIsocCompletion, IOUSBCompletion, IOUSBDevRequest, IOUSBDevRequestDesc

This long list in USB.h might become the most troublesome part, as these are raw structures where the first ones are more or less function-based objects.

Wouter1, 2016-06-07:
guessing the new versions of old structures
| old | new |
| IOUSBLowLatencyIsocFrame |IOUSBHostIsochronousFrame|
| IOUSBLowLatencyIsocCompletion | IOUSBHostIsochronousCompletion |
| IOUSBLowLatencyIsocCompletionAction | IOUSBHostIsochronousCompletionAction |

It looks like in general the USB.h file has been replaced with IOUSBHostIOSource.h

Wouter1, 2016-06-07:
These structures look comparable but have slight differences.

Eg 
```
struct IOUSBLowLatencyIsocFrame {
    IOReturn                        frStatus;
    UInt16                          frReqCount;
    UInt16                          frActCount;
    AbsoluteTime		    frTimeStamp;
};

```
versus

```
struct IOUSBHostIsochronousFrame
{
    IOReturn     status;
    uint32_t     requestCount;
    uint32_t     completeCount;
    AbsoluteTime timeStamp;
};

```

digdelu, 2016-06-08:
Big thx Man! good news in morning)
am on Yosmite(last upd) and its works great for me. am a guitar player a long time and my EMU0404 USB is my friend from 2004) many years playing in many clubs in russia and nothing cant stop my EMU.
I want to say THANK YOU!!! for Your job for us.
cheers)

Wouter1, 2016-06-08:
@digdelu thanks! On Yosemite the driver indeed was already working fine. This ticket is a working ticket for me to get it compile on El Capitan, in an attempt to get the low-latency connection back with El Capitan.

Wouter1, 2016-06-14:
apple changed the low latency 'busy' status from kUSBLowLatencyIsochTransferKey ('llit') to kIOReturnInvalid. It would not surprise me if this were the heart of the problem

Wouter1, 2016-06-14:
Trying to find substitute for IOUSBInterface#getInterfaceNumber()

Checking Apple's source code 

http://opensource.apple.com//source/IOUSBFamily/IOUSBFamily-205.3.5/IOUSBFamily/Classes/IOUSBInterface.cpp

where I see
```
UInt8  
IOUSBInterface::GetInterfaceNumber() 
{ 
    return _bInterfaceNumber; 
}
```

```
    _iInterface = _interfaceDesc->iInterface;
```

Checking .h file

```
class IOUSBInterface : public IOUSBNub
{

...
protected:
    const IOUSBInterfaceDescriptor *	_interfaceDesc;
```

Wouter1, 2016-06-14:
The interfaceNumber now is in InterfaceDescriptor.bInterfaceNumber.

Wouter1, 2016-06-14:
mBus in EMUUSBAudioDevice was not even used. Removed.

Wouter1, 2016-06-16:
Trouble replacing FindNextPipe was in IOUSBInterface but not in USBHostInterface. 
In IOUSBHostInterface it says 

    // Replacement: getInterfaceDescriptor and StandardUSB::getNextAssociatedDescriptorWithType to find an endpoint descriptor,
    // then use copyPipe to retrieve the pipe object

which is too vague to use directly

Wouter1, 2016-06-16:
I will try to use code from 
https://github.com/RehabMan/OS-X-BrcmPatchRAM/blob/master/BrcmPatchRAM/USBHostDeviceShim.cpp

Wouter1, 2016-06-16:
the old struct specifying the required pipe properties was this
```
typedef struct {
    UInt8 type;
    UInt8 direction;
    UInt16 maxPacketSize;
    UInt8 interval;
} IOUSBFindEndpointRequest;
```

The suggested getNextEndpointDescriptor takes 
```
    struct ConfigurationDescriptor
    {
        uint8_t     bLength;
        uint8_t     bDescriptorType;
        uint16_t    wTotalLength;
        uint8_t     bNumInterfaces;
        uint8_t     bConfigurationValue;
        uint8_t     iConfiguration;
        uint8_t     bmAttributes;
        uint8_t     bMaxPower;
    }

```

Wouter1, 2016-06-16:
This is the source code of the old FindNextPipe 

http://opensource.apple.com//source/IOUSBFamily/IOUSBFamily-210.4.5/IOUSBFamily/Classes/IOUSBInterface.cpp

```

IOUSBPipe*
IOUSBInterface::FindNextPipe(IOUSBPipe *current,
                                IOUSBFindEndpointRequest *request)
{
    const IOUSBController::Endpoint *	endpoint;
    IOUSBPipe *				pipe;
    int					numEndpoints;
    int					i;

    numEndpoints = _bNumEndpoints;

    if (request == 0)
        return NULL;

    if(current != 0) 
    {
        for(i=0;i < numEndpoints; i++) 
        {
            if(_pipeList[i] == current) 
            {
		i++; // Skip the one we just did
                break;
            }
	}
    }
    else
        i = 0;	// Start at beginning.

    for ( ;i < numEndpoints; i++) {
        pipe = _pipeList[i];
        if(!pipe)
            continue;
        endpoint = pipe->GetEndpoint();

        // check the request parameters
        if (request->type != kUSBAnyType &&
            request->type != endpoint->transferType)
            pipe = 0;		// this is not it
        else if (request->direction != kUSBAnyDirn &&
            request->direction != endpoint->direction)
            pipe = 0;		// this is not it

        if (pipe == 0)
            continue;

        request->type = endpoint->transferType;
        request->direction = endpoint->direction;
        request->maxPacketSize = endpoint->maxPacketSize;
        request->interval = endpoint->interval;
        return(pipe);
    }

    return(0);
}
```

Wouter1, 2016-06-16:
All calls in the code use first argument NULL, so they are just to find the FIRST pipe meeting the requirements. I will stub that out to ensure it's always null, this simplifies the code

Wouter1, 2016-06-16:
not all searches use all 4 fields of the request.
one only type and direction

Wouter1, 2016-06-16:
The original code only checked if type and direction match! The rest is apparently ignored

Wouter1, 2016-06-16:
And our driver does not use the details that the original code filled in after finding a matching pipe, so we can leave that out too.

Wouter1, 2016-06-17:
DeviceRequest (previously IOUSBDevRequest) does not have a pData field anymore

There is a function deviceRequest in IOUSBHostDevice that takes dataBuffer as argument
```
 IOReturn deviceRequest(IOService* forClient, StandardUSB::DeviceRequest& request, void* dataBuffer, IOUSBHostCompletion* completion, uint32_t completionTimeoutMs = kUSBHostDefaultControlCompletionTimeoutMS);
```

previously it was this
```
 IOReturn DeviceRequest(IOUSBDevRequest	*request,
                                IOUSBCompletion	*completion = 0);
```

and there also was this one, defined in EMUUBAudioMuteControl

```
typedef IOReturn (*USBDeviceRequest)(IOUSBDevRequest * request, void * refCon , Completion * completion );
```

the latter is puzzling, if that one exists, why is it not devlared in

Wouter1, 2016-06-17:
refCon was a parameter coming in in the create call to EMUUSBAudioMuteControl. Not clear what it is

The create parameters do not match those in the parent class IOAudioToggleControl

Wouter1, 2016-06-18:
I need to figure out who is calling EMUUSBAudioMuteControl::init and ::create

The parameters do not match the IOAudioToggleControl::init and ::create function parameters

The USBDeviceRequest which is part of the actual init() needs to be changed for generalization

Wouter1, 2016-06-18:
the init is called from the create. 
I think the create is done from EMUUSBAudioDevice :+1: 
```
					IOAudioToggleControl*	muteControl = IOAudioToggleControl::createMuteControl(deviceCur, channelNum, 0, featureUnitID, usage);

```

This function is deprecated but still is available. No documentation.

Checked the source code 

http://opensource.apple.com//source/IOAudioFamily/IOAudioFamily-1.1fc9/IOAudioToggleControl.cpp

Nothing special there, and no sign of the parameters that EMUUSBMuteControl::create takes.

Wouter1, 2016-06-18:
Wow, I removed the entire create() from the EMUUSBAudioMuteControl and the driver just works fine, including muting the audio.

Wouter1, 2016-06-18:
Removed the entire EMUUSBAudioMuteControl files. Driver still compiles and works fine.

Wouter1, 2016-06-18:
The EMUUSBAudioLevelControl also uses the same create() function.

Wouter1, 2016-06-18:
Removed the EMUUSBAudioLevelControl class and code. Driver still works fine (also checked with Hallab ; checked recording; checked control panel).

Wouter1, 2016-06-18:
That explains the un-understandable EMUUSBAudioMuteControl *create definition and clears the way for generalizing the deviceRequest call

Wouter1, 2016-06-23:
There's only 1 call to IOUSBDevice DeviceRequest. All others are to IOUSBInterface.

Wouter1, 2016-06-23:
The moment I include USBAudioObject.h int the IOUSBDevice.h, the whole class IOUSBDevice1 becomes invisible to the compiler. But I need it to resolve SET_CUR etc.

Wouter1, 2016-06-23:
I have this in USBAudioObject.h
#include <IOUSBInterface.h>

Guess that's where things screw up

Wouter1, 2016-06-24:
The calls to IOUSBInterface::DeviceRequest are hard to generalize. The new version is IOUSBHostInterface:deviceRequest.

There are two issues to work around

1. The first argument, IOUSBDevRequest and IOUSBDevRequestDesc, has changed structurally. The fields have different names. One field has been removed. 

2.The second argument,  Completion, is optional. If it is null, then the call is blocking and we could arrange all required structures internally in our generalized stub. But if Completion is not null, the call will not be blocking and the required structures have to persist when the call to deviceRequest returned, until the Completion has been called or maybe longer. This means the caller of our stub is to be the owner of the structures. Probably he should create those too.

That means we can not just stub the function call, we must stub the involved structures too

Wouter1, 2016-06-24:
I see only two places that actually use IOUSBDevRequest.

1. IOUSBInterface.h devRequest. Here I  already had  a IOUSBDevice#devRequest(..., completion) function. It made a local IOUSBDevRequest . I should remove completion, it can't work properly if I craete a local IOUSBDevRequest on the stack locally. It must be that completion =0 always when this is called.

2. IOUSBDevice.h devRequestSampleRate. Here, completion =0.

Wouter1, 2016-06-24:
XCode search function appears to be not working correctly. I do a search for IOUSBDevRequestDesc and XCode fails to deliver the occurence in EMUUSBAudioDevice line 2181. It can have severe impact on my refactoring if I miss some cases.

Wouter1, 2016-06-24:
Several optional "completion" arguments in devRequest functions were never used. Removed those, this removes a lot of the issues mentioned above

Wouter1, 2016-06-26:
Need to mimic FindNextInterface in 10.11

http://opensource.apple.com/source/IOUSBFamily/IOUSBFamily-516.4.1/IOUSBFamily/Classes/IOUSBDevice.cpp

This is the original code, thanks Apple for making it open source :)
```
/*
 * findNextInterface
 * This method should really be rewritten to use iterators or
 * be broken out into more functions without a request structure.
 * Or even better, make the interfaces and endpoint objects that
 * have their own methods for this stuff.
 *
 * returns:
 *   next interface matching criteria.  0 if no matches
 */
IOUSBInterface *
IOUSBDevice::FindNextInterface(IOUSBInterface *current, IOUSBFindInterfaceRequest *request)
{
    IOUSBInterfaceDescriptor 		*id = NULL;
    IOUSBInterface 			*intf = NULL;
    
    if (current)
    {
        // get the descriptor for the current interface into id
        
        // make sure it is really an authentic IOUSBInterface
        if (!OSDynamicCast(IOUSBInterface, current))
            return NULL;
		while (true)
		{
			if (FindNextInterfaceDescriptor(NULL, id, request, &id) != kIOReturnSuccess)
				return NULL;
			if (GetInterface(id) == current)
				break;
		}
    }
    
    while (!intf)
    {
		// now find either the first interface descriptor which matches, or the first one after current
		if (FindNextInterfaceDescriptor(NULL, id, request, &id) != kIOReturnSuccess)
		{
			return NULL;
		}
		
		intf =  GetInterface(id);
		// since not all interfaces (e.g. alternate) are instantiated, we only terminate if we find one that is
    }
    return intf;
    
}

```

Wouter1, 2016-06-26:
Previously, IOUSBInterface#deviceRequest could return a kIOUSBPipeStalled

But that constant seems not to exist anymore as return value.

I can't seem to find IOUSBHostInterface source code. The opensource.apple.com most recent stuff is IOUSBFamily-630.4.5/ eg

http://opensource.apple.com/source/IOUSBFamily/IOUSBFamily-630.4.5/IOUSBFamily/Headers/

and this seems NOT el capitan....

Wouter1, 2016-06-26:
Checked here https://opensource.apple.com/release/os-x-10112/ but this seems not to include the IOKit

Wouter1, 2016-06-26:
I hit on IOSyncer in 10.11! https://github.com/opensource-apple/xnu/blob/10.11/iokit/IOKit/IOSyncer.h

I always thought this had been deprecated.

So it's still there but I can't seem to include it.

Wouter1, 2016-06-26:
Checking the BrcmPatchRAM file at https://github.com/RehabMan/OS-X-BrcmPatchRAM/blob/master/BrcmPatchRAM/BrcmPatchRAM.cpp . That works supposedly in El Capitan and DOES use kIOUSBPipeStalld.

Wouter1, 2016-06-26:
Found it, it's just in IOKit/usb/USB.h. Not sure what was the reason I didn't include it in USB.h but it now seems fine.

Wouter1, 2016-06-27:
This is deprecated already in 10.9 and not available anymore. But no substitution is proviced by apple
```
 virtual IOReturn Read(IOMemoryDescriptor *	buffer,
                       IOUSBCompletion *	completion = 0,
			  IOByteCount *		bytesRead = 0);

```

Wouter1, 2016-06-27:
This was the source code of that 
```

// Transferring Data
IOReturn 
IOUSBPipe::Read(IOMemoryDescriptor *buffer, IOUSBCompletion *completion, IOByteCount *bytesRead)
{
    USBLog(7, "IOUSBPipe[%p]::Read #1", this);
    return Read(buffer, 0, 0, completion, bytesRead);
}
```

so the values were framestart = numframes = 0.

Wouter1, 2016-06-28:
Tried to load the 10.11 compiled driver. The output is a bit messy but looks like something is not linked properly
```
Jun 28 21:18:35 wouters-MBP sudo[504]:   wouter : TTY=ttys001 ; PWD=/Volumes/prive/EMU-driver/10.11 ; USER=root ; COMMAND=/sbin/kextunload /System/Library/Extensions/EMUUSBAudio.kext
Jun 28 21:18:35 wouters-MBP kernel[0]: Kext com.emu.driver.EMUUSBAudio not found for unload request.
Jun 28 21:18:35 wouters-MBP sudo[507]:   wouter : TTY=ttys001 ; PWD=/Volumes/prive/EMU-driver/10.11 ; USER=root ; COMMAND=/bin/rm -r /System/Library/Extensions/EMUUSBAudio.kext
Jun 28 21:18:35 wouters-MBP sudo[509]:   wouter : TTY=ttys001 ; PWD=/Volumes/prive/EMU-driver/10.11 ; USER=root ; COMMAND=/bin/cp -r EMUUSBAudio.kext /System/Library/Extensions
Jun 28 21:18:35 wouters-MBP sudo[511]:   wouter : TTY=ttys001 ; PWD=/Volumes/prive/EMU-driver/10.11 ; USER=root ; COMMAND=/bin/rm -rf /Library/Audio/MIDI Drivers/EMUMIDIDriver.plugin
Jun 28 21:18:35 wouters-MBP sudo[513]:   wouter : TTY=ttys001 ; PWD=/Volumes/prive/EMU-driver/10.11 ; USER=root ; COMMAND=/bin/mkdir -p /Library/Audio/MIDI Drivers/
Jun 28 21:18:35 wouters-MBP sudo[515]:   wouter : TTY=ttys001 ; PWD=/Volumes/prive/EMU-driver/10.11 ; USER=root ; COMMAND=/bin/cp -R EMUMIDIDriver orig.plugin /Library/Audio/MIDI Drivers/EMUMIDIDriver.plugin
Jun 28 21:18:35 wouters-MBP sudo[517]:   wouter : TTY=ttys001 ; PWD=/Library/Audio/MIDI Drivers ; USER=root ; COMMAND=/usr/bin/chgrp -R wheel EMUMIDIDriver.plugin
Jun 28 21:18:35 wouters-MBP sudo[519]:   wouter : TTY=ttys001 ; PWD=/System/Library/Extensions ; USER=root ; COMMAND=/usr/bin/chgrp -R wheel EMUUSBAudio.kext
Jun 28 21:18:35 wouters-MBP sudo[521]:   wouter : TTY=ttys001 ; PWD=/System/Library/Extensions ; USER=root ; COMMAND=/sbin/kextload -v EMUUSBAudio.kext
Jun 28 21:18:35 wouters-MBP com.apple.kextd[43]: Cache file /System/Library/Caches/com.apple.kext.caches/Directories/System/Library/Extensions/KextIdentifiers.plist.gz is out of date; not using.
Jun 28 21:18:35 wouters-MBP com.apple.kextd[43]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"
Jun 28 21:18:36 wouters-MBP com.apple.kextd[43]: kext signature failure override allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"
Jun 28 21:18:36 wouters-MBP kernel[0]: kxld[com.emu.driver.EMUUSBAudio]: The following symbols are unresolved for this kext:
Jun 28 21:18:36 wouters-MBP kernel[0]: kxld[com.emu.driver.EMUUSBAudio]: 	__ZN11StandardUSB15getEndpointTypeEPKNS_18EndpointDescriptorE
Jun 28 21:18:36 wouters-MBP kernel[0]: kxld[com.emu.driver.EMUUSBAudio]: 	__ZN11StandardUSB18getEndpointAddressEPKNS_18EndpointDescriptorE
Jun 28 21:18:36 wouters-MBP kernel[0]: kxld[com.emu.driver.EMUUSBAudio]: 	__ZN11StandardUSB20getEndpointDirectionEPKNS_18EndpointDescriptorE
Jun 28 21:18:36 wouters-MBP kernel[0]: kxld[com.emu.driver.EMUUSBAudio]: 	__ZN11StandardUSB25getNextEndpointDescriptorEPKNS_23ConfigurationDescriptorEPKNS_19InterfaceDescriptorEPKNS_10DescriptorE
Jun 28 21:18:36 wouters-MBP kernel[0]: kxld[com.emu.driver.EMUUSBAudio]: 	__ZN15IOUSBHostDevice9metaClassE
Jun 28 21:18:36 wouters-MBP kernel[0]: kxld[com.emu.driver.EMUUSBAudio]: 	__ZN18IOUSBHostInterface9metaClassE
Jun 28 21:18:36 wouters-MBP kernel[0]: kxld[com.emu.driver.EMUUSBAudio]: 	__ZN9IOUSBPipe5WriteEP18IOMemoryDescriptoryjP19LowLatencyIsocFrameP20LowLatencyCompletionj
Jun 28 21:18:36 wouters-MBP kernel[0]: kxld[com.emu.driver.EMUUSBAudio]: 	_utf8_encodestr
Jun 28 21:18:36 wouters-MBP kernel[0]: Can't load kext com.emu.driver.EMUUSBAudio - link failed.
Jun 28 21:18:36 wouters-MBP kernel[0]: Failed to load executable for kext com.emu.driver.EMUUSBAudio.
Jun 28 21:18:36 wouters-MBP kernel[0]: Kext com.emu.driver.EMUUSBAudio failed to load (0xdc008016).
Jun 28 21:18:36 wouters-MBP kernel[0]: Failed to load kext com.emu.driver.EMUUSBAudio (error 0xdc008016).
Jun 28 21:18:36 wouters-MBP com.apple.kextd[43]: Failed to load EMUUSBAudio.kext - (libkern/kext) link error.
Jun 28 21:18:36 wouters-MBP login[494]: DEAD_PROCESS: 494 ttys001
Jun 28 21:19:19 wouters-MBP kernel[0]: IO80211AWDLPeerManager::setAwdlOperatingMode Setting the AWDL operation mode from AUTO to SUSPENDED
Jun 28 21:19:19 wouters-MBP kernel[0]: IO80211AWDLPeerManager::setAwdlSuspendedMode() Suspending AWDL, enterQuietMode(true)
Jun 28 21:19:22 wouters-MBP kernel[0]: IO80211AWDLPeerManager::setAwdlOperatingMode Setting the AWDL operation mode from SUSPENDED to AUTO
Jun 28 21:19:22 wouters-MBP kernel[0]: IO80211AWDLPeerManager::setAwdlAutoMode Resuming AWDL
Jun 28 21:19:31 wouters-MBP TextMate[262]: [default] Failed to create bookmark with error: Error Domain=NSCocoaErrorDomain Code=260 "The file couldn’t be opened because it doesn’t exist."
Jun 28 21:19:31 --- last message repeated 1 time ---
Jun 28 21:19:31 wouters-MBP TextMate[262]: -[SFLMutableListItem resolveWithResolutionFlags:error:] Failed to create bookmark with error: Error Domain=NSCocoaErrorDomain Code=260 "The file couldn’t be opened because it doesn’t exist."
Jun 28 21:19:47 --- last message repeated 1 time ---
Jun 28 21:19:47 wouters-MBP com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.TextMate.262): Could not resolve origin of domain. XPC services in this domain's bundle will not be bootstrapped: error = 109: Invalid property list, taint = (null)
[... cut ...]
Jun 28 21:21:26 wouters-MBP kernel[0]: IO80211AWDLPeerManager::setAwdlOperatingMode Setting the AWDL operation mode from AUTO to SUSPENDED
Jun 28 21:21:26 wouters-MBP kernel[0]: IO80211AWDLPeerManager::setAwdlSuspendedMode() Suspending AWDL, enterQuietMode(true)
Jun 28 21:21:30 wouters-MBP kernel[0]: IO80211AWDLPeerManager::setAwdlOperatingMode Setting the AWDL operation mode from SUSPENDED to AUTO
Jun 28 21:21:30 wouters-MBP kernel[0]: IO80211AWDLPeerManager::setAwdlAutoMode Resuming AWDL

```

Wouter1, 2016-06-28:
No clue... Tried to load the extension into /Library/Extensions instead. Makes no difference.

Wouter1, 2016-06-30:
I can see these issues without actually installing the kext, using ```kextutil -n EMUUSBAudio.kext/```

```
Diagnostics for EMUUSBAudio.kext:
Warnings: 
    Dependency lacks appropriate value for OSBundleRequired and may not be availalble during early boot: 
        com.apple.kec.Libm - OSBundleRequired not set
        com.apple.vecLib.kext - OSBundleRequired not set
        com.apple.iokit.IOAudioFamily - OSBundleRequired not set

Code Signing Failure: not code signed
Warnings: 
    Dependency lacks appropriate value for OSBundleRequired and may not be availalble during early boot: 
        com.apple.kec.Libm - OSBundleRequired not set
        com.apple.vecLib.kext - OSBundleRequired not set
        com.apple.iokit.IOAudioFamily - OSBundleRequired not set

kxld[com.emu.driver.EMUUSBAudio]: The following symbols are unresolved for this kext:
kxld[com.emu.driver.EMUUSBAudio]: 	StandardUSB::getEndpointType(StandardUSB::EndpointDescriptor const*)
kxld[com.emu.driver.EMUUSBAudio]: 	StandardUSB::getEndpointAddress(StandardUSB::EndpointDescriptor const*)
kxld[com.emu.driver.EMUUSBAudio]: 	StandardUSB::getEndpointDirection(StandardUSB::EndpointDescriptor const*)
kxld[com.emu.driver.EMUUSBAudio]: 	StandardUSB::getNextEndpointDescriptor(StandardUSB::ConfigurationDescriptor const*, StandardUSB::InterfaceDescriptor const*, StandardUSB::Descriptor const*)
kxld[com.emu.driver.EMUUSBAudio]: 	IOUSBHostDevice::metaClass
kxld[com.emu.driver.EMUUSBAudio]: 	IOUSBHostInterface::metaClass
kxld[com.emu.driver.EMUUSBAudio]: 	IOUSBPipe::Write(IOMemoryDescriptor*, unsigned long long, unsigned int, LowLatencyIsocFrame*, LowLatencyCompletion*, unsigned int)
kxld[com.emu.driver.EMUUSBAudio]: 	_utf8_encodestr
Link failed (error code 5).
Check library declarations for your kext with kextlibs(8).

```

Wouter1, 2016-06-30:
the utf8_encodestr comes straight out of the "Replacement" recommendation from Apple, see IOUSHBHostDevice.h line 892

Wouter1, 2016-06-30:
utf8_encodestr is part of vfs_utfconv.c , apparently some bsd library

opensource.apple.com//source/xnu/xnu-1504.15.3/bsd/vfs/vfs_utfconv.c

Wouter1, 2016-06-30:
Maybe some OSBundleLibraries have to be specified in the plist?

Wouter1, 2016-06-30:
Just guessed that the lib would have the name com.apple.kpi.vfs with version 13.4 minimum

However that doesn't work

```
EMUUSBAudio.kext - dependency 'com.apple.kpi.vfs' not found.

    No kexts found for these libraries: 
        com.apple.kpi.vfs

```

Wouter1, 2016-06-30:
And I probably now need to link against IOUSBHostFamily, not IOUSBFamily, in the same plist OSBundleLibraries.

Wouter1, 2016-06-30:
Mmm, that gives

```
EMUUSBAudio.kext - no compatible dependency found for com.apple.iokit.IOUSBHostFamily.

...
    Only incompatible kexts found for these libraries: 
        com.apple.iokit.IOUSBHostFamily
```

Wouter1, 2016-06-30:
```kextstat ``` gives a list of the actually loaded kexts, with version nrs

Wouter1, 2016-06-30:
IOUSBHostFamil has version 1.0.1. Maybe that's why it didn't load
```
   38   13 0xffffff7f814af000 0x69000    0x69000    com.apple.iokit.IOUSBHostFamily (1.0.1) 4C8B5BB6-6AE4-313E-B79C-AC07A4E31A2D <37 7 5 4 3 1>
   39    5 0xffffff7f81555000 0x9a000    0x9a000    com.apple.iokit.IOUSBFamily (900.4.1) 7B5AC81A-D0B6-3F3D-87C7-AFD78F4686DB <38 12 7 5 4 3 1>
```

Wouter1, 2016-06-30:
That gives me
```
kxld[com.emu.driver.EMUUSBAudio]: 	IOUSBDevice::metaClass
kxld[com.emu.driver.EMUUSBAudio]: 	IOUSBInterface::metaClass
```

But the other errs are now gone. Maybe this is the right track and that I have some refs, or implemented some stub incorrectly

Wouter1, 2016-07-01:
There is a ref to IOUSBInterface in the plist. Maybe that should be IOUSBHostDevice on 10.11

Wouter1, 2016-07-01:
Trying to figure out where the IOUSBDevice reference comes from

Using nm I see these refs in this order
```
000000000002eee0 s __ZL11kDB_To_Gain
0000000000011680 t __ZN10Completion3setEPvPFvS0_S0_ijES0_
0000000000023ef0 T __ZN10StreamInfo14getNextFrameNrEv
0000000000023d50 T __ZN10StreamInfo4initEv
0000000000023e30 T __ZN10StreamInfo5resetEv
0000000000023d70 T __ZN10StreamInfo5startEy
                 U __ZN11IOUSBDevice9metaClassE
                 U __ZN11OSMetaClass21_RESERVEDOSMetaClass0Ev
                 U __ZN11OSMetaClass21_RESERVEDOSMetaClass1Ev
                 U __ZN11OSMetaClass21_RESERVEDOSMetaClass2Ev
                 U __ZN11OSMetaClass21_RESERVEDOSMetaClass3Ev
                 U __ZN11OSMetaClass21_RESERVEDOSMetaClass4Ev
                 U __ZN11OSMetaClass21_RESERVEDOSMetaClass5Ev
                 U __ZN11OSMetaClass21_RESERVEDOSMetaClass6Ev
                 U __ZN11OSMetaClass21_RESERVEDOSMetaClass7Ev
                 U __ZN11OSMetaClassC2EPKcPKS_j
                 U __ZN11OSMetaClassD2Ev
0000000000002820 t __ZN11OSMetaClassdlEPvm
0000000000023d00 t __ZN11OSMetaClassdlEPvm
0000000000024eb0 t __ZN11OSMetaClassdlEPvm
000000000000f980 t __ZN11OSMetaClassdlEPvm
00000000000246e0 t __ZN11OSMetaClassdlEPvm
0000000000006f10 t __ZN11OSMetaClassdlEPvm
000000000001a260 t __ZN11OSMetaClassdlEPvm
0000000000023650 t __ZN11RingBufferTIhEC2Ev
00000000000236d0 t __ZN11RingBufferTIjEC2Ev
0000000000012110 t __ZN12IOUSBDevice111isHighSpeedEv
0000000000019df0 t __ZN12IOUSBDevice114getFrameNumberEv

```

The StreamInfo refs probably come from StreamInfo.cpp

The next thing I recognise in the names is the IOUSBDevice1 code.

Could it be that the ending on "1" confuses the linker ?

Wouter1, 2016-07-01:
StreamInfo code in start could be generalized with

    ReturnIf(startUsbFrame < streamInterface->getDevice1()->getFrameNumber() + 10, kIOReturnTimeout);

Wouter1, 2016-07-01:
Seems I screwed up something, 10.11 does not compile anymore. 
The errors come from utfconv.h, this is a systme file?!?!?

Seems XCode damaged the file, the modification time is today. All other files are from 27 aug 2015

Wouter1, 2016-07-01:
Thanks to the open source on 
```
http://opensource.apple.com//source/xnu/xnu-3248.20.55/bsd/sys/utfconv.h?txt
```

I could figure out that 'utf8_encodestr' had been erased in that file, leaving jjust a list of parameters ....

I dont understand why these files are writable. Will try to change that...

[14 more replies]

closed 2016-07-01 by Wouter1