Wouter1/EMU-driver #89: Failed to load the kext on Yosemite

DMax525, 2016-07-25:
Hello!
I downloaded the .zip according to the instructions. When I run the installer without manufacturer drivers being installed, a terminal window shows this:

> EMU USB Installer by W.Pasman
> Use low latency setting (n/y)?: n
> 
> 
> cp: v9/EMUUSBAudio.kext: No such file or directory
> (Re)placing EMU kernel extension. We need the admin password for this.
> 
> Password:
> cp: EMUUSBAudio.kext: No such file or directory
> cp: EMUMIDIDriver orig.plugin: No such file or directory
> chgrp: EMUMIDIDriver.plugin: No such file or directory
> chgrp: EMUUSBAudio.kext: No such file or directory
> /System/Library/Extensions/EMUUSBAudio.kext failed to load - (libkern/kext) not found; check the system/kernel logs for errors or try kextutil(8).
> Waiting for the system to finish installation. Can take several minutes.
And nothing happens anymore. Kext files don't appear in S/L/E.

Then I tried to install your driver over the manufacturer's driver. In this case it shows this:

> EMU USB Installer by W.Pasman
> Use low latency setting (n/y)?: n
> 
> 
> cp: v9/EMUUSBAudio.kext: No such file or directory
> (Re)placing EMU kernel extension. We need the admin password for this.
> 
> Password:
> cp: EMUUSBAudio.kext: No such file or directory
> cp: EMUMIDIDriver orig.plugin: No such file or directory
> chgrp: EMUMIDIDriver.plugin: No such file or directory
> chgrp: EMUUSBAudio.kext: No such file or directory
> /System/Library/Extensions/EMUUSBAudio.kext failed to load - (libkern/kext) not found; check the system/kernel logs for errors or try kextutil(8).
> Waiting for the system to finish installation. Can take several minutes.
> Rebuilding the kernel
> Waiting for lock to replace the old kernel
> Driver is installed and ready for use!
> logout

But still no result after that. Kexts do not appear in folders they must be.
Then I try to put them manually (EMUUSBAudio.kext in S/L/E and midi driver in Library/Audio/Midi Drivers). Then I used Kext Utility and rebooted the machine. Still nothing. System info>software>Kexts shows that the kext is not loaded and that there's dependency resolution failures:
> Dependency Resolution Failures:
>   Kexts already loaded for these libraries are not compatible with the requested version:	com.apple.kpi.bsd
>   No kexts found for these libraries:	com.apple.iokit.IOUSBHostFamily

And finally when I try to load the kext manually via the Terminal, it shows this:

> Requesting load of /System/Library/Extensions/EMUUSBAudio.kext.
> /System/Library/Extensions/EMUUSBAudio.kext failed to load - (libkern/kext) dependency resolution failure; check the system/kernel logs for errors or try kextutil(8).

DMax525, 2016-07-25:
UPD: I was manually installing the kext from v11 folder. Now I installed kext from v9 folder and it seems to work fine. All clicks and cracks that I had with manufacturer's driver are gone. Could you please tell me, what's the difference between v9 and v11?

Wouter1, 2016-07-25:
Thanks for your interest in my driver.

The error message says that the installer can not find the kernel extension it wants to install. 

Can you please check the zip file that you downloaded. Can you confirm (1) the installer program is in that directory (2) confirm that the v9 directory is in there too (3) the v9 directory contains EMUUSBAudio.kext? 

To answer your question, the V9 version is for osx9 and 10. The v11 version is for osx11. FAIK Yosemite needs v9.

DMax525, 2016-07-25:
Thanks for your reply!

Yes, everything is in its place, if get you right. 
[image]
[image]

BTW, as I added before, after manual installation of v9 kext it works fine, but only after I manually load the kext via the terminal app.

Wouter1, 2016-07-26:
It seems something goes wrong with the recent fix that was suggested by @vizh . It seems $PWD is not returning the actual directory where the script is placed but rather the user's home dir

Wouter1, 2016-07-26:
Trying to follow http://stackoverflow.com/questions/59895/can-a-bash-script-tell-which-directory-it-is-stored-in

But  ```readlink -f``` gives me 
```
readlink: illegal option -- f

```

Wouter1, 2016-07-26:
@DMax525  I rolled back a fix that was apparently not working correctly. The current version of the installer should work again for you.

Wouter1, 2016-07-26:
This means we have back the problem @vizh reported before. But we need to fix it  differently.

Wouter1, 2016-07-26:
previously vizh reported that he needed to use $PWD instead, because

>Installer won't work when dirname $0 returns just the dot. In this case cd "$pathname" has no effect.

Wouter1, 2016-07-26:
We seem to have this

| action | value of $PWD | value of $0 |
|----|-------|-------|
|click on script|$HOME|full path to app|
|terminal: ./script|current dir|path to app relative to current dir|

Wouter1, 2016-07-26:
So switching to the $0 directory and then using pwd seems the only solution that works always

```pathname=$( cd $(dirname $0) ; pwd -P ) ```

Wouter1, 2016-07-26:
@DMax525  @vizh  can you confirm the new script works fine for you ?

DMax525, 2016-07-26:
@Wouter1 yes, I re-downloaded the zip, deleted the old kext and everything seems to be installed perfectly. And also the kext loads automatically after reboot.
Thank you very much for your development! It's really amazing that you keep good old devices updated. Now I think I'm gonna update to El Capitan and try it there.

Wouter1, 2016-07-26:
@DMax525  Thanks for the feedback and the nice comments!

Other software also may break on El Capitan (eg MS office, adobe stuff). I'd recommend to keep your old installation as well so that you can easily switch back.

DMax525, 2016-07-26:
@Wouter1 I've just discovered another issue. Sometimes sound disappears, usually after I open a logic x project or run Guitar Rig 5. Restart of the card doesn't seem to solve the problem. And when I turn it off, it is still in the list of devices in system preferences>sound. I tried to unload the kext via the terminal, but it doesn't help. System reboot solves the problem.

Wouter1, 2016-07-26:
Thanks for reporting. Please post new issues in new tickets (after checking the existing issues)
I made new ticket #90 for this

closed 2016-07-26 by Wouter1