Wouter1/EMU-driver #90: sound disappears

Wouter1, 2016-07-26:
@DMax55 reported

Sometimes sound disappears, usually after I open a logic x project or run Guitar Rig 5. Restart of the card doesn't seem to solve the problem. And when I turn it off, it is still in the list of devices in system preferences>sound. I tried to unload the kext via the terminal, but it doesn't help. System reboot solves the problem.

Wouter1, 2016-07-26:
@DMax55 which OSX? Can you also report what the /var/log/system.log is saying when this happens?

I can't reproduce this as I dont have logic X nor Guitar Rig 5. Did you see it with reaper or audacity?

Wouter1, 2016-07-26:
I think I have seen this occasionally happening when I use the flash plugin. Might be related.

DMax525, 2016-07-26:
@Wouter1 It's Yosemite. I don't have Reaper nor Audacity. I'll check the log next time this happens and post it here.

Wouter1, 2016-07-26:
Thanks, first step is figuring out how to reproduce reliably.

DMax525, 2016-08-07:
@Wouter1 It seems like everything works fine so far. Thank you very much again for your drive

Wouter1, 2016-08-07:
@DMax525 Thanks for reporting back.   I didn't see the issue yet either.

I will then close this as we can't reproduce.

Please re-open and post the log and apple bug report if the issue re-appears.

DMax525, 2016-08-12:
@Wouter1 Good day! Seems like it's happening again.
Here's what the log says

> Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: +EMUUSBAudioEngine[0xffffff8019a0a000]::performAudioEngineStart ()
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: create output pipe 
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: check for associated endpoint
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: frameSizeQueue empty, guessing some queue size. May need fix..
Aug 12 19:58:32 --- last message repeated 127 times ---
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:32 --- last message repeated 1 time ---
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: frameSizeQueue empty, guessing some queue size. May need fix..
Aug 12 19:58:32 --- last message repeated 63 times ---
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:32 --- last message repeated 2 times ---
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:32 --- last message repeated 3 times ---
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:33 --- last message repeated 3 times ---
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:33 --- last message repeated 3 times ---
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:33 --- last message repeated 4 times ---
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: LowPassFilter::filter init 185759637
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: USB timer started
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: **** Output Hiccup!! firstSampleFrame=724, nextExpectedOutputFrame=0 bufsize=98304
Aug 12 19:58:34 MacBook-Air-dmax.local Logic Pro X[436]: There was an error parsing the Info.plist for the bundle at URL Info.plist -- file:///Users/dmax/
	 The data couldn’t be read because it isn’t in the correct format.
	 <CFBasicHash 0x600001a6f780 [0x7fff74a8eed0]>{type = immutable dict, count = 1,
	entries =>
		0 : <CFString 0x7fff74a59660 [0x7fff74a8eed0]>{contents = "NSDebugDescription"} = <CFString 0x600001a68200 [0x7fff74a8eed0]>{contents = "Cannot parse a NULL or zero-length data"}
	}
Aug 12 19:58:34 MacBook-Air-dmax.local Logic Pro X[436]: There was an error parsing the Info.plist for the bundle at URL Info.plist -- file:///Users/dmax/
	 The data couldn’t be read because it isn’t in the correct format.
	 <CFBasicHash 0x600001a6e000 [0x7fff74a8eed0]>{type = immutable dict, count = 1,
	entries =>
		0 : <CFString 0x7fff74a59660 [0x7fff74a8eed0]>{contents = "NSDebugDescription"} = <CFString 0x600001a62c40 [0x7fff74a8eed0]>{contents = "Cannot parse a NULL or zero-length data"}
	}
Aug 12 19:58:34 MacBook-Air-dmax.local Logic Pro X[436]: There was an error parsing the Info.plist for the bundle at URL Info.plist -- file:///Users/dmax/
	 The data couldn’t be read because it isn’t in the correct format.
	 <CFBasicHash 0x608001a7b8c0 [0x7fff74a8eed0]>{type = immutable dict, count = 1,
	entries =>
		0 : <CFString 0x7fff74a59660 [0x7fff74a8eed0]>{contents = "NSDebugDescription"} = <CFString 0x608001a7b900 [0x7fff74a8eed0]>{contents = "Cannot parse a NULL or zero-length data"}
	}
Aug 12 19:58:34 MacBook-Air-dmax.local Logic Pro X[436]: There was an error parsing the Info.plist for the bundle at URL Info.plist -- file:///Users/dmax/
	 The data couldn’t be read because it isn’t in the correct format.
	 <CFBasicHash 0x600001a6cf40 [0x7fff74a8eed0]>{type = immutable dict, count = 1,
	entries =>
		0 : <CFString 0x7fff74a59660 [0x7fff74a8eed0]>{contents = "NSDebugDescription"} = <CFString 0x600001a6d080 [0x7fff74a8eed0]>{contents = "Cannot parse a NULL or zero-length data"}
	}
Aug 12 19:58:34 MacBook-Air-dmax.local Logic Pro X[436]: There was an error parsing the Info.plist for the bundle at URL Info.plist -- file:///Users/dmax/
	 The data couldn’t be read because it isn’t in the correct format.
	 <CFBasicHash 0x600001a6d100 [0x7fff74a8eed0]>{type = immutable dict, count = 1,
	entries =>
		0 : <CFString 0x7fff74a59660 [0x7fff74a8eed0]>{contents = "NSDebugDescription"} = <CFString 0x600001a6d280 [0x7fff74a8eed0]>{contents = "Cannot parse a NULL or zero-length data"}
	}
Aug 12 19:58:34 MacBook-Air-dmax.local Logic Pro X[436]: There was an error parsing the Info.plist for the bundle at URL Info.plist -- file:///Users/dmax/
	 The data couldn’t be read because it isn’t in the correct format.
	 <CFBasicHash 0x608001a7b900 [0x7fff74a8eed0]>{type = immutable dict, count = 1,
	entries =>
		0 : <CFString 0x7fff74a59660 [0x7fff74a8eed0]>{contents = "NSDebugDescription"} = <CFString 0x608001a7b8c0 [0x7fff74a8eed0]>{contents = "Cannot parse a NULL or zero-length data"}
	}
[... cut ...]
> Aug 12 20:03:54 MacBook-Air-dmax kernel[0]: EMUUSBAudioEngine::convertInputSamples err reading ring: e00002e7
> Aug 12 20:03:54 MacBook-Air-dmax kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP

So it happens the following way:
I open Logic X, making some recording and then the sound just suddenly disappears. The Logic keeps being sure that EMU is still there, connected and working well, but there's no sound. Trying to open system preferences>sound hangs the system. And I can't reboot it after that by pressing apple>reboot. Only **sudo shutdown -r now** reboots the system.

Wouter1, 2016-08-12:
@DMax525  Thanks for re-opening and posting the log. 

In your report I also see this

> Aug 12 19:58:44 MacBook-Air-dmax.local spindump[423]: Saved spin report for Logic Pro X version 10.2.0 (3987.29) to /Library/Logs/DiagnosticReports/Logic Pro X_2016-08-12-195844_MacBook-Air-dmax.spin

It looks like Logic Pro is crashing or became nonresponsive. Maybe because it can't read Info.plist -- file:///Users/dmax/ ? Maybe we can check the .spin report?

DMax525, 2016-08-14:
@Wouter1 here it is

Wouter1, 2016-08-14:
@DMax525  Thanks for posting the spindump. I'll check it to see if I can make something of it

Wouter1, 2016-08-14:
Thinning out the log file
```
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: +EMUUSBAudioEngine[0xffffff8019a0a000]::performAudioEngineStart ()
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: create output pipe
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: check for associated endpoint
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: frameSizeQueue empty, guessing some queue size. May need fix..
Aug 12 19:58:32 --- last message repeated 127 times ---
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:32 --- last message repeated 1 time ---
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: frameSizeQueue empty, guessing some queue size. May need fix..
Aug 12 19:58:32 --- last message repeated 63 times ---
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:32 --- last message repeated 2 times ---
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: RingBufferDefault::push warning. Ignoring overrun
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:32 --- last message repeated 3 times ---
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: RingBufferDefault::push warning. Ignoring overrun
Aug 12 19:58:32 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:33 --- last message repeated 3 times ---
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: RingBufferDefault::push warning. Ignoring overrun
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:33 --- last message repeated 3 times ---
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: RingBufferDefault::push warning. Ignoring overrun
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 19:58:33 --- last message repeated 4 times ---
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: RingBufferDefault::push warning. Ignoring overrun
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: LowPassFilter::filter init 185759637
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: USB timer started
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: RingBufferDefault::push warning. Ignoring overrun
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Aug 12 19:58:33 MacBook-Air-dmax kernel[0]: **** Output Hiccup!! firstSampleFrame=724, nextExpectedOutputFrame=0 bufsize=98304

Aug 12 19:58:34 MacBook-Air-dmax.local Logic Pro X[436]: There was an error parsing the Info.plist for the bundle at URL Info.plist -- file:///Users/dmax/ .....
Aug 12 19:58:44 MacBook-Air-dmax.local spindump[423]: Saved spin report for Logic Pro X version 10.2.0 (3987.29) to /Library/Logs/DiagnosticReports/Logic Pro X_2016-08-12-195844_MacBook-Air-dmax.spin

Aug 12 19:58:51 MacBook-Air-dmax.local powerd[48]: Failed to read current rating(0xe00002f0)
...
Aug 12 19:59:04 MacBook-Air-dmax kernel[0]: Sound assertion in AppleHDAWidget at line 2197
...
Aug 12 19:59:25 MacBook-Air-dmax kernel[0]: Google Chrome He (map: 0xffffff801f352a50) triggered DYLD shared region unnest for map: 0xffffff801f352a50, region 0x7fff90600000->0x7fff90800000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
....

Aug 12 20:03:54 MacBook-Air-dmax kernel[0]: EMUUSBAudioEngine::convertInputSamples err reading ring: e00002e7
Aug 12 20:03:54 MacBook-Air-dmax kernel[0]: EMUUSBAudioEngine::convertInputSamples READ HICKUP
Aug 12 20:03:54 MacBook-Air-dmax kernel[0]: RingBufferDefault::push warning. Ignoring overrun
Aug 12 20:03:54 --- last message repeated 4 times ---
Aug 12 20:03:54 MacBook-Air-dmax kernel[0]: warning way-out ring wrap position
Aug 12 20:03:54 MacBook-Air-dmax kernel[0]: RingBufferDefault::push warning. Ignoring overrun
```

Wouter1, 2016-08-14:
It's weird that we do not get any messages from the EMU driver between 19:58:33 and  20:03:54. Normally, the driver shwos also this *immediately* after the output hickup

kernel[0]: RingBufferDefault<USBInputRing>::push warning. Ignoring overrun

2e7 is a data underrun.

Wouter1, 2016-08-14:
@DMax525 it is hard to tell for sure what's going on here. Can you consistently reproduce this or is this  happening infrequently?

Is it possible that the logic pro crash *causes* the problem? 

It seems it is trying to read Info.plist -- file:///Users/dmax/ which fails. Could that be causing the crash? 

Also I see messages from google chrome interspersed in the log. Can this be related? Eg, do you also have the problem if chrome is not running?

DMax525, 2016-08-14:
@Wouter1 this happens quite infrequently. I doubt it's Logic that causes the problem, because this kind of trouble never happened with the manufacturer's driver.
I also thought that Chrome might be related, but I haven't yet tried to run Logic without Chrome.
It happens infrequently, but when it happens, it happens several times in a row and then again long time no see :D

Wouter1, 2016-08-14:
@DMax525  thanks for the info. 
Do you have any idea about the error message about the Info.plist file from Logic Pro?

Maybe you have a plugin re-located or told Logic a wrong address? Or maybe a plugin that is broken? Or something else that you once had in //Users/dmax but then moved it ?

https://logicprohelp.com/forum/viewtopic.php?t=119992&p=627414

DMax525, 2016-10-31:
@Wouter1 seems like everything is fine now. Thank you and sorry for not answering for so long

Wouter1, 2016-10-31:
@DMax525  great that it's working!

For future reference if others have the same problem, do you have any explanation on what was causing the problem and/or how you solved it

DMax525, 2016-10-31:
@Wouter1 it seems like it's worked out itself

Wouter1, 2016-10-31:
Oh well, sometimes these things happen. I'll close this then. Thanks!

closed 2016-10-31 by Wouter1