Wouter1/EMU-driver #91: test driver and installer on Sierra

Wouter1, 2016-10-13:
I need to test if the driver works on sierra.

One user reported

>It's just stuck in the middle with no error codes and without system utilization. But I did not test it much.

It may be that he forgot to enable kexts, or that he did not wait long enough (it can take several minutes before OSX finishes the installation).

denwijn, 2016-10-14:
I upgraded today from Yosemite, with a working EMU 0404 and your (older) driver, to Sierra.
In Sierra my 0404 was not recognized anymore in Audio/MIDI configuration.  
So I downloaded your latest driver, safeboot to terminal "csrutil enable --without kext" gave error message during install about unable to load kext, so I used "csrutil disable" this worked well.
Error disappeared, sound card is recognized again.  Thanks.

Wouter1, 2016-10-15:
@denwijn 

Thanks for reporting this. So if I understand you right 

```csrutil enable --without kext```

Is not working anymore in Sierra, but tHe more general

```csrutil disable```

still works and works as intended.

Does ```csrutil enable --without kext``` give you an error message or do you get the error message only later when you run the install script?

Wouter1, 2016-10-15:
I tried the command ```csrutil enable --without kext``` on Sierra 10.11.6 (15G31)

It works all fine here.

The driver (which was already installed in Capitan) still works fine.

Wouter1, 2016-10-15:
Ran the installer on Sierra. It works all fine here. It takes about 5 minutes for OSX to install the driver so you have to wait long enough.

Wouter1, 2016-10-15:
@denwijn 
>gave error message during install about unable to load kext

Are you referring to this message in the system.log file
>com.apple.kextd[43]: kext signature failure override allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"

This seems normal, it will later report 

> com.apple.kextcache[563]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext EMUUSBAudio.kext

denwijn, 2016-10-15:
I just did a re-install and walked through all steps again.
When I put in Terminal "csrutil enable --without kext" on Sierra, it returns the following message:
"csrutil: requesting an unsupported configuration. This is likely to break in the future and leave your machine in an unknown state.  Succesfully enabled System Integrity Protection. Please restart the machine for the changes to take effect."
So I rebooted anyway and tried to install. This worked for me too, without any error message in the install script.  I was patient enough and waited for 5 minutes, it seems to work fine now.  
Thanks and keep up the good work !

denwijn, 2016-10-15:
So the install script doesn't close automatically ?  A suggestion: maybe to mention in the installer to close the install script manually after waiting a few minutes?  Thanks a lot for the great work.

Wouter1, 2016-10-15:
@denwijn 

> it returns the following message: "csrutil: requesting an unsupported configuration. This is likely to break in the future and leave your machine in an unknown state. Succesfully enabled System Integrity Protection. Please restart the machine for the changes to take effect."

Yes this is a standard message. I dont know why Apple shows "may break in the future" as part of that ok message. 

>So the install script doesn't close automatically ?

Confirmed: I think the user wants to read the final OK and if I close the window he can't read it.

> A suggestion: maybe to mention in the installer to close the install script manually after waiting a few minutes?

You mean something like
>you can close this window now

 after the message 

>Driver is installed and ready for use

closed 2016-10-25 by Wouter1