Wouter1/EMU-driver #92: plugins

Wouter1, 2016-10-25:
@sookiesookie6 wrote (in #65)

This works for me except that a lot of plugins no longer work, when I switch back to built in audio they start working again. Just talking about built in plugins like the Pitch plugins etc.

Is there a setting I need to change perhaps?

Wouter1, 2016-10-25:
@sookiesookie6 

Is this on El Capitan?

Which plugins are you referring to? Built-in into what?

sookiesookie6, 2016-10-25:
Sorry I mean in Logic X (10) - I tried rebooting the machine (iMac on El Cap) and the plugins reappeared on the tracks I had created but then if I create a new track with the same plugins they are "missing" - googling online suggests it's a core audio issue - I am outputting audio to the EMU 0404 as well as inputting. Switching output to core audio resolves the issue - but I need to output to EMU. Any ideas?

Thank you for the amazing work on your driver, very much hoping not to have to buy new hardware and this has been a real life saver for getting audio into Logic!

Wouter1, 2016-10-25:
@sookiesookie6  I don't have Logic so I can't say much on that side.

Disappearing Logic plugins is a Logic issue, not a core audio issue or a driver issue.

The EMU driver seems even further away from Logic. Logic is working together with core audio. The EMU driver is just on the other side of the core audio fence. So they are never talking directly to each other, but through core audio. 

Did you try the normal-latency driver? Do you have enough buffer size in Logic? These are the main issues that can cause audio troubles in user land.

Are there other symptoms or logs besides plugins just disappearing?

Wouter1, 2016-10-25:
Are the symptoms mentioned here
https://support.native-instruments.com/hc/en-us/articles/209544429-OS-X-10-11-El-Capitan-Compatibility-News
relevant for your case?

sookiesookie6, 2016-10-25:
Ah I see ok, so EMU utilises core audio rather than serving as an alternative, I will have to wait to have another look until later this week - unfortunately working a lot this week so haven't got much chance to investigate, tonight was supposed to be a songwriting night but got hijacked by tech issues - first time using Logic X - my G5 died and I had to upgrade both computer and logic (new version very different so I'm still getting my head around it!)  - thank you very much for your help! I'll let you know how I get on!

closed 2016-11-18 by Wouter1