Wouter1/EMU-driver #94: Installation script

DevNulPavel, 2016-11-16:
awk from installation script has freezing on iMac, but on MacMini - everything ok

Wouter1, 2016-11-16:
@DevNulPavel 
Thanks for the reporting.

* What exactly freezes? 
* Did you wait long enough. The system can take several minutes (up to 10 I guess)  to install the extension
* What OS version?
* Did you enable unsigned kext's ?
* What do you see before the script stops working?

DevNulPavel, 2016-11-16:
Driver works fine (kext), i has not problems with it (low latency, no noise) - thanks for driver! :-) 👍 
I did wait for 20 minutes at awk installation step (after kext uploading), but nothing changes in terminal (no new messages in terminal, mac didn't freeze)

PS - Sorry for my english

Wouter1, 2016-11-16:
@DevNulPavel 

Thanks :)

Can you be more precise? There is no message "Please wait" coming from my installer.  But there are several "waiting for" messages. 

Can you please post the exact output in the terminal after some 5 minutes of waiting?

DevNulPavel, 2016-11-16:
Ok, i will paste terminal log, when i'll be at home

DevNulPavel, 2016-11-16:
`EMU USB Installer by W.Pasman
Use low latency setting (n/y)?: ny

Adjusting driver for low latency
(Re)placing EMU kernel extension. We need the admin password for this.

Password:
Waiting for the system to finish installation. Can take several minutes.`

for 20+ minutes, no changes

Wouter1, 2016-11-16:
@DevNulPavel  thanks. 
It looks like the system is not taking up the loading of the kernel. 

Can you post the ```/var/log/system.log``` file from the moment that you entered the password till 5 minutes later? I'm expecting to see text like this

Nov 16 22:30:36 ... Rescanning kernel extensions ...
Nov 16 22:30:38 .... locked; waiting for lock ...
.....

Maybe some error message appears instead?

DevNulPavel, 2016-11-18:
I've not found any lines similar lines after 10 minutes on `Waiting for the system to finish installation. Can take several minutes.`

I have used "grep"
`cat /var/log/system.log | grep -i "Rescanning"`
`cat /var/log/system.log | grep -i "waiting for lock"`

[system.log.zip](https://github.com/Wouter1/EMU-driver/files/599140/system.log.zip)

Wouter1, 2016-11-18:
Thanks for posting. 

I searched through the log file 

There are a few refrences to the installer if I grep the word ```EMU-driver-master```, but I do not see any other log messages related to the installer...
```
Nov 18 08:34:00 iMac iTerm2[681]: Time to encode state for window <PseudoTerminal: 0x7ffc5b60e8a0 tabs=2 window=<PTYWindow: 0x7ffc5b416e90 frame=NSRect: {{1546, 706}, {941, 711}} title=1. /Users/pavelershov/Desktop/EMU-driver-master/installer  /Users/pavelershov (bash) alpha=1.000000 isMain=0 isKey=0 isVisible=1 delegate=0x7ffc5b60e8a0>>: 0.08133399486541748
Nov 18 08:34:25 iMac iTerm2[681]: Time to encode state for window <PseudoTerminal: 0x7ffc5b60e8a0 tabs=2 window=<PTYWindow: 0x7ffc5b416e90 frame=NSRect: {{1592, 703}, {941, 711}} title=1. /Users/pavelershov/Desktop/EMU-driver-master/installer  /Users/pavelershov (bash) alpha=1.000000 isMain=0 isKey=0 isVisible=1 delegate=0x7ffc5b60e8a0>>: 0.002295970916748047
Nov 18 08:37:17 iMac iTerm2[681]: Time to encode state for window <PseudoTerminal: 0x7ffc5b60e8a0 tabs=2 window=<PTYWindow: 0x7ffc5b416e90 frame=NSRect: {{1592, 703}, {941, 711}} title=1. /Users/pavelershov/Desktop/EMU-driver-master/installer  /Users/pavelershov (bash) alpha=1.000000 isMain=1 isKey=1 isVisible=1 delegate=0x7ffc5b60e8a0>>: 0.002292990684509277
Nov 18 08:47:10 iMac iTerm2[681]: Time to encode state for window <PseudoTerminal: 0x7ffc5b60e8a0 tabs=2 window=<PTYWindow: 0x7ffc5b416e90 frame=NSRect: {{1592, 703}, {941, 711}} title=1. /Users/pavelershov/Desktop/EMU-driver-master/installer  /Users/pavelershov (bash) alpha=1.000000 isMain=0 isKey=0 isVisible=1 delegate=0x7ffc5b60e8a0>>: 0.002459049224853516
```

There are no sudo actions logged at all. FAIK all sudo actions executed should be logged there.  The installer executes several of them before reaching the "Waiting for the system to finish installation" message. Is it possible that all sudo actions are being skipped without warnings? (If you type the wrong password, the system should re-ask the password)

But the log file is hogged with messages from GoogleSoftwareUpdateDeamon. Could it be that this deamon is somehow hindering proper install?

Is your sudo command working? If you try sudo su, and enter your pass, does it give you a superuser prompt? (use exit to exit the superuser prompt)

The only other log file we might check is the kernel.log file (/var/log/asl or /var/log/kernel.log).

DevNulPavel, 2016-11-18:
i wrote valid password and logged in as root after. I've tried install driver with valid password every times (no messages about wrong password).

i'll try to see kernel.log messages when will be install driver next time

Wouter1, 2016-11-18:
@DevNulPavel  we can also try this: can you try to execute the relevant statements in the script manually? That way  we may be able to see errors that maybe are somehow not being reported? Can you read the script or do you need more specific instructions for this?

DevNulPavel, 2016-11-18:
Yes, i can add some text in script if needed

Wouter1, 2016-11-18:
@DevNulPavel  I mean executing the script manually, not modifying the script.

DevNulPavel, 2016-11-18:
oh, sorry

yes, i can try to execute script line by line manually

Wouter1, 2016-11-18:
>i'll try to see kernel.log messages when will be install driver next time
Probably there's no need to re-run the install. We just need to check the kernel.log file

>What purpose of awk step with Waiting for the system to finish installation. Can take several minutes. text in installation script??

The purpose is to keep the user informed about the system work on it until the installation is really done. This to avoid users to assume too quickly that it has been installed and then complaining that it does not work.

DevNulPavel, 2016-11-18:
@Wouter1 thank you for explanation

Wouter1, 2016-11-18:
@DevNulPavel  did you maybe turn off some specific logging to system.log?  The fact that the sudo commands do not appear in the system.log may be the same problem as the problem that the script can not detect the system's progress (the script is also checking the system.log file for that)

Wouter1, 2016-11-18:
@DevNulPavel  another question, does the driver actually install and work, even though the script does not say so?

DevNulPavel, 2016-11-18:
@Wouter1  i do not remember, that i disabled some logging on computer - possibly, everything have default settings

DevNulPavel, 2016-11-19:
i've found only this messages in system console application (find word - emu)

`20:02:13.574252 +0300	kextd	kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"

20:02:13.831104 +0300	kextd	kext signature failure override allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"

20:02:22.607355 +0300	kextcache	kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext EMUUSBAudio.kext
`

Wouter1, 2016-11-19:
It looks like the system IS accepting the kext.
Did you check if the system installed the kext after all?

DevNulPavel, 2016-11-19:
yes, the kext has installed yet

Wouter1, 2016-11-19:
Ok.  Glad to hear that it was still installed properly. Is it working properly too?

So it seems then that something regarding the logging has been disabled. And the installer waits for system.log messages that have been disabled. 

I don't know much about this. Would you be able to try a bit following this link to see if you can check or maybe turn back on the logging ?

https://developer.apple.com/reference/os/1891852-logging#1682426

DevNulPavel, 2016-11-19:
Yes, driver works great 👍 

i will read logging reference later and will check logging level

hansduy, 2016-11-19:
Today I had the same problem. The terminal freezed and nothing was installed. I copy'd the whole EMU-driver-master in the root directory and changed the rights to: everyone can write and read. Then I changed the rights of the installer file to the same and than I executed it.
It works!!!

The terminal output:

EMU USB Installer by W.Pasman
Use low latency setting (n/y)?: ny

Adjusting driver for low latency
(Re)placing EMU kernel extension. We need the admin password for this.

Password:
rm: /System/Library/Extensions/EMUUSBAudio.kext: No such file or directory
Waiting for the system to finish installation. Can take several minutes.
Rebuilding the kernel
Waiting for lock to replace the old kernel
Waiting for lock to replace the old kernel
Inserting the new kernel
Driver is installed and ready for use!
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...none found.

[Proces voltooid]

Wouter1, 2016-11-20:
I just tried the installer on El Capitan 10.11.6 15G31

It works fine here and this appears in system.log
```
Nov 20 21:56:00 wouters-MBP sudo[626]:   wouter : TTY=ttys000 ; PWD=......./EMU-driver/EMU-driver ; USER=root ; COMMAND=/sbin/kextunload -q /System/Library/Extensions/EMUUSBAudio.kext
Nov 20 21:56:00 wouters-MBP kernel[0]: Kext com.emu.driver.EMUUSBAudio not found for unload request.
Nov 20 21:56:00 wouters-MBP sudo[628]:   wouter : TTY=ttys000 ; PWD=......./EMU-driver/EMU-driver ; USER=root ; COMMAND=/bin/rm -r /System/Library/Extensions/EMUUSBAudio.kext
Nov 20 21:56:00 wouters-MBP sudo[630]:   wouter : TTY=ttys000 ; PWD=......./EMU-driver/EMU-driver ; USER=root ; COMMAND=/bin/cp -r EMUUSBAudio.kext /System/Library/Extensions
Nov 20 21:56:00 wouters-MBP sudo[632]:   wouter : TTY=ttys000 ; PWD=......./EMU-driver/EMU-driver ; USER=root ; COMMAND=/bin/rm -rf /Library/Audio/MIDI Drivers/EMUMIDIDriver.plugin
Nov 20 21:56:00 wouters-MBP sudo[634]:   wouter : TTY=ttys000 ; PWD=......./EMU-driver/EMU-driver ; USER=root ; COMMAND=/bin/mkdir -p /Library/Audio/MIDI Drivers/
Nov 20 21:56:00 wouters-MBP sudo[636]:   wouter : TTY=ttys000 ; PWD=......./EMU-driver/EMU-driver ; USER=root ; COMMAND=/bin/cp -R EMUMIDIDriver orig.plugin /Library/Audio/MIDI Drivers/EMUMIDIDriver.plugin
Nov 20 21:56:00 wouters-MBP sudo[638]:   wouter : TTY=ttys000 ; PWD=/Library/Audio/MIDI Drivers ; USER=root ; COMMAND=/usr/bin/chgrp -R wheel EMUMIDIDriver.plugin
Nov 20 21:56:00 wouters-MBP sudo[640]:   wouter : TTY=ttys000 ; PWD=/System/Library/Extensions ; USER=root ; COMMAND=/usr/bin/chgrp -R wheel EMUUSBAudio.kext
Nov 20 21:56:00 wouters-MBP sudo[642]:   wouter : TTY=ttys000 ; PWD=/System/Library/Extensions ; USER=root ; COMMAND=/sbin/kextload -v 0 EMUUSBAudio.kext
Nov 20 21:56:00 wouters-MBP com.apple.kextd[43]: Cache file /System/Library/Caches/com.apple.kext.caches/Directories/System/Library/Extensions/KextIdentifiers.plist.gz is out of date; not using.
Nov 20 21:56:00 wouters-MBP com.apple.kextd[43]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"
Nov 20 21:56:00 wouters-MBP com.apple.kextd[43]: kext signature failure override allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"
Nov 20 21:56:03 wouters-MBP com.apple.CDScheduler[40]: Thermal pressure state: 0 Memory pressure state: 0
Nov 20 21:56:03 wouters-MBP com.apple.CDScheduler[267]: Thermal pressure state: 0 Memory pressure state: 0
Nov 20 21:56:03 wouters-MBP com.apple.xpc.launchd[1] (com.apple.FileStatsAgent): Service only ran for 9 seconds. Pushing respawn out by 1 seconds.
Nov 20 21:56:03 wouters-MacBook-Pro.local cloudd[312]: Could not create primary backing account
Nov 20 21:56:05 wouters-MBP com.apple.xpc.launchd[1] (com.apple.FileStatsAgent): Service only ran for 0 seconds. Pushing respawn out by 10 seconds.
Nov 20 21:57:38 wouters-MBP ntpd[220]: time set +1.290260 s
Nov 20 21:59:00 wouters-MBP com.apple.kextd[43]: Cache file /System/Library/Caches/com.apple.kext.caches/Startup/KextPropertyValues_OSBundleHelper_x86_64.plist.gz is out of date; not using.
Nov 20 21:59:00 wouters-MBP com.apple.kextd[43]: Rescanning kernel extensions.
Nov 20 21:59:00 wouters-MBP com.apple.kextd[43]: Cache file /System/Library/Caches/com.apple.kext.caches/Startup/IOKitPersonalities_x86_64.ioplist.gz is out of date; not using.
Nov 20 21:59:01 wouters-MBP kernel[0]: Resetting IOCatalogue.
Nov 20 21:59:01 wouters-MBP com.apple.kextcache[657]: rebuilding //System/Library/PrelinkedKernels/prelinkedkernel
Nov 20 21:59:03 wouters-MBP com.apple.kextcache[658]: kext file:///System/Library/Extensions/JMicronATA.kext/ is in hash exception list, allowing to load
Nov 20 21:59:05 wouters-MBP com.apple.kextcache[658]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext EMUUSBAudio.kext
Nov 20 21:59:06 wouters-MBP com.apple.kextcache[660]: / locked; waiting for lock.
Nov 20 21:59:20 wouters-MBP com.apple.kextcache[658]: Created old kernelcache copy "/System/Library/Caches/com.apple.kext.caches/Startup/kernelcache"
Nov 20 21:59:20 wouters-MBP com.apple.kextcache[658]: Created prelinked kernel "/System/Library/PrelinkedKernels/prelinkedkernel"
Nov 20 21:59:20 wouters-MBP com.apple.kextcache[658]: Created prelinked kernel using "/System/Library/Kernels/kernel"
Nov 20 21:59:20 wouters-MBP com.apple.kextcache[657]: /System/Library/PrelinkedKernels/prelinkedkernel not cached.
Nov 20 21:59:20 wouters-MBP kernel[0]: hfs: mounted Recovery HD on device disk0s5
Nov 20 21:59:20 wouters-MBP mds[59]: (Volume.Normal:2464) volume:0x7faae3028000 ********** Bootstrapped Creating a default store:0 SpotLoc:(null) SpotVerLoc:(null) occlude:0 /Volumes/Recovery HD
Nov 20 21:59:20 wouters-MBP fseventsd[44]: Logging disabled completely for device:1: /Volumes/Recovery HD
Nov 20 21:59:20 wouters-MBP sandboxd[122] ([668]): DesktopServicesH(668) System Policy: deny file-write-mode /Volumes/Recovery HD/.Trashes/501
Nov 20 21:59:21 wouters-MBP com.apple.kextcache[657]: Successfully updated disk0s5.
Nov 20 21:59:21 wouters-MBP kernel[0]: hfs: unmount initiated on Recovery HD on device disk0s5
Nov 20 21:59:21 wouters-MBP com.apple.kextcache[660]: Lock acquired; proceeding.
Nov 20 21:59:21 wouters-MBP com.apple.kextcache[660]: /: helper partitions appear up to date.

```

Wouter1, 2016-11-20:
Tried to install a 2nd time to check thta there is no problem if the ext was already installed before.
But it still works fine the 2nd time.

Wouter1, 2016-11-20:
Tried to use the commands on https://developer.apple.com/reference/os/1891852-logging#1682426

But these are not working on el capitan. Available on macOS 10.12 and later only

Wouter1, 2016-11-20:
To test if asl.conf might be responsible for this misbehaviour, I edited /etc/asl.conf to

```
# Rules for /var/log/system.log
> system.log mode=0640 format=bsd rotate=seq compress file_max=5M all_max=50M
? [= Sender kernel] [<= Level error] file system.log
? [<= Level error] file system.log
? [= Facility auth] [<= Level error] file system.log
? [= Facility authpriv] [<= Level error] file system.log
```

This indeed suppresses the SUDO logs. I now only get to see

```
Nov 20 22:37:50 wouters-MBP com.apple.kextd[43]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"
Nov 20 22:37:50 wouters-MBP com.apple.kextd[43]: kext signature failure override allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"
Nov 20 22:38:18 wouters-MBP mds[59]: (DiskStore.Normal:1705) Starting health check...
```

Wouter1, 2016-11-20:
@DevNulPavel  maybe you can post the contents of ```/etc/asl.conf```. It might contain instructions to suppress logging.

Wouter1, 2016-11-20:
@hansduy  I think you have a different problem. If I change the downloaded files to read-only the installer will give warnings

```
cp: ./EMUUSBAudio.kext/Contents/Info.plist: Permission denied
cp: ./EMUUSBAudio.kext/Contents/MacOS/EMUUSBAudio: Permission denied
cp: ./EMUUSBAudio.kext/Contents/Resources/en.lproj/InfoPlist.strings: Permission denied
```

DevNulPavel, 2016-11-21:
@Wouter1 yes i will try to see `/etc/asl.conf`, when will be at home

DevNulPavel, 2016-11-21:
[asl.conf.zip](https://github.com/Wouter1/EMU-driver/files/603576/asl.conf.zip)

Wouter1, 2016-11-21:
<table>
<tr>
<td>
##
 # configuration file for syslogd and aslmanager
 ##
 
 # authpriv messages are root/admin readable
 ? [= Facility authpriv] access 0 80
 
 # remoteauth critical, alert, and emergency messages are root/admin readable
 ? [= Facility remoteauth] [<= Level critical] access 0 80
 
 # broadcast emergency messages
 ? [= Level emergency] broadcast
 
 # save kernel [PID 0] and launchd [PID 1] messages
 ? [<= PID 1] store
 
 # ignore "internal" facility
 ? [= Facility internal] ignore
 
 # save everything from emergency to notice
 ? [<= Level notice] store
 
 # Rules for /var/log/system.log 
 > system.log mode=0640 format=bsd rotate=seq compress file_max=5M all_max=50M
 ? [= Sender kernel] file system.log
 ? [<= Level notice] file system.log
 ? [= Facility auth] [<= Level info] file system.log
 ? [= Facility authpriv] [<= Level info] file system.log
 
 # Facility com.apple.alf.logging gets saved in appfirewall.log
 ? [= Facility com.apple.alf.logging] file appfirewall.log file_max=5M all_max=50M
 
  

</td>
<td>
!# configuration file for syslogd and aslmanager

!# authpriv messages are root/admin readable
? [= Facility authpriv] access 0 80

!# remoteauth critical, alert, and emergency messages are root/admin readable
? [= Facility remoteauth] [<= Level critical] access 0 80

# broadcast emergency messages
? [= Level emergency] broadcast

# save kernel [PID 0] and launchd [PID 1] messages
? [<= PID 1] store

# ignore "internal" facility
? [= Facility internal] ignore

# save everything from emergency to notice
? [<= Level notice] store

# Rules for /var/log/system.log 
> system.log mode=0640 format=bsd rotate=seq compress file_max=5M all_max=50M
? [= Sender kernel] file system.log
? [<= Level notice] file system.log
? [= Facility auth] [<= Level info] file system.log
? [= Facility authpriv] [<= Level info] file system.log

# Facility com.apple.alf.logging gets saved in appfirewall.log
? [= Facility com.apple.alf.logging] file appfirewall.log file_max=5M all_max=50M
</td>
</tr>
</table>

Wouter1, 2016-11-21:
My

Wouter1, 2016-11-21:
I added first line to my asl file. Then it seems that sudo actions from the commandlineare not logged to system.log anymore.

However the script still works ok.

Wouter1, 2016-11-21:
@DevNulPavel you can try commenting out that first line in asl.conf. You probably need to reboot after the change in order to have that any effect.

You could also try @hansduy suggestion above, even though I don't see any logic to it.

Other than that, I'm out of ideas. Something has broken part of the logging to system.log but I have no idea.

If someone has an idea, I'd love to hear it.

Wouter1, 2016-11-21:
I found here 
http://superuser.com/questions/932013/asl-conf-doesnt-take-an-effect-on-my-enviroment-os-x-yosemite

that asl.conf can be overridden by other files in /etc/asl . Maybe ther is something in /etc/asl directory 

Maybe com.apple.authd ? ON my machine it contains this
```
? [= Sender com.apple.authd] claim only
* file /var/log/authd.log mode=0640 compress format=bsd rotate=seq file_max=5M all_max=20M
? [<= Level error] file /var/log/system.log
? [<= Level error] store
```

what's in your file?

Or maybe you have other files in /etc/asl that are overriding ?

Wouter1, 2016-12-06:
Tried the installer on Siera  and indeed the logging is not appearing in system.log. Generally a lot of logging seems to be suppressed or maybe routed to somewhere else?
I thought that on El Capitan it was working fine, but I never checked the latest El Capitan

Wouter1, 2016-12-06:
Maybe this is the problem: apple changed the logging system

https://developer.apple.com/reference/os/1891852-logging

Wouter1, 2016-12-13:
Just coming back on this, apple removed that reference about the changes on logging! I will attach the page here 
[Logging - os | Apple Developer Documentation.pdf](https://github.com/Wouter1/EMU-driver/files/650166/Logging.-.os.Apple.Developer.Documentation.pdf)

Wouter1, 2016-12-13:
Yes, the messages do appear in the console.app. 
But the new logging system is filtering them out and only keeps them in memory. never writes it to disk 

```
default	22:26:56.680827 +0100	sudo	  wouter : TTY=ttys000 ; PWD=/Volumes/prive/EMU-driver/EMU-driver ; USER=root ; COMMAND=/sbin/kextunload -q /System/Library/Extensions/EMUUSBAudio.kext
default	22:26:56.686694 +0100	kernel	Kext com.emu.driver.EMUUSBAudio not found for unload request.
default	22:26:56.698293 +0100	sudo	  wouter : TTY=ttys000 ; PWD=/Volumes/prive/EMU-driver/EMU-driver ; USER=root ; COMMAND=/bin/rm -r /System/Library/Extensions/EMUUSBAudio.kext
default	22:26:56.717273 +0100	sudo	  wouter : TTY=ttys000 ; PWD=/Volumes/prive/EMU-driver/EMU-driver ; USER=root ; COMMAND=/bin/cp -r EMUUSBAudio.kext /System/Library/Extensions
default	22:26:56.732397 +0100	sudo	  wouter : TTY=ttys000 ; PWD=/Volumes/prive/EMU-driver/EMU-driver ; USER=root ; COMMAND=/bin/rm -rf /Library/Audio/MIDI Drivers/EMUMIDIDriver.plugin
default	22:26:56.746994 +0100	sudo	  wouter : TTY=ttys000 ; PWD=/Volumes/prive/EMU-driver/EMU-driver ; USER=root ; COMMAND=/bin/mkdir -p /Library/Audio/MIDI Drivers/
default	22:26:56.760214 +0100	sudo	  wouter : TTY=ttys000 ; PWD=/Volumes/prive/EMU-driver/EMU-driver ; USER=root ; COMMAND=/bin/cp -R EMUMIDIDriver orig.plugin /Library/Audio/MIDI Drivers/EMUMIDIDriver.plugin
default	22:26:56.776612 +0100	sudo	  wouter : TTY=ttys000 ; PWD=/Library/Audio/MIDI Drivers ; USER=root ; COMMAND=/usr/bin/chgrp -R wheel EMUMIDIDriver.plugin
default	22:26:56.793981 +0100	sudo	  wouter : TTY=ttys000 ; PWD=/System/Library/Extensions ; USER=root ; COMMAND=/usr/bin/chgrp -R wheel EMUUSBAudio.kext
default	22:26:56.808771 +0100	sudo	  wouter : TTY=ttys000 ; PWD=/System/Library/Extensions ; USER=root ; COMMAND=/sbin/kextload -v 0 EMUUSBAudio.kext
default	22:26:56.957105 +0100	kextd	kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"
default	22:26:57.458111 +0100	kextd	kext signature failure override allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"
default	22:27:04.630371 +0100	kextcache	kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext EMUUSBAudio.kext

default	22:27:01.808915 +0100	kextd	Rescanning kernel extensions.
default	22:27:08.025425 +0100	kextcache	/ locked; waiting for lock.
default	22:27:23.761805 +0100	kextcache	/ locked; waiting for lock.
default	22:27:30.667542 +0100	kextcache	Lock acquired; proceeding.
default	22:27:30.672009 +0100	kextcache	Lock acquired; proceeding.
default	22:27:30.671753 +0100	kextcache	/: helper partitions appear up to date.
default	22:27:30.675245 +0100	kextcache	/: helper partitions appear up to date.

```

Wouter1, 2016-12-13:
Process name is as shown above (sudo, kextd, kextcache)

The Library name of kextd is IOKit. The libname of kextcache is also kextcache.

Wouter1, 2016-12-13:
```log stream``` seems to give me the info I need.

Wouter1, 2016-12-13:
@DevNulPavel  this has been fixed! Thanks for reporting.

closed 2016-12-13 by Wouter1

DevNulPavel, 2016-12-14:
Nice 👍