Wouter1/EMU-driver #95: Control panel cannot find skins

emrecelikten, 2016-11-29:
I am using a fresh install of OS X Sierra with case-sensitive FS enabled.

Control panel is not displayed properly - I presume it might be caused by case sensitivity of the file system. Nothing is displayed under Skins menu.

Here is what it looks like: https://i.imgur.com/UBSN7xI.png

emrecelikten, 2016-11-29:
It seems it is not very easy to convert the file system back to case insensitive (and back again), so I'd rather not do that.

I tried renaming Skins folder to skins and SKINS, but it did not fix the problem. Moved it to an exFAT external drive, still the same problem. There are no errors on terminal whatsoever.

exFAT should be case insensitive, perhaps the issue is something else. I guess we will see if this is a common issue or not in a couple of months. Are you using Sierra?

Wouter1, 2016-11-29:
@emrecelikten  Thanks for trying some potential workarounds.

I don't have Sierra yet so I can't test this here.

It's also a problem that there are no errors appearing. Did you also look in system.log and kernel.log?

emrecelikten, 2016-11-29:
tail -f on /var/log/system.log shows these sometimes after launching, doesn't really seem related to the application:

`Nov 29 14:56:32 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc27630a90, 3, 0)
Nov 29 14:56:32 ammit com.apple.xpc.launchd[1] (com.apple.imfoundation.IMRemoteURLConnectionAgent): Unknown key for integer: _DirtyJetsamMemoryLimit
Nov 29 14:56:32 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc29021ea0, 3, 0)
Nov 29 14:56:32 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc2756e6d0, 3, 0)
Nov 29 14:56:37 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc29152e60, 3, 0)
Nov 29 14:56:37 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc2915b8f0, 3, 0)
Nov 29 14:56:37 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc2934ad30, 3, 0)
Nov 29 14:56:37 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc27749b10, 3, 0)
Nov 29 14:56:37 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc2915b8f0, 3, 0)
Nov 29 14:56:37 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc29350e10, 3, 0)
Nov 29 14:56:37 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc29350b50, 3, 0)
Nov 29 14:56:37 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc2756d9a0, 3, 0)
Nov 29 14:56:37 ammit logd[54]: _handle_cache_delete_with_urgency(0x7fcc29350bf0, 3, 0)
...`

 Nothing on kernel log either.

The driver works completely fine though, so this is not really an issue for me, at least.

Wouter1, 2016-11-29:
@emrecelikten  If it's related at all, no it does give much clues about what's the problem. 

I think it's a problem: how can you now check the data rate and settings of the driver?

emrecelikten, 2016-11-29:
True. But it is possible to configure CoreAudio settings using audio hosts, which is what I do. For the rest, I personally don't mind default settings. Of course that would not work for everyone.

Let us see if someone else encounters the same problem.

Wouter1, 2016-11-29:
@emrecelikten  I'll leave this open and give it a try when I get to Sierra

emrecelikten, 2016-11-29:
Thanks. Please let me know if I can help in any way.

Wouter1, 2016-12-01:
@emrecelikten  did you run the ControlPanel in its original directory ? Or did you copy it to a new location? If you copied, did you copy the Skins directory along with it?

emrecelikten, 2016-12-01:
It's original:

```bash
➜ ControlPanel pwd
/Users/emre/Downloads/EMU-driver-4.0.0/ControlPanel
➜ ControlPanel tree
.
├── E-MU\ USB\ Audio\ Control\ Panel.app
│   └── Contents
│       ├── Info.plist
│       ├── MacOS
│       │   └── E-MU\ USB\ Audio\ Control\ Panel
│       ├── PkgInfo
│       └── Resources
│           ├── EMU.icns
│           └── English.lproj
│               ├── InfoPlist.strings
│               └── MainMenu.nib
└── Skins
    ├── BlueSilver
    │   ├── Background.png
    │   ├── CheckBox.png
    │   ├── ComboBox.png
    │   ├── InfoBar.png
    │   ├── LargeBackground.png
    │   ├── StatusLight.png
    │   ├── TextBox.png
    │   ├── bg_AboutBox.png
    │   ├── skin.xml
    │   └── strings.xml
    ├── Classic
    │   ├── Background.png
    │   ├── CheckBox.png
    │   ├── ComboBox.png
    │   ├── InfoBar.png
    │   ├── LargeBackground.png
    │   ├── StatusLight.png
    │   ├── TextBox.png
    │   ├── bg_AboutBox.png
    │   ├── skin.xml
    │   └── strings.xml
    ├── Default
    │   ├── Background.png
    │   ├── CheckBox.png
    │   ├── ComboBox.png
    │   ├── InfoBar.png
    │   ├── LargeBackground.png
    │   ├── StatusLight.png
    │   ├── TextBox.png
    │   ├── bg_AboutBox.png
    │   ├── skin.xml
    │   └── strings.xml
    └── ProteusX
        ├── Background.png
        ├── CheckBox.png
        ├── ComboBox.png
        ├── InfoBar.png
        ├── LargeBackground.png
        ├── StatusLight.png
        ├── TextBox.png
        ├── bg_AboutBox.png
        ├── skin.xml
        └── strings.xml

10 directories, 46 files```

Wouter1, 2016-12-01:
Yes that looks fine.

emrecelikten, 2016-12-01:
Very strange, it now works properly when started from the terminal. I was trying to see if I could find anything by setting a breakpoint to functions like opendir with LLDB.

I wonder what I did differently last time as I remember seeing the same issue when run from terminal.

Currently, double clicking the application does not work, but running it from the terminal works.

Wouter1, 2016-12-02:
Runs fine from terminal but not double-clicked? Might it be something with the run settings (Finder File/Get Info)?

Wouter1, 2017-01-10:
@emrecelikten do you still have this problem?

I'm on Sierra now and here the control panel works fine. my FS is case insensitive and I would not know even how to change that. 

On sierra, we need to use the Console application to see reported errors, as most errors are not logged anymore to system.log (Apple changed this since Sierra).  Maybe you can check if you see some errors in Console?

emrecelikten, 2017-11-27:
Sorry for the very late reply.

Since it can be run from the terminal, it became a non-issue for me.

I have reinstalled the OS with non-case sensitive file system now. I do not remember any errors in the console though. I haven't installed the sound drivers yet, will report if I have any problems.

Thanks again.

Wouter1, 2017-11-27:
@emrecelikten  thanks for coming back to this. I close this issue !

Wouter1, 2019-08-28:
Doing tests on Mojave now.

This skin issue appears here too.

If you open the Controlpanel package contents (right click) and then go to Contents/MacOS/EMU USB Audio Control Panel, it does start with the skin correctly

Also, yo can make an alias to this file and then the alias also works correctly.

This all sounds like the system does not set up the home directory properly or something else that is there when run from terminal but not when run as app