Wouter1/EMU-driver #96: (libkern/kext) dependency resolution failure

camilodelvasto, 2016-11-30:
I'm trying to install it on El Capitan 10.11.1 but I'm getting this:
```
/System/Library/Extensions/EMUUSBAudio.kext failed to load - (libkern/kext) dependency resolution failure; check the system/kernel logs for errors or try kextutil(8).
```

In the end it says "Driver is installed and ready for use!", but there's no control panel and the device is not recognized in the audio preferences.

I ran `csrutil status` and got:
```
System Integrity Protection status: enabled (Custom Configuration).

Configuration:
	Apple Internal: disabled
	Kext Signing: disabled
	Filesystem Protections: enabled
	Debugging Restrictions: enabled
	DTrace Restrictions: enabled
	NVRAM Protections: enabled
```

Is there something else I need to do on my system before attempting the installation? Any help is much appreciated, thanks.

Wouter1, 2016-11-30:
@camilodelvasto  Thanks for using my driver and for reporting the issue.

Can you please  report the relevant messages in system/kernel logs here, to see which dependency is missing.

The csrutil status seems fine to me.

camilodelvasto, 2016-11-30:
@Wouter1 Thanks for the reply. I'm not really sure how to check the logs, could you give me a hint?

camilodelvasto, 2016-11-30:
I was able to get this:
```
Nov 30 15:51:58 mb05 sudo[5170]: camilodelvasto : TTY=ttys003 ; PWD=/Users/camilodelvasto/Downloads/EMU-driver-master ; USER=root ; COMMAND=/sbin/kextunload -q /System/Library/Extensions/EMUUSBAudio.kext
Nov 30 15:51:58 mb05 kernel[0]: Kext com.emu.driver.EMUUSBAudio not found for unload request.
Nov 30 15:51:58 mb05 sudo[5172]: camilodelvasto : TTY=ttys003 ; PWD=/Users/camilodelvasto/Downloads/EMU-driver-master ; USER=root ; COMMAND=/bin/rm -r /System/Library/Extensions/EMUUSBAudio.kext
Nov 30 15:51:58 mb05 sudo[5174]: camilodelvasto : TTY=ttys003 ; PWD=/Users/camilodelvasto/Downloads/EMU-driver-master ; USER=root ; COMMAND=/bin/cp -r EMUUSBAudio.kext /System/Library/Extensions
Nov 30 15:51:58 mb05 sudo[5176]: camilodelvasto : TTY=ttys003 ; PWD=/Users/camilodelvasto/Downloads/EMU-driver-master ; USER=root ; COMMAND=/bin/rm -rf /Library/Audio/MIDI Drivers/EMUMIDIDriver.plugin
Nov 30 15:51:58 mb05 sudo[5178]: camilodelvasto : TTY=ttys003 ; PWD=/Users/camilodelvasto/Downloads/EMU-driver-master ; USER=root ; COMMAND=/bin/mkdir -p /Library/Audio/MIDI Drivers/
Nov 30 15:51:58 mb05 sudo[5180]: camilodelvasto : TTY=ttys003 ; PWD=/Users/camilodelvasto/Downloads/EMU-driver-master ; USER=root ; COMMAND=/bin/cp -R EMUMIDIDriver orig.plugin /Library/Audio/MIDI Drivers/EMUMIDIDriver.plugin
Nov 30 15:51:58 mb05 sudo[5182]: camilodelvasto : TTY=ttys003 ; PWD=/Library/Audio/MIDI Drivers ; USER=root ; COMMAND=/usr/bin/chgrp -R wheel EMUMIDIDriver.plugin
Nov 30 15:51:58 mb05 sudo[5184]: camilodelvasto : TTY=ttys003 ; PWD=/System/Library/Extensions ; USER=root ; COMMAND=/usr/bin/chgrp -R wheel EMUUSBAudio.kext
Nov 30 15:51:58 mb05 sudo[5186]: camilodelvasto : TTY=ttys003 ; PWD=/System/Library/Extensions ; USER=root ; COMMAND=/sbin/kextload -v 0 EMUUSBAudio.kext
Nov 30 15:51:58 mb05 com.apple.kextd[47]: Cache file /System/Library/Caches/com.apple.kext.caches/Directories/System/Library/Extensions/KextIdentifiers.plist.gz is out of date; not using.
Nov 30 15:51:58 mb05 com.apple.kextd[47]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext "/System/Library/Extensions/EMUUSBAudio.kext"
Nov 30 15:51:58 mb05 com.apple.kextd[47]: EMUUSBAudio.kext - no compatible dependency found for com.apple.kpi.bsd.
Nov 30 15:51:58 mb05 com.apple.kextd[47]: Can't load EMUUSBAudio.kext - failed to resolve dependencies.
Nov 30 15:51:59 mb05 com.apple.CoreSimulator.CoreSimulatorService[5168]: Requested watch is is not available for pairing.  Assuming it has the capability.
Nov 30 15:51:59 mb05 com.apple.CoreSimulator.CoreSimulatorService[5168]: Requested phone is is not available for pairing.  Assuming it has the capability.
Nov 30 15:51:59 mb05 com.apple.CoreSimulator.CoreSimulatorService[5168]: Requested watch is is not available for pairing.  Assuming it has the capability.
Nov 30 15:51:59 mb05 com.apple.CoreSimulator.CoreSimulatorService[5168]: Requested phone is is not available for pairing.  Assuming it has the capability.
Nov 30 15:51:59 mb05 com.apple.CoreSimulator.CoreSimulatorService[5168]: Requested watch is is not available for pairing.  Assuming it has the capability.
Nov 30 15:51:59 mb05 com.apple.CoreSimulator.CoreSimulatorService[5168]: Requested phone is is not available for pairing.  Assuming it has the capability.
Nov 30 15:51:59 mb05 com.apple.CoreSimulator.CoreSimulatorService[5168]: Requested watch is is not available for pairing.  Assuming it has the capability.
Nov 30 15:51:59 mb05 com.apple.CoreSimulator.CoreSimulatorService[5168]: Requested phone is is not available for pairing.  Assuming it has the capability.
Nov 30 15:52:03 mb05 com.apple.kextd[47]: Cache file /System/Library/Caches/com.apple.kext.caches/Startup/KextPropertyValues_OSBundleHelper_x86_64.plist.gz is out of date; not using.
Nov 30 15:52:03 mb05 com.apple.kextd[47]: Rescanning kernel extensions.
Nov 30 15:52:03 mb05 com.apple.kextd[47]: Cache file /System/Library/Caches/com.apple.kext.caches/Startup/IOKitPersonalities_x86_64.ioplist.gz is out of date; not using.
Nov 30 15:52:04 mb05 kernel[0]: Resetting IOCatalogue.
Nov 30 15:52:04 mb05 com.apple.kextcache[5191]: rebuilding //System/Library/PrelinkedKernels/prelinkedkernel
Nov 30 15:52:05 mb05 com.apple.kextcache[5193]: kext file:///System/Library/Extensions/JMicronATA.kext/ is in hash exception list, allowing to load
Nov 30 15:52:07 mb05 com.apple.kextcache[5193]: EMUUSBAudio.kext - no compatible dependency found for com.apple.kpi.bsd.
Nov 30 15:52:07 mb05 com.apple.kextcache[5193]: EMUUSBAudio.kext is missing dependencies (including anyway; dependencies may be available from elsewhere)
Nov 30 15:52:07 mb05 com.apple.kextcache[5193]: kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext EMUUSBAudio.kext
Nov 30 15:52:09 mb05 GPUToolsAgent[5202]: Mac Agent inited !!!
Nov 30 15:52:09 mb05 com.apple.kextcache[5209]: / locked; waiting for lock.
Nov 30 15:52:17 mb05 com.apple.kextcache[5193]: EMUUSBAudio.kext - no compatible dependency found for com.apple.kpi.bsd.
Nov 30 15:52:22 --- last message repeated 1 time ---
Nov 30 15:52:22 mb05 com.apple.kextcache[5193]: Prelink failed for com.emu.driver.EMUUSBAudio; omitting from prelinked kernel.
Nov 30 15:52:29 mb05 com.apple.kextcache[5193]: Created old kernelcache copy "/System/Library/Caches/com.apple.kext.caches/Startup/kernelcache"
Nov 30 15:52:29 mb05 com.apple.kextcache[5193]: Created prelinked kernel "/System/Library/PrelinkedKernels/prelinkedkernel"
Nov 30 15:52:29 mb05 com.apple.kextcache[5193]: Created prelinked kernel using "/System/Library/Kernels/kernel"
Nov 30 15:52:30 mb05 com.apple.kextcache[5191]: /: no supported helper partitions to update.
Nov 30 15:52:30 mb05 com.apple.kextcache[5209]: Lock acquired; proceeding.
Nov 30 15:52:30 mb05 com.apple.kextcache[5209]: /: no supported helper partitions to update.
Nov 30 15:52:30 mb05 login[5070]: DEAD_PROCESS: 5070 ttys003
```
Hope that's useful :/

Wouter1, 2016-11-30:
Thanks, yes that's what I was looking for

> com.apple.kextcache[5193]: EMUUSBAudio.kext - no compatible dependency found for com.apple.kpi.bsd.

Wouter1, 2016-11-30:
If I run ```kextlibs EMUUSBAudio.kext```  on the v11 driver I get

```
For all architectures:
    com.apple.iokit.IOAudioFamily = 1.9.7f2
    com.apple.kpi.bsd = 13.4
    com.apple.kpi.iokit = 13.4
    com.apple.kpi.libkern = 13.4
    com.apple.kpi.mach = 13.4

For x86_64:
    9 symbols not found in any library kext.

```

Wouter1, 2016-11-30:
All 10.11 El Capitan machines should have Darwin version 15. Do I get 13.4 because I run the kextlibs on Mavericks?

Wouter1, 2016-11-30:
@camilodelvasto  my first guess is that I have compiled on a newer version of El Capitan and that it has a newer version of com.apple.kpi.bsd. Is it possible for you to upgrade your El Capitan and try again?

Wouter1, 2016-11-30:
found here that http://stackoverflow.com/questions/14381495/what-version-of-libraries-to-link-against-in-a-kexts-osbundlelibraries

>
    The Darwin major number is the OSX minor number plus 4 - so OSX 10.8 is Darwin 12, 10.7 is 11, 10.6 is 10, etc. (this is at least true going back to OSX 10.4, I don't have any experience with 10.3 or older)

>    The Darwin minor number is the OSX revision number - so OSX 10.8.2 is Darwin 12.2, OSX 10.6.8 is Darwin 10.8, OSX 10.7 is Darwin 11.0 etc.

So bsd 13.4 corresponds to OSX 10.9.4.. weird dependency, this was compiled on capitan which is 10.11

Wouter1, 2016-11-30:
Checking the info-11.plist, I see different info: OSBundleLibraries 15.4.0

This corresponds to OSX 11.4 which seems correct

Wouter1, 2016-11-30:
@camilodelvasto  I think I found another possible solution that we can try, that does not require you to update your system.

Maybe you can try this:

Before running the installer, Edit the Info.plist file in ```v11/EMUUSBAudio.kext/Contents/```

There you find this text near the bottom

```
		<key>com.apple.kpi.bsd</key>
		<string>15.4.0</string>

```

Can you change that to

```
		<key>com.apple.kpi.bsd</key>
		<string>15.0.0</string>
```

so only that number changes.
save the file and then run the installer as usual

camilodelvasto, 2016-12-01:
@Wouter1 that worked! I'm still unable to see a control panel (don't really know if there should be one), but I was able to play audio through the device and could even record a piece of music using GarageBand and my recently acquired Rhodes :) - thanks a lot!!!!!

Wouter1, 2016-12-01:
@camilodelvasto good to hear that :)

For my information, what did you do to make this work ? did you change the Info.plist or did you upgrade el Capitan?

The control panel is coming separately in the download, in the EMU-driver/ControlPanel directory. What happens if you double click it?

camilodelvasto, 2016-12-01:
@Wouter1 I just modified the version number in Info.plist. That seemed easier to me :) - Thaks for the great effort put into this.

Wouter1, 2016-12-01:
@camilodelvasto  Thanks for the update! Great to know this. I will fix the code accordingly to get this fixed permanently

Wouter1, 2016-12-01:
@camilodelvasto the installer was fixed, you should work now on El Capitan 10.11.1 without workaround. Thanks

closed 2016-12-01 by Wouter1