Wouter1/EMU-driver #97: Sierra 10.12.1 (16B2555)

digdelu, 2016-12-09:
1. Install doesn't work for me in many ways: disable/enable SIP, csrutill enable/disable/--without kext...not work.
2.record sound distored, and some sounds works like a driver from emu.com.
3.i can't understand how it works...it's only freeze on installation in terminal. i try in a lot of times and read the issues^ but not working!
4.HELP ME PLEASE!

Console: 
kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext EMUUSBAudio.kext 

any manipulations become to this messege!

Wouter1, 2016-12-09:
@digdelu  Strange, here playback works all fine on Sierra. I did not yet try to record though in Sierra but it worked fine on El Capitan.

>1 csrutill enable/disable/--without kext...not work

How can you then have "distorted sound"? I mean, if it didn't work, you won't even be able to play or record.

Also 
> kext-dev-mode allowing invalid signature -67062 0xFFFFFFFFFFFEFA0A for kext EMUUSBAudio.kext

proves that it IS working

>2.  record sound distored, and some sounds works like a driver from emu.com. 
Maybe the crystal in the EMU is getting bad. Can you try recording say 10 minutes of sound, and see if the clicking gets less over time?

>3 .i can't understand how it works...it's only freeze on installation in terminal. i try in a lot of times and read the issues^ but not working!
known issue  #94  with installer messages. It installs anyway.

>4 just post here :)

Wouter1, 2016-12-13:
@digdelu #94 has been fixed, installer should now properly message you.  I still didn't test recording (no time yet). Still having the recording issue?

Wouter1, 2016-12-18:
Digdelu tested this on Sierra 10.12.1 (16B2555)

Wouter1, 2016-12-18:
@digdelu  I tested all sample rates on Sierra 10.12.1 and it works all perfect. 

Do you still have the problem(s)?

Wouter1, 2017-01-10:
I did not hear anything back, I suppose @digdelu  got it working

closed 2017-01-10 by Wouter1

digdelu, 2017-03-07:
Im very sorry^ for the long time silence.
All work just magic!
please sorry and thank you again!

Wouter1, 2017-03-07:
@digdelu  great to hear, thanks for reporting!