Wouter1/EMU-driver #100: Can't install the drivers Mac os Sierra 10.12.3

Jerrelly, 2017-02-18:
when trying to instal is says : 
[image]
can not do nothing..but really need. please help
i have tried a lot of times already..

Wouter1, 2017-02-20:
@Jerrelly  I just tried to install, for me it works all right.

Looks like your unzipped file is damaged. Can you confirm that your unzipped download contains a v11 directory? Did you start the installer by double clicking installer?

Jerrelly, 2017-02-20:
@Wouter1 yes it has all the files, i tried different zip-archiver,but nothing work
yes i just double tap on installer 
also forgot to say that i have disabled SIP. 
hope for your soon help

Wouter1, 2017-02-20:
The first error message about a missing "/Contents" suggests your download is broken. There really should be this directory, it's the core of the kext

Let's first check that your download is correct

execute 
```
openssl sha1 <your downloaded EMU driver.zip> 
```
It should give this number

```
8c083ddd4e29c654d8896bfffa419aa57512feb9
```

Jerrelly, 2017-02-20:
@Wouter1 so I've checked, it is absolutely the same

Wouter1, 2017-02-20:
@Jerrelly  great. Then, if you right-click on the EMUUSBAudio.kext inside v11, and select "Show package contents", do you see a "Contents" folder?

Jerrelly, 2017-02-20:
@Wouter1 yes, and it contain : 
[image]

Wouter1, 2017-02-20:
Seems the script is not switching to the correct working directory

Can you edit the install script and add one line

```
pathname=$( cd $(dirname $0) ; pwd -P ) 
cd "$pathname"
```

Maybe you can make that into

```
pathname=$( cd $(dirname $0) ; pwd -P ) 
cd "$pathname"
echo "$pathname"
```

and report back what that echo prints in the console.

Jerrelly, 2017-02-20:
just add it to end ?

Wouter1, 2017-02-20:
No, add that echo right under the cd "#pathname"

maxkutcher, 2017-04-19:
Can not do nothing..but really need. please help!!! Please (macos sierra 10.12.4)
[image]

Wouter1, 2017-09-04:
@Jerrelly  I didn't hear anything anymore. Was this resolved?

soupergrouper, 2017-09-06:
Had the same problem. Solution was to use default zip file name

Wouter1, 2017-09-07:
@soupergrouper  thanks for chiming in with a suggestion. So you say that if you rename the zip file and then unzip the renamed file, something goes wrong in the unzipping?

soupergrouper, 2017-09-07:
@Wouter1 In my case I had old version in 'Downloads' folder. So the name of new download was 'EMU-driver-master 2' by default. I deleted the old one and renamed new to 'EMU-driver-master' and it worked.

Wouter1, 2017-09-07:
@soupergrouper  thanks for the additional info. I added a do not rename warning to the download instructions. 

I didn't hear back from @Jerrelly  so I suppose this is fixing this.

closed 2017-09-07 by Wouter1