Wouter1/EMU-driver #101: Can't install the drivers Mac os Sierra 10.12.4

maxkutcher, 2017-04-19:
Can't install the drivers Mac os Sierra 10.12.4! Please Help! After installation, the computer doesn't see sound card (e-mu 0204)
[image]

Wouter1, 2017-04-19:
@maxkutcher  Can you please post the log file?
Did you run csrutil ?

maxkutcher, 2017-04-19:
When csrutil disable - audio computer see sound card!!!

maxkutcher, 2017-04-19:
When csrutil disable - computer see sound card!!!

maxkutcher, 2017-04-19:
But when enable doesn't see

Wouter1, 2017-04-19:
@maxkutcher  Of course, this is according to the installation instructions. Does this solve your issue?

maxkutcher, 2017-04-19:
Works Everything! Thanks!!!

Wouter1, 2017-04-19:
@maxkutcher  Great, thanks for the feedback!

closed 2017-04-19 by Wouter1