Wouter1/EMU-driver #102: Problem on OSX Sierra 10.12.4

sataraEora, 2017-05-03:
Hello. Your driver was working perfectly on previous versions. But after the update 10.12.4.. I use EMU 0204 for recording musical instruments (electric guitar). So, after the few seconds when i play, sound disappears. I attached a screenshot of track panel. I wasnt playing, when i did that screenshot, but Cubase says me, that sound is pretty loud. When i turn off EMU and then turn on again, everything repeats again - i can play just few seconds and then sound disappears.

Sorry for my bad english and mistakes. Eng is not my native language, but i wish u can understand what i was talking about.

Wouter1, 2017-05-03:
@sataraEora  Thanks for reporting. I don't have latest Sierra yet so I can not confirm this.

Is there any message in the log files? 

I saw some reports of audio issues on Sierra 12.4 here https://discussions.apple.com/thread/7906975?start=0&tstart=0 and here https://discussions.apple.com/message/31572584#31572584
Maybe Apple again broke the USB driver stack? 

Did you try a non-powered USB or plugging directly in your mac? There are some reports that some hubs are causing issues on latest Sierra...

Wouter1, 2017-05-03:
@sataraEora  if you have some HD space to try the 10.12.5 Beta version: several users reported that it fixes the issue.

sataraEora, 2017-05-03:
@Wouter1 Im a noob in computers, programming, administrating etc, so i need some help to give u information about log files (i dont know where to find them), sry x_x im just a musician. But if u will say me where i can take the information (log files or smth what u need), i will give it to u.

"Maybe Apple again broke the USB driver stack?" - sounds like a true.

I plug EMU directly in my mac with usb cable. 

I will try that beta and i hope it can solve the problem x_x

Wouter1, 2017-05-03:
@sataraEora if you're a noob, then I recommend not to mess with the beta versions and instead roll back to sierra 12.4 where your driver still worked and wait till apple fixes it. 

The log can be inspected with the Console.app in /Applications/Utilities. It shows the log entries in real time, so best is to open the Console.app first and then use the EMU and check what appears in the console. You can ignore other system reports like from opendirectoryd, CommCenter, etc.

Wouter1, 2017-09-04:
@sataraEora  was this fixed?