Wouter1/EMU-driver #104: Not even detect 0404 in 10.12.5 (

gjoob, 2017-06-07:
MBPR 13 + Sierra 10.12.5, kext is loaded (I've checked it), but when I plug-in/power-on my 0404 nothing happens - Audition, System Sound, EMU Control Panel feel nothing.

Here is log part from Console.app:

default	17:02:33.643270 +0300	powerd	Received power source(psid:5000) update from pid 117: <private>
default	17:02:33.733335 +0300	icdd	#ICDebug - 23:{ICWiredBrowser.m} (USB Device first match)
default	17:02:33.734177 +0300	icdd	#ICDebug - 388:{ICWiredBrowser.m} (10 USB Descriptions Managed)
default	17:02:33.736098 +0300	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add E-MU 0404 | USB - 0x0/0x0/0x0 - 0x14200000 - ICDeviceDescriptionSUQuery)
default	17:02:33.750380 +0300	icdd	#ICDebug - 213:{ICResourceManager.m} (452D4D55-2D37-342D-3346-30342D303744|E-MU 0404 | USB|MANUFACTURER:E-MU Systems, Inc.;MODEL:E-MU 0404 | USB|SW=FALSE|)
default	17:02:33.752036 +0300	icdd	#ICDebug - 457:{ICDDMessageCenter.m} (+Add E-MU 0404 | USB - 0x0/0x0/0x0 - 0x14200000 - ICDeviceDescriptionAdded)
default	17:02:33.757913 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.758395 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.758697 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.758861 +0300	ocspd	0x7fe73cf00170 opened /private/var/db/crls/crlcache2.db: 29660 bytes
default	17:02:33.810774 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.811232 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.811521 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.818992 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.819449 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.819741 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.826405 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.826845 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.827157 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.835683 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.836247 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.836568 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.848760 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.849299 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.849632 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.876294 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.876757 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.877058 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.884209 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.884628 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.884890 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.891271 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.891995 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.892338 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.901863 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.902662 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.902992 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.914296 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.914879 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.915330 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.915341 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.915858 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.916020 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.916311 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.916517 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.916880 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.918085 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.918564 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.918873 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.922403 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.922792 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.923047 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.931576 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.932138 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.932613 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.933795 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.934040 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.934308 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.934727 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.934967 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.934975 +0300	trustd	asynchronously fetching CRL (http://crl.apple.com/codesigning.crl) for client (kextd[54])
default	17:02:33.935340 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.935451 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.935463 +0300	trustd	asynchronously fetching CRL (http://www.apple.com/appleca/root.crl) for client (kextd[54])
default	17:02:33.945299 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	17:02:33.945874 +0300	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
[... 412 lines cut ...]
default	17:02:34.550497 +0300	kernel	Default alternate setting ID is 17
default	17:02:34.550497 +0300	kernel	
default	17:02:34.551011 +0300	kernel	Default sample rate is 0
default	17:02:34.551012 +0300	kernel	
default	17:02:34.551014 +0300	kernel	Default alternate setting ID is 255
default	17:02:34.551014 +0300	kernel	
default	17:02:34.551018 +0300	kernel	- IOAudioEngine[0xffffff8022cf3480]::start(0xffffff801a3dc240, 0xffffff801a3dc240)
default	17:02:34.551020 +0300	kernel	-EMUUSBAudioEngine[0xffffff8022cf3480]::start (0xffffff801a3dc240) = 0
default	17:02:34.551020 +0300	kernel	
default	17:02:34.551025 +0300	kernel	- IOAudioDevice[0xffffff801a3dc240]::activateAudioEngine(0xffffff8022cf3480, 1) returns 0xFFFFFFFFE00002BC

here is "system_profiler SPUSBDataType" return:
...
        E-MU 0404 | USB:

          Product ID: 0x3f04
          Vendor ID: 0x041e  (Creative Labs)
          Version: 1.00
          Serial Number: E-MU-74-<...>-STATION 01
          Speed: Up to 480 Mb/sec
          Manufacturer: E-MU Systems, Inc.
          Location ID: 0x14200000 / 10
          Current Available (mA): 500
          Current Required (mA): 2
          Extra Operating Current (mA): 0

Wouter1, 2017-06-27:
@gjoob  Sorry for the late reply, I am fully busy with other projects and I won't have much time coming weeks.

I briefly checked the code that causes the "line 2050 goto Exit" , but it's a part that I never worked on so I have no idea what that part of the code is about. My guess is that it's trying to find a USB connection for a control interface.

OTOH it seems the audio device is engaging after some failed attempts. 

What does "STATION 01" mean in the system profiler info?

Can you try to disconnect all USB devices including hubs etc, and connect only the EMU directly?

Wouter1, 2017-06-27:
@gjoob  Did you have have your EMU0404 working before? What OS ?

Kokonutz, 2017-07-03:
Hi!
I think I have the same issue. I have a mac OS Sierra 10.12.5 (MacBook Pro 15' 2016). No sign of life of the EMU 0404 USB. I have followed the install instructions... The last OS it worked on is a MacBook Pro 15' mid 2012 running Mac OS 10.9.5 I will try to run it via parallels on windows 10. thank you for your help.

Wouter1, 2017-07-03:
@Kokonutz  did it never work for you on Sierra, or did you simply not try it on earlier versions of Sierra? I have it working properly here on Sierra 12.2.

Kokonutz, 2017-07-03:
Never tried on earlier versions of Sierra as the laptop was already updated.

alaija, 2017-09-10:
Hey I have the same problem with 0202, after install driver it worked, but it stoped after reconnect.

`Sound assertion "0 == thisUnitIDNum" failed in "/Volumes/prive/EMU-driver/EMU-driver/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit`

Wouter1, 2017-09-10:
@alaija thanks for reporting including the error message. 
The message indicates that the software is receiving a NULL unit number from the USB connection. 

Do you have the device connected directly to the computer?

alaija, 2017-09-10:
@Wouter1 yes it is directly connected.

Wouter1, 2017-09-10:
@alaija  @Kokonutz  This will need some debugging on 10.12.5. Unfortunately this is not going to happen soon, lack of time, and no place on my HD to upgrade and keeping the old system as well.

alaija, 2017-09-11:
@Wouter1 I forgot to notice that I have 10.12.6 actually, and it going to be 10.13 in a few days, so will pray for it will work on it.

Kokonutz, 2017-09-11:
@Wouter1 @alaija thank you for the feedbacks. If updates solve the issue, will report it 🤘

Wouter1, 2017-09-30:
I saw that Serg Podtynnyl cloned and made a few changes to this driver. I mailed him what he fixed

He replied

>I changed nothing, It was just cleanup from old version and references for latest xcode.

Kokonutz, 2017-10-02:
@Wouter1 cool! where can we find that Serg Podtynnyl cloned driver? Thanks

Wouter1, 2017-10-02:
I can't find him back now, he had a github account with another clone of my driver. He made some changes to the project (most cloners don't) and I asked.

Wouter1, 2017-10-14:
@gjoob  I read again through the log file you reported above. But it seems all is going fine till the end. The driver is reporting startup 4 lines before the end of the log.
activateAudioEngine IS returning some error message but there is nothing in the log yet indicating how the driver reacts on this. 
Do you have more log information after that point?

Wouter1, 2017-10-14:
0xFFFFFFFFE00002BC is "general error" from iokit

gjoob, 2017-10-19:
@Wouter1 I don't know what happened, but I tried to plug in my 0404 just now and it started to work. It seems, there is no s/pdif's, but stereo out (and stereo in, I guess) works well. Control center detected my device, as well as E-mu control panel, system sound works, Logic also plays sounds well. Thank you)

Wouter1, 2017-10-20:
@gjoob good to hear! 

So it works on 10.12.5 after all

Just for others that might have same issue, any idea what you might have done that kicked it into action?

vizh, 2019-04-26:
Tried to comment out `FailIf(NULL == thisUnitIDNum, Exit)` command at `EMUUSBAudioDevice.cpp:2050`. Device was detected by system after compiling and installing patched Kext, but sound was terrible (but exists) and console was full of errors.

Console is full of errors and debug messages from kext. indeed very many, here is just small part of:

```
error	03:07:45.537588 +0500	coreaudiod	HALS_IOA1Engine.cpp:365:EndWriting:  HALS_IOA1Engine::EndWriting: got an error from the kernel trap, Error: 0xE00002EE
default	03:07:45.548026 +0500	kernel	+ IOAudioEngineUserClient[<private>]::performWatchdogOutput(<private>, 573) - (25,354)
default	03:07:45.548065 +0500	kernel	- IOAudioEngineUserClient[<private>]::performWatchdogOutput(<private>, 573) - (25,554)
default	03:07:45.549365 +0500	kernel	  performClientOutput(d00e5a3,18d4) - missed samples (25,554). fCurrentLoopCount=25
```

------

```
default	03:07:38.382030 +0500	kernel	frameSizeQueue empty, guessing some queue size. May need fix..
default	03:07:38.382052 +0500	kernel	frameSizeQueue empty, guessing some queue size. May need fix..
default	03:07:38.382057 +0500	kernel	frameSizeQueue empty, guessing some queue size. May need fix..
```

------

```
default	03:07:37.678866 +0500	kernel	+-IOAudioEngine[<private>]::setState(0x1. oldState=0)
default	03:07:37.678868 +0500	kernel	- IOAudioEngine[<private>]::getTimerInterval()
default	03:07:37.678875 +0500	kernel	- IOAudioEngine[<private>]::startAudioEngine() returns 0x0
default	03:07:37.678877 +0500	kernel	- IOAudioEngine[<private>]::incrementActiveUserClients() - 1 returns 0
default	03:07:37.678878 +0500	kernel	- IOAudioEngine[<private>]::startClient(<private>) returns 0x0
default	03:07:37.678879 +0500	kernel	  engine started
default	03:07:37.678879 +0500	kernel	  bufferSet <private>
default	03:07:37.678880 +0500	kernel	  output clientBuffer <private>
default	03:07:37.678881 +0500	kernel	+ IOAudioStream[<private>]::addClient(<private>)
default	03:07:37.678883 +0500	kernel	- IOAudioStream[<private>]::addClient(<private>) returns 0x0
default	03:07:37.678884 +0500	kernel	- IOAudioEngineUserClient[<private>]::startClient() - 1 returns 0x0
default	03:07:37.678885 +0500	kernel	- IOAudioEngineUserClient::externalMethod returns 0x0
```

vizh, 2019-04-26:
Oh! Finally... Already thought to postpone the question for tomorrow, but realized that the problem is that kext could not determine the sampling rate of the device during initialization. 

Started the application "Audio Devices" and installed it manually. The error messages in the console disappeared and the sound became normal.

Wouter1, 2019-04-27:
@vizh thanks for reporting what might be causing this and for suggesting a workaround.
Where is the "Audio Devices" application?

vizh, 2019-04-27:
This is the "/Applications/Utilities/Audio MIDI Setup.app", sorry. Finally, check this out: https://github.com/vizh/EMU-driver/commit/fa84605901bb09e8c9c46b4e924e6042bcdbba38. It's not a final solution, but i have no time for now to go deeper and find exact problem and clean solution for it. But is works in my case.

Only inconvenience that user must set sampling frequency manually right after install.

Wouter1, 2019-04-27:
@vizh  I looked at your fix, you  outcommented a FailIf error check.  

But I fail to see how your fix would help. If the FailIf is in place, the code may jump to Exit which will just return curPath. If the FailIf is disabled, then tempPath is released and curPath is returned just as well. So the only difference is release of tempPath? 

I can see that releasing temPath avoids some memory leak. But not how it fixes issues with non-recognised devices.

Can you explain a bit more about your fix?

BTW the break at the end of the case statement seems unreachable, maybe I should remove it, maybe it's just to be sure or so

Wouter1, 2019-04-27:
@vizh would using the Audio MIDI setup app by itself help users to get around problem, even without your fix in the driver code?

vizh, 2019-04-27:
> Can you explain a bit more about your fix?

The fix is crazy. Yes. Unfortunately, I'm not very good at C++ but I was also surprised that this option worked. I double checked this. This hack is working on my workstation. Without it my card was not even listed in System Preferences / Sound.

I'm afraid I just got my finger in the sky.

I'm trying to find a way to see values of all variables in this function. But it can take some time. Teaching C++ as necessary. If you could help to find a way to log content of OSArray variables, then I can see what is going on.

> @vizh would using the Audio MIDI setup app by itself help users to get around problem, even without your fix in the driver code?

No. Because card not detected and not visible in Audio MIDI Setup.

Wouter1, 2019-04-28:
@vizh  thanks for the explanation. 

Seems that your card somehow got in an unknown/impossible state that forced the driver to exit early. And that your messing with the midi settings got tht fixed, after you did this "hard" bypassing of some safety checks. My guess is that your card would now work fine with the normal driver, after this forced reset.

>way to log content of OSArray variables
I think you have to loop over the array elements and log each. Because it depends on what is inside the array elements.

Wouter1, 2020-04-24:
There does not seem any action needed on the driver. I'm closing this.

gjoob, 2020-04-26:
Hi, I discovered what was happening some time ago. If I start the player, which puts the card in 192/24 mode (for example, Audirvana), and then turn off the power of the card, the card will not be detected after turning it on. I found two ways to revive her after that.
- Turn off the power and usb, send a 44.1/16 signal to the toslink, then turn on the card and then connect usb. After that, you need to explicitly switch the mode in the control panel.
- Find a Windows computer with an original driver (I was lucky to have Windows 7 under Fusion), switch the card to 44.1/16 and then turn it off. After that, everything works as usual. Until the next rash launch of Audirvana))

Wouter1, 2020-04-26:
@gjoob thanks for the update on this. I'll try to reproduce.
I assume by "card" you refer to the EMU0404 USB.

I understand that it does not matter HOW you put the card in 192/24 mode and I will use the standard EMU control panel to test this. Then turn the EMU off, and then turn it on again. If I understand you correctly, the EMU would then not be detected?

Wouter1, 2020-04-26:
@gjoob I can not reproduce this problem. My EMU does turn on properly and it is still in 192k mode after turning on. I suspect there is some issue with either the EMU hardware or maybe some software in the EMU itself.

But we can do some further testing to see if this is a software or hardware issue

If you say it is not detected, can you confirm that the device does not even appear to the operating system after turning on after turning it off in 192k? If you go to About this mac (top left apple on the screen), /SystemReport/USB and then browse through the USB Device tree (the top panel) do you see ```E-MU 0404 | USB``` ?

 Your original issue ```kernel Sound assertion "0 == thisUnitIDNum``` implies that the driver IS detected but that the driver fails to launch itself. Is that message still there? I ask because that message is coming from the driver and that implies the system IS detecting the card but fails to launch my driver

Wouter1, 2020-04-26:
I just noticed in this document which talks about the Windows driver and control panel
https://www.avisoft.com/E-MU%200404%20Installation.pdf

>Unfortunately, the most recent firmware update of the E-MU 0404 does not save this setting, which means that you need to do this each time after powering on the unit.

Is there a firmware update for this? 
Not saving the 192k setting? Might that be related to this issue?

gjoob, 2020-04-26:
Hello!
0404 is not with me right now unfortunately, but I have to say two important things:
- Latest firmware very often turn the card to a brick, as described in many forums, so nobody use it, i guess. There is the oldest one in my card, anyway. Also I still use MacOS 10.12.6 (Sierra).
- Audirvana uses exclusive access mode for the card and may be this is the root of the problem.

Wouter1, 2020-04-26:
@gjoob is it possible to test this problem with the control panel as I just did? If the problem does not occur then, this must be something specific with Audirvana and then you will have to contact them; then I can close this issue

Wouter1, 2020-04-26:
On this page
https://support.creative.com/Products/ProductDetails.aspx?prodID=15186&prodName=E-MU+0202+USB

they say

>E-MU® 0202 / 0404 / Tracker Pre USB Web Install
>This software contains the latest versions of the Mac OS X application, driver, and firmware for the E-MU 0202, 0404 USB, Tracker Pre audio interfaces. Details weergeven »

So there indeed seems to be some firmware. 

Digging into the software, there are 4 packages (plus some overall installer, readme etc)

EMUTrackerPreDocs.pkg
EMUUSB0202Docs.pkg
EMUUSB0404Docs.pkg
EMUUSBSoftware.pkg

There is an Archive.pax.gz in EMUUSBSoftware.pkg

That archive contains 3 apps:

E-MU USB Audio Uninstaller
EMUUSBAudioControlPanel (with skins folder)
Updater

The control panel is for powerpc so really old and un-runnable.

The updater indeed seems to contain a firmware updater, with this documentation inside
```
===================================
Firmware Changes in this Release
===================================

* S/PDIF input no longer muted when internally synced
S/PDIF input is no longer muted if the system is set to
internal clock source and the S/PDIF input clock is the
same as the system clock.

* Persistent settings no longer store 176.4kHz and 192kHz
The 176.4kHz and 192kHz samples rates are no longer stored
on power-off. Now, the system defaults to 44.1kHz.

* Firmware revision numbers are now unified for all E-MU Audio
USB products.

=========
Copyright
=========

Copyright (c) 2007 E-MU Systems. All rights reserved.
```

Wouter1, 2020-04-26:
The firmware updater app also probably is for powerpc only, it has a red 'no entry' sign over the icon. The text also seems not so inviting, why would you not store the selected 176k and 192k rates. And if you use SP/dif, what's the problem with using that as the clock

Wouter1, 2020-04-26:
The 0404 also has a page here https://support.creative.com/Products/ProductDetails.aspx?prodID=15185&prodName=0404%20USB%202.0 that mentions 'firmware'.

The downloaded dmg is identical to the one you get with the 0202 link

Wouter1, 2020-04-27:
Note, that firmware upgrader comes with the creative driver installer it seems, so you may get the firmware upgrade without further notice.

Wouter1, 2020-04-27:
@gjoob I searched extensively but could not find a single report of "bricking" the EMU with their firmware upgrade. Do you have any refs comfirming that?