Wouter1/EMU-driver #105: Soft limit on controller app isn't working and skins doesn't show

sametsafak, 2017-06-11:
I am using OSX 10.12.5. Creative's original driver installed for any hope :) 

Today I downloaded yours driver, I did everything on installing instructions. 
Then I copy control panel into the my applications folder and I see these issues.

1- There is no skin on app. But I solve it, I already copied skins folder into the app but it doesn't show skins. Then I cut the folders inside of skins folder and move them to parent folder. 
Folder tree is like this:

[image]

If I delete skins folder (while actual skins like BlueSilver is in this folder) app doesn't show skins. 

2- In first time, in skinless app, I clicked soft limit button and I saw SL led (yellow) turned on. Then I closed it. After that I tried reopen it and write spotlight emu, then I clicked wrong control panel... It was the original emu control panel application. And I closed it. After that I opened yours control panel application. Now soft limit button seems working in app but the yellow led on device doesn't turned on anymore.

I tried uninstall original emu driver, reinstall yours driver. Still same.
I tried uninstall yours driver, restart computer and reinstall yours driver. Still I can't turn soft limit led on.

Wouter1, 2017-06-11:
@sametsafak  

Do I understand correctly that the driver is working, but the control panel not?

Does the control panel show the Device ?

Did you try running the control panel from the original folder where you unzipped it?

sametsafak, 2017-06-11:
@Wouter1 

-Yes, the driver is working. Control panel was working too. 

-Yes I can see the device in control panel. It's working correctly about showing device.

-I tried both. Inside the original folder and inside of my applications folder.

Wouter1, 2017-06-11:
@sametsafak 
>*In addition, when I try to change sample rate, app crashes sometimes. Not always. (I have the crash report if you want to see)

Thanks. Yes I have noticed that it sometimes crashes. But I dont have the source code of the control panel (iit's the original panel) so I can't fix it anyway.

Wouter1, 2017-06-11:
@sametsafak 

>Control panel was working too

Do I understand that the 
* control panel skins worked before, but stopped working at some point? When did it stop working?
* the control panel now shows the device but not the "soft limit" button?

Wouter1, 2017-06-11:
@sametsafak 

> Then I closed it. After that I tried reopen it and write spotlight emu, then I clicked wrong control panel... It was the original emu control panel application. And I closed it. After that I opened [...]

The control panel was never changed, the original and my version are the same thing. So it should behave the same.

Wouter1, 2017-06-11:
I checked on my machine, I never use the soft limit and I even didn't know that there is a yellow light on the EMU

I can toggle the Soft Limit button in the control panel but I don't see the SL led change on the EMU either.

sametsafak, 2017-06-11:
**control panel skins worked before, but stopped working at some point? When did it stop working?**
Not the skins, the button of soft limit was working. Skin problems are solved actually when I change folders like I tell above.

**the control panel now shows the device but not the "soft limit" button?**
I can see the actual button. When I clicked it first time (before the solve skin problem with changing folders locations), it was worked. Then I opened original controller application as a mistake and clicked some controls. It didn't work. Then I opened your control center app, It's sound clip button didn't work either (I checked skinless folder structure again like it was working, It didn't work)

**The control panel was never changed, the original and my version are the same thing. So it should behave the same.**
Good to know that, thanks.I will consider it if I test another Mac.

**I can toggle the Soft Limit button in the control panel but I don't see the SL led change on the EMU either.**
Like I said, It was worked before I opened the original control center app. I don't know why. When I use my condenser mic, I use this feature and it helps limiting audio. 

I will try to do same things on my other Mac and write here the results.

Wouter1, 2017-06-11:
@sametsafak  Regarding the skins, you must have missed part of the installation instructions 

>If you like you can also copy the control panel into your applications directory. You also have to copy the Skins folder along with the app.

Seems that's resolved then :) 

Appparently the real issue here is only the soft limit button. I will change the title

Wouter1, 2017-06-11:
@sametsafak 

>When I clicked it first time (before the solve skin problem with changing folders locations), it was worked. 

Are you sure it worked on 10.12 with my driver and control panel? I tried several times but the SL led never turns on.

I'm confused now, maybe you ment that it worked on a previious version of OSX or that the SL was already on. Maybe you can indicate on which system you had it working? Which machine, which version of the driver? And what exactly was working, were you actually switching the SL on and off frequently with the control panel, or do you mean by "working" that the SL was on before you opened the control panel?

sametsafak, 2017-06-12:
Yes I am sure. Because it was worked on skinless version. I saw yellow led turns on and off when I clicked the button on controller repeatedly. I don't remember SL was on or off before I pressed the button in first time but it worked. 

All of the drivers, including yours is latest versions. My Mac has latest version OSX 10.12.5 It is an iMac mid 2011 27".

Wouter1, 2017-06-12:
@sametsafak  thanks for these clarifications. I'll have to debug what's going on, this will take some time.

sametsafak, 2017-06-12:
I make more tests to help you when I will be free.