Wouter1/EMU-driver #106: OS X 10.8.5 dependency resolution failure;

sjuktstarkgrogg, 2017-08-26:
Hi, so I read that another user had the same 'dependency resolution failure' but on a much more recent version of OS X... I tried to modify the .plist file a bit by changing versions etc. but nothing worked...

I'm running it on OS X 10.8.5 and I'm getting following error:

```
Adjusting driver for low latency
(Re)placing EMU kernel extension. We need the admin password for this.

rm: /System/Library/Extensions/EMUUSBAudio.kext: No such file or directory
/System/Library/Extensions/EMUUSBAudio.kext failed to load - (libkern/kext) dependency resolution failure; check the system/kernel logs for errors or try kextutil(8).
Waiting for the system to finish installation. Can take several minutes.
Rebuilding the kernel
Waiting for lock to replace the old kernel
Driver is installed and ready for use!
You can close this window.
logout
```

Wouter1, 2017-09-04:
@sjuktstarkgrogg  Sorry for the slow response, I have no time left at all unfortunately...

Can you post the message in the  system/kernel logs. There should be a message stating specifically which dependency can not be resoloved. This might be just a version issue, eg I might be requiring too new a library where an older version might also work.

Wouter1, 2018-01-17:
@sjuktstarkgrogg  
I'm not sure about the first line. Maybe something is not up to date and it might be related to problem

it looks like my driver is asking for a newer version of the  libraries (IOAudioFamily, IOUSBFamily,..., mach)

Which versions of these libraries do you have? Then I can check against the code.

sjuktstarkgrogg, 2018-01-18:
This is what's written about the concerned libraries in the following file: _KextIdentifiers.plist.gz_
```
		<dict>
			<key>CFBundleIdentifier</key>
			<string>com.apple.iokit.IOAudioFamily</string>
			<key>CFBundleVersion</key>
			<string>1.9.2fc7</string>
			<key>OSBundlePath</key>
			<string>IOAudioFamily.kext</string>
		</dict>
```
```
		<dict>
			<key>CFBundleIdentifier</key>
			<string>com.apple.iokit.IOUSBFamily</string>
			<key>CFBundleVersion</key>
			<string>635.4.0</string>
			<key>OSBundlePath</key>
			<string>IOUSBFamily.kext</string>
		</dict>
```
```
		<dict>
			<key>CFBundleIdentifier</key>
			<string>com.apple.kpi.iokit</string>
			<key>CFBundleVersion</key>
			<string>12.6.0</string>
			<key>OSBundlePath</key>
			<string>System.kext/PlugIns/IOKit.kext</string>
		</dict>
```
```
		<dict>
			<key>CFBundleIdentifier</key>
			<string>com.apple.kpi.libkern</string>
			<key>CFBundleVersion</key>
			<string>12.6.0</string>
			<key>OSBundlePath</key>
			<string>System.kext/PlugIns/Libkern.kext</string>
		</dict>
```
```
		<dict>
			<key>CFBundleIdentifier</key>
			<string>com.apple.kpi.mach</string>
			<key>CFBundleVersion</key>
			<string>12.6.0</string>
			<key>OSBundlePath</key>
			<string>System.kext/PlugIns/Mach.kext</string>
		</dict>
```

Wouter1, 2018-01-18:
@sjuktstarkgrogg 
This is in the bundle target properties of the driver (EMUUSBAudio-info.plist)

IOAudioFamily 1.9.7f2
IOUSBFamily 683.4
iokit 13.4
libkern 13.4
mach 13.4

So that explains why it does not load.

We would need to test if lowering the requirements to your machine versions would give a working driver. 

Can you try the following:

Before installing the driver, edit the Contents/Info.plist inside the v9 version of the driver (EMU-driver/v9/right click on EMUUSBAudio.kext and 'show package contents'/Contents/Info.plist). The values are at the far bottom of the plist. Make sure you dont change only the numbers, not the wrapper structures (</string> etc). Change the values in OSBundleLibraries according to the values you reported above. This can be done with xcode or plain text editor. Save. After that, run the installer as usual.

sjuktstarkgrogg, 2018-01-18:
Thanks for your rapid reply! I'm gonna try that straight away!

sjuktstarkgrogg, 2018-01-18:
Update

Wouter1, 2018-01-18:
@sjuktstarkgrogg  Thanks for trying. it looks like the version numbers I put in are indeed right and that it will not work with older versions of OSX than mavericks. 

Does the original driver from creative not work for 10.8 either?

sjuktstarkgrogg, 2018-01-19:
Thanks a lot for your time and help Wouter! 
It's a real bummer because I get so much clicks and pops when recording audio with the standard drivers and I really hoped that yours would work solve that problem. I think I might have to buy a new sound card then :/
Anyhow, again.. Thanks for all your help! (Don't hesitate to contact me if you think of some other solution that might make the drivers work on OS X 10.8 ;) )

Wouter1, 2018-01-19:
Yes it would require at least some code fixing and back porting, but I'm 
not going to put work in that.

Why don't you upgrade to 10.9 (or later)? Then my driver should work fine

Wouter

On 1/19/18 6:00 PM, sjuktstarkgrogg wrote:
> Thanks a lot for your time and help Wouter!
> It's a real bummer because I get so much clicks and pops when recording audio with the standard drivers and I really hoped that yours would work solve [...]

closed 2018-04-11 by Wouter1