Wouter1/EMU-driver #107: E-MU 0204, Driver does not work, macOS High Sierra 10.13

Chertolyas, 2017-10-14:
Hello! I tried to install your Driver but it does not work:
1. In Control Panel I see my E-MU 0204. I can change "Sample Rate".
2. In Programm (GarageBand) or in "System settings - Audio - Input" does not any signal.
3. On E-MU there is input signal and I hear sound in my headphones (which connected to E-MU).

If I install old original Driver:
1. In Control Panel I see my E-MU 0204. I can't change "Sample Rate" because System reboot with error.
2. In Programm (GarageBand) or in "System settings - Audio - Input" there is signal. But in result I have record with many loud noises.

Can you give any comments about this?) You are a legend in Internet))
I didn't use your driver on old versions macOS. I used original Driver and Logic Pro and I had any noises some times but very rarely.

Wouter1, 2017-10-14:
@Chertolyas sorry to hear it doesn't work for you.
Did you see ticket #104, could this be the same issue? 
Are there messages in the log files that might indicate what the problem is?

BTW, FAIK the original drivers worked only on very old versions of OSX, like 10.4. Did you have it working on a more recent system?

Chertolyas, 2017-10-15:
Thank you for fast reaction!)

I think that is not the same. But I'm noob so I'll try to give some information for you.

In issue #104 he wrote: "EMU Control Panel feel nothing". I see E-MU 0204 in Control Panel. Sorry, can I add pictures in this text?) How?)

**E-MU 0204 | USB:**

  ID продукта:	0x3f19
  ID производителя:	0x041e  (Creative Labs)
  Версия:	1.00
  Серийный номер:	E-MU-44-3F19-07DB0215-0E521-8740AT2A
  Скорость:	До 480 Мбит/с
  Производитель:	E-MU Systems, Inc.
  ID размещения:	0x14100000 / 2
  Допустимый ток (мА):	500
  Потребляемый ток (мА):	500
  Избыточный рабочий ток (мА):	0

**Part of log from Console:**

по умолчанию	10:39:44.898946 +0500	kernel	warning way-out ring wrap position
по умолчанию	10:39:44.898949 +0500	kernel	
по умолчанию	10:39:44.923596 +0500	sharingd	Nearby stop advertising of type: 16
по умолчанию	10:39:44.923700 +0500	sharingd	Nearby start advertising with data: Advertising request of type 16, priority 1, UseFG 432 (270.00 ms), data <0b00>, connectable 1, options {
} priority 0 mode 432 options (null)
по умолчанию	10:39:44.923861 +0500	wirelessproxd	Stop advertising for process sharingd (298) of type 16
по умолчанию	10:39:44.923972 +0500	wirelessproxd	Nothing left to advertise, stopping
по умолчанию	10:39:44.924671 +0500	kernel	003861.056399 Bluetooth USB Host Controller@1d181300: IOUSBHostDevice::forcePowerGated: kPowerStateOn timeout 0ms
по умолчанию	10:39:44.924854 +0500	wirelessproxd	Start advertising for process sharingd (298) of type 16 with advertising interval 432 (270.00 ms)
по умолчанию	10:39:44.925012 +0500	wirelessproxd	advertisingRulesOSX - advertisements: (
        {
        kCBAdvAppleMfgTypeKey = 16;
        kCBAdvDataAppleMfgData = <07ff4c00 10020b00>;
        kCBAdvOptionUseFGInterval = 0;
    }
)
по умолчанию	10:39:44.925435 +0500	wirelessproxd	Requesting to start advertising for clients 16  with (
        {
        kCBAdditionalAppleMfgAdvertisements =         (
        );
        kCBAdvDataAppleMfgData = <07ff4c00 10020b00>;
        kCBAdvOptionUseFGInterval = 0;
        kCBMsgArgIsPrivilegedDaemon = 1;
    }
)
по умолчанию	10:39:44.925860 +0500	analyticsd	[<private>] no observers; dropped.
по умолчанию	10:39:44.926334 +0500	analyticsd	[<private>] no observers; dropped.
по умолчанию	10:39:44.927031 +0500	analyticsd	[<private>] no observers; dropped.
по умолчанию	10:39:44.944400 +0500	kernel	warning way-out ring wrap position
по умолчанию	10:39:44.944417 +0500	kernel	
по умолчанию	10:39:44.944426 +0500	kernel	warning way-out ring wrap position
по умолчанию	10:39:44.944429 +0500	kernel	
по умолчанию	10:39:44.951527 +0500	trustd	could not enable test hierarchy: no UAT pinning preferences set
по умолчанию	10:39:44.956004 +0500	kernel	003861.087722 BRCM20702 Hub@1d181000: IOUSBHostDevice::powerStateWillChangeToGated: kPowerStateOn
по умолчанию	10:39:44.956040 +0500	kernel	003861.087774 BRCM20702 Hub@1d181000: IOUSBHostDevice::setPowerStateGated: kPowerStateSuspended -> kPowerStateOn
по умолчанию	10:39:44.956054 +0500	kernel	003861.087790 BRCM20702 Hub@1d181000: IOUSBHostDevice::setPowerStateGated: resuming pipes
по умолчанию	10:39:44.956086 +0500	kernel	003861.087821 BRCM20702 Hub@1d181000: IOUSBHostDevice::powerStateWillChangeToGated: AppleUSB20HubPort@1d181300 will inhibit idling
по умолчанию	10:39:44.956095 +0500	kernel	003861.087830 BRCM20702 Hub@1d181000: IOUSBHostDevice::updateIdlePolicyGated: consensus idle policy timeout 0ms
по умолчанию	10:39:44.990909 +0500	kernel	warning way-out ring wrap position
по умолчанию	10:39:44.990930 +0500	kernel	
по умолчанию	10:39:44.990939 +0500	kernel	warning way-out ring wrap position
по умолчанию	10:39:44.990941 +0500	kernel	
по умолчанию	10:39:45.009231 +0500	kernel	003861.140954 Bluetooth USB Host Controller@1d181300: IOUSBHostDevice::powerStateWillChangeToGated: kPowerStateOn
по умолчанию	10:39:45.009290 +0500	kernel	003861.141023 Bluetooth USB Host Controller@1d181300: IOUSBHostDevice::setPowerStateGated: kPowerStateSuspended -> kPowerStateOn
по умолчанию	10:39:45.009306 +0500	kernel	003861.141043 Bluetooth USB Host Controller@1d181300: IOUSBHostDevice::setPowerStateGated: resuming pipes
по умолчанию	10:39:45.009333 +0500	kernel	003861.141066 Bluetooth USB Host Controller@1d181300: IOUSBHostDevice::forcePowerGated: kPowerStateOn completed with 0x00000000 after 84ms
по умолчанию	10:39:45.009396 +0500	kernel	003861.141130 Bluetooth USB Host Controller@1d181300: IOUSBHostDevice::powerChangeDone:
по умолчанию	10:39:45.009415 +0500	kernel	003861.141148 Bluetooth USB Host Controller@1d181300: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@2 [<private>] has an idle policy of 500ms
по умолчанию	10:39:45.009427 +0500	kernel	003861.141161 Bluetooth USB Host Controller@1d181300: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@0 [<private>] has an idle policy of 500ms
[... 1000 lines cut ...]
по умолчанию	10:39:55.350129 +0500	analyticsd	[<private>] no observers; dropped.
по умолчанию	10:39:55.350759 +0500	analyticsd	[<private>] no observers; dropped.
по умолчанию	10:39:55.380780 +0500	kernel	003871.512718 BRCM20702 Hub@1d181000: IOUSBHostDevice::powerStateWillChangeToGated: kPowerStateOn
по умолчанию	10:39:55.380861 +0500	kernel	003871.512808 BRCM20702 Hub@1d181000: IOUSBHostDevice::setPowerStateGated: kPowerStateSuspended -> kPowerStateOn
по умолчанию	10:39:55.380878 +0500	kernel	003871.512826 BRCM20702 Hub@1d181000: IOUSBHostDevice::setPowerStateGated: resuming pipes
[... cut ...]
по умолчанию	10:39:55.412430 +0500	kernel	003871.544378 Bluetooth USB Host Controller@1d181300: IOUSBHostDevice::updateIdlePolicyGated: consensus idle policy timeout 500ms
по умолчанию	10:39:55.412443 +0500	kernel	003871.544391 BRCM20702 Hub@1d181000: IOUSBHostDevice::updateIdlePolicyGated: consensus idle policy timeout 0ms
по умолчанию	10:39:55.433258 +0500	wirelessproxd	Started to advertise for type 16 with error (null)
по умолчанию	10:39:55.434388 +0500	analyticsd	[<private>] no observers; dropped.
по умолчанию	10:39:55.439685 +0500	kernel	warning way-out ring wrap position

Chertolyas, 2017-10-15:
And yes, I worked with original driver on Yosemite. With Logic Pro. And had some little problems time to time. Not critical for home recording.

Wouter1, 2017-10-15:
@Chertolyas  thanks for posting the log. 
It looks like the only error message from the EMU is "kernel warning way-out ring wrap position"
I'm missing the startup of the device so I can't see if you're trying to play or record.
Also it's heavily cluttered with BlueTooth USB messages so it's a bit difficult to see what's going on. 

Just to be sure, you have the EMU connected directly with your computer?

The message suggests that the EMU is partially working at least. 
From this I would expect at least some recording input signal. 

What happens if you play sound through the EMU?

Wouter1, 2017-10-15:
@Chertolyas 
>And yes, I worked with original driver on Yosemite. With Logic Pro. And had some little problems time to time. Not critical for home recording.

Interesting. for me the original driver already failed on El Capitan! I mean crashing the system and heavy clicking in recorded sound if it didn't crash before that point.

Chertolyas, 2017-10-15:
OK, let's go) I found some interesting now.

1. I connected my microphone and headphones to E-MU. I connected E-MU to USB-port of my MacBook. I turned on E-MU and I see:
[image].
In Console I see:
[Turn on E-MU.pdf](https://github.com/Wouter1/EMU-driver/files/1385289/Turn.on.E-MU.pdf)

**Important!** First time I have a signal in "System Setting - Audio":
[image]

But after 1-2 minutes or some actions I haven't signal in "System Setting - Audio" and in Console I see that (it's only part of log):
[warning way-out ring wrap position.pdf](https://github.com/Wouter1/EMU-driver/files/1385292/warning.way-out.ring.wrap.position.pdf)

And to tell the truth - yes, I had the same:

> Interesting. for me the original driver already failed on El Capitan! I mean crashing the system and heavy clicking in recorded sound if it didn't crash before that point.

Wouter1, 2017-10-15:
@Chertolyas  checked the turn on log file. All seems to go fine .  Can you record or play when the EMU shows up in the audio settings?

the way-out ring wrap position may be normal, depending on what you are doing.
Never found a way to get that error message fixed.

 What hardware are you using?

xanview, 2017-11-13:
Btw, just wanted to add a note that the driver is working beautifully for me in High Sierra.

Wouter1, 2017-11-13:
@xanview  thanks for the input!

Chertolyas, 2017-11-13:
@Wouter1 Thank you for informing! I use MacBook Pro (13-inch, Mid 2012). It's may be the reason? Now I'll try to install driver one more time.

Wouter1, 2017-11-13:
@Chertolyas  it's a bit older machine, after the driver installs you might try first to use the control panel to switch to a low rate, say 44kHz, and see first if that works. Try recording first, then play back (play back is more complex technically)