Wouter1/EMU-driver #108: High Sierra, EMU Tracker Pre working great (10.13.2)

air-10, 2018-01-11:
1) writing down "csrutil enable --without kext" failed with a message "this config is not supported or something, restarting csr".
2) but installation was successful and all works none the less fine! thanks a lot!
3)PS: there is no explanation what are low-latency settings and how to switch it on later, after the installing.

thanks a lot !

Wouter1, 2018-01-12:
@air-10   Good to hear that it works.

interesting, I never heard this before and many people use my driver on 10.13. csrutil should be still the right tool for this in 10.13.
Also, if the driver works then the SIP must have been disabled properly.
Are you sure you did not make a typo?
Maybe you can try again and report the exact error message?

> there is no explanation what are low-latency settings

They are here
https://github.com/Wouter1/EMU-driver/blob/master/Latency.md

You can not switch the latency settings after installing. Just uninstall and re-install with the proper settings...

air-10, 2018-01-17:
@Wouter1 I have turned SIP off and on before, for the SwitchResX to work (and probably turned it on again later) but the response from terminal now was:

> csrutil --without kext 
> Requesting an unsupported configuration. This is likely to break in the future and leave your machine in an unsupported state.
> 
> Successfully enabled System Integrity Protection. Please restart you machine for the changes to take effect."

Althouh I got a proper condition now as "csrutil status" says: 

> macs-MacBook-Pro:~ mac$ csrutil status
> System Integrity Protection status: enabled (Custom Configuration).
> 
> Configuration:
> 	Apple Internal: disabled
> 	Kext Signing: disabled
> 	Filesystem Protections: enabled
> 	Debugging Restrictions: enabled
> 	DTrace Restrictions: enabled
> 	NVRAM Protections: enabled
> 	BaseSystem Verification: enabled
> 
> This is an unsupported configuration, likely to break in the future and leave your machine in an unknown state.

Wouter1, 2018-01-17:
@air-10  Ah now I see what you mean. That message "Requesting an unsupported configuration. This is likely to break in the future and leave your machine in an unsupported state." is not a failure message, it's in fact a OK message. Apple seems just to be warning about "the future". If you believe that message is a bug by itself, you need to contact Apple :) 

So it's the normal message from SIP. I'll close this ticket.

closed 2018-01-17 by Wouter1