Wouter1/EMU-driver #109: macOS 10.12.6 Sierra - confirmed working

log4x4, 2018-01-21:
Hello! It's not and issue, just want to confirm the work of driver in macOS 10.12.6 Sierra. There is a bug sometimes, when playing audio a short 0.5 second glitch, but i think that's all.
I have hackintosh in configuration like this:
```
Intel Xeon E3-1270 V4 3.0Ghz
Gigabyte GA-Z97-D3H rev 1.0
NVIDIA Geforce G100 256MB
4GB RAM
500GB HDD
```
Also maybe you have a donation account or something? For your work a few bucks from people will be useful :)

Wouter1, 2018-01-21:
@log4x4  Thanks for confirming!

log4x4, 2018-01-23:
Also i remember one strange thing about drivers.. I am using the EMU 0404 on Mac (hack) machine, and when i connected device with Windows system with last beta drivers available - the drivers promted window where was message about firmware update of the audiocard itself! I agreed, and when process of firmware update ended, audiocard rebooted itself (lights on it where ON and a second OFF, then turned ON again).
The thing i have to say - some strange errors with your drivers on other people device may accure because firmware mismatch.. I recommend to every user of EMU 0202/0204/0404 to connect their devices to Windows system with latest original driver from Creative and check the version of firmware that available for device.

Wouter1, 2018-04-11:
@bezromval  reported (another ticket)

Confirm working in MacOs High Sierra 10.13.3 with E-mu Tracker Pre:

I am here just to say the greatest thank you in the world. Driver totally works with playing and recording

Wouter1, 2018-04-11:
@longeee posted (another ticket)

macOS High Sierra 10.13.4 - E-MU Tracker Pre - Working Driver! 
 I confirm that the driver works flawlessly with E-MU Tracker Pre on Hackintosh macOS High Sierra 10.13.4

Thumbs up for your effort ;)

Wouter1, 2018-04-11:
@JasF reported in another ticket
10.12.4 macOS Sierra works ultimately good!

fotoknet, 2018-04-27:
I have hackintosh on asus p8z77 mb. 
Two cards are working fine. (44.1-192khz playing music from audirvana,  working mic ang guitar preamps)
But only if I do full install of proprietary creative driver before this drivers. 
In Sierra and high Sierra. 
First creative driver and after that wouter1's driver!
Cards are e-mu 0204 and 0202.

roddy20, 2018-06-23:
Works OK
Device 0x3f02 EMU 0202, FL1100 USB card, macOS 10.12.6 Sierra,

Wouter1, 2018-07-16:
A russian user reported 23 june that 0202 works properly on all frequencies.

aleksandrphp, 2018-09-14:
Hardware:
```
Gigabyte GA-B75-D3H
I5-3330
E-MU 0404 USB (connect to USB 2.0 port)
```
First install Sierra and official Creative drivers (up to 96 kHz). Then find this project and install it. All frequencies unblocked, work fine. Then update to High Sierra and still good;)
Previous users write about 0,5 sec lag. It present, but not serious issue. Thank you for project!

Wouter1, 2019-10-26:
@timmyleej reported (in issue #123)

>@Wouter1 I have E-MU 0204 and 0404 both connected to my hackintosh installed on chinese motherboard, Xeon E5-1650, C602 Chipset. Connected in usb 2.0 ports, both worked well in Sierra and Mojave.

Wouter1, 2019-11-07:
On #123 @grimrian  reported EMU works fine with version 4.0.1 of the driver.

Wouter1, 2019-11-07:
@grimrian  thanks for reporting. It's great to hear that it works fine on yor mac. Can you detail on your setup like what mac you have (eg macbook pro 2013) and how you connect the EMU to the machine (direct, USB2 port, or via an adapter, etc).

grimrian, 2019-11-07:
Hi, first of all thanks for a super driver and instructions. It is great to be able to continue using the E-MU, which is a very good ADC/DAC.

Machine is this one:

E-MU 0404  is plugged directly in USB port on Mac. It also works through a USB extender.

Software was initially installed on Sierra, I believe, all auto upgrades up to Catalina 10.15.1 went fine, no issues.

Not sure what else to add…

Regards, Tomas

Wouter1, 2019-11-07:
@grimrian what is the hardware?

grimrian, 2019-11-07:
> 
> @grimrian <https://github.com/grimrian> what is the hardware?
Did you see my EARLIER response re. Mac… +The E-Mu is a 0404. 

This was the process

On El Capitan and later:

reboot in recovery mode (hold cmd-R while booting)
go to utilities-terminal
enter this command: csrutil enable --without kext
reboot your machine
Run the install script 
there was a strange sound unit eq2mac or something which appeared after the installation. I deleted it. 

Regards,

Tomas

Wouter1, 2019-11-07:
@grimrian with hardware I mean the mac hardware. What do you see when you click on "about this mac" in the apple menu in the top left corner

grimrian, 2019-11-07:
Strange, I have sent you a screenshot of that particular hardwareinfo twice - maybe the github mail system does not allow images, so here is the text:

Hardware Overview:

  Model Name:	iMac
  Model Identifier:	iMac18,3
  Processor Name:	Quad-Core Intel Core i5
  Processor Speed:	3,4 GHz
  Number of Processors:	1
  Total Number of Cores:	4
  L2 Cache (per Core):	256 KB
  L3 Cache:	6 MB
  Memory:	24 GB
  Boot ROM Version:	178.0.0.0.0
  SMC Version (system):	2.41f2
  Serial Number (system):	C02VG690J1GG
  Hardware UUID:	FF6A4CC6-5B37-5FC8-A6AA-D0019E4F7C84

It is an iMac 27” from 2017.

Hopefully it helps,

regards,

Tomas

Wouter1, 2019-11-08:
@grimrian  thanks for the details

Github does allow screenshots, you can just drag in the image onto the github webpage, at least that works for me.

Do I understand correctly that your EMU is connected to the USB-3 port (not the USB-C port)? Does it also work ok if you connect it to USB-C through an adapter?

Wouter1, 2019-11-09:
@jakubzwierzycki commented in #120 
After updating to catalina everything works like a charm, Ableton usage, emu0404 usb midi config and everything else. On previous  osx updates i had to reinstall this driver to restore possibility to  choose emu0404 usb midi out / in in ableton preferences. After update to  catalina there is no issue like that, so nice work:)
--

Wouter1, 2019-11-09:
@jacubzwierzycki thanks for the report, and great to hear it works properly.
Can you give some hardware details, and also how your EMU is connected -direct into USB2 port, direct into USB3 port, through an adapter for USB-C, or whatever?

jakubzwierzycki, 2019-11-09:
@Wouter1 i am using MacBook pro 13.3 Retina early 2015. It seems to work on all USB inputs in this version of macbook for me. Tomorrow i will try with external monitor usbs/usb hub, just when multiple devices thru USB hub as i remember i had issues with my keyboards / midi controllers also on windows with official driver, so i always prefered to have emu connected directly to laptop/macbook

Wouter1, 2019-11-10:
@jakubzwierzycki  thanks for the info. You then have two "old-style" USB 3 ports without need for a adapter. Good to hear that that works. Please let us know what happens if you use some hub or adapter to connect the EMU.

jakubzwierzycki, 2019-11-10:
When i connect EMU thru Logilink USB 2.0 4port hub then unfortunately it is not visible in Audio midi setup. same as using dell P2415Q monitor usbs.
For test i installed emu0404 usb beta driver for windows on probook 6560b with windows 10, and then connection thru same logilink usb2.0 hub works.

Wouter1, 2019-11-11:
@jakubzwierzycki  thanks for the update! 

It's really surprising that it is not even visible when you connect the hub because visibility information is not time critical at all.

Is it visible in the hardware overview (about this mac/system report/usb) ? If not then something seems really broken, either in macos or in that hub

log4x4, 2019-11-11:
> When i connect EMU thru Logilink USB 2.0 4port hub then unfortunately it is not visible in Audio midi setup. same as using dell P2415Q monitor usbs.
> For test i installed emu0404 usb beta driver for windows on probook 6560b with windows 10, and then connection thru same logilink usb2.0 hub works.

Really i prefer not to use any hubs since it is a potential source of weird problems with USB devices, especially if it is not flash drive or something.

Wouter1, 2019-11-11:
@log4x4 Yes sure. But on the latest macs you don't have a 'normal' USB port  anymore. It's all USB-C and you can't connect the EMU directly anymore. Unless I'm missing something

Wouter1, 2019-11-11:
@log4x4  but the idea might be good to avoid any additional electronics.

It looks like this one for instance

Varta USB-C - USB 57946101401 USB-adapter

does not have any additional circuitry and is just a straightforward socket adapter

jakubzwierzycki, 2019-11-11:
@Wouter1  yes, in system report / usb EMU is  also visible when connected via hub 
https://www.dropbox.com/s/135ki2z4mtzhq2h/Zrzut%20ekranu%202019-11-11%2014.38.29.png?dl=0

Wouter1, 2019-11-11:
@jakubzwierzycki  ok then probably the driver is somehow not finding the connection and therefore the EMU never is registered as audio device (it remains an unknown USB device). For now I assume this is not an issue as the direct connection works. 

Thanks for reporting!

Emptyfruit, 2019-11-15:
Playback is definitely working on Catalina 15.1. Will try recording sometime next week.

Wouter1, 2019-11-15:
@Emptyfruit  thanks for reporting. If playback works then recording will work too as the driver requires the recording stream to play (even if you don't record)

timmyleej, 2019-11-30:
@Wouter1  Hi! Sorry for being quiet. I've solved my playback/record issues with Catalina with 0404 and 0204, not sure about certain thing that fixes issues, but i've changed my motherboard and then updated straint to Catalina 10.15.1 (instead of previosly tested 10.15.0 version). So now I have BOTH last macos version and working sound)

Wouter1, 2019-11-30:
@timmyleej  thanks for the update and good to hear that it works properly now!

timmyleej, 2019-12-01:
@Wouter1 But ive got new issue now) Audio stops working after sleep.

Wouter1, 2019-12-01:
@timmyleey this ticket is to report proper working. For issues please either add to an existing ticket that matches your case or make a new ticket

Kokonutz, 2019-12-02:
Hi @Wouter1 ,
Just want to thank you for the continuous updates.
I have installed your latest installer and instructions 3 days ago on MacOS Catalina 10.15.1 on a MacBook Pro mid 2012 and all seams to be running smooth even with the control panel and skins :)
Thanks 🤘

Wouter1, 2019-12-02:
@Kokonutz  thanks for reporting, good to hear!

hanggard, 2020-01-31:
I confirm that the sound is working for me at macbook pro 2017 latest catalina. Just installed the driver.

Wouter1, 2020-01-31:
@hanggard  thanks for the report!

TheGreatKat, 2020-03-27:
Hey guys!
I can't do it on my Mac with Catalina... I try to do everything like in instruction of the installation but I am not sure how to give full disk access to driver installer> I can easily find to give it to Terminal but what you mean by driver installer? And then sudo mount -uw / is not recognized in Terminal...
Hope you all are doing well, regards!

grimrian, 2020-03-27:
Hi Maybe you could use the previous version, the kext installer?

Cheers

Wouter1, 2020-03-27:
Please only post *proper working* reports in this ticket. No issue discussions here. Make a new issue if you have an issue or post in the appropriate existing ticket related to your problem

popowsergei, 2020-05-21:
10.14.6
0204
Native EMUU_MacAppDrv_US_1_50_07 works, but the choice is only 44.1, 48, 96.
The panel works, volume control, MIDI. Everything works except for 176.4 and 192 kHz.

Wouter1/EMU-driver v11
Volume control doesn't work – it freezes. 
All frequencies are available, but on 176.4 and 192 there is noise and freezes after 5 minutes.

wtnh, 2020-08-04:
Works: EMU 0404 USB with Macbook Pro "Late 2013" running Catalina. Control panel only works if invoked via terminal as mentioned elsewhere, but this is a minor issue for me. I am using the 0404 for audio amplifier distortion testing and the driver works perfectly. See attached REW screenshot for the 0404 connected in a loopback mode for FFT testing.
[image]

roddy20, 2021-01-13:
EMU 0202, Big Sur, works OK
kext installed to /L*/E*

Wouter1, 2021-01-13:
@roddy20  thanks for reporting!

roddy20, 2021-01-13:
> @roddy20 thanks for reporting!

Thank you for the driver, I can do more tests if needed
I played a guitar with Waves GTR, sound is OK, no hearable latency

Wouter1, 2021-01-13:
@roddy20  yes, maybe you can give some info in ticket #132 , there were some troubles installing on Big Sur in the past. Did you just run the installer? Did you need to do something different from the standard? Or did you follow what  jefazo2k5  wrote?

roddy20, 2021-01-13:
> @roddy20 yes, maybe you can give some info in ticket #132 , there were some troubles installing on Big Sur in the past. Did you just run the installer? 
I did

Smuteee, 2021-01-16:
Okay, here is my report. Macbook Pro mid 2014, 15’ retina, Core i7 2.8 Ghz running Catalina with all the latest updates.
After finding your driver via google search on github i decided to dust off my old EMU 404 usb that was basically useless since Win xp days and gave it a go because i remember its sound to be quite good and (driver issues aside) to be of good quality (and because i have not much else to do during lockdown) ; ). Some try and error on a hackintosh machine core i7 back in 2016 did never really work out for me, in fact it was a pain i. t. a and my failure back then has bothered me ever since. Bought two used original apple MAC on the cheap last year. One MacPro (umbrellastand)which is my productive system, all updated and no hacks and the Macbook for experiments. Following your instructions it worked right out of the box (in fact better than the original creative installation with 3pages of disclaimers and installation windows hiding behind windows hardware lab warnings) it all works like a charm, recording, listening to apple music and playing on a midi keyboard a vst synth at the same time, switching buffers and sample rates, even the s/pdif ports i remember never worked for me in the past are now okay, even when i remove the usb connection the interface still works in 48khz as a connection between two other dac ( one via recording line inputs and the other via s/pdif input )and my active speakers as a switchable volume control. Only issue i have is the control panel not running with skins if not started in terminal. Thank you again Wouter1 for all the amazing work and continued support to reanimate this nice peace of abandonware. ; )

Smuteee, 2021-01-16:
As per original instructions, use good quality usb cables, no hubs !

Wouter1, 2021-01-16:
@Smuteee thanks!

bezromval, 2021-02-02:
Can somebody confirm of working with M1 chip?

Wouter1, 2021-02-04:
@bezromval  From what I read here https://developer.apple.com/documentation/apple_silicon/about_the_rosetta_translation_environment

Rosetta can translate most Intel-based apps, including apps that contain just-in-time (JIT) compilers. However, Rosetta doesn’t translate ..    Kernel extensions

the answer seems no.

djkondakov, 2021-02-27:
Thanks to the author for the driver, everything works perfectly on Mac OS Mojave, except for S / Pdif. I installed the original drivers, put your drivers on top. I connected the player to the Analog to Digital convector. Tosklink cable optics. Ext on the sound card blinks but no sound. I switched Direct Monitor to S / Pdif mode, maybe my Direct Monitor volume control is broken, because it scrolls. I tried to connect via Coaxial, the result is the same, only Ext does not flash. The convector looks good, but why there is no sound it did not work out, perhaps the convector has incompatible data transfer protocols, it is the simplest.

djkondakov, 2021-02-27:
S/PDIF Optical work. Yeah )))) Analog to Digital Audio Converter output to E-MU 0404 USB 48 kHZ External Clock work, great =)

djkondakov, 2021-02-27:
Direct monitor: Main

djkondakov, 2021-02-27:
Sync Status: Int + Ext

Wouter1, 2021-03-01:
@djkondakov don't install the "original driver". Just install the driver I provide. 
So you got it all working properly?

Wouter1, 2021-04-27:
@thelostprophet21  In #138  reported

Hello. For me it's working perfectly. I'm using the sample rate at 48khz with the EMU 0202.
Small note: I had to open the installer with terminal in order to have it run.

macOS BigSur v 11.2.3

2.4 Ghz 8- Core Intel i9

[6 more replies]