Wouter1/EMU-driver #112: DAW and bugreporting

fuuuckREG, 2018-06-30:
hi
0204 works not so good for coreaudio and asio.
i test it on sierra and now on hisierra, it the same.
1. 44khz it dont work anyway...only for listening in player, not any DAW.
2. also it lag many times and i need go to driver cont.panel and change freq to different to give it "reset".
2.1 for tets was be notice that best one is only switch on 192khz.
3. i test also recording compensate latency with LogicPro I/O tools. it lag many time and anytime give random ping..i just switch freq or buffer, back and it show new numbers..
i put here (down) some numbers that i test, but it take some time and makes me crazy so its not very full.

so if i can help you to fix it i give you all logs and do any tests that i can do.
so let me know what you need and what i must to do.
best regards.

```
i/o tools test (- is minus, number is samples)
192khz
1k -63
32 -65
64 -65
128 -60
256 -64

176.4khz
512 +310
32 -57 (random)
256 -56 (random)

96khz
1k -246
32 -275 (-553)
128 -19 (-292)
256 -291 (main?)
512 -19 (-291)

88khz
32 -255
128 -312
256 - 312
1k -312

48khz
32 -1
64 -114
256 - 116
512 -114
1024 -138
```

Wouter1, 2018-07-01:
@fuuuckREG 

>0204 works not so good for coreaudio and asio

My driver is only for coreaudio. I think programs needing asio are on their own, they need to fix it themselves.

There should be no need to compensate for latency. I fine-tuned the driver so that it reports the actual latency very accurately. If that does not work properly, it must be due to something more fundamental in the driver being not working as it should.

Your user name is a bit offending. I suppose you got frustrated that you need to register to post here (I can't help that...) but your point then gets to the wrong person (me)

>44khz it dont work anyway...only for listening in player, not any DAW.

Now that's odd. Playback is the most complex part of the driver, it needs the recording part to work to start with. 

Does the driver work properly with "the player" at all frequencies? 

Did you install driver with standard latency settings?

Can you play and record with audacity?

I have no idea what the io tools output means nor do I see how it would help to fix the issue.

I suggest that you look into the log file to see if it logs important errors

fuuuckREG, 2018-07-01:
my nick its not about you, dont worry, is fucking site dont accept 5 mail.
i lost many time becouse of crazy admins here..
and now it delete accepted one, becouse it not accept mails from site..ofcourse, i dont need this shit..stupid admins. i create it only for this topic///
anyway..sorry for it, i just was very angry, put regform again and again..

1. so i/o toos it pugin in logic pro x - using for calculate phicycal latency for recording.
it must be same becouse it calculate how long cables etc..but is random..
if you dont know look this https://www.youtube.com/watch?v=7oNIq10gA0g
2. where is log file and which name one you need? or all? say me place i give you everything.
3. audacity i dont have it if you about DAW.
4. driver works in players in any freq. just for playback, but in daw - no.
5. when install i choose yes for qestion about small latency.

Wouter1, 2018-07-01:
@fuuuckREG  Sorry to hear that. But please change your language, I don't like it and the whole world can view it 

@1 sorry I don't have time for that and it may be irrelevant 
@2 you need to open the console application (Applications folde) and then you can see the logs. 
@3 google it, you can download it for free.
@4 sounds then like a daw problem. Recording also works fine?
@5 maybe you can also try normal latency although low latency should work (unless you have old slow compu?)

fuuuckREG, 2018-07-02:
sorry i dont understand "change language"? im not english speaker and dont learn it.

i take a detailed video for you
so as i did it detailed, i notice that i was wrong about players..
i just never open panel and think that is 44khz by default..but it was not..
if you wanna know about players, its not important, i use this, but the same for all..
DAW also the same, cubase with asio - no difference.
so here is the video
https://mega.nz/#!EQ0CWYYL!6oIRByGBq9owl5UyEdDzTC2pPBHqbv7ArbXwFWmJQnQ
1. sad to hear it..its important for recording vocals..
2. ok today i come with video, logs i look little later
3. but i dont use it and dont wont, even if it good work. all my projects in logic and cubase..if you can take some info with it i do? its really can give some logs or etc?...
4. it was my mistake, look in video, both not works
5. so sad, if i just unistall and install again it ok? or i can do it without unistall just run installer in master?

Wouter1, 2018-07-02:
@fuuuckREG 

>sorry i dont understand "change language"? im not english speaker and dont learn it.

You were using rude words. I ment that you should change that and not use them here. 

You should really open the control panel to see what it says. Can you try that and post pic of control panel here?

@3 audacity is using core audio which is what my driver does. I do not know about asio, FAIK asio means that the application itself will do the audio, not my driver. 
What happens if you just play audio through quicktime player?

@4 What exactly is not working? BTW Tried to  play your video but the online player is not working properly. I hear sound though so something seems working. I can't seem to download it. But if you say it works, then I don't need to hear it.

@5  yes uninstall and then install again. And yes, installing again will work too.

fuuuckREG, 2018-07-02:
3. logicpro is too using coreaduio...asio using only cubase
quicktime i write before, it same for any player, please look video..
4. i dont understand BTW? why you cant download o-O?? just click downloadn it without cloud, u dont need some players..it h265 maybie its problem?
but you need to look it, its not work, you hear sound of another, i show many things very detailed test.
5. i'll try it after i find log and you see this video, its important.

Wouter1, 2018-07-02:
@3 I can't help with logicpro, I'm using only coreaudio (quicktime, audacity) and sometimes reaper. I only hear sound with the online video, it stutters, and there's nothing to see 

@4 can't download https://mega.nz/#!EQ0CWYYL!6oIRByGBq9owl5UyEdDzTC2pPBHqbv7ArbXwFWmJQnQ, 
If I use tor the download starts and seems to complete properly but I get a file of 0 bytes. 

Firefox does not get to the player website, it says 
```
MEGA failed to load because of 

Filename: blob:https://mega.nz/854214b2-6c5e-c54a-9dfa-9d3fc656b026:61279
Exception: SecurityError: The operation is insecure.

Please click OK to refresh and try again.
```

How about my questions? eg "Can you try that and post pic of control panel here?"

fuuuckREG, 2018-07-02:
aa, i know this error, its becouse u use some blockers...im my case it CanvasBlocker..
off it or try another brouther w/o block plugins, thereis no viruses.

i try to put it here, so delete .zip for all files, but no first one and unarchive.
here limit for 10MB i cant upload 1 video...

[emu-h264.zip](https://github.com/Wouter1/EMU-driver/files/2154998/emu-h264.zip)
[emu-h264.z01.zip](https://github.com/Wouter1/EMU-driver/files/2154999/emu-h264.z01.zip)
[emu-h264.z03.zip](https://github.com/Wouter1/EMU-driver/files/2155005/emu-h264.z03.zip)
[emu-h264.z02.zip](https://github.com/Wouter1/EMU-driver/files/2155006/emu-h264.z02.zip)

3. please read good, Logic pro use COREAUDIO TOO!! Asio - is only for cubase, but in my case it not important becouse it the same results!
4. your qes, also write before "so i dont have any log options in panel and in app container no any log files..
maybe you mean log is just little messages in app window? if yeas so thereis no errors//
anyway i show everthing in video"

Wouter1, 2018-07-02:
@fuuuckREG  I have no blockers.

I can download the files all right but Stuffit expander says "A failure has occured, Unexpected stuffit Engine internal error". gunzip says "unknown suffix. ignored" and I don't see anything unzipped. 

Maybe you have special settings used in Logic Pro that are incorrect. I just don't know. I know that you can set problematic settings in Reaper like impossible small buffer sizes.

@4 I don't understand your answer. I'm asking you to post screenshot of the control panel. Not about logs. 
The logs you can see in the Console.app in Applications/Utilities. It's standard tool coming with the OS installation.

fuuuckREG, 2018-07-02:
i write "delete .zip for all files, but no first one and unarchive"
this site i cant put .z01 z02 z03 , so i rename it in .z01.zip, just delete zip, but not touch first one..
in logic no any problem settings, and i special use big buffer, but it not important..
anyway which way i can send you it? in video all you need to know..its crazy...

4. aa, you must write it before, i look into EMU control panel, that you tell me before, there is no log.

Wouter1, 2018-07-02:
You wrote "i try to put it here, so delete .zip for all files, but no first one and unarchive. here limit for 10MB i cant upload 1 video..."
I didn't understand that, thanks for your re-explanation. 

Great, I can see your video now.

Seems your main problem is with 44kHz? On other rates it seems all working fine?  And I see some speedup problem at 192 in your daw. 

Can you work around it by using the 48kHz or the 92kHz?

And the control panel is crashing. I know, but nothing I can do about that, creative never released the source code of that. It seems to do the job...

Wouter1, 2018-07-02:
Tried to attach your video here but it also refuses >10MB and says "Yowza, that's a big file". It's only 32MB, that's not big ...

Wouter1, 2018-07-02:
@fuuuckREG  "aa, you must write it before, i look into EMU control panel, that you tell me before, there is no log."

I'm totally not following you. I never said that there is a log in the control panel. I asked for 2 things: (1) shot of the control panel (which you gave in above screenshot) and (2) log info from console application

fuuuckREG, 2018-07-02:
> Tried to attach your video here but it also refuses >10MB and says "Yowza, that's a big file". It's only 32MB, that's not big ...
i tell you its crazy site and you dont belive me ;)

Wouter1, 2018-07-02:
@fuuuckREG  

Sounds like you can do most work with a workaround. Maybe I can see something in the logs to get a clue what's going on at 44. Please post the relevant info from the logs.

Yes you open the console and then switch the driver with control panel to 44kHz. The console will show a lot of messages, also from other system tasks. You can set filters to select the EMU, or just have luck if there is no other system task messaging at the same time. Scan the log for relevant info that might point to the problem. It does NOT save anything, you have to copy/paste FAIK

Wouter1, 2018-07-02:
@fuuuckREG  BTW video looks good, nice job on that. Bit weird thought that it only plays in VLC not in Quicktime.

fuuuckREG, 2018-07-02:
thank you! i did it for show everything i can to kill any qestions.
ill try console tomorrow, now is night, im from japan.
just say how put filter for EMU, its command or i can see something like this in options?

Wouter1, 2018-07-02:
@fuuuckREG  Just start it up , switch the audio rate with control panel and check the messages coming up in the console. Check the task name associated with the messages you need and then add a filter for that task name.

Wouter1, 2018-07-02:
@fuuuckREG  I'm wondering about that performance warning you got in the video.  What compu do you have (type, MHz, processor)?

fuuuckREG, 2018-07-03:
>  I'm wondering about that performance warning

this one was only becouse it error with diffirent freq driver and daw, daw force it back and 44khz not works so it error, in normal case there is no problem, i have new model its skylake i7-6700.

so here is log
i switch to 44/48/192 etc..no so many info, think its can help..
[emu-log.txt](https://github.com/Wouter1/EMU-driver/files/2157915/emu-log.txt)

Wouter1, 2018-07-03:
@fuuuckREG  thanks for posting the log. Unfortunately you captured the wrong device. This is the log of the control panel. I need the log of the driver itself. When you change the rate in the control panel, the kernel task will respond to that.

Wouter1, 2018-07-03:
@fuuuckREG  >skylake i7-6700

Seems fast enough then, unless they really screwed up with other system components like cheap memory. Is this a mac or a hackintosh?

fuuuckREG, 2018-07-03:
ou, problem;((
is hackintosh, but setting not by newbie, work hard on it, memory ddr4 2133ghz

so i dont saw driver name, i look again, but to be shure can you say real name of driver?

Wouter1, 2018-07-03:
@fuuuckREG  Ok. So it might be some compatibility problem of for instance the USB chipset...

Anyway maybe the log says something useful.

It's something with EMUUSBAudioDevice I believe. I don't have the device here but just turn the filter off and check the messages

fuuuckREG, 2018-07-03:
ok, after mb 3-4hours i come back with it.
p.s. please look also your mail then you put in profile.

fuuuckREG, 2018-07-03:
so there is no another EMU process..
the one most near to it that come if i switch freq is coreaudio.

the first i just filter for you lines, this two comes when switch on 48 after 44
```
error	02:27:53.924297 +0900	coreaudiod	HALB_UCObject.cpp:399:SetProperty:  HALB_UCObject::SetProperty: got an error from the IORegistry, Error: 0xE00002BC
error	02:27:57.674760 +0900	coreaudiod	HALB_UCObject.cpp:399:SetProperty:  HALB_UCObject::SetProperty: got an error from the IORegistry, Error: 0xE00002BC
```

and next there is full log of it
when open panel it was 192, so i start switch down 176-96-88-48-44-192
[emu-log.txt](https://github.com/Wouter1/EMU-driver/files/2158929/emu-log.txt)

Wouter1, 2018-07-03:
@fuuuckREG  No coreaudiod not the one, that's something of osx. I'm pretty sure it's something wit EMU and it spits a lot of logs when you change the rate.

fuuuckREG, 2018-07-03:
but thereis no another emu...
i can put all log after but is many prosess sims to bi too big..

Wouter1, 2018-07-03:
@fuuuckREG  Maybe kernel then? I dont have device here so I can't help you further. Disable other stuff , disable internet etc so that the log output in general is only few lines per second. Then you can see lot of output n log after you switch rate. Should be stupid simple.

Wouter1, 2018-07-03:
@fuuuckREG  I saw the log with logs of coreaudiod. It only has HALS_* messages, this is from the core HAL system, not of my driver. If 10s is too big, then post 5s *from moment you switch the rate*.

fuuuckREG, 2018-07-03:
i was fast as wind! ..if now is right i happy
[emu-log2.txt](https://github.com/Wouter1/EMU-driver/files/2159223/emu-log2.txt)

Wouter1, 2018-07-03:
```
default	06:05:10.417520 +0900	SafeEjectGPUStartupDaemon	notify_post returned 0
default	06:05:10.592883 +0900	tccd	PID[861] is checking access for target PID[653]
default	06:05:10.598490 +0900	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
default	06:05:11.464480 +0900	SafeEjectGPUStartupDaemon	notify_post returned 0
default	06:05:11.982906 +0900	apsd	<private>: Received request from <private> to generate token for topic "<private>" and identifier ""
default	06:05:11.983023 +0900	apsd	Certificate not yet generated
default	06:05:11.983102 +0900	apsd	<private>: Rescheduling request to generate token for topic "<private>" and identifier ""
default	06:05:12.417405 +0900	SafeEjectGPUStartupDaemon	notify_post returned 0
default	06:05:13.045278 +0900	apsd	<private>: Received request from <private> to generate token for topic "<private>" and identifier ""
default	06:05:13.045395 +0900	apsd	Certificate not yet generated
default	06:05:13.045468 +0900	apsd	<private>: Rescheduling request to generate token for topic "<private>" and identifier ""
default	06:05:13.427035 +0900	SafeEjectGPUStartupDaemon	notify_post returned 0
default	06:05:13.471685 +0900	E-MU USB Audio Control Panel	-[NSMenuItem accessibilityIsAttributeSettable:]: unrecognized selector sent to instance 0x60c0000aa620
error	06:05:13.472744 +0900	E-MU USB Audio Control Panel	An uncaught exception was raised
error	06:05:13.472778 +0900	E-MU USB Audio Control Panel	<private>
error	06:05:13.477108 +0900	E-MU USB Audio Control Panel	<private>
default	06:05:14.088282 +0900	E-MU USB Audio Control Panel	-[NSMenuItem accessibilityIsAttributeSettable:]: unrecognized selector sent to instance 0x60c0000aa440
error	06:05:14.088486 +0900	E-MU USB Audio Control Panel	An uncaught exception was raised
error	06:05:14.088547 +0900	E-MU USB Audio Control Panel	<private>
error	06:05:14.091538 +0900	E-MU USB Audio Control Panel	<private>
default	06:05:14.417477 +0900	SafeEjectGPUStartupDaemon	notify_post returned 0
default	06:05:14.600832 +0900	kernel	+ IOAudioStream[<private>]::setProperties(<private>)
default	06:05:14.600850 +0900	kernel	  24: streamFormat->fBitWidth = 24
default	06:05:14.600850 +0900	kernel	  0: streamFormat->fAlignment = 0
default	06:05:14.600852 +0900	kernel	  1: streamFormat->fByteOrder = 1
default	06:05:14.600853 +0900	kernel	  2: streamFormat->fNumChannels = 4
default	06:05:14.600854 +0900	kernel	  0x6c70636d: streamFormat->fSampleFormat = 0x6c70636d
default	06:05:14.600855 +0900	kernel	  0x73696e74: streamFormat->fNumericRepresentation = 0x73696e74
default	06:05:14.600856 +0900	kernel	  24: streamFormat->fBitDepth = 24
default	06:05:14.600857 +0900	kernel	  24: streamFormat->fBitWidth = 24
default	06:05:14.600858 +0900	kernel	  0: streamFormat->fAlignment = 0
default	06:05:14.600859 +0900	kernel	  1: streamFormat->fByteOrder = 1
default	06:05:14.600860 +0900	kernel	  2: streamFormat->fNumChannels = 4
default	06:05:14.600861 +0900	kernel	  0x6c70636d: streamFormat->fSampleFormat = 0x6c70636d
default	06:05:14.600862 +0900	kernel	  0x73696e74: streamFormat->fNumericRepresentation = 0x73696e74
default	06:05:14.600864 +0900	kernel	  24: streamFormat->fBitDepth = 24
default	06:05:14.600865 +0900	kernel	  24: streamFormat->fBitWidth = 24
default	06:05:14.600866 +0900	kernel	  0: streamFormat->fAlignment = 0
default	06:05:14.600866 +0900	kernel	  1: streamFormat->fByteOrder = 1
default	06:05:14.600867 +0900	kernel	  2: streamFormat->fNumChannels = 4
default	06:05:14.600869 +0900	kernel	  0x6c70636d: streamFormat->fSampleFormat = 0x6c70636d
default	06:05:14.600870 +0900	kernel	  0x73696e74: streamFormat->fNumericRepresentation = 0x73696e74
default	06:05:14.600871 +0900	kernel	  24: streamFormat->fBitDepth = 24
default	06:05:14.600871 +0900	kernel	  24: streamFormat->fBitWidth = 24
default	06:05:14.600873 +0900	kernel	  0: streamFormat->fAlignment = 0
default	06:05:14.600874 +0900	kernel	  1: streamFormat->fByteOrder = 1
default	06:05:14.600874 +0900	kernel	  2: streamFormat->fNumChannels = 4
default	06:05:14.600876 +0900	kernel	  0x6c70636d: streamFormat->fSampleFormat = 0x6c70636d
default	06:05:14.600878 +0900	kernel	  0x73696e74: streamFormat->fNumericRepresentation = 0x73696e74
default	06:05:14.600878 +0900	kernel	  24: streamFormat->fBitDepth = 24
default	06:05:14.600893 +0900	kernel	  24: streamFormat->fBitWidth = 24
default	06:05:14.600894 +0900	kernel	  0: streamFormat->fAlignment = 0
default	06:05:14.600895 +0900	kernel	  1: streamFormat->fByteOrder = 1
default	06:05:14.600896 +0900	kernel	  2: streamFormat->fNumChannels = 4
default	06:05:14.600898 +0900	kernel	  0x6c70636d: streamFormat->fSampleFormat = 0x6c70636d
default	06:05:14.600899 +0900	kernel	  0x73696e74: streamFormat->fNumericRepresentation = 0x73696e74
default	06:05:14.600900 +0900	kernel	  24: streamFormat->fBitDepth = 24
default	06:05:14.600901 +0900	kernel	  24: streamFormat->fBitWidth = 24
default	06:05:14.600922 +0900	kernel	  0: streamFormat->fAlignment = 0
default	06:05:14.600937 +0900	kernel	  1: streamFormat->fByteOrder = 1
default	06:05:14.600941 +0900	kernel	  4: streamFormat->fNumChannels = 4
default	06:05:14.600942 +0900	kernel	  0x6c70636d: streamFormat->fSampleFormat = 0x6c70636d
default	06:05:14.600943 +0900	kernel	  0x73696e74: streamFormat->fNumericRepresentation = 0x73696e74
default	06:05:14.600946 +0900	kernel	  24: streamFormat->fBitDepth = 24
default	06:05:14.600947 +0900	kernel	  24: streamFormat->fBitWidth = 24
[... cut ...]
error	06:05:20.221478 +0900	deleted	fsctl error: Invalid argument, using HARDCODED low disk threshold 10737418240 for volume_size: 63478689792
error	06:05:20.221527 +0900	deleted	fsctl error: Invalid argument, using HARDCODED low disk threshold 5368709120 for volume_size: 63478689792
error	06:05:20.221566 +0900	deleted	fsctl error: Invalid argument, using HARDCODED very low disk threshold 3221225472 for volume_size: 63478689792
default	06:05:20.221603 +0900	deleted	CACHE_DELETE_VOLUME_SIZE : 63478689792

```

Wouter1, 2018-07-03:
@fuuuckREG  yes this is good, it was kernel as I guessed 2nd time.

The EMU control panel reports that it crashed. I can also see you switch from 192k to 44kHz.

The number of input channels is switched from 2 to 4. 

Then we can see what probably is the problem

```
default	06:05:14.601001 +0900	kernel	Bug? EMUUSBAudioConfigObject::GetEndpointPollInterval stream not opening
default	06:05:14.601008 +0900	kernel	4 channel 24 bit @ 44100 Hz is not supported. Suggesting alternate setting 255
```

Then it looks you are switching AGAIN, now to 48kHz?

I'm not sure why something thinks the device supports 4 channels. I would need to dig very deep in the code (it's not my code that does this part).

fuuuckREG, 2018-07-03:
i swtitch from 192 to 44 and after 2 sec to 48 yes. becouse its bouth not work

soo what is reasult? i dont understand,,,that means you can nothing?

Wouter1, 2018-07-03:
@fuuuckREG  that means that it probably can be fixed if I had the time to debug this further AND a 0204. Maybe just guessing where the problem is (without 0204) might be sufficient but highly inconvenient. 

Can you try if the workaround mentioned in #71 works for you?

fuuuckREG, 2018-07-03:
i hope you will get time...
1. but it have 2 mono output 1headphone and 1 stereo, mb stereoout is #4?
2. i dont understand what i need to try? i cant see in this post which one i need..

Wouter1, 2018-07-03:
@fuuuckREG  

1. I don't get what you are saying. What is "mb". 
2. Check the first pic in #71. You need to set "Audio-Midi settings" from 4 outs to 2 outs 

I have very little time at least for the coming 4 months.

Wouter1, 2018-07-03:
The message "Need to adjust channel controls" comes from EMUUSBAudioEngine::performFormatChange.

We  have a note there

>            // FIXME this gives errors when switching to 4 channels. Maybe remove it?
>            // The error may be because beginConfigurationChange() was not yet called.
>            // this may be related to disabled code at end of this function?
>            // usbInputStream.audioStream->setFormat(newFormat,false);

Wouter1, 2018-07-03:
That sets the number of channels for the input stream. On 0204 it's nonsense to set it to 4 as there are only 2 input streams. 

The request that we get from the IOKit (OSX kernel) is containing this info. So we're just getting wrong request from OSX. 

Now performFormatChange does not even have an option to set different number of channels for input and output???

Wouter1, 2018-07-03:
The performFormatChange works on just 1 channel. So we should get two calls, one for the usbInputStream and one for the usbOutputStream.

Wouter1, 2018-07-03:
There are more code warnings in that area. Line 1476

            // Wouter: following test is broken, it always will succeed!

The call to performFormatChange=0, which is the OUTPUT direction, and that one has 4 channels. So the call from OSX is correct. However the "special hack" decides taht if we're working on output, there is a new sampling rate AND the number of channels differs, then we  need to change the 

Checked very old code, it always was like this except that I added comments on the weirdness of the code at some point.

Wouter1, 2018-07-03:
The only thing that the weird code around "Need to adjust input number of channels" does is to set needToChangeChannels. And that variable is de-facto not used because the code that kicked in in that case (line 1592) is disabled. So whatever happens there is not the cause of the current problem.

Wouter1, 2018-07-03:
The performFormatChange fails but it seems the problem is caught. But the code to get the alternative setting is broken (as already reported in code). Maybe we can figure out what that code was supposed to do?

fuuuckREG, 2018-07-04:
wow i think you write like memo for yoursels, its too difficult for me i cant get it so fast ww
1. mb is maybe
2. this pic crazy lang, are you russian? i dont understand it lol, but ok i try change there for 2 channels.

Wouter1, 2018-07-04:
@fuuuckREG  yes I was investigating and I write down the details. That's because there can be a long time between moments I have time to work on it and I dont want to redo the investigations.

Yes please try that, it might work for 44 & 48kHz.

Can you please post the logs also for the moment you TURN ON the EMU? The first 5 secs or so. I need it to see which data transfer modes the 0204 supports in order to fix some code deep down.

fuuuckREG, 2018-07-04:
i tried it, so it looks like it works for player, but for daw stay same as i show in video.
also it not easy to switch, it come back to prewios freq many times, so we need to find one from it can switch on 44 and 48.
ok, later i post log of turn on.

sauio, 2018-11-04:
nice to admins restore it! so can you tell your time map to start this, just about is ok

Wouter1, 2018-11-04:
@sauio  Nothing happened since july. Maybe I can find some time in a weekend this or next month.

sauio, 2018-11-05:
ok, come to check this topic 1-2 month after, if you need some more info or logs let me know here i support you with this everything i can do.

sauio, 2018-12-17:
hiiii
there is already new year coming soon.. still no santa present for us? T________T

Wouter1, 2018-12-17:
@sauio sorry no news here. All my time goes into another project at this moment.

Wouter1, 2019-03-17:
Checked the code again.

Problem probably is in EMUUSBAudioEngine somewhere in line 1490-1504. 

Situation is : performFormatChangeInternal for a new OUTPUT channel.
We first need to change the input channel and the code tries the newAlternateSettingsID for that which was however for the output channel. On 0204 no such ID is available for the input. VerifySampleRateIsSupported thus fails

I think this was as intended for 0204, and that they did some recover action in that case in line 1493 -.

Wouter1, 2019-03-17:
The first step in findAltInterfaceWithSettings is FindNextAltInterfaceWithNumChannels. 
That fails, because there IS no interface with 4 channels input. 

This call is wrong in this case, we need another interface with the requested SAMPLE RATE, not with the same #channels.

Wouter1, 2019-03-17:
FindAltInterfaceWithSettings is used in many places. 
In other places, they check the result and try different calls, we might try the same

like trying explicitly with lower number of channels in EMUUSBAudioEngine line 1475

The test in line 1498 should be changed, I think what was intended is and AND instead of the OR:

```
if ((1 != mPollInterval) && (8 != mPollInterval))
```

Wouter1, 2019-03-17:
git push (from xcode) is failing

```
SSL error: error:1407742E:SSL routines:SSL23_GET_SERVER_HELLO:tlsv1 alert protocol version (-1)
```

https://stackoverflow.com/questions/48938019/git-pull-push-unable-to-access-https-ssl-routines-seem-to-be-down

Wouter1, 2019-03-17:
pushed manually from command line...

Wouter1, 2019-03-18:
Let's switch to develop branch for testing fixes for this as I can't test myself.

sdfsssdf, 2019-04-06:
@Wouter1
yeah, finally it comes! im here now, what can i do to test it?
there is old bunch in downloads, so if download master from top page will be ok?

Wouter1, 2019-04-06:
@sdfsssdf Thanks for offering to help testing. But nothing significant happened yet, just some additional documentation and searching.

sdfsssdf, 2019-04-06:
@Wouter1, but you wrote T_T
"Let's switch to develop branch for testing fixes"..

[33 more replies]