Wouter1/EMU-driver #113: Hacintosh Sierra and High Sierra, all right)

aleksandrphp, 2018-09-14:
Hardware:
```
Gigabyte GA-B75-D3H
I5-3330
E-MU 0404 USB (connect to USB 2.0 port)
```
First install Sierra and official Creative drivers (up to 96 kHz). Then find this project and install it. All frequencies unblocked, work fine. Then update to High Sierra and still good;)
Previous users write about 0,5 sec lag. It present, but not serious issue. Thank you for project!

closed 2018-09-14 by aleksandrphp

Wouter1, 2018-09-14:
@aleksandrphp  thanks for reporting!
Great to hear it all works on sierra.

But 0.5 seconds latency? That is not good. I measure 0.0049 seconds with buffer size 64 on 192kHz.
Did you follow the measurement instructions indicated here
https://github.com/Wouter1/EMU-driver/blob/master/Latency.md

aleksandrphp, 2018-09-15:
I mean 0,5 glitch. First comment 
https://github.com/Wouter1/EMU-driver/issues/109#issue-290276576

Wouter1, 2018-09-15:
@aleksandrphp   I'm not following.  0.5 glitch? As in half a glitch? There's no such thing as 'half a glitch' just like there is no 'half a drop of water'.

aleksandrphp, 2018-09-16:
0,5 sec, ofcourse)

Wouter1, 2018-09-16:
@aleksandrphp  0,5 sec of WHAT? How did you measure it?