Wouter1/EMU-driver #115: Control Panel GUI Problem

Amaan05, 2018-10-24:
[image]

The control panel looks like this (see attached)

Skins are copied in the control panel folder. It looks the same regardless of where I launch it from.

Amaan05, 2018-10-25:
MacBook Pro 13-inch, mid-2012 with High Sierra 10.13.2

I did run the csrutil command in the recovery mode and the installation did say "You can close this window" etc. I'd be willing remove and do it all again if you suggest? 

Thanks for such a speedy reply!

Wouter1, 2018-10-25:
It should just work on high sierra. 

you have a EMU0404?  
Did you connect the EMU directly to the mac? 
It looks like the driver is unreachable. Is the driver itself working?
Do you see some hints to the problem in the system log?

virtualoftheday, 2018-10-26:
I have same issue with my 0204

[image]

Thats for device took off. 
For active 0204 panel looks like this:
[image]

I think drivers were installed correctly, I can see 0204 in Sound and Midi panels. 
Adding screenshots about my configuration

[image]

[image]

Amaan05, 2018-10-26:
@virtualoftheday  . Run the E-Mu control panel through Terminal instead of the usual way. 

@Wouter1  Sorry, checked some of the closed issues and found that by running through terminal gets the skins working properly. However, my driver isn't detecting my device (which is a usb 0404) . 
I haven't had time to fiddle around with it yet, but will try some more things and send you an update and my log report when I get time.

But yeah @virtualoftheday , running the panel through the terminal should solve all the problems in your case at least.

virtualoftheday, 2018-10-26:
Yes, thanks! 
It works through Terminal, can change settings and frequency.
Thanks again!
(In the meantime I prepared log of how control panel works when opening as usual, maybe it has some useful info for you)
[image]

Amaan05, 2018-10-26:
Right, I still haven't got my driver to detect my 0404 which i am connecting directly via USB. 
I don't get any messages from icdd making me think my device is not connecting with the computer. 

I am attaching the log ... (the -67062 error from what i google looks like a general thing and not to do with the driver but you be the judge) 
[New Log.pdf](https://github.com/Wouter1/EMU-driver/files/2519326/New.Log.pdf) << Log

I got half a mind to getting my usb cable checked as I have never used this device and cable before. 

Thanks a lot guys! For any help, whatsoever!

Wouter1, 2018-10-26:
@virtualoftheday  thanks. That post does not show any message from the EMU driver, it's just the system doing things related to the EMU such as configuring the USB port.

But yes from the devicesaudio output screenshot it looks like the EMU is working.

Wouter1, 2018-10-26:
@Amaan05  I checked your log file but I see hardly any messages from the EMU usb driver. It's mostly the kernel talking about a IOAudioEngine and coreaudiod talking about HAL and SoundFlower.

Could soundflower be causing problems here?

I also see a message from the *control panel* in the log. 

So from this log it seems you were starting the control panel, not the driver. The driver starts up when you turn on the connected EMU. It has nothing to do with the control panel. 

What do you see it in the USB device tree? (under the 'system report' button, see pic of @virtualoftheday)

I see a MacOS error -67062 but I have no clue what that could be referring to.

Amaan05, 2018-10-26:
dont see anything to do with the EMU.. so i guess it's safe to assume it's not connecting to the machine itself? I will see if i can test this USB cable or try one i know for sure works. 

[image]

_as for the -67062 error.. from what I gather from googling its seems to be a common error across Sierra systems. So I am not holding it responsible for our problem as of yet.. will look into this one some other day with someone more adept by my side._

Wouter1, 2018-10-26:
@Amaan05  Yes it's not showing up at all, the machine does not even detect that the EMU was attached to the USB port. Maybe it's the cable. Lights come up on the EMU when you turn it on?

Amaan05, 2018-10-26:
Yup. the 'Power On' light comes on .. 

(that's the only one that comes on. None of the other ones do no matter what I do.. I  am assuming they will be functional once the EMU interacts with the machine)

Wouter1, 2018-10-26:
@Amaan05  So far it looks like a hardware problem. Yes please try another cable, another port on the compu etc to rule out the cable. It also might be a defective port on your EMU

Wouter1, 2018-10-26:
@Amaan05  btw this also would explain why your panel lost its skin. It seems the skin only appears after the device connects with the driver.

Wouter1, 2018-10-26:
@virtualoftheday do you also have this missing skin if the device is on and the driver loaded properly? FAIK it's like this:

* double click control panel starts with skin if driver loaded correctly (so EMU connected and on, driver installed properly etc)
* the skin does not load if you double click the control panel but driver is not loaded
* if you run from terminal, skin loads also without driver (but of course you can't do anything).

Wouter1, 2018-10-26:
I added that info to the help/faq section.

virtualoftheday, 2018-10-27:
@Wouter1  >do you also have this missing skin if the device is on and the driver loaded properly?

I think so. I could run control panel from terminal and use it - change frequency, channel numbers etc - so driver works. Also I can control 0204 in Audio/Midi Setup.
But loading control panel by double-click makes skinless window every time.

Wouter1, 2018-10-27:
@virtualoftheday 
can you try ( you have the driver already installed, so we test only the control panel here)
* download fresh driver. Do not rename downloaded file. Unzip if necessary (double click zipped file)
* connect & turn on the EMU
* double click the downloaded&unzipped control panel

virtualoftheday, 2018-10-29:
Sorry, still no skin. But EMU works. 
[image]

Wouter1, 2018-10-29:
@virtualoftheday thanks for testing! 
Strange, Driver is working, using fresh downloaded control panel...  I can't reproduce this, maybe it is something only for the 0204 but I can't test that. Your previous log did not show any errors from the ControlPanel either.

I don't see much more options to fix this, without exception or way to reproduce. But maybe you can to try a different un-zip program? Some time ago someone had a unzipper that seemed to damage some files.

virtualoftheday, 2018-10-29:
I tried Unarchiver and Archive utility, same result.
What should I look for in my log file to help you clarify this?

Wouter1, 2018-10-29:
@virtualoftheday

Thanks for testing that. At this point I have no new suggestions.

 From your previous log,  "E-MU USB Audio Control..." seems to be the name that the control panel uses. But I thought you already posted the log above?

virtualoftheday, 2018-10-29:
I posted entries containing 'E-MU' string, so maybe some info was omitted there

camilodelvasto, 2018-12-02:
@virtualoftheday @Wouter1 How does one run the control panel from the terminal? I assume my driver is working (I can select the device in the control panel and can reproduce/record sound using GarageBand), but the control panel is showing this:
[image]

I tried the following from my downloads folder, but it's doing the same thing, so I assume that's not what you mean by 'opening it from the terminal':
```open E-MU\ USB\ Audio\ Control\ Panel.app```

Or perhaps is my driver not loading completely? Thanks for any help rescuing this awesome device...

Wouter1, 2018-12-02:
@virtualoftheday 

* start terminal (Applications/Utilities/Terminal)
* cd EMU-driver-master/ControlPanel
* E-MU\ USB\ Audio\ Control\ Panel.app/Contents/MacOS/E-MU\ USB\ Audio\ Control\ Panel

The "open" command from the terminal gives identical result to double-clicking the app so that doesn't help

Wouter1, 2018-12-02:
Added note on main page about the issues with 02204 and referring to this ticket

YangHuang, 2018-12-10:
I am having the same issue as @camilodelvasto on a 0404, solved by launching the app from terminal.

germanamv, 2019-04-03:
Anyone having trouble launching the control panel app normally, try this:
`sudo xattr -dr com.apple.quarantine <LOCATION OF .APP FILE>`
e.g.
`sudo xattr -dr com.apple.quarantine /Applications/ControlPanel/E-MU\ USB\ Audio\ Control\ Panel.app/`
This will remove special attribute (com.apple.quarantine) which is added during extraction of the zip to all files. Solved the problem for me.

Robertbadr, 2021-01-31:
Hi! I removed special attribute as @germanamv said and now Control Panel works good. Big Sur, 0204. But works only when 2 channels selected (via MIDI settings, there are no channel settings in Control Panel).

Wouter1, 2021-02-01:
@Robertbadr  Thanks,  seems I need to look into changing that attr, maybe in the installer. 

The 2-channel issue was a known issue, but it is not cleare to me whether this is always needed. The installer could be clarified on this.

bettercomb, 2021-09-03:
I had the same issue with ControlPanel appearing as noted at the top of this Issue. 
Reading through this thread, found the reference to suspected 'com.apple.quarantine' attribute and used 'xattr -rD' against the .app directory, which resolved the issue.

I"m happy to take a whack at modifying the installer script to silently perform this action. Might be something as simple as the following before exiting the script:

`#remove com.apple.quarantine attribute (adding during unzip?) for correct ControlPanel execution`
`if [ -d "$pathname/ControlPanel" ] ; then`
`  xattr -dr com.apple.quarantine "$pathname/ControlPanel"`
`fi`

Wouter1, 2021-09-04:
@bettercomb Thanks for looking into this!

But I don't quite understand, if the control panel appears, then the control panel DID start. If it would be quarantined, it would not even start. So why do you think tha setting the attribute can resolve this issue?

bettercomb, 2021-09-07:
It's not entirely clear to me how or why com.apple.quarantine affects Control Panel execution. But, I'll apply a bit more rigor to debugging a clean installation on another Mac to try to shed some light on the observed behavior. More to come.

Wouter1, 2021-09-07:
@bettercomb thanks, looking forward to it! Also please report if you find other relevant issues related to this ticket