Wouter1/EMU-driver #116: problem about 44/48khz not works in coreaudio

sauio, 2018-11-02:
hi, 
crazy admins delete my old account with all posts becouse i setup that i dont need a mails from them..
just come to ask did you save info that you need to fix problem about 44/48khz not works in coreaudio, 
or you need me to create some logs again?
if you save all, how's your plans about release? maybie you alredy have something or not even start?

Wouter1, 2018-11-02:
@sauio which ticket are you talking about? Maybe you can just post there instead?

sauio, 2018-11-03:
>Maybe you can just post there instead?
i wrote that admins delete it all! no thread anymore.
it was last topic here, mb 3mounts ago, i did video for you and many logs, about 44/48khz in logicproX.
you've said that it can be fixed but 3mounts you not have a time..

Wouter1, 2018-11-03:
@sauio  strange, I did not notice that anything got deleted. Yes I remember some video but It's not so easy finding back thee ticket. What was your previous user name?

sauio, 2018-11-03:
you cant back it, it was deleted with account, also this too can be deleted becouse it block everything that cant get emails...first come pop-up that account marked becouse block emails, and after delete it with all messages and threads, they crazy. 
previous name was something like fuuckreg

Wouter1, 2018-11-03:
@sauio  you might be right, it seems issue 112 disappeared

https://github.com/Wouter1/EMU-driver/issues/112

Not good, i'll check with the admins here what's going on. 

BTW I did not have much time yet, but the ticket contained my first analysis and details on where I need to work.

Wouter1, 2018-11-03:
@sauio  I contacted github to see what they are doing, through https://github.com/contact

Wouter1, 2018-11-03:
@sauio  GitHub restored ticket #112, it was indeed the missing ticket that you were referring to.
I close this one, as I understand it was only a request to get an update. If you have further questions you can post them in #112.

closed 2018-11-03 by Wouter1