Wouter1/EMU-driver #117: Hi

timmyleej, 2018-11-03:
IS there any way to disable 4 channel mode for emu0204? Every time I open Logic Pro X, It switches emu 0204 to 4 channel and every Im forced to switch it back cause 4 channel mode doesnt work with 0204 anyway.

Wouter1, 2018-11-04:
@timmyleej  I don't know but I think not. I can't reproduce as I don't have 0204. See also #112.

Wouter1, 2019-04-19:
@timmyleej you could edit the .plist inside the kernel driver (this is tricky, only for experts) or use the midi setup tools (applications/Utilities) to set it up manually

Also you might try the develop version that I hope is going to fix issues with 0204, see #112.

closed 2019-04-19 by Wouter1

timmyleej, 2019-04-20:
@Wouter1 #112 develop version (v11) did nothing to fix this issue, will try plist editing.

Wouter1, 2019-04-20:
@timmyleej  thanks for testing. Can you report what the EMU driver is logging? 

I'm particularly looking for this message and the messages around it 
"XXX channel YY bit @ ZZ Hz is not supported. Suggesting alternate setting XX"