Wouter1/EMU-driver #118: emu 0202 issues

Wouter1, 2019-02-12:
@maxelroad wrote (originally ticket #114 

new attempt to install the driver e-mu.
MacBook Pro 2015 16GB, Mojave
E-MU 0202 CSS
the computer does not see the device. The installation was carried out in accordance with the description.
I bring text from the terminal windows in accordance with the sequence of operations

recovery mode:
-bash-3.2# csrutil enable --without kext
csrutil: requesting an unsupported configuration. This is likely to break in the future and leave your machine in an unknown state.
Successfully enabled System Integrity Protection. Please restart the machine for the changes to take effect.

reboot

after reboot:
Last login: Tue Feb 12 09:44:19 on console
MacBook-Pro15:~ axel$ sudo spctl --master-desable
Password:
spctl: unrecognized option `--master-desable'
System Policy Basic Usage:
spctl --assess [--type type] [-v] path ... # assessment
spctl --add [--type type] [--path|--requirement|--anchor|--hash] spec ... # add rule(s)
spctl [--enable|--disable|--remove] [--type type] [--path|--requirement|--anchor|--hash|--rule] spec # change rule(s)
spctl --status | --master-enable | --master-disable # system master switch

Kernel Extension User Consent Usage:
spctl kext-consent ** Modifications only available in Recovery OS **
status
Print whether kernel extension user consent is enabled or disabled.
enable
Enable requiring user consent for kernel extensions.
disable
Disable requiring user consent for kernel extensions.
add
Insert a new Team Identifier into the list allowed to load kernel extensions without user consent.
list
Print the list of Team Identifiers allowed to load without user consent.
remove
Remove a Team Identifier from the list allowed to load kernel extensions without user consent.

reboot

after reboot:
Last login: Tue Feb 12 10:08:03 on ttys000
MacBook-Pro15:~ axel$ /Users/axel/Downloads/EMU-driver-master/installer ; exit;

EMU USB Installer by W.Pasman
Use low latency setting (n/y)?: ny

Adjusting driver for low latency
(Re)placing EMU kernel extension. We need the admin password for this.

Password:
Waiting for the system to finish installation. Can take several minutes.
Rebuilding the kernel
Waiting for lock to replace the old kernel
Driver is installed and ready for use!
You can close this window.
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

Wouter1, 2019-02-12:
@maxelroad what's the thing with ```sudo spctl --master-desable```?

The driver seems installed properly. So what happens when you try to use it? Do you see it in te USB device overview? (Apple menu/About this mac/Overview/System Report/USB)

maxelroad, 2019-02-12:
Computer driver shows.
There are no sound devices in the list of 0202
And I do not understand how to install the control panel from drivers 0202.

maxelroad, 2019-02-12:
This is what I see in System Report.

E-MU 0202 | USB:

  Product ID:	0x3f02
  Vendor ID:	0x041e  (Creative Labs)
  Version:	1.00
  Serial Number:	E-MU-03-3F02-07DA0C05-0E92F-8740T2A
  Speed:	Up to 480Mb/sec
  Manufacturer:	E-MU Systems, Inc.
  Location ID:	0x14100000 / 6
  Current Available (mA):	500
  Current Required (mA):	450
  Extra Operating Current (mA):	0

Devices:

Built-in Microphone:

  Default Input Device:	Yes
  Input Channels:	2
  Manufacturer:	Apple Inc.
  Current SampleRate:	44100
  Transport:	Built-in
  Input Source:	Internal Microphone

Built-in Output:

  Default Output Device:	Yes
  Default System Output Device:	Yes
  Manufacturer:	Apple Inc.
  Output Channels:	2
  Current SampleRate:	44100
  Transport:	Built-in
  Output Source:	Internal Speakers

Soundflower (2ch):

  Input Channels:	2
  Manufacturer:	ma++ ingalls for Cycling '74
  Output Channels:	2
  Current SampleRate:	44100
  Transport:	Unknown
  Input Source:	Default
  Output Source:	Default

Sonarworks True-Fi:

  Input Channels:	2
  Manufacturer:	SIA Sonarworks
  Output Channels:	2
  Current SampleRate:	44100
  Transport:	Unknown
  Input Source:	Default
  Output Source:	Default

Wouter1, 2019-02-12:
@maxelroad 

Ok the system notices the USB device and it has USB2 connection.

You don't need to install the control panel. You can just double click the "E-MU USB Audio Control Panel.app" inside the Control Panel inside the zip.

What happens if you do that?

And, do you see the EMU now if you go to system preferences/Sound?

maxelroad, 2019-02-12:
Not

С уважением, Михаил.
тел: +79169013595

Wouter1, 2019-02-12:
@maxelroad  Huh? Please refer to the questions and answer each

maxelroad, 2019-02-12:
The control panel does not open correctly; the selection points for the quantization frequency are not visible
just a gray window

device not visible in system settings

Wouter1, 2019-02-12:
@maxelroad  can you please collect and report the system log coming from the EMU driver? Start the log a second before turning on and stop a few seconds after turning on.

maxelroad, 2019-02-13:
Hi!
Look
Install terminal text:

EMU USB Installer by W.Pasman
usage: dirname path
Use low latency setting (n/y)?: ny

cp: v11/EMUUSBAudio.kext: No such file or directory
Adjusting driver for low latency
/Users/axel/Downloads/EMU-driver-master (1).zip Folder/EMU-driver-master/installer: line 52: cd: /Users/axel/EMUUSBAudio.kext/Contents/: No such file or directory
(Re)placing EMU kernel extension. We need the admin password for this.

Password:
Sorry, try again.
Password:
rm: /System/Library/Extensions/EMUUSBAudio.kext: No such file or directory
cp: EMUUSBAudio.kext: No such file or directory
cp: EMUMIDIDriver orig.plugin: No such file or directory
chgrp: EMUMIDIDriver.plugin: No such file or directory
chgrp: EMUUSBAudio.kext: No such file or directory
/System/Library/Extensions/EMUUSBAudio.kext failed to load - (libkern/kext) not found; check the system/kernel logs for errors or try kextutil(8).
Waiting for the system to finish installation. Can take several minutes.
Rebuilding the kernel
Waiting for lock to replace the old kernel
Driver is installed and ready for use!
You can close this window.
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

I don't like what the terminal says

Wouter1, 2019-02-13:
@maxelroad  I asked for the system log, not the text from the installer.

But the installer does not work prperly. If you want to reinsall you have to download the zip file fresh

maxelroad, 2019-02-13:
I am sorry, but how to make this log. I have never done this before.
The installer is downloaded from this link: https://github.com/Wouter1/EMU-driver.git <https://github.com/Wouter1/EMU-driver.git>

on Windows, which is installed in the program Рarallels everything works. The device is working.

Wouter1, 2019-02-13:
@maxelroad It's a long time ago so I can't remember exactly. Try using the Console application in Utilities and google how to filter the EMU messages

Andreaux, 2019-11-11:
Hello there. I seem to have a similar issue. Installed the drivers, they installed without any error. I see the E-MU 0404 USB in System Information, but it is not available as an audio device anywhere within the system. I am on Mojave, latest version.

Wouter1, 2021-01-18:
Mark Verduin mailed me

Na het afstoffen van mijn EMU 2002 volg ik met veel belangstelling je Github EMU-driver repository https://github.com/Wouter1/EMU-driver. Daar ik niet over een Github-account beschik, stuur ik je mijn vraag per mail - vanzelfsprekend mag je de vraag eventuele oplossingen gebruiken voor de community.

iMac (21,5 inch late 2009)
macOS Sierra versie 10.12.6 (16G2136)

De instructies voor installatie heb ik gevolgd (https://github.com/Wouter1/EMU-driver/blob/master/Install.md)
Ik loop tegen precies hetzelfde issue aan als #118 https://github.com/Wouter1/EMU-driver/issues/118
In de bijlage heb ik de log file bijgevoegd.

Zou je me willen helpen mijn EMU weer werkend te krijgen?  
```
standaard	09:22:11.500902 +0100	kernel	+ IOAudioDevice[<private>]::init(<private>)
standaard	09:22:11.500931 +0100	kernel	- IOAudioDevice[<private>]::init(<private>) returns 1
standaard	09:22:11.501003 +0100	kernel	+ IOAudioDevice[<private>]::start(<private>)
standaard	09:22:11.501051 +0100	kernel	- IOAudioDevice[<private>]::start(<private>)
standaard	09:22:11.501097 +0100	kernel	Sound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
standaard	09:22:11.501126 +0100	kernel	+ IOAudioDevice[<private>]::setPowerState(1, <private>)
standaard	09:22:11.501129 +0100	kernel	- IOAudioDevice[<private>]::setPowerState(1, <private>) returns 0x0
standaard	09:22:11.522132 +0100	kernel	Sound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
standaard	09:22:11.543189 +0100	kernel	+ IOAudioDevice[<private>]::setDeviceName(<private>)
standaard	09:22:11.543205 +0100	kernel	+ IOAudioDevice[<private>]::setDeviceModelName(<private>)
standaard	09:22:11.543211 +0100	kernel	- IOAudioDevice[<private>]::setDeviceModelName(<private>)
standaard	09:22:11.543215 +0100	kernel	- IOAudioDevice[<private>]::setDeviceName(<private>)
standaard	09:22:11.543223 +0100	kernel	+ IOAudioDevice[<private>]::setManufacturerName(<private>)
standaard	09:22:11.543229 +0100	kernel	- IOAudioDevice[<private>]::setManufacturerName(<private>)
standaard	09:22:11.543236 +0100	kernel	+findPipe: dir=1, type = 3
standaard	09:22:11.543241 +0100	kernel	endpoint found: epDirection = 1, epType = 3
standaard	09:22:11.543302 +0100	kernel	-findpipe: success
standaard	09:22:11.543557 +0100	kernel	+-IOAudioDevice[<private>]::initHardware(<private>)
standaard	09:22:11.544407 +0100	kernel	+ IOAudioEngine[<private>]::init(0x0)
standaard	09:22:11.544410 +0100	kernel	  properties(0x0) == NULL
standaard	09:22:11.544443 +0100	kernel	+-IOAudioEngine[<private>]::setState(0x0. oldState=0)
standaard	09:22:11.544449 +0100	kernel	- IOAudioEngine[<private>]::init(0x0)
standaard	09:22:11.544459 +0100	kernel	+ IOAudioDevice[<private>]::activateAudioEngine(<private>, 1)
standaard	09:22:11.544501 +0100	kernel	+EMUUSBAudioEngine[<private>]::start (<private>)
standaard	09:22:11.544526 +0100	kernel	Adding notification with custom matching dictionary
standaard	09:22:11.544572 +0100	kernel	+ IOAudioEngine[<private>]::start(<private>, <private>)
standaard	09:22:11.544780 +0100	kernel	Default sample rate is 96000
standaard	09:22:11.544784 +0100	kernel	Default alternate setting ID is 7
standaard	09:22:11.544836 +0100	kernel	There are 12 alternate stream interfaces on interface 2
standaard	09:22:11.544845 +0100	kernel	Interface 2, Alt 2 has a
standaard	09:22:11.544847 +0100	kernel	24 bit interface,
standaard	09:22:11.544856 +0100	kernel	2 channels, and
standaard	09:22:11.546928 +0100	kernel	++about to set output : ourInterfaceNumber = 1 & newAlternateSettingID = 8
standaard	09:22:11.546952 +0100	kernel	  0x6c70636d: streamFormat->fSampleFormat = 0x6c70636d
standaard	09:22:11.546958 +0100	kernel	  24: streamFormat->fBitDepth = 24
standaard	09:22:11.546971 +0100	kernel	  0x73696e74: streamFormat->fNumericRepresentation = 0x73696e74
standaard	09:22:11.546989 +0100	kernel	  24: streamFormat->fBitDepth = 24
standaard	09:22:11.546992 +0100	kernel	  24: streamFormat->fBitWidth = 24
standaard	09:22:11.546994 +0100	kernel	  0: streamFormat->fAlignment = 0
standaard	09:22:11.546996 +0100	kernel	  1: streamFormat->fByteOrder = 1
standaard	09:22:11.547000 +0100	kernel	- IOAudioStream[<private>]::validateFormat(<private>, 0x0, <private>) returns 1
standaard	09:22:11.547008 +0100	kernel	+ IOAudioStream[<private>]::setFormat(<private>, <private>)
standaard	09:22:11.547034 +0100	kernel	  2: streamFormat->fNumChannels = 2
standaard	09:22:11.547037 +0100	kernel	  0x6c70636d: streamFormat->fSampleFormat = 0x6c70636d
standaard	09:22:11.547040 +0100	kernel	  0x73696e74: streamFormat->fNumericRepresentation = 0x73696e74
standaard	09:22:11.547043 +0100	kernel	  24: streamFormat->fBitDepth = 24
standaard	09:22:11.547045 +0100	kernel	  24: streamFormat->fBitWidth = 24
standaard	09:22:11.547046 +0100	kernel	  0: streamFormat->fAlignment = 0
standaard	09:22:11.547048 +0100	kernel	  1: streamFormat->fByteOrder = 1
standaard	09:22:11.547051 +0100	kernel	  2: streamFormat->fNumChannels = 2
standaard	09:22:11.547053 +0100	kernel	  0x6c70636d: streamFormat->fSampleFormat = 0x6c70636d
standaard	09:22:11.547056 +0100	kernel	  0x73696e74: streamFormat->fNumericRepresentation = 0x73696e74
standaard	09:22:11.547059 +0100	kernel	  24: streamFormat->fBitDepth = 24
[... cut ...]
standaard	09:22:34.465583 +0100	kernel	+- IOAudioEngine[<private>]::removeUserClient(<private>) returns 0x0
standaard	09:22:34.465593 +0100	kernel	+- IOAudioEngine::removeUserClientAction(<private>, <private>) returns 0x0
standaard	09:22:34.465594 +0100	kernel	+- IOAudioEngine[<private>]::clientClosed(<private>)
standaard	09:22:34.465596 +0100	kernel	+- IOAudioEngineUserClient[<private>]::clientDied() returns 0x0
standaard	09:22:34.465699 +0100	kernel	+- IOAudioEngineUserClient[<private>]::stop(<private>)
```

Wouter1, 2021-01-18:
Mark it seems that the driver was starting up properly at 09:22:23.
And then at 09:22:30 you just turned off the EMU?

Can you open the control panel (from terminal/command line) and see what it says

Wouter1, 2021-01-18:
Mark replied

Yes, EMU turned off at 09:22:30.

Terminal says nothing, control panel just opens, but not right, see scr shot:

[image]

Wouter1, 2021-01-18:
Seems you ran the control panel by double clicking, as it has no skin. 
You should run it from the terminal.

Additionally, can you confurm that the extension loaded (about this mac/system report/extensions)

Wouter1, 2021-01-18:
Mark replied

I'm a novice in command line and did this:

open E-MU\ USB\ Audio\ Control\ Panel.app

in /Applications/Controlpanel/EMU-driver-master

It seems it doesn't load a skin. Am I doing something wrong?

Further the system reports says:
```
E-MU USB Audio Control Panel:

  Versie:1,50
  Afkomstig van:Onbekend
  Laatst gewijzigd:14-01-21 07:43
  Soort:Intel
  64-bits (Intel):Ja
  Locatie:/Applications/EMU-driver-master/ControlPanel/E-MU USB Audio Control Panel.app
```

Wouter1, 2021-01-18:
I tried this:

[image]

In terminal

/Applications/EMU-driver-master/ControlPanel/E-MU\ USB\ Audio\ Control\ Panel.app/Contents/MacOS/E-MU\ USB\ Audio\ Control\ Panel ; exit;

And this shows up:
[image]

But preferences haven been grayed out:

[image]

Wouter1, 2021-01-18:
It seems all working fine. You even got it working at 192k. Congratulations :-)

Wouter1, 2021-01-18:
Mark wrote

Still not good working. 

I unistalled
Reinstalled
then i get this in system logs:

Jan 18 14:12:46 Mark2 com.apple.xpc.launchd[1] (com.apple.quicklook[968]): Endpoint has been activated through legacy launch(3) APIs. Please switch to XPC or bootstrap_check_in(): com.apple.quicklook
Jan 18 14:13:08 Mark2 com.apple.xpc.launchd[1] (com.apple.imfoundation.IMRemoteURLConnectionAgent): Unknown key for integer: _DirtyJetsamMemoryLimit
Jan 18 14:13:38 --- last message repeated 1 time ---

Wouter1, 2021-01-18:
Mark

FAIK quicklook has nothing to do with the EMU driver. Maybe something with the control panel but I doubt it.

What is the status of the extension (About this mac/System Report/Extensions)

Wouter1, 2021-01-19:
Mark wrote
The control panel says now: No EMU Devices Found. 

[image]

I also noticed that the green Stereo/Mono leds on the hardware/device aren't working anymore.

Wouter1, 2021-01-19:
Mark, please undo your most recent actions to the place where it worked. Control panel should show what it showed you before, with sample rate and device ID

Wouter1, 2021-01-20:
Mark wrote
Undo the changes... I didn't klnow when it happened. So I uninstalled and reinstalled.

But whatever I try it says No E-MU Devices found.

In System Information I see:
```
E-MU 0202 | USB:

  Productcode:0x3f02
  Fabrikantcode:0x041e  (Creative Labs)
  Versie:1.00
  Serienummer:E-MU-FA-3F02-07D60A15-0EA98-STATION 01
  Snelheid:Maximaal 480 Mb/sec
  Fabrikant:E-MU Systems, Inc.
  Locatiecode:0x24100000 / 2
  Beschikbare stroom (mA):500
  Vereiste stroom (mA):450
  Extra stroom in werking (mA):0
```

This is was the console says when i start up the EMU.
```
standaard15:09:54.683994 +0100kernelSound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
standaard15:09:54.705025 +0100kernelSound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
standaard15:09:54.725366 +0100kernel+findPipe: dir=1, type = 3
standaard15:09:54.725371 +0100kernelendpoint found: epDirection = 1, epType = 3
standaard15:09:54.725438 +0100kernel-findpipe: success
standaard15:09:54.726243 +0100kernel+EMUUSBAudioEngine[<private>]::start (<private>)
standaard15:09:54.726267 +0100kernelAdding notification with custom matching dictionary
standaard15:09:54.726292 +0100kernel+EMUUSBAudioEngine::audioDevicePublished (<private>, 0x0, <private>)
standaard15:09:54.726299 +0100kernel++EMUUSBAudioEngine[<private>]: found device (<private>) for Audio Engine (<private>)
standaard15:09:54.726306 +0100kernel++EMUUSBAudioEngine[<private>]: found device (<private>) for Audio Engine (<private>)
standaard15:09:54.726310 +0100kernel-EMUUSBAudioEngine::audioDevicePublished (<private>, 0x0, <private>)
standaard15:09:54.726324 +0100kernel+EMUUSBAudioEngine[<private>]::initHardware (<private>)
standaard15:09:54.726825 +0100kernelDefault sample rate is 0
standaard15:09:54.726829 +0100kernelDefault alternate setting ID is 255
standaard15:09:54.726840 +0100kernel-EMUUSBAudioEngine[<private>]::start (<private>) = 0
standaard15:31:22.250355 +0100kernelSound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
standaard15:31:22.271402 +0100kernelSound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
standaard15:31:22.292513 +0100kernel+findPipe: dir=1, type = 3
standaard15:31:22.292519 +0100kernelendpoint found: epDirection = 1, epType = 3
standaard15:31:22.292582 +0100kernel-findpipe: success
standaard15:31:22.293604 +0100kernel+EMUUSBAudioEngine[<private>]::start (<private>)
standaard15:31:22.293630 +0100kernelAdding notification with custom matching dictionary
standaard15:31:22.293653 +0100kernel+EMUUSBAudioEngine::audioDevicePublished (<private>, 0x0, <private>)
standaard15:31:22.293659 +0100kernel++EMUUSBAudioEngine[<private>]: found device (<private>) for Audio Engine (<private>)
standaard15:31:22.293674 +0100kernel++EMUUSBAudioEngine[<private>]: found device (<private>) for Audio Engine (<private>)
standaard15:31:22.293680 +0100kernel-EMUUSBAudioEngine::audioDevicePublished (<private>, 0x0, <private>)
standaard15:31:22.293695 +0100kernel+EMUUSBAudioEngine[<private>]::initHardware (<private>)
standaard15:31:22.294157 +0100kernelDefault sample rate is 0
standaard15:31:22.294159 +0100kernelDefault alternate setting ID is 255
standaard15:31:22.294170 +0100kernel-EMUUSBAudioEngine[<private>]::start (<private>) = 0
standaard15:36:44.308143 +0100kernelSound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
standaard15:36:44.329190 +0100kernelSound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
standaard15:36:44.350315 +0100kernel+findPipe: dir=1, type = 3
standaard15:36:44.350321 +0100kernelendpoint found: epDirection = 1, epType = 3
standaard15:36:44.350388 +0100kernel-findpipe: success
standaard15:36:44.351475 +0100kernel+EMUUSBAudioEngine[<private>]::start (<private>)
standaard15:36:44.351502 +0100kernelAdding notification with custom matching dictionary
standaard15:36:44.351525 +0100kernel+EMUUSBAudioEngine::audioDevicePublished (<private>, 0x0, <private>)
standaard15:36:44.351532 +0100kernel++EMUUSBAudioEngine[<private>]: found device (<private>) for Audio Engine (<private>)
standaard15:36:44.351539 +0100kernel++EMUUSBAudioEngine[<private>]: found device (<private>) for Audio Engine (<private>)
standaard15:36:44.351543 +0100kernel-EMUUSBAudioEngine::audioDevicePublished (<private>, 0x0, <private>)
standaard15:36:44.351557 +0100kernel+EMUUSBAudioEngine[<private>]::initHardware (<private>)
standaard15:36:44.352022 +0100kernelDefault sample rate is 0
standaard15:36:44.352025 +0100kernelDefault alternate setting ID is 255
standaard15:36:44.352035 +0100kernel-EMUUSBAudioEngine[<private>]::start (<private>) = 0
standaard15:42:21.801573 +0100kernelSound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
[... cut ...]
standaard16:03:08.034246 +0100kernel++EMUUSBAudioEngine[<private>]: found device (<private>) for Audio Engine (<private>)
standaard16:03:08.034250 +0100kernel-EMUUSBAudioEngine::audioDevicePublished (<private>, 0x0, <private>)
standaard16:03:08.034266 +0100kernel+EMUUSBAudioEngine[<private>]::initHardware (<private>)
standaard16:03:08.034727 +0100kernelDefault sample rate is 0
standaard16:03:08.034731 +0100kernelDefault alternate setting ID is 255
standaard16:03:08.034741 +0100kernel-EMUUSBAudioEngine[<private>]::start (<private>) = 0
```

Would you please tell me what to do?

Wouter1, 2021-01-20:
EMU seems starting up OK. You broke the log output halfway

Default Sample rate is set to 0. Don;t know what's going on .

Please use control panel to set it properly.

Wouter1, 2021-01-22:
Mark

This might be a problem similar to a known issue on the EMU 0204 as reported on the main wiki page:

>There is a known issue with the 0204 (control panel not working and device switching to 4 channels out which does not work); For now please check issue 115 for workarounds.

Did you try that?

yaglo, 2021-01-24:
I'm having a similar issue. The card worked when I first installed the driver, but then I switched it off and back on, and now it's not visible in Audio & MIDI Setup or System Preferences. It's in the list of devices in System Information though (USB section).

What I've noticed in the log is this:

```
default	21:45:17.938692 +0000	kernel	Can't unload kext com.emu.driver.EMUUSBAudio; classes have instances:
default	21:45:17.938705 +0000	kernel	    Kext com.emu.driver.EMUUSBAudio class EMUUSBAudioDevice has 11 instances.
default	21:45:17.938708 +0000	kernel	    Kext com.emu.driver.EMUUSBAudio class EMUUSBAudioEngine has 11 instances.
```

Could it be the issue?

There's also this line when it starts

```
default	21:35:54.729426 +0000	kernel	Sound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
default	21:35:54.750405 +0000	kernel	Sound assertion "0 == thisUnitIDNum" failed in "/Users/wouter/Desktop/EMU-driver_subframe_sync/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
```

Wouter1, 2021-01-25:
@yagio 
Are you on mojave?
What mac is this 
Which EMU do you have ?
Is the EMU connected directly to the mac?

yaglo, 2021-01-25:
It's Hight Sierra 10.13.6, iMac 27" (Mid 2011), E-MU 0202, connected directly to the Mac.

Wouter1, 2021-01-28:
@yaglo  

The message "default	21:45:17.938692 +0000	kernel	Can't unload kext com.emu.driver.EMUUSBAudio; classes have instances:" seems to say

that your EMU is currently on and that the driver therfore can not be installed/replaced

Can you retry installing after turning off the EMU and rebooting

Miro265, 2022-07-06:
same issue here, installed e-mu 0202 on macOS version 10.14.6 and says NO E-MU Devices Found. 
could you please help?