Wouter1/EMU-driver #119: Device now visible in Yosemite

the-M2, 2019-02-15:
I've used this driver for a while and it worked fine on Yosemite. But there were cracks while recording. And recently there were cracks when listening to any sound source (browser, player, etc). I've decided to install the driver without low latency mode. And now the device is not visible in sound settings. I've tried installing with low latency and without, tried different USB ports, rebooted several times, but device won't show up. How can I fix it?

I'm on Mac OS 10.10.5 and the card is EMU Tracker Pre.

the-M2, 2019-02-15:
Well, I've surfed a little bit more for the answer, and the solution was to execute:
sudo nvram boot-args="kext-dev-mode=1"
and reboot

Wouter1, 2019-02-15:
@the-M2 sounds like you reset the PRAM or something. Yes if you get this 'not code signed' the developer extensions are not enabled.

Good that you got it sorted out!

closed 2019-02-15 by Wouter1