Wouter1/EMU-driver #120: Acceptance test T2 chip

Wouter1, 2019-08-29:
There appear to be lots of problems with recording audio over USB in  OSX due to the new T2 security chip

I want to do the acceptance tests https://github.com/Wouter1/EMU-driver/blob/master/Acceptance.md with this new chip now that my new macbook pro has such a chip.

Wouter1, 2019-08-29:
| Rate | Accepted |

| 48000 | ok |

Wouter1, 2019-08-29:
I read here 
https://www.extremetech.com/computing/285971-apples-latest-macs-have-a-serious-usb-audio-problem

a suggested relation with date&time panel. But I could not reproduce that. I got just some system sounds through my recording but no clicks or crackle.

Wouter1, 2019-08-29:
Interesting is that I can only reproduce some crackle so far at 88200. Nothing at any other rates.

Wouter1, 2019-08-29:
I checked a video here 
https://www.youtube.com/channel/UCNNkY4viZz43lB9r_Gcx8Ww
he has a buzz like sound every 10 seconds. Very different from what I have at 88200 Hz

Wouter1, 2019-08-30:
I re-routed the cable mess and checked the connections. Then I did another run of the tests

Wouter1, 2019-08-30:
Conclusion

There are issues with glitches in the recording. 
They seem to occur only in the 'odd' rates: 88200, 176400 and sometimes more often than others.

The noise in the previous test seems to be a cabling/amp setting issue and was solved in the 2nd  test.

Wouter1, 2019-08-30:
I tried to find messages in system.log.

However I ran the acceptance test at 88200Hz for 20 minutes and there was not a single glitch ?? Is the console.app that I need to run to see the system log somehow changing system behaviour?

Wouter1, 2019-09-02:
What I can still try is to disable "set Date and Time automatically" in system preferences as suggested here

https://noisegate.com.au/apples-latest-macs-workaround-for-audio-performance-bug/

Wouter1, 2019-09-18:
Testing in different configurations now. 

Nest test, Testing with power cord attached. EMU attached to 4K Dell 3214Q attached to jp adapter attached to USBC port of mac

88200

Wouter1, 2019-09-18:
As you can see in the zoomed in clicks, these clicks seem not natural. Particularly there is a ripple BEFORE the click. The clicks have identical shape, just reversed. The ripple has a frequency exactly half the sampling frequency. 

[image]

[image]

Wouter1, 2019-09-18:
The driver internally is not doing any filtering. So this appears to be some filter deep inside OSX.

Wouter1, 2019-09-19:
Next test, MacBook Pro (Retina, 15-inch, Late 2013) clamshell mode, 2.3 GHz Intel Core i7, 16 GB 1600 MHz DDR3, 10.12.6 (16G29)
Testing with power cord attached. EMU attached to Apple USB-C to USB, HDMI, USB-C adapter attached to USBC port of mac. Power adapter mouse and keyboard is attached to HP adapter that is  connected but to another USB-C port on the mac. Fresh install of OSX. No additional extensions have been installed, only the EMU ext. Set reaper to request 128byte buffer. Keyboard and mouse also connected to display. OSX 'special effect' sounds of finder are going to laptop speakers, not to EMU (this is different from preivous tests). I had to mess a bit with the finder because it kept forgetting my wallpaper.

Wouter1, 2019-09-19:
Next test, MacBook Pro (Retina, 15-inch, Late 2013) laptop mode, 2.3 GHz Intel Core i7, 16 GB 1600 MHz DDR3, 10.12.6 (16G29)
Testing with power cord attached. EMU attached to HP adapter  attached to USBC port of mac, with a cable without ferrite coil. 
Power adapter  connected to another USB-C port on the mac. 

Fresh install of OSX. No additional extensions have been installed, only the EMU ext. Set reaper to request 128byte buffer. Keyboard and mouse also connected to display. OSX 'special effect' sounds of finder are going to laptop speakers, not to EMU. 

| Rate | Accepted |
| ------------- | ------------ |
| 88200 | heavy clicking (>50) in first minute |

Wouter1, 2019-09-19:
Next test, MacBook Pro (Retina, 15-inch, Late 2013) laptop mode, 2.3 GHz Intel Core i7, 16 GB 1600 MHz DDR3, 10.12.6 (16G29)
Testing with power cord attached. EMU attached with a cable without ferrite coil to USB2.0 input 1 of  conrad PL-US3H3123 ,   attached to HP adapter, connected to USBC port of mac. 
Power adapter  connected to another USB-C port on the mac. 

Fresh install of OSX. No additional extensions have been installed, only the EMU ext. Set reaper to request 128byte buffer. Keyboard and mouse also connected to display. OSX 'special effect' sounds of finder are going to laptop speakers, not to EMU. 

| Rate | Accepted |
| ------------- | ------------ |
| 88200 | ok |
| 96000 | |

Wouter1, 2019-09-19:
internal speaker output rate changed to 88400Hz, otherwise same as previous

Does not make difference, 

| Rate | Accepted |
| ------------- | ------------ |
| 48000 | 5 ticks spread out |

Wouter1, 2019-09-19:
Doing a test on High sierra with same system

Next test, MacBook Pro (Retina, 15-inch, Late 2013) clamshell mode, 2.3 GHz Intel Core i7, 16 GB 1600 MHz DDR3,  10.13.6 (17G8030)
Testing with power cord attached. EMU attached with a cable without ferrite coil to HP adapter, connected to USBC port of mac.
Power adapter connected to another USB-C port on the mac. Nothing else attached to mac.

Wouter1, 2019-09-20:
In high sierra the ticks look different. Still synthetic but now symmetric
[image]

Wouter1, 2019-09-20:
[image]

Wouter1, 2019-09-20:
Something seems not to work with the reaper audio settings , if I request 192 kHz from reaper I still get 96kHz, even though the reaper panel shows 192. In the menu, reaper still shows 96

Wouter1, 2019-09-20:
It works perfect under sierra. 
Assuming high sierra did not do any major modifications in the usb or audio system, this must be something related to the USB-C system

Wouter1, 2019-09-21:
Read up a bit on USB3.
From https://en.wikipedia.org/wiki/USB_hub

> Note that USB 3.0 hubs do not currently[14][citation needed] perform transaction translation to super-speed for USB 2.0 devices. 

Also on the hardware layer there is separate wires for the USB2 data.

So it seems that USB3 is backwards compatible with USB2 by just having a a competely separate hardware layer for USB2. 

This makes the problem even weirder as this would suggest all the old USB2 driver software *could* still be inplace , there would just have to be a separate USB driver added to it?

I need to look at how OSX integrates the two (USB2 and USB3)

Wouter1, 2019-09-22:
I can try a few more things
* See if EMU driver logs something when the ticks occur. 
* See if something else logs issues, like the "Audio Overload" reported elsewhere.

Wouter1, 2019-09-23:
There are no EMU messages when the ticks happen. So the EMU is properly receiving and processing.

Wouter1, 2019-09-23:
Again had a run at 88.2kHz with large number of ticks in first 30secs. I don't see special messages in the console (no filters are on)

Wouter1, 2019-09-23:
There seem 2 possibilities remaining
1. The USB stream delivers bad bits occasionally without reporting. 
2. The USB stream works perfect but it gets screwed up later eg in kernel or in reaper

Wouter1, 2019-09-23:
Checking if the problem might be in Reaper.

Tested twice with quicktime / Audacity combo. No ticks.

Wouter1, 2019-09-23:
Tried to record and play back with audacity directly. It gives a bad effect where the 550Hz beep is truncated in blocks of 0.23s with a 0.005s silence between subsequent blocks.  Not sure what is going on

Wouter1, 2019-09-23:
Next test, MacBook Pro (Retina, 15-inch, Late 2013) clamshell mode, 2.3 GHz Intel Core i7, 16 GB 1600 MHz DDR3,  10.14.6 (18G95)
Testing in clamshell mode.4K display attached directly to one USBC port. network cable, external mouse and keyboard and. EMU attached with a cable without ferrite coil to the display.  Display and power cable connected to HP adapter, connected to USBC port of mac.

| Rate | Accepted |
| ------------- | ------------ |
| 96000 | About 30 clicks, most between 0 and 6 minutes, one at 9:30. Clicks are now very short, 2 high peaks and one small peak up, followed by a tiny ripple of maybe 4 samples before and after the sample.  |

Wouter1, 2019-09-28:
I just found out that if you turn OFF in security&privacy panel the access of audacity to the microphone, also the EMU usb audio gets blocked.

I thought that all this security was handled through the T2 chip,
That would be weird though, is USB audio routed through the T2 chip?

Wouter1, 2019-09-28:
In 2018 Ableton tech reported
https://cdm.link/2019/02/apple-2018-glitch/#comment-4344129785
>Unchecking the time sync only reduces the dropouts. You need to also uncheck the location sync in the Time Zone tab, but even then you're only reducing the dropouts. Any time the T2 chip tries to sync, it will overload the USB2.0 bus, causing dropouts. There is no way any musician should go on stage with a 2018 MBP or 2018 MacMini and a USB2.0 audio interface directly connected to it. The only solutions are 1) Get a thunderbolt or USB3.0 audio interface, or 2) Use a Certified Thunderbolt 3 powered hub (not a bus-powered or cheap non-certified hub) - Cable Matters has a solution for $169 on Amazon, but most of those hubs are at least $200-300. Using a certified TB3 hub creates a separate external USB2.0 hub in which you can run your USB2.0 audio interface. The cheap hubs do not work because they don't work on TB3, and therefore they don't create their on separate USB2 bus, which means you're going to get dropouts.

Wouter1, 2019-09-28:
The T2 firmware seems to be set up through the apple configurator : see this reset procedure for T2

https://help.apple.com/configurator/mac/2.7.1/#/apd0020c3dc2

The configurator seems a very broad tool
https://help.apple.com/configurator/mac/2.7.1/#/cadf1802aed

Wouter1, 2019-09-28:
There are almost no hardware diagrams available providing more insight in the hardware configuration of the USB ports. 

There is one here https://forums.macrumors.com/threads/did-someone-say-block-diagram.2096445/ that suggests that the Mac Pro (not macbook) has USB ports connected to CPU with only a USB-C /Thunderbold "mixer" (my name).

[image]
 

If you look at the USB info on the System report, it  it seems that BridgeOS is communicating through a USB-like connection called "iBridge bus"
The other devices like USB keyboard, mouse, and EMU are connected with a separate USB bus, not going through the BridgeOS.

Wouter1, 2019-09-28:
It might be possible to determine if the problem is in the upstream (computer to USB) or downstream (USB to computer) if we use a different signal generator. 
Then we can test if the INPUT works properly

Having a proper input is highest priority because the output is depending on proper input (not the other way round)

Wouter1, 2019-09-28:
Playback from Audacity is just terrible, soundling like turning on and off briefly 4 times per second.  This is with EMU set to 48k and audio project in audacity also 48k.

The log shows the EMU messages but you have to enter "EMU" as search term. Maybe there are too many other messages otherwise.

It shows a huge amount of these
{{{
	default	EMUUSBAudio	14:36:37.312775 +0200	kernel	frameSizeQueue empty, guessing some queue size. May need fix..
}}}

and some of
{{{
	default	EMUUSBAudio	14:36:37.399085 +0200	kernel	warning way-out ring wrap position

Wouter1, 2019-09-28:
I quit and restart audacity, and suddenly play works all fine. 
And then I restart again, and problem re-appears.

Wouter1, 2019-09-28:
Maybe it is related to logging the errors? Logging too many errors might cause the driver to slow down, losing more packets and causing the problem to worsen?

Wouter1, 2019-09-28:
It appears Apple has again changed their core level audio interface #122 
It's now not possible to change the error logging . 

Also that change might be directly related to the audio issues.

Wouter1, 2019-09-28:
Trying if XCode 7.3.1 (latest 7) can be installed on Mojave to get around this

Wouter1, 2019-09-28:
Download fails about half an hour into download (of 10+ hours, download rate is like 400kb/s only)

Wouter1, 2019-09-28:
XCode 7.3 does not start up

{{{
Termination Reason:    DYLD, [0x4] Symbol missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Symbol not found: _OBJC_IVAR_$_NSTextViewIvars.sharedData
  Referenced from: /Volumes/VOLUME/Xcode7.app/Contents/MacOS/../SharedFrameworks/DVTKit.framework/Versions/A/DVTKit
  Expected in: /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
 in /Volumes/VOLUME/Xcode7.app/Contents/MacOS/../SharedFrameworks/DVTKit.framework/Versions/A/DVTKit

}}}

Wouter1, 2019-09-29:
Another idea is to use the built-in audio output to generate the test tone. If the build-in works properly. this can generate an input signal for testing the EMU input separately

Wouter1, 2019-09-29:
Testing in this configuration
Setup:
clamshell mode.
MacBook Pro 2019 touchbar, 2.8GHz quad core i7.
macOS Mojave 10.14.6
13', 16GB 2133MHz LPDDR3 RAM,
Iris Plus Graphics 655
Dell 3214Q 4K display

Generic HP-Type-C to HDMI 4K Gigabit Ethernet RJ45 Port USB 3.0 Adapter Converter
EMU0404 ->  Dell 3214Q USB port->  HP adapter-> USB-C on mac
mac headphone out -> EMU0404 Line input 

Also attached during test:
USB mouse and keyboard -> 4K screen
Apple Power adapter. -> HP adapter
Ethernet -> HP adapter

and another USB-C from mac -> 4K screen to drive the display

| Rate | Accepted |
| ------------- | ------------ |
| 48000 | ok |

Wouter1, 2019-09-29:
Great, so the *recording* part of the driver seems to work fine. 

It seems that there is a problem with the playback buffering.

Wouter1, 2019-09-29:
Checing the "way-out ring wrap position" error in the code

It occurs inside the getCurrentSampleFrame and may explain the dropouts (the small silent periods. 
GetCurrentSampleFrame is used by the IOAudioEngine to determine how much of the mix buffer have been sent out and can be re-used (nulled in fact). If getCurrentSampleFrame returns a wrong number, part of the buffer will be nulled which is what we hear actually. THe error code indicates wer are returning a wrong number.

We use the system nanosecond clock to estimate the position from the last USBinput wrap. See UsbInputRing::estimatePositionAt(offset). 

EMUUSBAudioEngine.getCurrentSampleFrame() returns the estimate -4ms. By giving 4ms earlier point, the idea is that the engine keeps a safe 4ms behind the actual read position, avoiding premature overwrites if the USB would lag a bit.

Something here seems going wrong, we are getting excessive latency, more than one frame.

Wouter1, 2019-09-29:
Let's first get my old compu out of the moth balls so that I can edit the code... Then let's try to see if pos<0 or pos>1 in EMUUSBAudio line 878 where the warning occurs.

Wouter1, 2019-09-30:
Tried to reproduce. The clicking is easy, half of the time it has lots of ticks. But reproducing the error message not. It seems I get these errors now only when starting playback, about 10 times max,  which is normal.

Wouter1, 2019-09-30:
Tried to enable several loggers together: debugIOLogPC  debugIOLogAS  debugIOLogW  
But then the driver does not load, I get no error message on console and driver remains nonloaded. I also tried installing the new driver with original installer. Nothing. Not even message that driver is invalid or so. overview of extensions shows the driver, but Loaded:NO

Wouter1, 2019-09-30:
Huh??? I do this
```
sudo mv EMUUSBAudio.kext /System/Library/Extensions
```

and the result is the /System/Library/Extensions contains now only a folder Contents ???
Seems I screwed up the system

Wouter1, 2019-10-01:
The problem reproduces best with EMU and project both on 48kHz 
The console shows the messages, filter with "IOAudioFamily". Sometimes it does not show anything, you may need to restart console, sometimes even twice. I did not yet see any useful error messages,messing now with the debug output settings

Wouter1, 2019-10-01:
I think I turned on all debug messages. But console.app is showing only some turn on and off messages. Not a single timing, USB state etc message. 

Apparently you need to reboot before the audio driver really loads... New in mojave maybe. Anyway I now get messages,

Wouter1, 2019-10-01:
I hear  a good amount of clicking but see
* only 1 "output hiccup"
* no "way-out ring wrap position"
* only thing I notice is that clipOutputSamples is calling 50% of the time twice, 50% once. But this seems normal, I also see this at 44kHz where we have just a few clicks

Wouter1, 2019-10-02:
Next day... kext suddenly not loading anymore if I turn on the EMU. 

I don't even see it in the USB device tree. 

Trying different cable seems to help. Cable broken?

Wouter1, 2019-10-02:
Trying to use the old script but it needs to be modified because columns have changed in the console output. Anyway, there is a large distance now between the read and write actions to USB. Check this
```
EMUUSBAudio	clipOutputSamples firstSampleFrame=6388, numSampleFrames=1804, currentHead =3120	18:19:06.630879 +0200	kernel
EMUUSBAudio	clipOutputSamples firstSampleFrame=0, numSampleFrames=1268, currentHead =3124	18:19:06.630960 +0200	kernel
EMUUSBAudio	+ readCompleted framelist 0x2  result 0	18:19:06.661548 +0200	kernel
EMUUSBAudio	+GatherInputSamples 0	18:19:06.661556 +0200	kernel
EMUUSBAudio	+ read frameList 2	18:19:06.661614 +0200	kernel
EMUUSBAudio	READ framelist 2 in framenr 1169993 at 1170862973221	18:19:06.661642 +0200	kernel
EMUUSBAudio	- readCompleted currentFrameList=0x2	18:19:06.661644 +0200	kernel
EMUUSBAudio	WRITE framenr 1169929 at 1170862996786	18:19:06.661665 +0200	kernel
EMUUSBAudio	+ readCompleted framelist 0x3  result 0	18:19:06.693511 +0200	kernel
EMUUSBAudio	+GatherInputSamples 0	18:19:06.693520 +0200	kernel
EMUUSBAudio	+ read frameList 3	18:19:06.693578 +0200	kernel
EMUUSBAudio	READ framelist 3 in framenr 1170025 at 1170894938659	18:19:06.693608 +0200	kernel
EMUUSBAudio	- readCompleted currentFrameList=0x3	18:19:06.693610 +0200	kernel
EMUUSBAudio	WRITE framenr 1169961 at 1170894964499	18:19:06.693633 +0200	kernel
EMUUSBAudio	clipOutputSamples firstSampleFrame=1268, numSampleFrames=3072, currentHead =6191	18:19:06.694851 +0200	kernel
EMUUSBAudio	+ readCompleted framelist 0x0  result 0	18:19:06.725511 +0200	kernel
EMUUSBAudio	+GatherInputSamples 0	18:19:06.725520 +0200	kernel
EMUUSBAudio	+ read frameList 0	18:19:06.725577 +0200	kernel
EMUUSBAudio	READ framelist 0 in framenr 1170057 at 1170926939241	18:19:06.725609 +0200	kernel
EMUUSBAudio	- readCompleted currentFrameList=0x0	18:19:06.725612 +0200	kernel
EMUUSBAudio	WRITE framenr 1169993 at 1170926963180	18:19:06.725632 +0200	kernel
EMUUSBAudio	+ readCompleted framelist 0x1  result 0	18:19:06.757513 +0200	kernel
EMUUSBAudio	+GatherInputSamples 0	18:19:06.757521 +0200	kernel
EMUUSBAudio	filter 1170937787365 -> 1170937850014	18:19:06.757540 +0200	kernel
EMUUSBAudio	+ read frameList 1	18:19:06.757573 +0200	kernel
EMUUSBAudio	READ framelist 1 in framenr 1170089 at 1170958929443	18:19:06.757599 +0200	kernel
```

so 
```
READ framelist 2 in framenr 1169993 at 1170862973221	18:19:06.661642 +0200	kernel
WRITE framenr 1169993 at 1170926963180	18:19:06.725632 +0200	kernel
```

The time difference is .725632-.661642 = 0.06399s. Seems way too much.

Wouter1, 2019-10-02:
I checked the time differences between read and write, for instance we see in #70
906758436083 - 906694455291
= 63980792 ms. That is actualy very close to what we have now.

Wouter1, 2019-10-02:
The old simplistic script 
```
cat untitled.txt | awk '// { if ($2=="WRITE") { L=$4; B1=$6; } else if ($2=="READ" && L==$7) { B2=$9; print "wrap diff " B1-B2 } }'
```
does not work because we are now logging more write events and the distance between read and writes is too large.

Wouter1, 2019-10-02:
Here's a new script that works with the new output
```
cat untitled.txt | awk '// { if ($2=="READ") {  time[$7]=$9;  } else if ($2=="WRITE") { print "wrap diff " $6 - time[$4]  }  }'
```

It shows pretty constant values  around 64ms. Snap from it
```
wrap diff 64120693
wrap diff 64120823
wrap diff 64088895
wrap diff 63914440
wrap diff 63963349
wrap diff 64061287
wrap diff 64131025
wrap diff 64068299
wrap diff 63875420
wrap diff 63943246
wrap diff 64079866
wrap diff 64026878
wrap diff 63941144
wrap diff 64003724
wrap diff 64096842
wrap diff 63907339
wrap diff 64019925
wrap diff 64009067
```

Wouter1, 2019-10-02:
Still this large value seems suspicious. 64ms is about 2800 samples at 48k. I thought we had buffer of 4ms only.

Wouter1, 2019-10-04:
I finally figured out how to debug&compile the code on a late 2018 mac. You can still install High Sierra on it and there you can run XCode7 which builds the driver out of the box.

Wouter1, 2019-10-04:
Trying to find where we initialize the streams to see if they are sufficiently in sync at launch time.

Wouter1, 2019-10-04:
notifyWrap is reporting buffer wrap events to the EMUUSBAudioEngine. Its timestamp can be up to 1ms off because of the nature of the USB system. This is why we filter the timestamps. 

The idea is that the wrap time happens at different places within the millisecond all the time, so that we can estimate it.

The filter might have problems if the wraps fall every time at the same place in the millisecond. 

This will happen if the buffer accomodates an exact number of milliseconds of audio.

[188 more replies]

closed 2020-04-24 by Wouter1