Wouter1/EMU-driver #123: Catalina no sound

timmyleej, 2019-10-19:
Catalina, 10.15 (19A602). No sound on both 0204 and 0404. No playback sound and no recording as well

Wouter1, 2019-10-19:
@timmyleej  thanks for reporting. Sorry to hear it doesn't work.
You can see the device in your player and select it?
Are you trying to play music? 

I do not have Catalina and probably will not upgrade any time soon. So I can't reproduce/confirm this.

timmyleej, 2019-10-19:
@Wouter1  yeah, os x see both devices, I can change sample-rate and channel settings by midi-settings app (emu control panel is gray). But still no sound. HDMI and other sound devices work well. I've reverted back to Mojave. But if U can give some advices how to fix it I can test it on another disk drive.

Wouter1, 2019-10-19:
@timmyleej  Did you check if there are messages from the EMU device in the console (Applications/Utilities)?

Wouter1, 2019-10-19:
@timmyleej  one more thing, did you give microphone and speaker permission to the sound app you use? I don't know if is exists speaker  permission in catalina but there is mic permission needed for recording in mojave. And technically, mic needs to be working for playback

totencham, 2019-10-23:
So, it's working on Catalina?

Wouter1, 2019-10-23:
@totencham  I don't know. One totencham reported an issue on his machine but the problem was not reproduced. Do you have more input on this?

totencham, 2019-10-23:
No, i'm scared to update because i need working interface...

Wouter1, 2019-10-23:
@totencham  I can understand. There are about 15 visitors each day so I assume most downloads today are for Catalina and I got only 1 issue report for Catalina so far. But nobody confirmed proper working yet beyond high sierra.

I'm at this moment more concerned about the new laptops with USB-C interface and the required adapters. But that's also because I stick with Mojave for now.

timmyleej, 2019-10-23:
@Wouter1  Made a fresh install of catalina on separate hard drive, searched for a special permissions for audio devices, there is none. 
No sound, here is console log ,maybe it can help u
[image]
[image]
[image]
[image]

Wouter1, 2019-10-23:
@timmyleej 

Generally EMU message come from the *kernel* task so other threads can be ignored

First screen shot contains HAL message. It's from a system service connecting the EMU, not from the EMU driver.

The second one shows some possibly serious failures. 
I notice that there is something about "privacy acknowledgment" that might hint towards my suggestion to try

Third one, looks like some more system messages unrelated to EMU

Fourth one, one message from EMU that the engine is starting.

Wouter1, 2019-10-23:
Checked again, the second one has some weird messages, as if the output channel mOutput.init() has not been called. 

it is called in initHardware

start is called in startUSBStream

According to the manuals https://developer.apple.com/documentation/kernel/ioaudioengine/1812206-inithardware start should call initHardware.

bezromval, 2019-10-25:
catalina didn't support 32bit app. E-mu driver is 32 bit. This is the only reason that stops me to upgrade to catalina.

Wouter1, 2019-10-25:
@bezromval  interesting, I did not  realize this.
I'll look if there is a 64 bit setting or compiler option

bezromval, 2019-10-25:
Thanks. Waiting for good news from you.

Wouter1, 2019-10-25:
@bezromval why do you think it's 32 bit? I looked into the compiler settings but everything is already set to 64 bit.  The only reference to something 32 bit is

"Apple LLVM 7.1 Warnings : Implicit Conversion to 32 bit type"

And from the docu this is just a compiler warning setting, not anything influencing the build

Wouter1, 2019-10-25:
Additionally the file command also says it's 64 bit:

```
file v11/EMUUSBAudio.kext/Contents/MacOS/EMUUSBAudio 
v11/EMUUSBAudio.kext/Contents/MacOS/EMUUSBAudio: Mach-O 64-bit kext bundle x86_64

```

bezromval, 2019-10-25:
Sorry for that. I just look into about this mac system report software and find out that uninstaller only is 32 bit. I don't understand in this much. I just want to e-mu works fine with catalina.

Wouter1, 2019-10-25:
And the screenshot from timmyleej proves that it actually loaded as the EMU is reporting that it starts the engine.

Wouter1, 2019-10-25:
@bezromval  uninstaller? I only provide an installer. And it is just a shell script, for which 32/64 bit should be irrelevant.

Wouter1, 2019-10-25:
@timmyleej can you please report your hardware details, and how you have the EMU connected, and the type of the USB ports on your hardware? 

I still have not heard of anyone else having problems on Catalina so it may be just a problem by 1 user?

timmyleej, 2019-10-26:
@Wouter1  I have E-MU 0204 and 0404 both connected to my hackintosh installed on chinese motherboard, Xeon E5-1650, C602 Chipset. Connected in usb 2.0 ports, both worked well in Sierra and Mojave.

Wouter1, 2019-10-26:
@timmyleej  thanks for the details. Direct connection with USB2 ports seems fine.  But it might be an issue with Catalina?

Did you already check the security settings for audio?

timmyleej, 2019-10-26:
@Wouter1 i think its catalina related issues.
Yeah, I've checked settings. There's nothing input/output related.

Wouter1, 2019-10-26:
@timmyleej  can you be more specific. There *should* be something input/output related otherwise the security manager will block the audio (at least the input, maybe the output as well in cataline) . What recorder/player program do you use and what are the security settings for audio in/out for that program?

timmyleej, 2019-10-26:
@Wouter1 I'll check again this evening, I use stock music player (ex-itunes) and youtube in Safari for checking audio playback. And Logic Pro X for record/playback.

totencham, 2019-10-31:
@Wouter1 is a chance that it will work in Catalina?

Wouter1, 2019-10-31:
@totencham  sure, there is only one user so far that reported a problem in Catalina.

totencham, 2019-10-31:
> @totencham sure, there is only one user so far that reported a problem in Catalina.

But this is someone who works?

Wouter1, 2019-10-31:
@totencham  who is "this"?

totencham, 2019-10-31:
Has anyone confirmed that it works in Catalina?

Wouter1, 2019-10-31:
No reports at all yet, #109.

grimrian, 2019-11-07:
Hello, I upgraded to Catalina without realising there might have been an issue. Fortunately everything works, E-MU interface, Midi, all sample rates. Input, output OK. Turns up everywhere as an audio inout or output option. Version is 4.0.1 originally installed in April. Imac 2017. If anyone needs info from system, let me know.

jakubzwierzycki, 2019-11-09:
After updating to catalina everything works like a charm,
Ableton usage, emu0404 usb midi config and everything else. On previous osx updates i had to reinstall this driver to restore possibility to choose emu0404 usb midi out / in in ableton preferences. After update to catalina there is no issue like that, so nice work:)

Wouter1, 2019-11-09:
@timmyleej any news here? 
@totencham  two users reported it all works fine in catalina.  The issue with timmyleej might also be hardware related (he has a hackintosh)

totencham, 2019-11-25:
Hi! I updated to Catalina on MacBook Pro and i have problem with driver's installation. 
In step. 2 "In a terminal do sudo mount -uw /" returns: 
[image]

When i start installer script: 
[image]
 and nothing changes...

Anyone can help me?

Wouter1, 2019-11-25:
@totencham  did you see and follow the additional installation instructions to massage Catalina 

https://github.com/Wouter1/EMU-driver/blob/master/Install.md

totencham, 2019-11-25:
> @totencham did you see and follow the additional installation instructions to massage Catalina
> 
> https://github.com/Wouter1/EMU-driver/blob/master/Install.md

How was I supposed to not see if I have a problem with this? I did step.1 but on step.2 i was stopped :)

Wouter1, 2019-11-25:
Thanks for posting the pics.
I see they also have "filesystem protection" now.

Can you try with ```csrutil disable``` insttead of ```csrutil enable --without kext```

totencham, 2019-11-25:
> Thanks for posting the pics.
> I see they also have "filesystem protection" now.
> 
> Can you try with `csrutil disable` insttead of `csrutil enable --without kext`

You're great! Now, it's working fine! Thank you!

Wouter1, 2019-11-26:
@totencham  thanks for reporting back! I'll fix the instructions page accordingly.

Wouter1, 2019-11-26:
Updated the installation docu

closed 2019-11-26 by Wouter1

djjuank, 2020-03-08:
Hi, where can I download the driver for the emu 0404  to works with Mac OS Catalina?

Thanks

grimrian, 2020-03-08:
It does actually work with Catalina on my iMac 2017. 
Regards, Tomas

Sent from mobile phone

Wouter1, 2020-03-09:
@djjuank  FAIK this driver also works on Catalina. I updated the info page to include Catalina.