Wouter1/EMU-driver #124: Catalina: Install not working - csrutil issue

dyemyhair, 2019-11-10:
It seems like "csrutil enable --without kext" is not enough in Catalina to install the driver. Disable SIP completely doesn't work either.

Installer is not able to write in /Extensions:

> rm: /System/Library/Extensions/EMUUSBAudio.kext: No such file or directory
> cp: /System/Library/Extensions/EMUUSBAudio.kext: Read-only file system
> cp: EMUUSBAudio.kext: unable to copy extended attributes to /System/Library/Extensions/EMUUSBAudio.kext: Read-only file system
> cp: /System/Library/Extensions/EMUUSBAudio.kext/Contents: No such file or directory
> cp: EMUUSBAudio.kext/Contents: unable to copy extended attributes to /System/Library/Extensions/EMUUSBAudio.kext/Contents: No such file or directory
> cp: /System/Library/Extensions/EMUUSBAudio.kext/Contents/MacOS: No such file or directory
> cp: EMUUSBAudio.kext/Contents/MacOS: unable to copy extended attributes to /System/Library/Extensions/EMUUSBAudio.kext/Contents/MacOS: No such file or directory
> cp: /System/Library/Extensions/EMUUSBAudio.kext/Contents/MacOS/EMUUSBAudio: No such file or directory
> cp: /System/Library/Extensions/EMUUSBAudio.kext/Contents/Resources: No such file or directory
> cp: EMUUSBAudio.kext/Contents/Resources: unable to copy extended attributes to /System/Library/Extensions/EMUUSBAudio.kext/Contents/Resources: No such file or directory
> cp: /System/Library/Extensions/EMUUSBAudio.kext/Contents/Resources/en.lproj: No such file or directory
> cp: EMUUSBAudio.kext/Contents/Resources/en.lproj: unable to copy extended attributes to /System/Library/Extensions/EMUUSBAudio.kext/Contents/Resources/en.lproj: No such file or directory
> cp: /System/Library/Extensions/EMUUSBAudio.kext/Contents/Resources/en.lproj/InfoPlist.strings: No such file or directory
> cp: /System/Library/Extensions/EMUUSBAudio.kext/Contents/Info.plist: No such file or directory
> chgrp: EMUUSBAudio.kext: No such file or directory
> /System/Library/Extensions/EMUUSBAudio.kext failed to load - (libkern/kext) not found; check the system/kernel logs for errors or try kextutil(8).

Wouter1, 2019-11-11:
@dyemyhair It seems you do not have permissions to write in the /System/Library directory. 

I think you need to manage some permissions but can't help you with the details

Here is some more info

https://www.howtogeek.com/443611/how-macos-catalinas-new-security-features-work/

dyemyhair, 2019-11-16:
That website's information aren't contributing to any progress but I'm working on it.

Wouter1, 2019-11-16:
Does your account have admin rights ? You need to install it from an account that has permissions to write there (after entering your pass of course). 

http://osxdaily.com/2016/08/03/turn-standard-to-administrator-account-mac-os/

dyemyhair, 2019-11-16:
> Does your account have admin rights ? You need to install it from an account that has permissions to write there (after entering your pass of course).
> 
> http://osxdaily.com/2016/08/03/turn-standard-to-administrator-account-mac-os/

Catalina runs on a separate read-only volume:
"With macOS Catalina, you can no longer store files or data in the read-only system volume, nor can you write to the "root" directory ( / ) from the command line, such as with Terminal."
Source: https://support.apple.com/en-us/HT210650

So, it seems like being admin is not enough to gain writing permission on this volume. I can't figure out what to do yet but I believe this issue is connected to this change.

Wouter1, 2019-11-16:
It already was like that, you can write there only after disabling SIP. 
so it seems like SIP still enabled, or something changed. 

From what I read there I will not be able to even try Catalina as they suggest there can be only one Catalina installe, on a partition that Apple determines.

Wouter1, 2019-11-16:
@dyemyhair  I have  question, I understand fromn that page that Catalina is on a separate partition

Can you do ```ls /``` ? Is there a System folder still there?

dyemyhair, 2019-11-16:
Finally! Yes! I got it! Thank you for your support!
I'm not sure all of the following is necessary but that's what I've done:

- disable SIP completely in secure mode: "csrutil disable"
- under "System Preferences > Security and Privacy > Privacy" grant full disk access to terminal and the driver installer
- perform in terminal:
"sudo mount -uw /"
"sudo killall Finder"
- run installer

Wouter1, 2019-11-17:
@dyemyhair  thanks for the update. Great to hear you got it working.

Looks like I need to write some additional explanations.

Do you really need the full ```csrutil disable``` or would only disabling kext be enough?
Why do you need to kill the Finder?

dyemyhair, 2019-11-17:
That's the point, I don't know what steps are crucial.

I _believe_ `csrutil disable --without kext `, `sudo mount -uw /` and `sudo killall Finder` should be enough.

Killing the Finder makes it restart to load the changes.

I might deinstall and reinstall tomorrow to verify this if needed?

Wouter1, 2019-11-17:
@dyemyhair 
>Killing the Finder makes it restart to load the changes.
But you don't need the finder for the install?

dyemyhair, 2019-11-17:
Yes, but it relaunches instantly.

Wouter1, 2019-11-18:
I Added the extra install instructions on the installation page.

For now I did not add the csrutil disable and the killall Finder as these seem unnecessary and make the machine needlessly less secure

closed 2019-11-18 by Wouter1

dyemyhair, 2019-11-19:
> I Added the extra install instructions on the installation page.
> 
> For now I did not add the csrutil disable and the killall Finder as these seem unnecessary and make the machine needlessly less secure

I think that's a good decision for now. Thank you for your time:)

Wouter1, 2019-11-19:
@dyemyhair  And thanks for reporting this issue and detailing on the fix :-)