Wouter1/EMU-driver #127: not installed on Catalina

MaximZar, 2020-01-02:
I do everything you wrote. But in the end, I always hang up the installation process....... Screenshot attach. Help me, please =(( I try to solve but I cannot(
[image]

MaximZar, 2020-01-02:
To clarify: I tried to enter "n", "nn", "ny". It doesn't change anything. All permissions gave, csrutil disable did.

Wouter1, 2020-01-03:
@MaximZar  Seems it's installing properly but waiting for the kernel to finish. How long did you wait?

MaximZar, 2020-01-03:
I waited about 30 minutes

MaximZar, 2020-01-03:
And nothing =(

MaximZar, 2020-01-03:
I started the installation again an hour ago. For an hour absolutely nothing has changed

MaximZar, 2020-01-03:
5 hours ago =D I think that Installation is frozen

Wouter1, 2020-01-03:
Yes, it can take several minutes but more than 15 would be excessive.

Has the extension been installed? Do you see it after a reboot in the list of available extensions (About this Mac/System Report/Extensions)? Does it work if you try to play back ?

MaximZar, 2020-01-03:
While I was waiting for a response, I could not restrain themselves and started logic. And lo and behold!!! it works! Apparently the plugin was installed but the console does not notify! If you have had the same problem as me: 
1. Just give an indefinite amount of time for installation. Wait for example 1 hour. 
2. Quietly go into logic or your daw and play!

MaximZar, 2020-01-03:
Thanks for trying to help me!

MaximZar, 2020-01-03:
and thank you for still helping a lot of people with our devices.

Wouter1, 2020-01-03:
Good to hear that it works anyway :)

Maybe you can help a bit figuring out why you don't get the OK  from the installer.
Could you report what the log files are reporting?

I suggest that you do the following
1. open a terminal and run ```log stream```
2. run the installer, start the installation and wait about 5 minutes (I suspect that this is enough to complete the install)
3. Copy the output from the log stream and post it for me to check

MaximZar, 2020-01-03:
Ok! Now I will do it

MaximZar, 2020-01-03:
I ran the installer, wait about 7 minutes and I got a  next log: [Uploading terminalForWouter.txt…]()

Wouter1, 2020-01-03:
@MaximZar  thanks. 

I searched the log file but I don't see the expected messages from the kernel that it starts rebuilding the kernel cache. Maybe they changed something in OSX and that it's either not needed anymore or that it needs to be forced. I need to do some research on that 

Can you try to execute the following command in another terminal window, right after you did a normal install. Just leave the install window open and see if you get any progress there after executing this command:

```sudo kextcache -u```

MaximZar, 2020-01-03:
Ok. I will do it

MaximZar, 2020-01-03:
My results: (I walked away for a long time, I hope this did not affect)
[image]

Wouter1, 2020-01-04:
@MaximZar  thanks! 

it did not work as they suggested on the forum where I found that...
I clearly should have tried it myself first, sorry for that. I'll have to figure out first how that works

Wouter1, 2020-01-04:
@MaximZar the command probably should have been ```sudo kextcache -i /``` so I missed the /

But maybe it does not work because the kext is already installed. 

So can you first remove the extension like this (double check your sudo rm command, removing the wrong stuff can kill your system)
```
sudo rm -rf /System/Library/Extensions/EMUUSBAudio.kext/
sudo kextcache -i /
```

and then wait till you get 'lock acquired; proceeding'

and then re-run the normal installer?

MaximZar, 2020-01-04:
Oh. I can no do it over the next 3 day, because im not at home. But when I'll come back I will do it

Wouter1, 2020-01-04:
@MaximZar no problem, we continue here when you have time for it.

MaximZar, 2020-01-07:
Hi
I did it 
And I got this
[image]
[image]

:

Wouter1, 2020-01-07:
@MaximZar  thanks
Did you first remove the EMU extension?  It may just dropped out of view on your screenshots.

Otherwise I have no clue at this point on why the system is not reporting the progress on rebuilding the kext cache...

Wouter1, 2020-01-07:
@MaximZar  on which version of MacOS is this?

MaximZar, 2020-01-07:
Yes. I deleteв. I did everything what you said. I use macOS Catalina

Wouter1, 2020-01-07:
On my machine, these messages are coming from the IOKit library.

Wouter1, 2020-01-07:
These are the messages from IOKit in the text file from Maxim
```
2020-01-04 01:06:13.135400+0400 0x274      Default     0x0                  96     0    kextd: (IOKit) [com.apple.kext:kextlog] Cache file /System/Library/Caches/com.apple.kext.caches/Directories/System/Library/Extensions/KextIdentifiers.plist.gz is out of date; not using.
2020-01-04 01:06:13.585049+0400 0x274      Error       0x0                  96     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67062) allowed: <OSKext 0x600002413540 [0x7fff89ea0090]> { URL = "file:///System/Library/Extensions/EMUUSBAudio.kext/", ID = "com.emu.driver.EMUUSBAudio" }
2020-01-04 01:07:38.150478+0400 0x37ab9    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection added: IOHIDEventSystemConnection uuid:F4DD5D98-389E-43F3-9D39-489034357B6E pid:5640 process:Terminal type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:07:38.151924+0400 0x37bb4    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection removed: IOHIDEventSystemConnection uuid:F4DD5D98-389E-43F3-9D39-489034357B6E pid:5640 process:Terminal type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:10:24.864301+0400 0x37f20    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection added: IOHIDEventSystemConnection uuid:8B31CE81-4DD1-42F2-BA5B-F0FE75CA623C pid:5640 process:Terminal type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:10:24.865924+0400 0x37f72    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection removed: IOHIDEventSystemConnection uuid:8B31CE81-4DD1-42F2-BA5B-F0FE75CA623C pid:5640 process:Terminal type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:11:14.352982+0400 0x38046    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection added: IOHIDEventSystemConnection uuid:E670C0F2-3F48-419C-B329-5EE89D4E1E3B pid:5390 process:QuickTime Player type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:11:14.354722+0400 0x38049    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection removed: IOHIDEventSystemConnection uuid:E670C0F2-3F48-419C-B329-5EE89D4E1E3B pid:5390 process:QuickTime Player type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:11:15.152335+0400 0x37f76    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection added: IOHIDEventSystemConnection uuid:5452E354-4566-4269-A5E3-345FB6CA0777 pid:423 process:Finder type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:11:15.153989+0400 0x38049    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection removed: IOHIDEventSystemConnection uuid:5452E354-4566-4269-A5E3-345FB6CA0777 pid:423 process:Finder type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:11:15.873614+0400 0x38046    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection added: IOHIDEventSystemConnection uuid:FBD13E7A-ACC7-4892-8FE9-F48D0EDBBCDD pid:1361 process:Electron type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:11:15.875297+0400 0x38049    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection removed: IOHIDEventSystemConnection uuid:FBD13E7A-ACC7-4892-8FE9-F48D0EDBBCDD pid:1361 process:Electron type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:12:17.787007+0400 0x38046    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection added: IOHIDEventSystemConnection uuid:EF0A8C3D-F30F-4492-8AFC-DDB35D0E7858 pid:5640 process:Terminal type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:12:17.788763+0400 0x3818c    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection removed: IOHIDEventSystemConnection uuid:EF0A8C3D-F30F-4492-8AFC-DDB35D0E7858 pid:5640 process:Terminal type:Passive entitlements:0x0 caller:HIToolbox: _TISCopyAttachedKeyboardLanguages + 82 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:14:06.020346+0400 0x38288    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection added: IOHIDEventSystemConnection uuid:1A8FF719-9731-4FA1-8895-378E2A03A4B8 pid:5733 process:com.apple.appkit.xpc.openAndSav type:Passive entitlements:0x0 caller:HIToolbox: ___GetIOHIDEventSystemClient_block_invoke + 26 attributes:(null) inactive:1 events:0 mask:0x0
2020-01-04 01:14:06.032085+0400 0x38288    Default     0x0                  146    0    hidd: (IOKit) [com.apple.iohid:default] Connection added: IOHIDEventSystemConnection uuid:3F32D3DE-AE06-4C76-B865-83FA64016330 pid:5733 process:com.apple.appkit.xpc.openAndSav type:Passive entitlements:0x0 caller:HIServices: ___GetIOHIDEventSystemClient_block_invoke + 26 attributes:(null) inactive:1 events:0 mask:0x0
```

Wouter1, 2020-01-07:
@MaximZar 

maybe I can use a different way to keep track of the progress.
Can you post the output of this command on your machine

```sudo kextcache -v 1 -i /```

MaximZar, 2020-01-07:
I did it
[image]

Wouter1, 2020-01-10:
On Mojave the command gives this
```
rebuilding //System/Library/PrelinkedKernels/prelinkedkernel
KernelCache ID: 47E4835D7225DD36413335059107D482
Created prelinked kernel "/System/Library/PrelinkedKernels/prelinkedkernel"
Created prelinked kernel using "/System/Library/Kernels/kernel"
Symlink "/System/Library/Caches/com.apple.kext.caches/Startup/kernelcache" -> "../../../PrelinkedKernels/prelinkedkernel"
/: no supported helper partitions to update.
Kext with invalid signatured (-67062) allowed: <OSKext 0x7fb7c36e93c0 [0x7fff8ef928e0]> { URL = "file:///System/Library/Extensions/EMUUSBAudio.kext/", ID = "com.emu.driver.EMUUSBAudio" }
Directory caches updated for /System/Library/Extensions.
Directory caches updated for /Library/Extensions.

```

Wouter1, 2020-01-10:
On High Sierra it gives
```
rebuilding //System/Library/PrelinkedKernels/prelinkedkernel
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/MBBActivateDriver.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7f91359fc020 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/MBBActivateDriver.kext/", ID = "com.MBB.driver.MBBActivateDriver" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiLogLevel.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7f91359d3160 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiLogLevel.kext/", ID = "com.huawei.driver.HuaweiLogLevle" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardECMControl.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7f91359a1f00 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardECMControl.kext/", ID = "com.huawei.driver.HuaweiDataCardECMControl" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7f91358def30 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/", ID = "com.huawei.driver.HuaweiDataCardDriver" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/MBBActivateDriver.kext does not appear in strict exception list for architecture: x86_64
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiLogLevel.kext does not appear in strict exception list for architecture: x86_64
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardECMControl.kext does not appear in strict exception list for architecture: x86_64
/Library/Extensions/HuaweiDataCardDriver_10_9.kext does not appear in strict exception list for architecture: x86_64
KernelCache ID: 25379BE02B7ED921AC8C3BA0DC530926
Created prelinked kernel "/System/Library/PrelinkedKernels/prelinkedkernel"
Created prelinked kernel using "/System/Library/Kernels/kernel"
Symlink "/System/Library/Caches/com.apple.kext.caches/Startup/kernelcache" -> "../../../PrelinkedKernels/prelinkedkernel"
/: no supported helper partitions to update.
Kext with invalid signatured (-67062) allowed: <OSKext 0x7fa1b305dc70 [0x7fff8583ab40]> { URL = "file:///System/Library/Extensions/EMUUSBAudio.kext/", ID = "com.emu.driver.EMUUSBAudio" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7fa1b32ce4f0 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/", ID = "com.huawei.driver.HuaweiDataCardDriver" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardACMControl.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7fa1b32e0d40 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardACMControl.kext/", ID = "com.huawei.driver.HuaweiDataCardACMControl" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardACMData.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7fa1b32e18e0 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardACMData.kext/", ID = "com.huawei.driver.HuaweiDataCardACMData" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardDMM.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7fa1b32f8df0 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardDMM.kext/", ID = "com.huawei.driver.HuaweiDataCardDMM" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardECMControl.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7fa1b3332f50 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardECMControl.kext/", ID = "com.huawei.driver.HuaweiDataCardECMControl" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardECMData.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7fa1b33793a0 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiDataCardECMData.kext/", ID = "com.huawei.driver.HuaweiDataCardECMData" }
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiLogLevel.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7fa1b33a80a0 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/HuaweiLogLevel.kext/", ID = " [... cut ...]
/Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/MBBEthernetData.kext does not appear in strict exception list for architecture: x86_64
Kext with invalid signatured (-67007) allowed: <OSKext 0x7fa1b33d96d0 [0x7fff8583ab40]> { URL = "file:///Library/Extensions/HuaweiDataCardDriver_10_9.kext/Contents/PlugIns/MBBEthernetData.kext/", ID = "com.MBB.driver.MBBEthernetData" }
Directory caches updated for /System/Library/Extensions.
Directory caches updated for /Library/Extensions.

```

Wouter1, 2020-01-10:
We can trigger on these keywords and it should work from High Sierra and higher
```
rebuilding
KernelCache ID
Directory caches updated
```

Wouter1, 2020-01-10:
@MaximZar  a new release 4.0.2 is now available that should fix your issue.

Can you please test it and confirm if it works?

MaximZar, 2020-01-14:
I did it and had a beauty result! It's solve my problem. Thanks!

Wouter1, 2020-01-14:
@MaximZar  thanks for the confirmation!

closed 2020-01-14 by Wouter1

TheGreatKat, 2020-03-30:
Hello! I still have a problem to do it on my mac with Catalina. Actually what should be the final result? Should it be just detected in the audio settings or also the app should be in installed in applications folder? Anyway it does not work in any way. I csrutil disable in terminal, rebooted, then I allowed full disk access to terminal (but not to drive installer as I dont know which installer does it mean? so maybe thats a gap in my process). And then sudo mount -uw / does not work at all. I tried to install drivers. The process looks fine, its writen its done and ready to use but it doesnt. Any advices, ideas? Thanks in advance!

Wouter1, 2020-03-30:
@TheGreatKat  Installer means the "installer" file in the root of the project. Downloaded as you download the entire project zip file - either using the "download" button on the main wiki, or the download buttons in the release page.. 

Final result after running the installer  should be

* system is enabled to run developer kext's
* EMU kext is installed in /System/Library/Extensions

As a result, the EMU kext should start up when the system detects it. The system than makes the EMU appear in all audio applications (eg the system Audio panel, the Midi panel, Audacity, Reaper) 

There is nothing installed in applications folder.

What do you mean by "does not work at all". You get an error message? Please show what you do and if there are error messages or so.