Wouter1/EMU-driver #128: S/PDIF output intermittently disabling

iambryanhaney, 2020-05-21:
Catalina 10.15.4
Using 48khz for general OS output (Spotify, Chrome, etc).

If there are no active audio streams for a few minutes, the S/PDIF output will cease to function when playback is resumed. I verified that the device is receiving the stream - the headphone output is functioning.

Toggling the sample rate back and forth eventually re-activates the S/PDIF output, but not dependably; it generally takes 1-10 toggles.

iambryanhaney, 2020-05-21:
Copy/paste of fault generated from toggling sample rate (from 48khz to 88.2khz).

Process:               E-MU USB Audio Control Panel [27665]
Path:                  /Users/USER/Downloads/*/E-MU USB Audio Control Panel.app/Contents/MacOS/E-MU USB Audio Control Panel
Identifier:            com.creative-professional.E-MU USB Audio Control Panel
Version:               1.50 (1.50.07)
Code Type:             X86-64 (Native)
Parent Process:        ??? [27190]
Responsible:           Terminal [1166]
User ID:               501

Date/Time:             2020-05-20 19:53:23.713 -0500
OS Version:            Mac OS X 10.15.4 (19E287)
Report Version:        12
Bridge OS Version:     4.4 (17P4281)
Anonymous UUID:        7B389D34-CDE4-9B1D-BDD8-0664756FC7DC

Sleep/Wake UUID:       7ECB50B2-A120-4E8D-B368-BD376BCF6E97

Time Awake Since Boot: 110000 seconds
Time Since Wake:       14000 seconds

System Integrity Protection: disabled

Crashed Thread:        6  Dispatch queue: com.apple.audio.CADispatchQueue.SerialQueue

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000018
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [27665]

VM Regions Near 0x18:
--> 
    __TEXT                 0000000100000000-000000010001b000 [  108K] r-x/rwx SM=COW  /Users/USER/Downloads/*/E-MU USB Audio Control Panel.app/Contents/MacOS/E-MU USB Audio Control Panel

Application Specific Information:
objc_msgSend() selector name: invalidate

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff6f252dfa mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff6f253170 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff353640b5 __CFRunLoopServiceMachPort + 247
3   com.apple.CoreFoundation      	0x00007fff35362b82 __CFRunLoopRun + 1319
4   com.apple.CoreFoundation      	0x00007fff35361ffe CFRunLoopRunSpecific + 462
5   com.apple.HIToolbox           	0x00007fff33f95abd RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff33f957d5 ReceiveNextEventCommon + 584
7   com.apple.HIToolbox           	0x00007fff33f95579 _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff325e0c99 _DPSNextEvent + 883
9   com.apple.AppKit              	0x00007fff325df4e0 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
10  com.apple.AppKit              	0x00007fff325d11ee -[NSApplication run] + 658
11  com.apple.AppKit              	0x00007fff325a2ff6 NSApplicationMain + 777
12  com.creative-professional.E-MU USB Audio Control Panel	0x00000001000013c4 start + 52

Thread 1:: AMCP Logging Spool
0   libsystem_kernel.dylib        	0x00007fff6f252e36 semaphore_wait_trap + 10
1   com.apple.audio.caulk         	0x00007fff68cebb16 caulk::mach::semaphore::wait() + 16
2   com.apple.audio.caulk         	0x00007fff68ceb9b2 caulk::semaphore::timed_wait(double) + 106
3   com.apple.audio.caulk         	0x00007fff68ceb7c4 caulk::concurrent::details::worker_thread::run() + 30
4   com.apple.audio.caulk         	0x00007fff68ceb1e4 void* caulk::thread_proxy<std::__1::tuple<caulk::thread::attributes, void (caulk::concurrent::details::worker_thread::*)(), std::__1::tuple<caulk::concurrent::details::worker_thread*> > >(void*) + 45
5   libsystem_pthread.dylib       	0x00007fff6f316109 _pthread_start + 148
6   libsystem_pthread.dylib       	0x00007fff6f311b8b thread_start + 15

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff6f252e36 semaphore_wait_trap + 10
1   com.apple.audio.caulk         	0x00007fff68cebb16 caulk::mach::semaphore::wait() + 16
2   com.apple.audio.caulk         	0x00007fff68ceb9b2 caulk::semaphore::timed_wait(double) + 106
3   com.apple.audio.caulk         	0x00007fff68ceb7c4 caulk::concurrent::details::worker_thread::run() + 30
[... 451 lines cut ...]
Graphics: kHW_IntelUHDGraphics630Item, Intel UHD Graphics 630, spdisplays_builtin
Graphics: kHW_AMDRadeonPro560XItem, Radeon Pro 560X, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2400 MHz, Micron, 8ATF1G64HZ-2G6E1
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, DDR4, 2400 MHz, Micron, 8ATF1G64HZ-2G6E1
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Feb 28 2020 15:24:56 version 9.30.357.35.32.5.47 FWID 01-9ce4adf3
Bluetooth: Version 7.0.4f6, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: BillBoard Device
USB Device: USB2.0 Hub
USB Device: USB Receiver
USB Device: E-MU 0404 | USB
USB Device: Hub
USB Device: Texas Instruments USB Billboard Device
USB Device: USB3.0 Hub
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.4
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.4

Wouter1, 2020-05-21:
The crashing control panel is  issue #81 . I can not do anything about that as Creative Labs never released the source code of the control panel.

Wouter1, 2020-05-21:
@iambryanhaney  we could move that stacktrace to the proper ticket #81  but it looks mostly identical to the stacktrace we already have there. Shall we  remove it here?

Wouter1, 2020-05-21:
@iambryanhaney 

>If there are no active audio streams for a few minutes, the S/PDIF output will cease to function when playback is resumed. I verified that the 0404 is receiving the stream - the headphone output continues to function.

What I suspect is that the stream is really stopped, and restarted when playback resumes. 

And maybe there is a bug in the driver that restarts the normal stream but not the spdif

Can you see any evidence of this theory in the log  ? And, do you see any errors in the logs?

iambryanhaney, 2020-05-22:
Removed the stacktrace and removed mention of issue #81 type crash.

iambryanhaney, 2020-05-22:
> What I suspect is that the stream is really stopped, and restarted when playback resumes.

>And maybe there is a bug in the driver that restarts the normal stream but not the spdif

>Can you see any evidence of this theory in the log ? And, do you see any errors in the logs?

My understanding of under-the-hood particulars is limited.  Are there other logs that I might inspect that would be of assistance?

An example scenario would be playing a YouTube video.

S/PDIF outputs normally -> Youtube video ends -> Click on another video -> S/PDIF is no longer outputting (but headphones are) -> Toggle sample rate repeatedly while video is playing -> S/PDIF resumes outputting.

As a stopgap to this behavior, I have been keeping Spotify running at an imperceptible volume in the background. As long as the OS has at least one active audio stream routing to the 0404, it appears to prevent S/PDIF failure.

Wouter1, 2020-05-22:
@iambryanhaney  ok thanks for removing the stacktrace.

Yes apparently something does not restart properly. 

If you open Applications/Utilities/Console you get to see the log files. I would like to hear if there are messages there that might indicate a problem with restarting the s/pdif

Wouter1, 2020-05-22:
searched for pdif in the sources. 

Binary file ./ControlPanel/E-MU USB Audio Control Panel.app/Contents/MacOS/E-MU USB Audio Control Panel matches
./src/EMUUSBAudio/EMUUSBAudioDevice.cpp:		if (mRealClockSelector->valueExists(kClockSourceSpdifExternal)) {
./src/EMUUSBAudio/EMUUSBAudioDevice.h:	EMUXUCustomControl*		mDigitalIOSPDIF;	// spdif settings
./src/EMUUSBAudio/EMUUSBDeviceDefines.h:	kClockSourceSpdifExternal
./src/EMUUSBAudio/EMUUSBDeviceDefines.h:	spdif,
./src/EMUUSBAudio/EMUUSBDeviceDefines.h:	kDigitalFormat			= 0x05,	// the spdif format

Wouter1, 2020-05-22:
We can't fix any control panel stuff so let' s hope this problem does not originate there.

The only reference in our code thus is the EMUUSBAudioDevice.cpp code.

But this code is disabled.

Wouter1, 2020-05-22:
Checked 
https://github.com/Wouter1/EMU-driver/blob/e573375c486c64b577f28776adb19ee317510b53/src/EMUUSBAudio/EMUUSBAudioDevice.cpp

This code has been disabled from the start. 

But that is only logging. What I suppose matters is ```mRealClockSelector->setProperty(kIOAudioSelectorControlClockSourceKey,"0");```.

Wouter1, 2020-05-22:
That is about the CLOCK SOURCE. The issue seems about the s/pdif OUTPUT, not the clock source.

Wouter1, 2020-05-22:
@iambryanhaney  a few checks
* What does control panel show when it works, and when it does not work. Please report all settings, also device, sample rate etc. 
* What do the sync status indicators show when it works and does not work
* you had this exact same setup working in Windows
* What happens if you toggle the direct monitor or s/pdif buttons on the EMU itself?

Wouter1, 2020-05-22:
The S/Pdif output requires channel 3 and 4 to be enabled. 
From the diagram I made long ago it looks that the computer has to send these channels separately. They would then be a copy of channels 1 and 2 ?

Wouter1, 2020-05-22:
The normal control panel suggests that SPDIF output is always on becaues you can not select 'off', only 'SPDIF" or "AES"

Wouter1, 2020-05-22:
@iambryanhaney  I tried to use the spdif on some old audio stuff but I get no signal at all. Also it seems I would have to transmit data to channel 3 and 4 of the EMU where default is 1+2. 

How do you play to SPDIF on mac?

iambryanhaney, 2020-09-14:
Sincere apologies for the belated reply.

I haven't done anything special with regard to audio setup - it would appear that channels 3/4 are always mirroring when the E-MU is being used as the system sound device.

[image]