Wouter1/EMU-driver #129: After installing 5500 XT, the sound in the EMU 0404 audio card disappeared

maksimlenskii, 2020-05-23:
Good afternoon! Tell me.. after installing 5500 XT, the sound in the EMU 0404 audio card disappeared: when starting a track on iTunes, the Play button does not work. Added ATI HDMI audio, which was not previously available. What can I do or how can I disable HDMI audio?

Before installing the video card the sound was in the system.

MacOS Catalina 10.15.4

maksimlenskii, 2020-05-23:
Maybe it has something to do with the fact that HDMI audio took the IRQ from the EMU?

Wouter1, 2020-05-23:
@maksimlenskii 

>What can I do or how can I disable HDMI audio?

I don't know, you can ask me about EMU driver but I don't know about 5500XT or HDMI. 

There is no "IRQ" used in the driver. Maybe the OS uses it internally but software should not use that directly

There are a large number of possibilities. Maybe you can check
If the EMU still detected? (check the System Report / USB)
Is the driver loading? (cehck in System report/Extensions). Scroll down to "EMUUSBAudio" and check the 'loaded" column after you start playback 
Is EMU selected as audio device? Check System Preferences/Audio. 

"the play button does not work". Is it disabled? Does itunes start playing (you see time progress) but no sound? How about other players like quicktime?

maksimlenskii, 2020-05-23:
1. EMU still detected
2. I see the EMU in the list
3. After installing a new ATI 5500 video card, the sound card is still visible in the system, but the sound does not come from it.when you select it in the list of players (any), the sound file or stream does not start (like iTunes). When switching from it to another sound card, the sound appears.
4. When EMU was selected as the main audio source, there was a delay. Before the advent of the video card, everything was fine.

Wouter1, 2020-05-23:
@maksimlenskii  Can you report what the EMU and kernel reports in the logs when you start playing sound to the EMU. System/Utilities/Console.

maksimlenskii, 2020-05-23:
How do I do this?

maksimlenskii, 2020-05-23:
I disabled all other sound cards to avoid conflict. But here's what I found: when I now select EMU, there is some delay when switching the Input and Output tabs. I will be very grateful if I can find out how much the installation of a new video card affected the conflict with sound.

Wouter1, 2020-05-23:
@maksimlenskii 

>How do I do this?
As indicated in the https://github.com/Wouter1/EMU-driver/blob/master/FAQ.md
under "How do I make a system log of a mal-functioning driver?"

So far I did not understand that your video card also ia an audo device and probably also has an audio driver.  
So it could also be a bug in that HDMI VK228  audio driver? 

Did you install a separate HDMI VK228 audio driver?

maksimlenskii, 2020-05-23:
default	21:41:23.964066+0200	kernel	handle_get_purgeable_stats:8636: total purgeable file count: 0 total purgeable size: 0
default	21:41:24.066612+0200	kernel	handle_get_purgeable_stats:8636: total purgeable file count: 42 total purgeable size: 1748992
default	21:41:24.396310+0200	kernel	handle_get_purgeable_stats:8636: total purgeable file count: 0 total purgeable size: 0
default	21:41:39.660443+0200	kernel	Sandbox: 10 duplicate reports for bluetoothd deny(1) mach-lookup com.apple.server.bluetooth
error	21:41:39.660448+0200	kernel	Sandbox: studentd(860) deny(1) file-read-metadata /Users/maksimlenskii/Library
default	21:41:39.904306+0200	kernel	ACM: createCredentialSet: Credential set created, CS[118]/2/0/222.
default	21:41:39.992146+0200	kernel	ACM: verifyPolicy: Verifying policy UserIdentificationWithBiometrics, checkKeybagUUID = No, CS[118].
error	21:41:40.150000+0200	kernel	Sandbox: com.apple.CloudP(861) deny(1) file-read-xattr /Users/maksimlenskii/Pictures/Photos Library.photoslibrary
default	21:41:40.365529+0200	kernel	PMRD: power event 14 args 0x0 0x0
default	21:41:40.365539+0200	kernel	PMRD: wrangler RootDomainPower (ps 4, 0->1)
default	21:41:40.365540+0200	kernel	PMRD: evaluatePolicy( 0, 0x0 )
default	21:41:40.369877+0200	kernel	ACM: verifyPolicy: Policy satisfied = No, preflight = Yes, CS[118].
default	21:41:40.537930+0200	kernel	ACM: deleteCredentialSet: Deleting credential set, CS[118].
default	21:41:40.701745+0200	kernel	er_state_obj_get_for_recovery:4639: No ER state object for volume Recovery - rolling is not happening, nothing to recover.
default	21:41:40.855671+0200	kernel	handle_mount:329: vol-uuid: 4157F0D6-7EDC-4ABF-B2B8-EF97AE3D2E60 block size: 4096 block count: 62430796 (unencrypted; flags: 0x1; features: 1.0.2)
default	21:41:40.917145+0200	kernel	handle_mount:342: setting dev block size to 4096 from 512
default	21:41:41.001372+0200	kernel	nx_volume_group_update:6352: Volume Recovery role 4 Not a System or data volume
default	21:41:41.052333+0200	kernel	apfs_vfsop_mount:1489: mounted volume: Recovery
default	21:41:41.106312+0200	kernel	ioqueue_depth = 256,   ioscale = 8
default	21:41:41.183101+0200	kernel	Sandbox: diskarbitrationd(88) System Policy: allow file-read-data /dev/rdisk1s3
default	21:41:41.183121+0200	kernel	disk1s3: device is not readable.
default	21:41:42.415280+0200	kernel	apfs_vfsop_unmount:1681: /dev/disk1: unmounting volume 'Recovery'
default	21:41:42.487045+0200	kernel	apfs_vfsop_unmount:1712: waiting for cleaners to finish: purgatory
default	21:41:42.571286+0200	kernel	nx_volume_group_update:6352: Volume Recovery role 4 Not a System or data volume
default	21:41:42.620165+0200	kernel	apfs: total mem allocated: 101624456 (97 mb);
default	21:41:42.699207+0200	kernel	apfs_vfsop_unmount:1929: all done.  going home.  (numMountedAPFSVolumes 4)
error	21:41:44.917188+0200	kernel	Sandbox: AssetCacheLocato(913) deny(1) file-read-data /Users/maksimlenskii/Library/Preferences/.GlobalPreferences.plist
default	21:41:46.995740+0200	kernel	tx_flush:1028: xid 261352 tx stats: # 100 finish 100 enter 71 wait 1 29469us close 0us flush 179704us
default	21:41:47.192203+0200	kernel	tx_flush:1028: xid 2837095 tx stats: # 4580 finish 4591 enter 248 wait 81 9292us close 133us flush 3004us
error	21:41:48.020690+0200	kernel	Sandbox: bluetoothd(119) deny(1) mach-lookup com.apple.server.bluetooth
default	21:41:53.933098+0200	kernel	handle_get_purgeable_stats:8636: total purgeable file count: 0 total purgeable size: 0
default	21:41:54.035748+0200	kernel	handle_get_purgeable_stats:8636: total purgeable file count: 42 total purgeable size: 1748992
default	21:41:54.342561+0200	kernel	handle_get_purgeable_stats:8636: total purgeable file count: 0 total purgeable size: 0
default	21:42:23.988339+0200	kernel	handle_get_purgeable_stats:8636: total purgeable file count: 0 total purgeable size: 0
default	21:42:24.310022+0200	kernel	handle_get_purgeable_stats:8636: total purgeable file count: 0 total purgeable size: 0
default	21:42:40.905673+0200	kernel	PMRD: power event 14 args 0x0 0x0
default	21:42:40.905698+0200	kernel	PMRD: wrangler RootDomainPower (ps 4, 0->1)
default	21:42:40.905699+0200	kernel	PMRD: evaluatePolicy( 0, 0x0 )
default	21:42:47.419100+0200	kernel	+ IOAudioDevice[<private>]::init(<private>)
default	21:42:47.419116+0200	kernel	- IOAudioDevice[<private>]::init(<private>) returns 1
default	21:42:47.419143+0200	kernel	+ IOAudioDevice[<private>]::start(<private>)
default	21:42:47.419171+0200	kernel	- IOAudioDevice[<private>]::start(<private>)
default	21:42:47.570081+0200	kernel	Sound assertion "0 == thisUnitIDNum" failed in "/Volumes/prive/EMU-driver/EMU-driver/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
default	21:42:47.572159+0200	kernel	
default	21:42:47.572195+0200	kernel	+ IOAudioDevice[<private>]::setPowerState(1, <private>)
default	21:42:47.572196+0200	kernel	- IOAudioDevice[<private>]::setPowerState(1, <private>) returns 0x0
default	21:42:47.576951+0200	kernel	015092.499067 VDCAssistant@(null): AppleUSBHostUserClient::openGated: could not open provider E-MU 0404 | USB. provider already opened for exclusive access by Microsoft Edge
default	21:42:47.903513+0200	kernel	Sound assertion "0 == thisUnitIDNum" failed in "/Volumes/prive/EMU-driver/EMU-driver/src/EMUUSBAudio/EMUUSBAudioDevice.cpp" at line 2050 goto Exit
default	21:42:47.905589+0200	kernel	
[... cut ...]
default	21:43:51.465543+0200	kernel	- IOAudioEngine[<private>]::stopAudioEngine() returns 0x0
default	21:43:53.990725+0200	kernel	handle_get_purgeable_stats:8636: total purgeable file count: 0 total purgeable size: 0
default	21:44:02.472491+0200	kernel	Sandbox: 2 duplicate reports for bluetoothd deny(1) mach-lookup com.apple.server.bluetooth
error	21:44:02.472507+0200	kernel	Sandbox: CoreSpotlightSer(2693) deny(1) process-info-dirtycontrol self
error	21:44:03.921775+0200	kernel	Sandbox: bluetoothd(119) deny(1) mach-lookup com.apple.server.bluetooth

maksimlenskii, 2020-05-23:
So it could also be a bug in that HDMI VK228 audio driver? - No, VK228 is a monitor on the HDMI output of ATI rx5500 XT video card

Wouter1, 2020-05-23:
@maksimlenskii  Yes thanks for the log. It looks like you send only part of the log, it stops halfway the init process.

What machine do you have actually (model, ports, USB devices, hubs, etc)/

I saw some USB errors in the logs. Maybe you can (1) connect the EMU directly to your machine (2) disconnect other USB devices. (3)  cnnect the monitor to the built-in port rather than the sdpecial graphics card, so that you do not need to use the new graphics card.

Can you explain how you play s/pdif. FAIK you need to stream to channels 3 and 4 but default you will stream to 1 and 2 with all players. I thought I already asked this but can't find my post?? I am sure I mentioned that I can't test this because I have no properly working SPDIF equipment

>VK228 is a monitor

Well earlier you wrote "Added ATI HDMI audio, which was not previously available". I suppose that means, you installed drivers ? The monitor apparently has speakers? And the speaker audio is streamed over HDMI. Is that monitor attached to your graphics card? THen the sound has somehow to be piped through that card. And that might need a driver.

maksimlenskii, 2020-05-23:
I have a Hackintosh. Previously, the sound card worked when the monitor was connected to an integrated Intel HD 630 video card. I installed a discrete ATI 5500 card, after which the EMU stopped working from all ports. both analog and digital.

maksimlenskii, 2020-05-23:
And the speaker audio is streamed over HDMI. Is that monitor attached to your graphics card? - Yes 
The monitor apparently has speakers? - Yes

maksimlenskii, 2020-05-23:
I thought at first that the problem was that the sound switched to the monitor. All right. I turned on the music on iTunes and heard it through the monitor speakers. Then I went to settings and selected EMU? after that, the playback stopped and the sound from the speakers connected to the EMU was not played. When you clicked the Play button on iTunes, the slider didn't move. It started moving and the sound went when I selected the sound from the monitor again. Previously, EMU could work in parallel with SB X-Fi . Now only SB and monitor works as sound. Yes, I disconnected SB from the system. It didn't help the EMU work.

Wouter1, 2021-01-22:
@maksimlenskii I was reading again through the still open tickets.

It seems this got never solved. 
Now I was wondering, maybe it's related to a recently reported issue with hackintoshes?

Check the bottom of the installation notes

https://github.com/Wouter1/EMU-driver/blob/master/Install.md