Wouter1/EMU-driver #130: Crackling Distortion for entire playback

tmbgfan1234, 2020-05-31:
I'm using a Tracker Pre and I have El Capitan 10.11.6. Whenever I record, the entire track has crackling distortion. All sample rates are effected and no amount of latency change fixes the issue. The acceptance test contains all of the distortion as well. I'm new to this, so I don't know what other information you would like.

Wouter1, 2020-06-01:
@tmbgfan1234  What machine is this on (type of mac, year, USB ports)?

Did you use the control panel to set the sample rate of the EMU? Which rate? Is the problem with all rates?

tmbgfan1234, 2020-06-01:
Thanks for the quick response! It's a MacBook Pro (Mid 2009). 2.26 GHz Intel 2 Duo. 5 GB 1067 MHz DDR3. USB 2.0. I used the control panel to set the rate and tried each rate, but every rate had the same problem.

Wouter1, 2020-06-02:
This might issue #120 that I fixed recently, or some other issues fixed a bit longer ago.
Unfortunately I can not compile it anymore for  el capitan. as it requires XCode 7 (I'm already running an older XCode 11 to compile for high sierra...). So the current distri contains an old version for osx < 10.11

One option would be that you compile it yourself. Would that be possible for you?
The build instructions are on https://github.com/Wouter1/EMU-driver/blob/master/Developer.md

Another option would be that you upgrade your OS, at least to 10.12 sierra

A guess would be to try to install the 10.12+ kext into the 10.11. I think it won't work but this might be a border case that might just work.

Wouter1, 2020-06-08:
@tmbgfan1234 were you able to apply one of my suggestions?

tmbgfan1234, 2020-06-08:
I am not quite sure how to compile things myself. I hit the threshold of being able to update my OS. So, to be honest, I just gave up and used my PC desktop instead. It's not as versatile, but it works.

Wouter1, 2020-06-09:
@tmbgfan1234 sorry that I can not help fixing old drivers anymore.  I have to close this as I do not see how this can be fixed then.

closed 2020-06-09 by Wouter1